machinekit/machinekit #677: undefined symbol rtapi_acos while executing script linuxcnc

rosaldorosa, 2015-06-04:
i gantry in menu
ii.Linux stajnia 3.16.0-38-generic #52~14.04.1-Ubuntu SMP Fri May 8 09:43:57 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux
iii.linux-image-3.8-1-xenomai.x86-amd64
linux-headers-3.8-1-xenomai.x86-amd64
iv.MACHINEKIT - 0.1
Machine configuration directory is '/home/stajnia/machinekit/configs/by_interface/parport/gantry'
Machine configuration file is 'gantry.ini'
Starting Machinekit...
/home/stajnia/machinekit/scripts/realtime: line 94: printf: Unity: invalid number
io started
halcmd loadusr io started
gantry.hal:11: insmod failed, returned -1:
do_load_cmd: dlopen: /home/stajnia/machinekit/rtlib/posix/tp.so: undefined symbol: rtapi_acos
rpath=/home/stajnia/machinekit/rtlib/posix:/home/stajnia/machinekit/lib
See /var/log/linuxcnc.log for more information.
Shutting down and cleaning up Machinekit...
/home/stajnia/machinekit/scripts/realtime: line 94: printf: Unity: invalid number
Cleanup done
v 
Jun  4 19:11:17 stajnia msgd:0: startup pid=5245 flavor=posix rtlevel=1 usrlevel=1 halsize=524288 shm=Posix gcc=4.8.2 version=unknown
Jun  4 19:11:17 stajnia msgd:0: ØMQ=4.0.4 czmq=2.2.0 protobuf=2.5.0
Jun  4 19:11:17 stajnia msgd:0: configured: sha=e600aeb
Jun  4 19:11:17 stajnia msgd:0: built:      Jun  4 2015 18:14:47 sha=e600aeb
Jun  4 19:11:17 stajnia msgd:0: register_stuff: actual hostname as announced by avahi='stajnia.local'
Jun  4 19:11:17 stajnia msgd:0: zeroconf: registering: 'Log service on stajnia.local pid 5245'
Jun  4 19:11:17 stajnia msgd:0: rtapi_app:5252:user accepting commands at ipc:///tmp/0.rtapi.a42c8c6b-4025-4f83-ba28-dad21114744a
Jun  4 19:11:17 stajnia rtapi:0: do_load_cmd: dlopen: /home/stajnia/machinekit/rtlib/posix/tp.so: undefined symbol: rtapi_acos
Jun  4 19:11:17 stajnia rtapi:0: rpath=/home/stajnia/machinekit/rtlib/posix:/home/stajnia/machinekit/lib
Jun  4 19:11:17 stajnia msgd:0: hal_lib:5252:rt GANTRYKINS: joints[0] = 0
Jun  4 19:11:17 stajnia msgd:0: hal_lib:5252:rt GANTRYKINS: joints[1] = 1
Jun  4 19:11:17 stajnia msgd:0: hal_lib:5252:rt GANTRYKINS: joints[2] = 2
Jun  4 19:11:17 stajnia msgd:0: hal_lib:5252:rt GANTRYKINS: joints[3] = 3
Jun  4 19:11:17 stajnia msgd:0: hal_lib:5252:rt GANTRYKINS: joints[4] = 4
Jun  4 19:11:17 stajnia msgd:0: hal_lib:5252:rt GANTRYKINS: joints[5] = 5
Jun  4 19:11:17 stajnia msgd:0: hal_lib:5252:rt GANTRYKINS: joints[6] = -1
Jun  4 19:11:17 stajnia msgd:0: hal_lib:5252:rt GANTRYKINS: joints[7] = -1
Jun  4 19:11:17 stajnia msgd:0: hal_lib:5252:rt GANTRYKINS: joints[8] = -1
Jun  4 19:11:17 stajnia msgd:0: rtapi_app:5252:user do_load_cmd: dlopen: /home/stajnia/machinekit/rtlib/posix/tp.so: undefined symbol: rtapi_acos
Jun  4 19:11:17 stajnia msgd:0: rtapi_app:5252:user rpath=/home/stajnia/machinekit/rtlib/posix:/home/stajnia/machinekit/lib
Jun  4 19:11:17 stajnia msgd:0: zeroconf: registered 'Log service on stajnia.local pid 5245' _machinekit._tcp 0 TXT "uuid=a42c8c6b-4025-4f83-ba28-dad21114744a" "instance=b630797c-0adc-11e5-ae3a-74d4358d3a1e" "service=log" "dsn=ipc:///tmp/0.log.a42c8c6b-4025-4f83-ba28-dad21114744a"
Jun  4 19:11:19 stajnia rtapi:0: unload: 'gantrykins' not loaded
Jun  4 19:11:20 stajnia msgd:0: rtapi_app exit detected - scheduled shutdown
Jun  4 19:11:22 stajnia msgd:0: msgd shutting down
Jun  4 19:11:22 stajnia msgd:0: zeroconf: unregistering 'Log service on stajnia.local pid 5245'
Jun  4 19:11:22 stajnia msgd:0: log buffer hwm: 0% (11 msgs, 852 bytes out of 524288)
Jun  4 19:11:22 stajnia msgd:0: normal shutdown - global segment detached
vi machinekit don't leave anything in dmesg

strahlex, 2015-06-05:
package install or from git repo? In both cases please update the problem was fixed some days ago.

rosaldorosa, 2015-06-06:
Package installed from GitHub

rosaldorosa, 2015-06-06:
Mistakenly closed

rosaldorosa, 2015-06-06:
I've check librtapi_math.so.0 and it exists in lib dir.
do_load_cmd: dlopen: /home/stajnia/machinekit/rtlib/posix/tp.so: undefined symbol: rtapi_acos
This error appears during execution of linuxcnc

mhaberler, 2015-06-06:
maybe the EXPORT_SYMBOL(rtapi_acos) is missing?

rosaldorosa, 2015-06-08:
I think the issue is, that .so library are dynamicaly linked, but rtapi_acos comes from statically linked.
SO it can not be used dynamically.It looks, that library .o was renamed to .so, thus making it useless

strahlex, 2015-06-08:
No, this error just means that librtapimath was not loaded by the rtapi_app_posix. It is not necessary to statically link this library anywhere. 

@rosaldorosa Can you please do a rebuild from machinekit/master

@mhaberler EXPORT_SYMBOL is for kernel modules only.

mhaberler, 2015-06-08:
@strahlex: no, it is for RTAPI, and required for kernel as well as userland exported functions

see the HAL library for an example - it is #ifdef RTAPI, not #ifdef BUILD_SYS_KBUILD

koppi, 2015-06-08:
I reproduced this issue: cloned a fresh rep from master, then the usual stuff:

```bash
$ sh autogen.sh
$ ./configure --prefix=/usr --sysconfdir=/etc --without-xenomai --without-xenomai-kernel
$ make
$ sudo make setuid
$ sudo make install
$ linuxcnc (selected axis sim mm . ini)
```
output:
```bash
$ linuxcnc
MACHINEKIT - 0.1
Machine configuration directory is '/usr/share/linuxcnc/examples/sample-configs/sim/axis'
Machine configuration file is 'axis_mm.ini'
Starting Machinekit...
io started
halcmd loadusr io started
core_sim.hal:7: insmod failed, returned -1:
do_load_cmd: dlopen: /usr/lib/linuxcnc/posix/tp.so: undefined symbol: rtapi_acos
rpath=/usr/lib/linuxcnc/posix
See /var/log/linuxcnc.log for more information.
Shutting down and cleaning up Machinekit...
```

RunningLight, 2015-06-08:
@koppi I don't see this is I build from fresh source on a virgin copy of Debian 8/Jessie but accept RIP defaults (e.g., `$./configure`). Machinekit/Linuxcnc launcher brings up `axis_mm.ini` without any complaint about undefined symbol. Clearly something is wrong between `configure` and `make` when chosing target directories.

On my RIP build,

```
$ find ~ -name *tp.so
/home/kreed/machinekit/rtlib/posix/tp.so
/home/kreed/machinekit/rtlib/rt-preempt/tp.so
/home/kreed/machinekit/rtlib/xenomai/tp.so

$ find ~ -name librtapi_math.so*
/home/kreed/machinekit/lib/librtapi_math.so
/home/kreed/machinekt/lib/librtapi_math.so.0
```

To continue,
```
nm -g /home/kreed/machinekit/lib/librtapi_math.so
0000000000205fd4 B __bss_start
                 w __cxa_finalize@@GLIBC_2.2.5
0000000000205fd4 D _edata
0000000000205fe0 B _end
0000000000004d18 T _fini
                 w __gmon_start__
0000000000001ef0 T __ieee754_acos
0000000000002220 T __ieee754_asin
0000000000002aa0 T __ieee754_atan2
0000000000004980 T __ieee754_fmod
00000000000046b0 T __ieee754_hypot
00000000000011d0 T __ieee754_pow
0000000000003150 T __ieee754_rem_pio2
0000000000001b60 T __ieee754_sqrt
0000000000000de8 T _init
                 w _ITM_deregisterTMCloneTable
                 w _ITM_registerTMCloneTable
                 w _Jv_RegisterClasses
0000000000002fc0 T __kernel_cos
00000000000034f0 T __kernel_rem_pio2
0000000000002e10 T __kernel_sin
0000000000003f00 T __kernel_standard
00000000000025a0 T __kernel_tan
0000000000205fd8 B libm_errno
0000000000205fd0 D _LIB_VERSION
0000000000004660 T __powidf2
0000000000002210 T rtapi_acos
0000000000002510 T rtapi_asin
0000000000002820 T rtapi_atan
0000000000002a90 T rtapi_atan2
0000000000004290 T rtapi_cbrt
0000000000001d80 T rtapi_ceil
0000000000004420 T rtapi_copysign
0000000000002ee0 T rtapi_cos
0000000000001b40 T rtapi_fabs
0000000000004460 T rtapi_finite
0000000000001060 T rtapi_floor
0000000000004d00 T rtapi_fmax
0000000000004d10 T rtapi_fmin
0000000000004cf0 T rtapi_fmod
0000000000004970 T rtapi_hypot
0000000000001b30 T rtapi_pow
0000000000004480 T rtapi_rint
0000000000003d70 T rtapi_scalbn
0000000000002d30 T rtapi_sin
0000000000001d70 T rtapi_sqrt
0000000000002520 T rtapi_tan
```
@rosaldorosa Do you see all these symbols in your copy of librtapi_math.so?

mhaberler, 2015-06-08:
somebody does not believe me...

you MUST EXPORT_SYMBOL(foo) for userland threads as well

depending on gcc/ld/ld.so version it might or might not work without.

RunningLight, 2015-06-08:
I believe! ...but I am weak in the ways of the RT Masters. I see all the math symbols exported from `krtapi_math.c` and I see all of them available as external symbols in librtapi_math.so/.so.0 (e.g., using the `-g` flag on `nm` above). What differentiates between exporting them for userland threads vs for kernel threads?

  Just call me *Grasshopper*

mhaberler, 2015-06-09:
just because a symbol shows up in 'nm' that doesnt mean it resolves for RT modules

see the details of the [EXPORT_SYMBOL](https://github.com/machinekit/machinekit/blob/master/src/rtapi/rtapi_export.h#L40-L41) macro - it uses symbols in a custom ELF section:

#define EXPORT_SYMBOL(x) __attribute__((section(".rtapi_export"))) \
    char rtapi_exported_##x[] = #x;

I recommend to study tests/symbol.0 and tests/symbol.1 to understand the exporting business, maybe modify test.0 with EXPORT_SYMBOL and see what happens

be aware this is criticial to functionality - a key requirement of rtapi_app is to separate its own symbol namespace from the namespace of the RT modules it loads, and "patching up" without understanding exactly what is going on is a good recipe for very subtle breakage due to symbol linkage - I spent a week on fixing one of those

rosaldorosa, 2015-06-09:
"just because a symbol shows up in 'nm' that doesnt mean it resolves for 
RT modules"
What do You mean by that?
Maybe the module is not loaded into kernel space with insmod

On 06/09/2015 11:40 AM, Michael Haberler wrote:
> just because a symbol shows up in 'nm' that doesnt mean it resolves 
> for RT modules

rosaldorosa, 2015-06-09:
Does it mean it never worked before?
Does it mean some design part is missing?

rosaldorosa, 2015-06-09:
I've never heard, that kernel communicate with user space via 
Export_SYMBOL()
It would break the whole system

strahlex, 2015-06-09:
@rosaldorosa Since you are using the posix flavor the `krtapi_math.h` header will not be used but the `librtapi_math.h`. I tried adding the EXPORT_SYMBOL macro there as well but got stuck with some weird compile errors.

mhaberler, 2015-06-09:
@rosaldorosa I think you are jumping to conclusions a tad early

neither is there a kernel module nor an insmod for userland threads; as far as modules go in the userland context, those are dynamic shared objects (aka "shared library") which are processed through the dlopen(3)/dlysym() API

@strahlex - any use of EXPORT_SYMBOL should be protected by #ifdef RTAPI, maybe that was the issue?

@koppi - can you report platform details (arch, gcc/ld version, os etc)?

rosaldorosa, 2015-06-09:
Is so then where is shared lib which contains rtapi_acos?
the only one existing is

|lib/librtapi_math.so

which is static library (at leasr ldd inform about it)
|

koppi, 2015-06-09:
@mhaberler 

```bash
$ uname -a
Linux x220 3.19.0-17-generic #17-Ubuntu SMP Wed May 6 16:46:12 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux
```
```bash
$ lsb_release -a
Distributor ID:	Ubuntu
Description:	Ubuntu 15.04
Release:	15.04
Codename:	vivid
```
```bash
$ gcc --version
gcc (Ubuntu 4.9.2-10ubuntu13) 4.9.2
Copyright (C) 2014 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
```bash
$ ld --version
GNU ld (GNU Binutils for Ubuntu) 2.25
Copyright (C) 2014 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or (at your option) a later version.
This program has absolutely no warranty.
```

mhaberler, 2015-06-09:
@strahlex - see the definition of EXPORT_SYMBOL, this actually emits data and should go into a .c file, not a .h file

mhaberler, 2015-06-09:
I just tried with commit 1c03d2219 on jessie, and cannot reproduce the effect, both xenomai and posix work fine
````
mah@nwheezy:~/machinekit-tutorial/rtlib/xenomai$ ld --version
GNU ld (GNU Binutils for Debian) 2.25

mah@nwheezy:~/machinekit-tutorial/rtlib/xenomai$ gcc --version
gcc (Debian 4.9.2-10) 4.9.2

mah@nwheezy:~/machinekit-tutorial/rtlib$ ldd posix/tp.so 
	linux-gate.so.1 (0xb7722000)
	libc.so.6 => /lib/i386-linux-gnu/i686/cmov/libc.so.6 (0xb753d000)
	/lib/ld-linux.so.2 (0xb7723000)
mah@nwheezy:~/machinekit-tutorial/rtlib$ ldd xenomai/tp.so 
	linux-gate.so.1 (0xb77b2000)
	libc.so.6 => /lib/i386-linux-gnu/i686/cmov/libc.so.6 (0xb75cd000)
	/lib/ld-linux.so.2 (0xb77b3000)
mah@nwheezy:~/machinekit-tutorial/rtlib$ cd ../libexec/
mah@nwheezy:~/machinekit-tutorial/libexec$ ldd rtapi_app_posix 
	linux-gate.so.1 (0xb776b000)
	liblinuxcncini.so.0 => /home/mah/machinekit-tutorial/lib/liblinuxcncini.so.0 (0xb7764000)
	liblinuxcncshm.so.0 => /home/mah/machinekit-tutorial/lib/liblinuxcncshm.so.0 (0xb7761000)
	libmtalk.so.0 => /home/mah/machinekit-tutorial/lib/libmtalk.so.0 (0xb7741000)
	liblinuxcnc-pb2++.so.0 => /home/mah/machinekit-tutorial/lib/liblinuxcnc-pb2++.so.0 (0xb7537000)
	librtapi_math.so.0 => /home/mah/machinekit-tutorial/lib/librtapi_math.so.0 (0xb7530000)
	libpthread.so.0 => /lib/i386-linux-gnu/i686/cmov/libpthread.so.0 (0xb74f2000)
	librt.so.1 => /lib/i386-linux-gnu/i686/cmov/librt.so.1 (0xb74e9000)
	libprotobuf.so.9 => /usr/lib/i386-linux-gnu/libprotobuf.so.9 (0xb73bd000)
	libczmq.so.3 => /usr/local/lib/libczmq.so.3 (0xb7307000)
	libzmq.so.3 => /usr/lib/i386-linux-gnu/libzmq.so.3 (0xb72a2000)
	libavahi-common.so.3 => /usr/lib/i386-linux-gnu/libavahi-common.so.3 (0xb7293000)
	libavahi-client.so.3 => /usr/lib/i386-linux-gnu/libavahi-client.so.3 (0xb727f000)
	libstdc++.so.6 => /usr/lib/i386-linux-gnu/libstdc++.so.6 (0xb718c000)
	libdl.so.2 => /lib/i386-linux-gnu/i686/cmov/libdl.so.2 (0xb7187000)
	libuuid.so.1 => /lib/i386-linux-gnu/libuuid.so.1 (0xb7181000)
	libm.so.6 => /lib/i386-linux-gnu/i686/cmov/libm.so.6 (0xb713b000)
	libgcc_s.so.1 => /lib/i386-linux-gnu/libgcc_s.so.1 (0xb711e000)
	libc.so.6 => /lib/i386-linux-gnu/i686/cmov/libc.so.6 (0xb6f73000)
	libjansson.so.4 => /usr/lib/i386-linux-gnu/libjansson.so.4 (0xb6f64000)
	/lib/ld-linux.so.2 (0xb776c000)
	libz.so.1 => /lib/i386-linux-gnu/libz.so.1 (0xb6f47000)
	libpgm-5.1.so.0 => /usr/lib/libpgm-5.1.so.0 (0xb6ef1000)
	libsodium.so.13 => /usr/lib/i386-linux-gnu/libsodium.so.13 (0xb6e72000)
	libdbus-1.so.3 => /lib/i386-linux-gnu/libdbus-1.so.3 (0xb6e1a000)
mah@nwheezy:~/machinekit-tutorial/libexec$ ldd rtapi_app_xenomai
	linux-gate.so.1 (0xb770b000)
	liblinuxcncini.so.0 => /home/mah/machinekit-tutorial/lib/liblinuxcncini.so.0 (0xb7704000)
	liblinuxcncshm.so.0 => /home/mah/machinekit-tutorial/lib/liblinuxcncshm.so.0 (0xb7701000)
	libmtalk.so.0 => /home/mah/machinekit-tutorial/lib/libmtalk.so.0 (0xb76e1000)
	liblinuxcnc-pb2++.so.0 => /home/mah/machinekit-tutorial/lib/liblinuxcnc-pb2++.so.0 (0xb74d7000)
	librtapi_math.so.0 => /home/mah/machinekit-tutorial/lib/librtapi_math.so.0 (0xb74d0000)
	libnative.so.3 => /usr/lib/libnative.so.3 (0xb74a5000)
	libxenomai.so.0 => /usr/lib/libxenomai.so.0 (0xb749d000)
	libpthread.so.0 => /lib/i386-linux-gnu/i686/cmov/libpthread.so.0 (0xb7480000)
	librt.so.1 => /lib/i386-linux-gnu/i686/cmov/librt.so.1 (0xb7477000)
	libprotobuf.so.9 => /usr/lib/i386-linux-gnu/libprotobuf.so.9 (0xb734c000)
	libczmq.so.3 => /usr/local/lib/libczmq.so.3 (0xb7296000)
	libzmq.so.3 => /usr/lib/i386-linux-gnu/libzmq.so.3 (0xb7231000)
	libavahi-common.so.3 => /usr/lib/i386-linux-gnu/libavahi-common.so.3 (0xb7221000)
	libavahi-client.so.3 => /usr/lib/i386-linux-gnu/libavahi-client.so.3 (0xb720d000)
	libstdc++.so.6 => /usr/lib/i386-linux-gnu/libstdc++.so.6 (0xb711b000)
	libdl.so.2 => /lib/i386-linux-gnu/i686/cmov/libdl.so.2 (0xb7116000)
	libuuid.so.1 => /lib/i386-linux-gnu/libuuid.so.1 (0xb7110000)
	libm.so.6 => /lib/i386-linux-gnu/i686/cmov/libm.so.6 (0xb70c9000)
	libgcc_s.so.1 => /lib/i386-linux-gnu/libgcc_s.so.1 (0xb70ac000)
	libc.so.6 => /lib/i386-linux-gnu/i686/cmov/libc.so.6 (0xb6f02000)
	libjansson.so.4 => /usr/lib/i386-linux-gnu/libjansson.so.4 (0xb6ef3000)
	/lib/ld-linux.so.2 (0xb770c000)
	libz.so.1 => /lib/i386-linux-gnu/libz.so.1 (0xb6ed6000)
	libpgm-5.1.so.0 => /usr/lib/libpgm-5.1.so.0 (0xb6e7f000)
	libsodium.so.13 => /usr/lib/i386-linux-gnu/libsodium.so.13 (0xb6e00000)
	libdbus-1.so.3 => /lib/i386-linux-gnu/libdbus-1.so.3 (0xb6da9000)

````

koppi, 2015-06-09:
In my setup there is no ```librtapi_math.so.0``` in ```rtapi_app_posix```:
```bash
$ ldd rtapi_app_posix 
	linux-vdso.so.1 =>  (0x00007ffd787fc000)
	liblinuxcncini.so.0 => /home/koppi/Desktop/cnc/machinekit/lib/liblinuxcncini.so.0 (0x00007efd1823f000)
	liblinuxcncshm.so.0 => /home/koppi/Desktop/cnc/machinekit/lib/liblinuxcncshm.so.0 (0x00007efd1803a000)
	libmtalk.so.0 => /home/koppi/Desktop/cnc/machinekit/lib/libmtalk.so.0 (0x00007efd17e1b000)
	liblinuxcnc-pb2++.so.0 => /home/koppi/Desktop/cnc/machinekit/lib/liblinuxcnc-pb2++.so.0 (0x00007efd17af0000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007efd17885000)
	libprotobuf.so.9 => /usr/lib/x86_64-linux-gnu/libprotobuf.so.9 (0x00007efd17557000)
	libczmq.so.1 => /usr/lib/x86_64-linux-gnu/libczmq.so.1 (0x00007efd1730d000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007efd16ffd000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007efd16df9000)
	libuuid.so.1 => /lib/x86_64-linux-gnu/libuuid.so.1 (0x00007efd16bf4000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007efd169dd000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007efd16613000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007efd1640b000)
	libavahi-common.so.3 => /usr/lib/x86_64-linux-gnu/libavahi-common.so.3 (0x00007efd161fe000)
	libavahi-client.so.3 => /usr/lib/x86_64-linux-gnu/libavahi-client.so.3 (0x00007efd15fed000)
	libjansson.so.4 => /usr/lib/x86_64-linux-gnu/libjansson.so.4 (0x00007efd15de0000)
	/lib64/ld-linux-x86-64.so.2 (0x00007efd18446000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007efd15bc4000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007efd158bc000)
	libsodium.so.10 => /usr/lib/libsodium.so.10 (0x00007efd15660000)
	libzmq.so.3 => /usr/lib/x86_64-linux-gnu/libzmq.so.3 (0x00007efd153ff000)
	libdbus-1.so.3 => /lib/x86_64-linux-gnu/libdbus-1.so.3 (0x00007efd151b7000)
	libpgm-5.1.so.0 => /usr/lib/x86_64-linux-gnu/libpgm-5.1.so.0 (0x00007efd14f6a000)
	libsodium.so.13 => /usr/lib/x86_64-linux-gnu/libsodium.so.13 (0x00007efd14d18000)
```

koppi, 2015-06-09:
I just did a ```make V=1```, see the output:
```bash
Linking rtapi_app_posix
g++ -Wl,-rpath,/home/koppi/Desktop/cnc/machinekit/rtlib/posix -Wl,-rpath,/home/koppi/Desktop/cnc/machinekit/lib \
    -o ../libexec/rtapi_app_posix objects/rtapi/posix/rtapi_app.o objects/rtapi/posix/rtapi_compat.o objects/rtapi/posix/rtapi_hexdump.o objects/rtapi/posix/rtapi_support.o ../lib/liblinuxcncini.so.0 ../lib/liblinuxcncshm.so ../lib/libmtalk.so.0 ../lib/liblinuxcnc-pb2++.so.0 ../lib/librtapi_math.so.0 -lpthread -lrt \
-lprotobuf -pthread -lpthread  -lczmq -lzmq  -lavahi-common -lavahi-client   -lstdc++ -ldl -luuid
```

koppi, 2015-06-09:
Adding ```-Wl,--no-as-needed``` to the linker flags worked for me:
```
g++ -Wl,-rpath,/home/koppi/Desktop/cnc/machinekit/rtlib/posix -Wl,-rpath,/home/koppi/Desktop/cnc/machinekit/lib     -o ../libexec/rtapi_app_posix objects/rtapi/posix/rtapi_app.o objects/rtapi/posix/rtapi_compat.o objects/rtapi/posix/rtapi_hexdump.o objects/rtapi/posix/rtapi_support.o ../lib/liblinuxcncini.so.0 ../lib/liblinuxcncshm.so ../lib/libmtalk.so.0 ../lib/liblinuxcnc-pb2++.so.0 -Wl,--no-as-needed ../lib/librtapi_math.so.0 -lpthread -lrt -lprotobuf -pthread -lpthread  -lczmq -lzmq  -lavahi-common -lavahi-client   -lstdc++ -ldl -luuid
```

```librtapi_math.so.0``` then shows up in the library dependencies:
```bash
$ ldd rtapi_app_posix 
	linux-vdso.so.1 =>  (0x00007ffd9d7d5000)
	liblinuxcncini.so.0 => /home/koppi/Desktop/cnc/machinekit/lib/liblinuxcncini.so.0 (0x00007eff0de0d000)
	liblinuxcncshm.so.0 => /home/koppi/Desktop/cnc/machinekit/lib/liblinuxcncshm.so.0 (0x00007eff0dc08000)
	libmtalk.so.0 => /home/koppi/Desktop/cnc/machinekit/lib/libmtalk.so.0 (0x00007eff0d9e9000)
	liblinuxcnc-pb2++.so.0 => /home/koppi/Desktop/cnc/machinekit/lib/liblinuxcnc-pb2++.so.0 (0x00007eff0d6be000)
	librtapi_math.so.0 => /home/koppi/Desktop/cnc/machinekit/lib/librtapi_math.so.0 (0x00007eff0d4b6000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007eff0d24c000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007eff0d044000)
	libprotobuf.so.9 => /usr/lib/x86_64-linux-gnu/libprotobuf.so.9 (0x00007eff0cd15000)
	libczmq.so.1 => /usr/lib/x86_64-linux-gnu/libczmq.so.1 (0x00007eff0cacb000)
	libzmq.so.3 => /usr/lib/x86_64-linux-gnu/libzmq.so.3 (0x00007eff0c86b000)
	libavahi-common.so.3 => /usr/lib/x86_64-linux-gnu/libavahi-common.so.3 (0x00007eff0c65e000)
	libavahi-client.so.3 => /usr/lib/x86_64-linux-gnu/libavahi-client.so.3 (0x00007eff0c44d000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007eff0c13e000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007eff0bf39000)
	libuuid.so.1 => /lib/x86_64-linux-gnu/libuuid.so.1 (0x00007eff0bd34000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007eff0ba2c000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007eff0b815000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007eff0b44b000)
	libjansson.so.4 => /usr/lib/x86_64-linux-gnu/libjansson.so.4 (0x00007eff0b23e000)
	/lib64/ld-linux-x86-64.so.2 (0x00007eff0e014000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007eff0b022000)
	libsodium.so.10 => /usr/lib/libsodium.so.10 (0x00007eff0adc6000)
	libpgm-5.1.so.0 => /usr/lib/x86_64-linux-gnu/libpgm-5.1.so.0 (0x00007eff0ab79000)
	libsodium.so.13 => /usr/lib/x86_64-linux-gnu/libsodium.so.13 (0x00007eff0a927000)
	libdbus-1.so.3 => /lib/x86_64-linux-gnu/libdbus-1.so.3 (0x00007eff0a6df000)
```

koppi, 2015-06-09:
in ```src/rtapi/Submakefile``` I see:
```
../libexec/rtapi_app_$(threads): \
  $(call TOOBJS, $(RTAPI_APP_SRCS)) \
  ../lib/liblinuxcncini.so.0 \
  ../lib/liblinuxcncshm.so \
  ../lib/libmtalk.so.0 \
  ../lib/liblinuxcnc-pb2++.so.0 \
  ../lib/librtapi_math.so.0
  $(ECHO) Linking $(notdir $@)
  @mkdir -p $(dir $@)
  $(Q)$(CXX) -Wl,-rpath,$(EMC2_RTLIB_DIR) $(RTAPI_APP_RPATH) \
      -o $@ $^ $(RT_LDFLAGS) \
  $(PROTOBUF_LIBS) $(CZMQ_LIBS) $(AVAHI_LIBS) $(LTTNG_UST_LIBS) -lstdc++ -ldl -luuid
```
Is this the right place to add the ```-Wl,--no-as-needed``` option?

mhaberler, 2015-06-10:
yes, all flavors of rtapi_app need this 

great find, congratulations! was not aware of this latest and greatest rebuggung ;)

I was wrong above insofar as this holds for symbol resolution _between RT modules_ but not between a RT module and a shared library linked into rtapi_app

mhaberler, 2015-06-10:
not sure how to test for the presence of this flag in ld, does ld ignore unknown flags?

we might actually need a configure test 

actually this might affect pretty much every program which uses dlopen()/dlsym() for plugins/loadable extensions/modules, not sure what the Canonical guys were thinking by enabling this _greatest and latest feature_; among others, it will be webtalk and it could actually be Python as well as that uses dlopen() for 'import'

rosaldorosa, 2015-06-10:
Could You put the exact line where You inserted this option?
is it

|||-Wl,--no-as-needed||
$(PROTOBUF_LIBS) $(CZMQ_LIBS) $(AVAHI_LIBS)||$(LTTNG_UST_LIBS)  -lstdc++ -ldl -luuid

|

koppi, 2015-06-10:
@rosaldorosa I just checked, and this worked for me: you can pass ```-Wl,--no-as-neede``` during the configure step like so:
```
env LDFLAGS="-Wl,--no-as-needed" ./configure ...
```

koppi, 2015-06-10:
@mhaberler I don't know anything about the autotools stuff, read a book once, "Die AutoTools", of course the tile was in german.

rosaldorosa, 2015-06-10:
After googling a bit I found it to be rather workaround then real fix.
Please look at:
http://stackoverflow.com/questions/10539153/how-to-follow-linking-order-when-linking-against-static-library-with-gnu-make
It shows wrong target in compilation ( instead of .o should be .so)

rosaldorosa, 2015-06-11:
Unfortunately still don't work for me.
I think the reason is return from ldd rtlib/posix/tp.so
     linux-vdso.so.1 =>  (0x00007ffd28105000)
     libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007ff54f693000)
     /lib64/ld-linux-x86-64.so.2 (0x00007ff54fc9f000)

librtapi is not on the list

strahlex, 2015-06-11:
rosaldorosa writes:

> Unfortunately still don't work for me.
> I think the reason is return from ldd rtlib/posix/tp.so
>      linux-vdso.so.1 =>  (0x00007ffd28105000)
>      libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007ff54f693000)
>      /lib64/ld-linux-x86-64.so.2 (0x00007ff54fc9f000)
>
> librtapi is not on the list
That's how it should be. librtapi_math.so is a shared library which will
be loaded by rtapi. It does not have to be linked against the rt libs.

> ---
> https://github.com/machinekit/machinekit/issues/677#issuecomment-111147039

rosaldorosa, 2015-06-12:
My configure command is env LDFLAGS="-Wl,--no-as-needed" ./configure 
--with-xenomai --with-boost=YES
But result is that ldd still don't show rtapi_math - ldd 
libexec/rtapi_app_xenomai

mhaberler, 2015-06-12:
@rosaldorosa - is there any compelling reason why you are using Ubuntu over Debian?

I think the website is pretty clear the platform of choice is debian, and ubuntu was dropped for good reasons - there is no perceivable upside for supporting this, that and the next distro of the week

so if you want something working fast, use debian or be prepared to work yourself through the mess of "improvements" which various distro packagers have unleashed

widening the range of distros which run machinekit _packages_ is a non-goal and a waste of time IMO, but if you come up with a PR that is fine

also, please clarify what you mean by "Package installed from GitHub" - there are no packages on github, just the source repo

rosaldorosa, 2015-06-13:
I've found that, env LDFLAGS="-Wl,--no-as-needed" ./configure ... don't work
I've found, that export LDFLAGS="-Wl,--no-as-needed"; ./configure ... don't work
The difference is that the second give output from echo $LDFLAGS
However when it comes to linking LDFLAGS is empty
But when I put manually above option into Submakefile I got minor success.
Ldd returns dependencies on librtapi_math.so
But still no progress with missing dependencies. Do You know at which compilation the rtapi_xenomai application is informed anout librtapi_math existance. Beacause other libs in libs dir are on the dependency list of main app (except math)

It's really hard for me to believe, that compilation issues are related to distro, because all use same compilation software. Also deb packages for ubuntu don't work

strahlex, 2015-06-13:
@rosaldorosa No, there are slight (and big in case of Ubuntu) differences in the parameters and tools that the distros supply.

rosaldorosa, 2015-06-14:
I think, that Ubuntu disclosed weak points of this distribution. If You want to findout yourself, just add default compiler flag -Wl,--as-needed. To make sure you added it correctly in every Makefile and Submakefile, where $(LDFLAGS) used, just put $(ECHO) LDFlags test $(LDFLAGS), to make sure Your change works

mhaberler, 2015-06-14:
@rosaldorosa - I'm not sure we are on the same page with respect to "weak points" - but I get the impression you have not fully grasped the scope and impact of this grandiose "feature" you are fighting against:

if the canonical guys in their quest for the (as they say) "latest and greatest ld support" decide to sink the vessel by this massive and not-backwards-compatible change of linker semantics, and with it very likely any plugin-based application which uses the dlopen(3) API starting from every single web browser out there, then they are certainly free to do so, but I would not concur in calling every application which cannot deal with this type of nonsense by day one "weak".  I am just waiting for Python to go tits up on this feature.

I would rather suggest this point of view: somebody has not done his homework, and it aint us.

That said: you want this on Ubuntu (for reasons I cannot discern), you get to file a pull request to track canonical "improvements". That is how all open source projects work, including this one. Telling other people to fix "the weak points" is unlikely to create much enthusiasm with them based on the line of argumentation so far.

As for a PR: adding flags sledgehammer-style throughout the build system as you suggest is likely to introduce more damage than upside. What you want to do is identify those binaries which need to provide library support which cannot be detected link-time as the modules to be loaded are not known link-time. Those are (in the minimum) rtapi_app and webtalk, and likely others. Then you fix the linking of those, and _only_ those.

rosaldorosa, 2015-06-14:
But the rtapi_exec_xenomai must be aware of librtapi_math presense. In normal circumstance it must informed by linking -rdynamic -lrtapi_math, which doesn happen, or I can not find

rosaldorosa, 2015-06-14:
../libexec/rtapi_app_$(threads): \
        $(call TOOBJS, $(RTAPI_APP_SRCS)) \
        ../lib/liblinuxcncini.so.0 \
        ../lib/liblinuxcncshm.so \
        ../lib/libmtalk.so.0 \
        ../lib/liblinuxcnc-pb2++.so.0 \
        ../lib/librtapi_math.so.0
Do You think above is correct for dynamic linking?

mhaberler, 2015-06-14:
@rosaldorosa - I guess you mean rtapi_app_<flavor>

this daemon _per se_ does NOT require rtapi_math as it does not use math functions
HOWEVER, the rt modules loaded with dlopen() do, which is why rtapi_math must be made visible to that process

here lies the stupidity of this default "as-needed" ld behavior: ld thinks it does a valuable job by optimizing away the use of rtapi_math because it is not referenced by any code in rtapi_app proper, when in fact it cannot tell link-time what is going to be required with modules coming in through dlopen()

and that is such a colossal blunder at a foundation level that I am dead sure this feature will affect other projects as well

now, on bringing in rtapi_math into rtapi_app:

that can either be done by linking into rtapi_app_<flavor> OR by linking it into the rtapi.so module, which is the first thing loaded

the easiest way to assure rtapi_app gets rtapi_math linked in as a shared lib is to add a dumy function in rtapi_app.cc which references a function exported by rtapi_math, and assure the optimizer gets no chance at eliminating this dead code - that can be verified by inspecting the build artefacts (.o) with nm

re your question. "Do You think above is correct for dynamic linking?": that line is a dependency rule. It will not cause linking, it forces building of prerequisite objects. The [actual linking happens here](https://github.com/machinekit/machinekit/blob/master/src/rtapi/Submakefile#L173-L175).

If you want to trace the build, try the 'V=1' and 'VV=1' make flags.

koppi, 2015-06-14:
> and that is such a colossal blunder at a foundation level that I am dead sure this feature will affect other projects as well
@mhaberler see https://lists.debian.org/debian-devel-announce/2011/02/msg00011.html

mhaberler, 2015-06-14:
@koppi - well over at debian they had their brains turned on:

````
Fix build failures with ld --as-needed
- --------------------------------------

[Note this is not turned on by default]                <------------
``````

koppi, 2015-06-14:
@mhaberler my guess is, that in Debian, ```ld --as-needed``` will become the default, once, this is fixed in all packages: https://bugs.debian.org/cgi-bin/pkgreport.cgi?tag=ld-as-needed;users=debian-gcc@lists.debian.org

rosaldorosa, 2015-06-15:
Thanks mhaberler
You advise was correct. I've added one of rtapi_math function to rtapi_app and it works
Very strange is that ldd for librtapi_math  returns "statically linked"
But ldd rtapi_app_xenomai returns librtapi_math.so.0 => machinekit/lib/librtapi_math.so.0 (0x00007f5c5764a000)
Would it be simpler to explicitly add -rdynamic -l rtapi_math into  tp.so compilation/linkage process and any other library using rtapi_math ?

mhaberler, 2015-06-15:
@rosaldorosa - re 'strange statically linked' - does not look strange to me
rtapi_math _is_ a shared library, but  doesnt reference _other shared libraries_, so that makes sense
whereas rtapi_app_* _does_ reference the rtapi_math shared library

"re simpler" - no, to the contrary - good recipe to create havoc: there is no reason to assume tp is the only module to reference math symbols

referencing rtapi_math from rtapi_app_* guarantees the rtapi_math symbols are exported to ALL modules

linking rtapi_math into tp.o only guarantees tp.o will see the rtapi_math symbols - so what's the point?

rosaldorosa, 2015-06-15:
Shared library is loaded only when needed. Your solution loads 
rtapi_math permanently making no difference ot static linkage
If You compile tp.so with g++ -shared ... -lrtapi_math the rtapi_math 
will be on dependency list and called when needed
This is the standard purpose of using shared libraries

rosaldorosa, 2015-06-18:
Do You know how to modify the order of compilation such that instead of machinetalk rtapi_math is first
It does not make sense to compile rtapi_math at the end. This library has no other lib dependecies so should be compiled first. 
I think John Morris is expert on this

rosaldorosa, 2015-06-18:
I've found it in Makefile  SRCDIR machinetalk should be moved down.
Now adding rtapi_math during compilation don't require no-as-needed flag anymore
PS this flag (as-need) is a huge step forward in code size and efficiency.
So debian people should get used to it

rosaldorosa, 2015-06-18:
During compilation this huge number of warnings should be cleaned.

rosaldorosa, 2015-06-18:
Can I push the changeset to Your review?

mhaberler, 2015-06-18:
@rosaldorosa - "modify the order of compilation" .."first" - first what? built - then a dependency; linked - a linking order

re 'huge number of warnings' - please give an example, and argue why this is relevant

re 'huge step forward in code size and efficiency' - please show evidence - are you sure you understand what shared libraries and code size have to do with each other?

re push - sure: http://www.machinekit.io/docs/contributing/

zultron, 2015-06-20:
This stuff is all related to #670 and the issues to which it links.

I'd like someone consider the possibility that the userland comp.so modules that use math functions should be linked to `rtapi_math.so`, and `rtapi_app_*` not be.

The only benefit I see to linking against only `rtapi_app_*` is avoiding some headache to figure out the best way to modify the build system, the problem in #336.

mhaberler, 2015-12-16:
seems stamina kind of evaporated... so I filed a PR

PR #832 (2015-12-16): rtapi/build: link rtapi_app_* with -Wl,--no-as-needed
fixes #677 on Ubuntu (at least 14.04)
commit 19ec016e (2015-12-16): rtapi/build: link rtapi_app_* with -Wl,--no-as-needed

fixes #677 on Ubuntu (at least 14.04)
closed 2015-12-16 by luminize