machinekit/machinekit #687: external synchronisation of HAL threads not possible

mhaberler, 2015-06-16:
this thread is about collecting clue on externally clocking HAL threads.

There is huge upside in having support this, as it would allow hardware to sample positions in next-to-exact time and then trigger the servo thread to go about its job, which in essence takes out host thread jitter completely as far as position sampling noise goes.

The requirement for RT thread now changes from "thread must commence execution at an exact point in time" to "thread must complete function chain within servo cycle", which is a major relaxation.

Note this is what the external phase locked loop scheme tries to achieve, but potentially better and easier (no PLL catch/locking issues): external hardware (Mesanet firmware, PRU..) could run a "base thread" and interrupt the host at a submultiple of the "hardware base rate". 

To see what the PLL scheme is already able to achieve, I steal two screenshots by Peter Wallace:

feedback noise (top signal) without PLL:

[image]

feedback noise (top signal) WITH PLL:

[image]

related topics: 

[synchronizing HAL threads to an external clock](https://groups.google.com/forum/#!topic/machinekit/drU1p_GNZ3E)

[RTDM GPIO driver to support external clocking on ARM/Xenomai](https://github.com/mhaberler/gpio-irq-rtdm)

[HAL triggers branch to eventually support external clocking ](https://github.com/mhaberler/machinekit/tree/hal-triggers)

related links and threads on the Xenomai list about RTDM-native which (ideally) would enable this scheme also for RT-PREEMPT and possibly vanilla kernels:

http://www.osadl.org/RTDM-native.howto-rtdm-native.0.html
https://git.xenomai.org/rtdm-native.git/
https://www.mail-archive.com/xenomai@xenomai.org/msg09182.html
https://www.mail-archive.com/xenomai@xenomai.org/msg09162.html
https://www.mail-archive.com/xenomai@xenomai.org/msg09117.html

RunningLight, 2015-06-16:
This is a very exciting evolutionary. I shy away from marketing phrases like "game changer" but synchronizing threads to external clocks has that feel to it. What a bonus if it could be made to work even with vanilla kernels.

+1

mhaberler, 2015-06-18:
I got a Mesanet 5i25 to generate clock interrupts, this should be sufficient to write a matching kernel driver to make Mesanet cards trigger HAL threads:

[Generating clock interrupts from Mesanet cards](https://github.com/mhaberler/asciidoc-sandbox/wiki/Generating-clock-interrupts-from-Mesanet-cards)

mhaberler, 2015-06-18:
since this starts to work, I'm pondering how to make this work with machinekit from a build perspective; I see two options (for now for userland threads only, RTAI can come later):

- a driver which is completely separate from hostmot2/hm2_pci and ONLY does IRQ posting - simple, standalone, usable outside of machinekit, but replicates code from al/drivers/mesa-hostmot2
- an integrated variant of the hm2_pci.c driver which takes a param like 'clocking=1'

NB hm2_pci is a _HAL component_, but the clock RTDM driver is not

In the second variant, the driver would not do anything at the pin fiddling level, but restricted to ONLY handling the IRQ; it would he harder to use outside machinekit, and have no duplicated code

not sure how to play this

mhaberler, 2015-06-22:
@automata posted this, pasted here because that's where we'll find it again ;):

For RTAI PCI/PCIe interrupts, we can look at the following site
http://pmatias.me/2015-05/rtai-measuring-irq-latency/

more details on PCIe IRQ source on:
https://github.com/thotypous/rtai-irq-latency/blob/master/PCIe_SW/rtai_pcie_interrupt_driver.c

However, as Michael said, if external interrupt works, the kernel thread scheduling latency will become irrelevant. (Interrupt servicing latency will become the more relevant number)

-automata

mhaberler, 2015-06-22:
I got RTDM-native compiled and the rtdmtest program to pass on 3.18.13-rt10: http://permalink.gmane.org/gmane.linux.real-time.xenomai.users/22854

next summit: getting this to work with hardware we care about; for instance, GPIO and parport

maybe I should try the gpio-rtdm-irq test driver using RTDM-native and a vanilla kernel on the BB, so we get some timing data

mhaberler, 2015-06-26:
besides interrupt handling I could imagine the following simpler scheme for syncing an RT thread to an external clock source:

the RT thread has an inner loop like so:
````
while (not exited) 
    sample clock source (eg GPIO/parport/PWM pin/IRQ request register)
    if not edge detected on clock source:
       continue
    run thread function chain

    wait dT (until just before the next edge is expected)
````

dT would be chosen such that it would be slightly larger than the maximum expected jitter, so on the order of 10-100uS.

This would sync the thread function chain to the clock edge, without needing a kernel driver but at the expense of busy-looping on the clock pin for a few percent of a typical servo cycle

assuming a 1mS servo cycle -
at dT=10uS up to 1% of the CPU cycles would be lost polling 
at dT=100uS up to 10% of the CPU cycles would be lost polling 

so that scheme would lend itself naturally to a multicore platform where on core could be dedicated to an RT thread

yishinli, 2015-06-28:
Assume we are going to run MKIT on RPi2, which is with limited expansion options in comparing with PCI or GPMC of BBB. Its has SPI interface to exchange data with external devices, say FPGA. Consider the following approach:
[image]
In this approach, the RPi2 has to fetch all the inputs from FPGA, do RT tasks, and output to FPGA , in one servo period. Assume we have the following registers to be exchanged from RPi2 and FPGA:
* Input Registers (total: 1536 bits)
  1. encoder: 8 * 32bit
  2. probed pos: 8 * 32bit
  3. index pos: 8 * 32bit
  4. analog input: 8 * 16bit
  5. digital input: 128 * 1bit
  6. misc: 512bit
* Output Registers (total: 256 bits)
  1. PWM/PULSE/DIR cmd: 8*16bit
  2. analog output: 4 * 16bit
  3. digital output: 64 * 1 bit
* With 10MHz spi data rate, it takes 179.2us for transceiving all the registers

We've implemented this on BBB and learned that the system performance is not good. The major reasons are we set the servo period as 655.36us, and the SPI interface is not at DMA mode. We observed 80% of CPU load for rtapi:0 (mainly SPI data exchanging).

With RPi2, we can offload the SPI task to a dedicate CPU core:
[image]
We observed the CPU load of rtapi:0 lower than 20%. However, since the SPI communication is running as a user space process, its latency is unstable. Therefore, it can not cache the hardware servo clock tick. I assume there are better implementation to resolve this issue.

Third approach ... to be continued ....

mhaberler, 2015-06-29:
@yishinli - yeah, that usermode scheduling is not the right thing to do for SPI I/O in the servo loop path

what would be required is a RTDM SPI driver (assuming Xenomai), or do the SPI job in an RT thread in a polling fashion, torching off a core for that job by binding that thread to a core

I saw one for the BB but never got the author to respond:  https://github.com/netico-solutions/xdrv-spi
I did not try it yet

mhaberler, 2015-06-29:
I discussed [the idea mentioned above](https://github.com/machinekit/machinekit/issues/687#issuecomment-115635345) with Peter Wallace, and it looks entirely feasible with Mesanet cards, this would work with just userland programming and without an IRQ/kernel driver and assorted kernel API vagaries:

`````
mhaberler: Q- besides the IRQ handler, I was thinking about a semi-polled solution
[5:09pm] mhaberler: can I just poll the IRQ register to detect the clock edge?
[5:09pm] mhaberler: and clear it once I have seen it go hi?
[5:10pm] mhaberler: or would you suggest some other way of polling the clock signal, like a pin value?
[5:11pm] pcw_home: you can read the timer register and wait for it to roll over
[5:12pm] mhaberler: and that would be which one? let pull the regmap
[5:15pm] pcw_home: Actually is called the DPLLPhaseError register
[5:15pm] mhaberler: ah 0x7100
[5:16pm] pcw_home: counts to 2^32 every period (not by ones though)
[5:17pm] mhaberler: so a write clears it? "writes load the accumulator"
[5:17pm] pcw_home: writes load it
[5:18pm] pcw_home: so for example when using the DPLL feature, you load it the first cycle to avoid a long lock-in time
[5:19pm] mhaberler: I see
[5:19pm] mhaberler: I guess I could view it as a free running ctr with DPLL base rate = (clocklow/2^DDSSize*DPLLPreScale)*BaseRateRegisterValue
[5:19pm] mhaberler: ?
[5:19pm] pcw_home: Yes
[5:20pm] mhaberler: so in that case all I need is to calculate the count at which to release the thread
[5:20pm] mhaberler: and when it's reached I write it with a zero
[5:21pm] pcw_home: free running as long as you do not read 0x7600 (this will correct the frequency based on the current phase error)
[5:21pm] mhaberler: I will abstain
[5:22pm] pcw_home: that is, read of 0x7600 are equivalent to reads of 0x7100 except they latch the error and use it
[5:23pm] mhaberler: so, last hurdle: figure the DPLL base rate
[5:23pm] mhaberler: ddssuze=0x7300 bits0..7
[5:24pm] mhaberler: DPLLPreScale?
[5:26pm] mhaberler: BaseRateRegisterValue == read 0x7000 I guess
[5:27pm] mhaberler: and 0x428	ClockLow	In Hz
[5:29pm] pcw_home: you could look at the hm2 driver part that supports the DPLL
[5:29pm] mhaberler: hold on
[5:31pm] mhaberler: ok, I guess hal/drivers/mesanet-hostmot2/dpll.c
[5:31pm] pcw_home: the prescale is chosen so the base rate is as big as possible but less than 2^31
[5:32pm] mhaberler: like here? https://github.com/machinekit/machinekit/blob/master/src/hal/drivers/mesa-hostmot2/dpll.c#L165-L173
[5:33pm] pcw_home: both for best frequency setting resolution and so that the DPLL gain is within range
[5:33pm] pcw_home: Yes thats it
[5:33pm] mhaberler: ah, ok, so within hostmot2 I have all I need then
[5:34pm] mhaberler: ah I see, DPLLPreScale is set by host side for best resolution based on thread rate and clock as read
[5:34pm] pcw_home: yeah, if you dont want to use the DPLL as a DPLL but as a master clock, you just don't read 0x7600 and then somehow
[5:35pm] pcw_home: use the accumulator as a timing reference (Either IRQs or whatever)
[5:36pm] pcw_home: theres a bug in that code also... if you set the base frequency too low (and hence run out of prescale range) the code does not notice
[5:36pm] mhaberler: ah ok
[5:36pm] pcw_home: (only happens around 100Hz so normally not an issue)
[5:36pm] mhaberler: well worst case I'd have to optionally disable using DPLL if detected
[5:37pm] mhaberler: and if not detected, complain it aint in the firmware?
[5:39pm] pcw_home: its detected like all hm2 devices (module tag in the IDROM) so if its not there, there will likely be a hal pin thats missing
[5:40pm] pcw_home: say base frequency KHz
[5:41pm] mhaberler: I guess this is the test: https://github.com/machinekit/machinekit/blob/master/src/hal/drivers/mesa-hostmot2/dpll.c#L52-L55
[5:41pm] pcw_home: will you have a generic "use hardware clock" option?
[5:43pm] mhaberler: generic for hm2 I would guess, and a matching option for the HAL thread
[5:43pm] mhaberler: do you mean "generic" to subsume DPLL as well as just using the card as external clock?
[5:45pm] mhaberler: or just for the external clocking scheme I'm cooking on?
[5:45pm] pcw_home: well it seems that having an external clock source means some upper level changes as well
[5:45pm] mhaberler: well definitely, the thread execution will be a tad different
[5:46pm] mhaberler: but that part I'm not much concerned about, it is more about how to integrate different external clock sources in a non-intrusive and extensible way
[5:49pm] mhaberler: it's this : https://github.com/machinekit/machinekit/blob/master/src/hal/lib/hal_thread.c#L17-L82
[5:50pm] mhaberler: what changes is - the rtapi_wait() in line 80 will be different
[5:51pm] mhaberler: I guess what I'll do to avoid polling all the rest of the servo time slot is to wait until just before the release point, say 50 or 100uS, then start polling
[5:51pm] mhaberler: so I waste 5-10% of a core max
[5:52pm] mhaberler: however, the drivers need to export a 'wait for clock edge' function which can be called from this point, and that is the API design job, but with the vtable feature this is now pretty easy:
[5:53pm] mhaberler: vtables avoid load-time symbol resolution for inter-module API's
[5:53pm] mhaberler: https://github.com/machinekit/machinekit/tree/master/src/hal/vtable-example
[5:53pm] pcw_home: right, a decent tradoff to remove jitter, though if positions are latched (as they can be by the timer or PRU)
[... cut ...]
[5:56pm] mhaberler: (I cannot just call an hm2_yadayada function from hal_threads.c because it is not a given a hm2 will be the timing source, but the vtable fixes that)
[5:56pm] mhaberler: a hm2 would not even be loaded say on a bb, and then loading the hal lib would fail because hm2_yadayada  would not resolve
[5:57pm] mhaberler: as usual just an extra indirection step
[5:59pm] mhaberler: actually the pre-release time window when to start polling can be auto-calibrated, by letting it run and observe if a servo window was missed due to jitter; start at zero window size, then increase until jitter reliably covered
````

luminize, 2015-07-07:
I got my workstation upgraded to Jessie (no xenomai kernel). It has an mesa 5i20 card. If I can do some testing let me know.

luminize, 2015-07-07:
Sorry, wrong button :(

mhaberler, 2015-12-19:
It just occurred to me (while commenting on #839) the PRU could be used as a timing source as well, which is a compromise solution

it could look like so:

- assume a servo system with encoders
- the noise will come from jitter in position sampling as outlined above
- now if a PRU can be coerced to a) read the encoder, or do the A/B decoding b) do so at a lower jitter than a host thread c) signal the host thread to run once a position sample is available, this is probably better than what we have now (full host jitter in pos sample) but not quite as good as external sampling and syning

the-snowwhite, 2016-05-17:
These are my findings on enabling a uio_irq with the generic unmodified version of the uio_pdrv_genirq driver built into the 4.1-ltsi-rt kernel  on the mksocfpga platform

https://github.com/the-snowwhite/mksocfpga/blob/master/docs/IRQ-mod.md

I will now create a do not merge before testing commit onto master.

mhaberler, 2016-05-20:
just outlining how this will be used:

so far, we used a periodic thread (like `newthread servo 1000000 fp`) and chained all RT component functions onto that

to use externally synced threads with the timer interrupt:

- we'll use a non-periodic thread which doesnt wait for a period to expire
- hm2_soc_ol.c gets the setup functions to enable timers (poking values into the FPGA space)
- thet setup code actually should test if the IRQ's are actually delivered and verify the timing
- I'll add a HAL function to hm2_soc_ol.c, tentatively called `waitirq` which basically does what uio_test.c does - and acknowledge every IRQ immediately after returning from the read

so in summary this will look like so:
`````
loadrt hostmot2
loadrt hm2_soc_ol <args>  # this step exports the 'waitirq' function
newthread servo 1000000 nowait fp # the period is meaningless if 'nowait' is given
addf waitirq servo  # blocks until next IRQ from firmware
addf <otherfunct> servo
...
start
````

dkhughes, 2016-05-23:
@mhaberler I'm testing out your commits that just went through. The value in timer=XXXX is specific to the clock frequency in the hardware design correct?

mhaberler, 2016-05-23:
yes, that timer value came straight from @the-snowwhite 's notes - an expression was not at hand

dkhughes, 2016-05-23:
@mhaberler Okay, I'll find out how it's calculated and submit a patch that is based off a desired period instead of the magic number.

I must have messed something up when I recompiled with your latest patch set, I'm getting usermod timeout errors now. Gotta check my configs...

dkhughes, 2016-05-24:
I havent been able to get the waitirq stuff working on my hardware yet. Here is my progress:

- I did track down the magic number - The top 8 bits are the prescale used by the dpll, the bottom 24 bits are bounding limits that the dpll phase correction must work inside. 0x0C (the top 8 bits of 211812352) sets a prescalar value of 12 (kind of funky divider for a 1kHz signal). This prescale defaults to 1. The four timers are only 16 bit, so I imagine this value should be set so that we don't overflow in the count period of a single interrupt - which means we calculate this from the reported reference clock feeding the dpll.

- It looks like base rate is also an input the dpll requires? When I read from the device after programming I get 0 for the base rate unlike @mhaberler 's example included with the waitirq commits. I think we are supposed to set this value as it controls the reference accumulator, but I might be missing something. Here is why base rate is important:
[https://github.com/machinekit/mksocfpga/blob/master/HW/hm2/hmtimers.vhd#L150](https://github.com/machinekit/mksocfpga/blob/master/HW/hm2/hmtimers.vhd#L150)
Are you manually setting your base rate somewhere that isn't in the irq test hal file? `setp`, perhaps?

dkhughes, 2016-05-24:
I pushed the DPLL setup from the hm2_soc_ol driver up to the hostmot2 register function. That way we have access to the dpll functions to set them up automagically and correctly based on the clock used by the dpll module in firmware. Now the timer1 value in the hal file is simply the period in nanoseconds that you want the dpll to run at. That is working nicely, but I need to add a read block at address 7600 to prevent the dpll capturing phase error and changing our (assumed) reference clock and causing drift. Here is the dpll all set up this way:

```
  Comp   Inst Type  Dir         Value  Name                             Epsilon         Flags
    75        float IN            100  hm2_jdcb.0.dpll.01.timer-us   	0.000010	0
    75        float IN            100  hm2_jdcb.0.dpll.02.timer-us   	0.000010	0
    75        float IN            100  hm2_jdcb.0.dpll.03.timer-us   	0.000010	0
    75        float IN            100  hm2_jdcb.0.dpll.04.timer-us   	0.000010	0
    75        float IN              1  hm2_jdcb.0.dpll.base-freq-khz 	0.000010	0
    75        u32   OUT    0x0000002A  hm2_jdcb.0.dpll.ddsize        			0
    75        float OUT             0  hm2_jdcb.0.dpll.phase-error-us	0.000010	0
    75        u32   IN     0x00400000  hm2_jdcb.0.dpll.plimit        			0
    75        u32   OUT    0x00000018  hm2_jdcb.0.dpll.prescale      			0
    75        u32   IN     0x0000A000  hm2_jdcb.0.dpll.time-const    			0
```
Unfortunately, still no joy on the IRQ front. Does this show func output look sane?
```
Exported Functions:
  Comp   Inst CodeAddr  Arg       FP   Users Type    Name
    66        b6705d31  00000000  NO       0 user    delinst
    75        b56264e1  0003cca8  NO       1 thread  hm2_jdcb.0.pet_watchdog
    75        b560fc49  0003cca8  YES      1 thread  hm2_jdcb.0.read
    75        b560fd79  0003cca8  YES      1 thread  hm2_jdcb.0.read_gpio
    75        b560fcd5  0003cca8  YES      1 thread  hm2_jdcb.0.write
    75        b560fda5  0003cca8  YES      1 thread  hm2_jdcb.0.write_gpio
    66        b6705c0d  00000000  NO       0 user    newinst
    75        b54ecc45  b54ff3e8  NO       1 xthread waitirq.0
```

mhaberler, 2016-05-24:
at first look - yes

this is the relevant part of the irqtest.hal test:

`````
machinekit@mksocfpga:~$ halrun -I machinekit/configs/hm2-soc-stepper/irqtest.hal
msgd:0 stopped
rtapi:0 stopped
halcmd: show pin hm2.0.
Component Pins:
  Comp   Inst Type  Dir         Value  Name                             Epsilon         Flags
    75        u32   OUT    0x00001BCE  hm2.0.irq-count               			0
    75        u32   OUT    0x00000001  hm2.0.irq-missed              			0
    75        u32   OUT    0x00000000  hm2.0.read-errors             			0
    75        u32   OUT    0x00000000  hm2.0.write-errors            			0

halcmd: show pin hm2.0.
Component Pins:
  Comp   Inst Type  Dir         Value  Name                             Epsilon         Flags
    75        u32   OUT    0x00002059  hm2.0.irq-count               			0
    75        u32   OUT    0x00000001  hm2.0.irq-missed              			0
    75        u32   OUT    0x00000000  hm2.0.read-errors             			0
    75        u32   OUT    0x00000000  hm2.0.write-errors            			0

halcmd:
`````
note increasing hm2.0.irq-count 

this is straight from the [image](http://deb.mah.priv.at/uploads/de0-nano/debian-8.4-machinekit-de0-armhf-2016-05-24/) which has current mk/mk master 

your dpll seems to work - it locks on the base thread (1 kHz)
what is strange is the zero phase error but then I dont see your hal setup - I get for irqtest.hal:

`````
halcmd: show pin hm2_5i25.0.dpll.phase-error-us
Component Pins:
  Comp   Inst Type  Dir         Value  Name                             Epsilon         Flags
    75        float OUT     -479.6491  hm2_5i25.0.dpll.phase-error-us	0.000010	0

halcmd: show pin hm2_5i25.0.dpll.phase-error-us
Component Pins:
  Comp   Inst Type  Dir         Value  Name                             Epsilon         Flags
    75        float OUT     -472.1971  hm2_5i25.0.dpll.phase-error-us	0.000010	0
`````
notice it's moving in the jitter range which makes sense

I would think it is zero because the hm2_jdcb.0.read function isnt run (I _think_ that is the one the dpll tracks)

mhaberler, 2016-05-24:
that said - let's think about this dpll thing  a bit

the purpose why the dpll exists is because linuxCNC has host-timing only

the dpll locks onto the host read function and reads the position values based on that 'proper' point in time, i.e. just a bit before the read function retrieves it

now for external syncing - what we are up to - the dpll does not make sense because the firmware is the timing master, no point in catching its own tail

A fixed sequence of cylic 'read pos values' - interrupt - present values is just fine

it'd be great if the driver could determine how old the sample is, i.e. the time delay from sampling to read - maybe reading a timer reg can tell that; that would enable very good interpolation on the host side

but I admit I'm not deep enough into the idea behind it to fully make sense of all this

dkhughes, 2016-05-24:
Right, we can block the DPLL from tracking the read function though, by not allowing a read to the 7600 register when IRQ is activated. This creates a free running reference timer that we want, and probably cause the phase error to go to zero since we never latch a read time. I moved the names of the irq pins in hm2_soc_ol to reflect the board they are from, maybe that broke something?

dkhughes, 2016-05-24:
Hmmm.....Is it not getting kicked the first time?

```
halcmd: show pin hm2_jdcb.0.irq.
Component Pins:
  Comp   Inst Type  Dir         Value  Name                             Epsilon         Flags
    75        u32   OUT    0x00000002  hm2_jdcb.0.irq.count          			0
    75        u32   OUT    0x00000000  hm2_jdcb.0.irq.missed         			0
```
If I punch it:
```
machinekit@mksocfpga:~$ sudo ./mksocmemio -w a00 2
    mksocfpgamemio: read write hm2 memory locatons by cmmandline arguments  
    /dev/uio0 opened fine OK 
    region mmap'ed  OK 
    mem pointer created OK
Program name: ./mksocmemio    input option = w 
read Write read  
Address 2560 	value = 3 
Address 2560 	former val = 0x00000003 	 wrote: --> 0x00000002 	 read: = 0x00000002 
```
The interrupt shows up:
```
halcmd: show pin hm2_jdcb.0.irq.
Component Pins:
  Comp   Inst Type  Dir         Value  Name                             Epsilon         Flags
    75        u32   OUT    0x00000002  hm2_jdcb.0.irq.count          			0
    75        u32   OUT    0x00000000  hm2_jdcb.0.irq.missed         			0
```

mhaberler, 2016-05-24:
pins: hard for me to tell - I claim MBA ignorance rights ;)

disabling the DPLL tracking if IRQ's are enabled makes sense to me

do you see a way how the read function can determine the elapsed time since IRQ raised? 

I mean all this would be really cool to have in theory - it just isnt used anywhere right now; but the extended thread API can [pass in much more information to a component funct](https://github.com/machinekit/machinekit/blob/master/src/hal/lib/hal_priv.h#L380-L398) than the legacy 'period' (which is thread setup-time constant); plus, this struct is extensible without breaking binary compatibility

but then nothing should keep us from doing it right this time around ;)

I guess we'll have to make http://freeby.mesanet.com/regmap a mk project document, git and all

dkhughes, 2016-05-24:
WIP for the additions I've been making is here:
[https://github.com/JDSquared/machinekit](https://github.com/JDSquared/machinekit)
I will get this figured out tonight and be caught up with you on the altera side.

> do you see a way how the read function can determine the elapsed time since IRQ raised?
The read function can query the timer values. Since we know the parameters of the timer we can calculate the elapsed time.

> I guess we'll have to make http://freeby.mesanet.com/regmap a mk project document, git and all
Yep, I will include the updated regmap when I can get some time to start putting the constants back in the firmware. I think that belongs in mksocfpga.

cdsteinkuehler, 2016-05-24:
On 5/24/2016 4:31 PM, Michael Haberler wrote:
> that said - let's think about this dpll thing a bit
> 
> the purpose why the dpll exists is because linuxCNC has host-timing only

Exactly.

Since you've added IRQ support to the driver and HAL, what you really
want (and what the hostmot2 code does even without the DPLL) is:

* hostmot2 samples current position values and generates an IRQ

* HAL driver wakes up and reads position values which were sampled at
exactly the right time, even though some arbitrary (and variable)
amount of time has now passed.

* HAL driver updates commanded position/velocity/whatever

The DPLL is a hack to do the above while tracking the software timing
(which will be close to but won't exactly match the hardware clocks)
so you don't do something silly like gain or loose a uS each servo
period, eventually getting dramatically out-of-sync between what the
hardware and software think the "current time" is.

-- 
Charles Steinkuehler
charles@steinkuehler.net

mhaberler, 2016-05-24:
@cdsteinkuehler so far understood (hey, that's straight from #687 ;)

the one thing I have never been able to answer for myself is: does the point in them when the HAL thread  gets around to update commanded values to the firmware have ANY impact on results?  it might be an academic question, and is really mere curiosity on my behalf

my working assumption is - once the HAL thread updates commanding values (pos, vel..), they take effect immediately as I view the firmware as a very fast running thread(s) controlled via shared memory

background of the question - there is a primary source of jitter and that is the event chain from sampling to host thread reaction (the one we are about to take out)

 there _might_ be a secondary source of jitter based on the time when new target values are applied 

if this is so, that secondary jitter might be eliminated by latching values at the end of the period, just before sampling the new values

not that it matters a lot though, and once we have proper observing pins that impact could actually be measured

dkhughes, 2016-05-24:
@mhaberler If the period value is not latched and passed to every component, but is instead recalculated before every function call, then yes, I would think components such as differentiation, controllers, etc. could theoretically suffer from invalid timestamps. That being said, if you drive your time period low enough, you can probably minimize these effects.

What I know from writing controllers for autonomous vehicles is that I want determinism. Jitter is what will kill you as noise in a controller is detrimental since I can't have my gains as stiff as I would like (the output tracks the noise which causes chatter). 

You mentioned having smarter thread functionality. What I'm hoping this dpll-as-a-free-running-timer offers us is the ability to latch data at a known point in time, then we latch the timestamp and use that delta t as the period being passed into the HAL functions. Maybe that will eliminate the joint following errors I'm seeing when hm2 stepgens are being used. But, it should also allow all of the timing relaxation you mention above. All of a sudden, we don't care what the processor clock says, the sensory input and calculated output are based on a clock that is accurate to ppm if the board is any good. Now, just make your processing happen in less than one cycle. C'est tres cool!

cdsteinkuehler, 2016-05-24:
On 5/24/2016 5:17 PM, Michael Haberler wrote:
> @cdsteinkuehler <https://github.com/cdsteinkuehler> so far understood (hey,
> that's straight from #687 <https://github.com/machinekit/machinekit/issues/687> ;)
>
> the one thing I have never been able [...]

Yes!

> my working assumption is - once the HAL thread updates commanding values (pos, 
> vel..), they take effect immediately as I view the firmware as a very fast 
> running thread(s) controlled via shared memory

Yep.

> background of the question - there is a primary source of jitter and that is the
> event chain from sampling to host thread reaction (the one we are about to take out)
>
> there /might/ be a secondary [...]

You will reduce the impact of thread jitter by queuing the
calculations and applying the updated commanded values at the start of
the next period, but that will introduce a delay in the feedback loop
that the controller will need to account for.  Ultimately, this will
approximately halve the effective servo thread period.

Alternately, you could calculate updated values based on the current
time (so amplifying any changes since it has less than a full servo
period time to be effective), which is likely an easier solution.  As
long as the absolute IRQ latency is substantially less than the servo
period, this should work *VERY* well.

-- 
Charles Steinkuehler
charles@steinkuehler.net

dkhughes, 2016-05-25:
> You will reduce the impact of thread jitter by queuing the
calculations and applying the updated commanded values at the start of
the next period, but that will introduce a delay in the feedback loop
that the controller will need to account for.  Ultimately, this will
approximately halve the effective servo thread period.

Shouldn't this effectively add 1 period worth of phase lag? The rate wouldn't be affected unless you update one period, calculate the next, update, calculate. The goal would still be to update and calculate, just apply the values at a known time, correct? Shoot, if your period is fast enough compared to the dynamics of your system you probably won't even notice the additional lag. I like this update then calculate approach. We always know when the control data hits...

cdsteinkuehler, 2016-05-25:
On 5/24/2016 7:29 PM, dkhughes wrote:
> You will reduce the impact of thread jitter by queuing the
>     calculations and applying the updated commanded values at the start of
>     the next period, but that will introduce a delay in the [...]

Yes, but in general with control theory the earlier you apply your
correction the better everything works.  If you want to intentionally
delay updates to the commanded position/velocity registers, you really
need to account for this in the SW that's generating the
position/velocity targets.

...but this is a gross over-simplification (as is my previous post),
so feel free to do as you'd like and deal with the consequences.  As
you mention, for many configurations it really won't make much
difference (since most are not running close to the ragged edge).

-- 
Charles Steinkuehler
charles@steinkuehler.net

dkhughes, 2016-05-25:
@cdsteinkuehler Ha, yep, thats why I mentioned determinism being the most important thing, you can compensate if need be, although I guess eventually we will hit a limit with the hardware we have at hand.

@mhaberler I found the hang, it's stuck at the read function waiting for an interrupt. Not amazing news, but now at least I know where to look - IRQ config probably.

dkhughes, 2016-05-25:
Okay, got it:
```
halcmd: show pin hm2_jdcb.0.irq.
Component Pins:
  Comp   Inst Type  Dir         Value  Name                             Epsilon         Flags
    75        u32   OUT    0x00002B1B  hm2_jdcb.0.irq.count          			0
    75        u32   OUT    0x00000000  hm2_jdcb.0.irq.missed         			0

halcmd: show pin hm2_jdcb.0.irq.
Component Pins:
  Comp   Inst Type  Dir         Value  Name                             Epsilon         Flags
    75        u32   OUT    0x00003C4B  hm2_jdcb.0.irq.count          			0
    75        u32   OUT    0x00000000  hm2_jdcb.0.irq.missed         			0
```
@mhaberler Your code waitirq code was 100% correct. The problem was interrupt polarity in the bitfile. Turns out, Xilinx interrupt controller needs an inverter from the hostmot2 interrupt signal. Go figure. I will submit pull requests with the changes eliminating the magic number....right after I test the latency with my real machine config :)

dkhughes, 2016-05-25:
I've instrumented my config with some minmax and latency components. The externally synchronized thread has removed the occasional "large" latency that I had been seeing with hal driven thread timing. Promising! 

The following error has suddenly become a very pretty trapezoid: a trapzeoid that is lagging the commanded position by exactly 1 servo period. Hmmmmmm....maybe this is the default dpll timer1 value of 100us causing the feedback position from a stepgen to lag because it is latched after we are reading. I'm not sure why this delay was designed into the drivers as defaults but I will investigate.

Oh, is there a way to make motmod not generate threads, or make servo-thread a nowait thread or should I patch in a way based on a config value like servo_thread_nowait=1 or something?

dkhughes, 2016-05-25:
Turned out to be a bit of a HAL gotcha that is more obvious to me now. In the hal file, the order of the functions called is first listed first served isn't it? I simply swapped the order of the addfs in my hal file from the references I had been using (an old config from one of the generators):
```
#This was bad!!
addf waitirq.0                             servo
addf motion.command-handler  servo
addf motion.controller                sevo
addf hm2.X.read                        servo
addf hm2.X.write                        servo
```
to...
```
#This is much better!!
addf waitirq.0 servo
addf hm2.X.read servo
addf motion.command-handler  servo
addf motion.controller servo
...
... other stuff that might effect hm2 state ...
...
addf hm2.X.write servo
```
and boom, no more joint following errors using the external time reference.

mhaberler, 2016-05-25:
@dkhughes yes, functs are executed in addf order (if you do not give it a position argument)

this is [where functs are linked to the thread](https://github.com/machinekit/machinekit/blob/master/src/hal/lib/hal_funct.c#L302-L344)
this is [where funct execution happens](https://github.com/machinekit/machinekit/blob/master/src/hal/lib/hal_thread.c#L17-L87)

mhaberler, 2016-05-25:
that sounds very promising indeed!

I'd love to have scoping pins we could wiggle in the kernel irq and in the thread executor
I did [something similar back then here ](https://github.com/mhaberler/gpio-irq-rtdm/blob/master/gpio_irq_rtdm.c#L58-L61) and it permitted [scoping the event chain](http://static.mah.priv.at/public/rtdm-chain.png): IRQ happens / kernel IRQ handler executes / userland executes

dkhughes, 2016-05-26:
@mhaberler Can you explain this check for me? What is the gotcha here?
[https://github.com/machinekit/machinekit/commit/61990da43611dfefd4f0874f03de30771d7ba27e#diff-dd71f67eb12dba345de304499fbfd5e6R4107](https://github.com/machinekit/machinekit/commit/61990da43611dfefd4f0874f03de30771d7ba27e#diff-dd71f67eb12dba345de304499fbfd5e6R4107)

As I've started moving towards gui integration I've noticed we are missing interrupts, and there is a very long delay before components start processing functions - around 2-3 secs at startup. When the py threads start running my jitter goes very poor, around 70us compared to 30us without external syncing.

mhaberler, 2016-05-26:
you might want to pull and rebase - this is not fatal any more

this used to happen: somebody has a single core cpu, specifies nowait and bang, your RT thread takes over the platform - solid freeze ;)

I did this for further mitigation: https://github.com/machinekit/machinekit/commit/cf7c8e3f30c45d1c6bcc20b02d1e178c3db802cc 

so if hal threads are stopped, the RT thread would still execute rtapi_wait() to leave some breathing time for non-RT threads

dkhughes, 2016-05-26:
Okay, that explanation makes sense. I'm pretty sure have the fix from [https://github.com/machinekit/machinekit/commit/cf7c8e3f30c45d1c6bcc20b02d1e178c3db802cc](https://github.com/machinekit/machinekit/commit/cf7c8e3f30c45d1c6bcc20b02d1e178c3db802cc). I'm not seeing any errors on startup.

I'm still trying to figure out where the thread is getting stuck at on startup. I edited the latency test script to use a blocking thread and the gui takes about 5s-10s for a first update. Then the jitter is huge (almost 2ms) because it is missing interrupts somewhere in that time frame. Once it's going the jitter settles to below 25us with an occasional missed interrupt.

dkhughes, 2016-05-26:
As a double check to make sure the dpll wasn't the culprit with the way I'm interfacing it, I wrote a new IRQ controller that just has it's own free running timer. Behavior is the same.

dkhughes, 2016-05-26:
@mhaberler I've put my lsa on the hardware and piped the irq out of the fpga. I can confirm that for multiple seconds at startup, no IRQs are generated so the hal is getting stuck somewhere before calling hm2_soc_ol init function which starts the thread moving. 

Once things get in motion, the IRQ is set at 1ms rate as expected. What's odd is the periodic short then long response time to clear the IRQ. We get a short, (75us -90us response) followed by a long (115us-130us response). After 20s or so, these stop and we are left with a really consistent 60-80us response time. *But*, check out this screen shot:

[image]

What do you make of this? This has to mean kernel misconfig right? RTAPI reporting interval time that is half of what it actually is? Clock governor misreporting CPU speed?

dkhughes, 2016-05-26:
My bad, the clock interval was misreported because of the frequency of the cpu set by the governor! I changed to the performance governor and the ns times match what I would expect. Plus, the interrupt response time seems to have settled at around 60-70us for the zynq. This is looking much better:
[image]

dkhughes, 2016-05-26:
Here are some tangible results from a configuration including motion, stepgens, etc..:
```
# Config running on hal periodic thread
87927          timedelta.0.jitter
1079748        timedelta.0.max
912073         timedelta.0.min
```
versus
```
# Config running on external irq
43580          timedelta.0.jitter
1039283        timedelta.0.max
956420         timedelta.0.min
```
Half the jitter in situ - that is pretty cool!

mhaberler, 2016-05-26:
very good!

I wish we could shave off that write() in the hm2_soc_ol handler, I'm sure it helps a bit.

Even if it means a kernel driver, we do now have the means with dkms to build and distribute, so not a drama. And the kernel scoping pin would have a place too (just not sure if we still will have a gpio driver, or it's going to be all hm2).

Is that startup delay still in place? 

I have a gut feeling - the watchdog function. Maybe thread the pet-watchdog function on the thread **before** the waitirq to make sure the watchdog is petted first thing.

dkhughes, 2016-05-26:
@mhaberler I need to run histogram but my feeling from watching the jitter is that typically it's less than 15us with an occasional outlier. Without being able to probe the times like you showed in your screenshot its tough to tell where the time is spent (context switch, interrupt controller response time, etc.). Eliminating the write would be a straightforward optimization, but I'm not sure how much actual latency we will gain. 

The delay is in place still. Occasionally, I am getting timeout errors back and forth with the rtapi on startup, and then I reboot and they go away. Something is stuck spinning somewhere for sure, I just can't track it down. Do you see any of these delays when you start up waitirq on an altera device?

The hm2 watchdog? I'm pretty sure it's fed automatically by the hm2_write function. If I add pet watchdog back in the hal file it complains about that being a deprecated function call.

mhaberler, 2016-05-26:
watchdog: automatic petting by write - I think they did this over at LinuxCNC, not sure it was ported over

during the delay period: what does /proc/interrupts have to to say? numbers increasing? never seen such a delay on the altera

by 'piped out' you mean you can scope the IRQ request on a pin?

this might mean we'll have to fork the uio_platform_irq driver once more and add scoping pins ... I dont mind adding the gpio functions, but usually have a hell of a time doing the related dts magic :-/

dkhughes, 2016-05-26:
> by 'piped out' you mean you can scope the IRQ request on a pin? 

Yep, the lsa screens above are from a physical pin off the fpga. 

> during the delay period: what does /proc/interrupts have to to say?

Frozen:
```
machinekit@mksocfpga:~$ cat /proc/interrupts | grep hm2-socfpg0
164:     733435          0       GIC  61 Edge      hm2-socfpg0
machinekit@mksocfpga:~$ cat /proc/interrupts | grep hm2-socfpg0
164:     733435          0       GIC  61 Edge      hm2-socfpg0
machinekit@mksocfpga:~$ cat /proc/interrupts | grep hm2-socfpg0
164:     733435          0       GIC  61 Edge      hm2-socfpg0
machinekit@mksocfpga:~$ cat /proc/interrupts | grep hm2-socfpg0
164:     734132          0       GIC  61 Edge      hm2-socfpg0
```

dkhughes, 2016-05-26:
Hmm. It's like the hm2 setup function isn't getting called. The lsa shows no action at all until latency-test unfreezes. I figured I would see 1 pulse (from the interrupt that is generated when we activate the controller and 1ms expires), but I see nothing for about 5 seconds, then it goes.

mhaberler, 2016-05-26:
hm, I'd look at IRQStatus (0x0A00) ClearIRQ(0x0B00) WatchDogStatus (0x0D00)

mmap[0x0A00] & 0x2 should be 2 (IRQ enabled)

does mmap[0x0A00] & 0x1 ever become 1 during the delay? I understand 1 as 'IRQ posted'

does writing to mmap[0x0B00] get things going? 

mmap[0x0D00] & 0x1 == 1 - watchdog has bitten

mhaberler, 2016-05-26:
is that 'multiple seconds at startup' fixed/repeatable or does it vary?

dkhughes, 2016-05-26:
> does mmap[0x0A00] & 0x1 ever become 1 during the delay? I understand 1 as 'IRQ posted'

That's correct the 1 does mean the IRQ is posted. That is the signal I am watching externally with the LSA - it remains low until the unfreeze.

> is that 'multiple seconds at startup' fixed/repeatable or does it vary?

Repeatable, happens every time for about 4.1 seconds. Doesn't matter if I start axis or latency-test or halrun. In halrun, if I query the pins I can see they are stalled until the delay expires.

During startup I can see the registers were written correctly (a80 is the period/prescale register for the new irq timer):
```
machinekit@mksocfpga:~$ sudo ./mksocmemio -r a00
    mksocfpgamemio: read write hm2 memory locatons by cmmandline arguments  
    /dev/uio0 opened fine OK 
    region mmap'ed  OK 
    mem pointer created OK
Program name: ./mksocmemio    input option = r 
Read only 
Address 2560 	value = 0x00000006 
machinekit@mksocfpga:~$ sudo ./mksocmemio -r a80
    mksocfpgamemio: read write hm2 memory locatons by cmmandline arguments  
    /dev/uio0 opened fine OK 
    region mmap'ed  OK 
    mem pointer created OK
Program name: ./mksocmemio    input option = r 
Read only 
Address 2688 	value = 0x0003E864 
```
The 6 in a00 is because I added a timer enable bit. If I manually write those values with mksocmemio and I have uio_util running I see an interrupt immediately.

mhaberler, 2016-05-26:
wild guess - could these 4.1seconds come from some timer or counter register wraparound?

could it be we're enabling IRQ's too early? what about moving the IRQ enable code to happen during the first execution of waitirq?

dkhughes, 2016-05-26:
> wild guess - could these 4.1seconds come from some timer or counter register wraparound?

Brilliant! I found a bug in the vhdl that would do just that. Recompiling now. The same startup bug is in the dpll code, it's just a shorter delay.

dkhughes, 2016-05-26:
Here is the issue (from hmtimers.vhd):
```
	if PrescaleCount = x"01" then
		PrescaleCount <= PreScale;
				...timer increments here..
	else
		PreScaleCount <= PreScaleCount -1;
	end if;  -- prescale;
```

PrescaleCount after a reset defaults to 0. That means we have to wait until the 7 bit counter wraps around before the accumulator begins to work. My delay with the new freerunning IRQ was longer because it was waiting for a 24 bit counter to do the same thing.

dkhughes, 2016-05-26:
That was it! Working well now. So, we have a choice. I can do either of the following:

1) Submit the changes for the dpll to disable error latching when IRQs are enabled
2) Submit the free running counter version of the IRQ controller. I could add this controller in the address gap, which would make the old IRQ generation still available for DPLL / external latching.

If we do (2) then the existing DPLL framework doesn't change, we just gain a second top level IRQ pin in the VHDL / dts.

dkhughes, 2016-05-26:
Last thing I need to break totally free of periodic threads is a way to stop motmod from making `servo-thread`. It has to be run at least as fast as any other thread (so 1ms updates in my case)or i get the following error:

```
msgd:0: hal_lib:2964:rt hal_create_xthread:182 HAL error: new thread period 1000000 is less than clock period 1000000000
```

`servo-thread`'s time duration is around 20us in the watch window. That plus context switching time is probably really hurting irq latency in the external driven thread.

mhaberler, 2016-05-26:
another legacy. Back then the idea was synchronization of shared data would happen through thread priorities plus non-preemption, and threads would have to be started in ascending or descending periods, dont remember which one

the multicore branch will put that to rest, but first I need to wrap up project housecleaning and this will take several moons

yes, ripping threads out of motion is a good cause. just make it a module option - it will affect tons of configs otherwise. Or I can look into it if you want me to.

dkhughes, 2016-05-26:
>  just make it a module option - it will affect tons of configs otherwise

You mean a compile time module option or a config string style option? Configs like example/reference ini/hal files correct? I wouldn't want to break anything, I was thinking of adding something like a servo_thread_nowait=1 or something and then not calling newthread explicity. I was hoping that would be the least intrusive method to get something running quickly.

>  Or I can look into it if you want me to.

I'll take any help you want to offer. If you don't have a chance it's no problem. Tonight I will poke the motmod with a stick until it does what I need it to, or I smash my local git tree and I have to revert to head again.

mhaberler, 2016-05-26:
have a smash to get going, I'll look into it tomorrow

yes, config - a module parameter

[73 more replies]