machinekit/machinekit #688: instcomp problem - maximum pin count not working correctly

strahlex, 2015-06-16:
I have discovered a problem in the instcomp utility. The per instance pincount features seems to work incorrectly. I will explain the problem using the `debounce` component where I discovered the problem.

This component has a `pincount` parameter initialized with 8 https://github.com/machinekit/machinekit/blob/master/src/hal/i_components/debounce.icomp#L10
Furthermore there are the two options `MAXCOUNT` and `DEFAULTCOUNT` https://github.com/machinekit/machinekit/blob/master/src/hal/i_components/debounce.icomp#L12 If I understand correctly these two options should be used to default and limit the maximum pincount.

However, I noticed that the the maximum pincount only depends on the default value that is specified within the pincount parameter. In this case for example a pincount greater than 8 will not work.

    newinst debounce debounce.0 pincount=12

results in a debounce instance with 8 pins only

ArcEye, 2015-06-16:
OK, I can see what is happening, I solved one problem but created another.

The testing for MAXCOUNT needs to be done earlier, or it will get set at DEFAULTCOUNT and cancel out later pin increases
MAXCOUNT needs to be used earlier to set maxpins, one is cancelling the other out
The pincount= parameter is getting through correctly and ip->localpincount is being set, but that in fact is worse, because the code could now overrun the number of pins.

Damn, I'll sort it tomorrow

ArcEye, 2015-06-17:
Fixed in  instcomp.g bugfix #690

commit ccc2e800 (2015-06-17): Merge pull request #690 from ArcEye/master

instcomp.g bugfix closes #688
closed 2015-06-17 by strahlex