machinekit/machinekit #689: machinekit lacks ROS integration capability

mhaberler, 2015-06-17:
machinekit would make a great realtime playout/device control layer for [ROS](http://www.ros.org). ROS could become an alternative driving layer, in parallel to the existing CNC control application (rs274 interpreter, task, motion).

What I think is needed is a way to either plug in natively into ROS, or use one of the _bridge_ approaches outlined here (taken from the [talk by Arne Haman/Bosch talk at i-ROS Stuttgart 06-09-2105](2015_06_09_ROS_embedded_Hamann.pdf):

[image]

The options for the interfacing strategy I currently see are:

- if nothing else, we can do rosbridge/JSON/websockets - we have the moving parts in place but I think 'native' is the way to go longer term. So I would prefer either
- rosc - a bit early and shaky, but native, ROS 1.0, or 
- ros2 DDS interaction - no client library but then we have a full linux environment on-target.
- any other options we could go for?

I guess rosserial is out of scope. I have currently no idea what the ros1/ros2 cutover timeline is.

For reference, this is where we are with machinetalk: 

[image]

possible locations to plug ROS into machinetalk:
-  haltalk, the scalar/message API daemon into HAL
-  or in front of the zeromq/protobuf stack
-  or in front of the websockets/JSON translator.

I have no preferences at this point but layering middlewares ontop of each other is a recipe for issues.

Some background links:

http://design.ros2.org/

intended as anchor issue for ROS integration

luminize, 2015-06-17:
While we're at it we might think about using [Gazebo](http://www.gazebosim.org) as a simulator. If we "talk" to ROS then I'm guessing we can also "talk" to Gazebo

gavanderhoorn, 2015-06-17:
@mhaberler: for the *native* category, have you considered using [cros](https://github.com/it-robotics/cros)? I think the use-cases driving its design are close to what you have in mind. [Main site](http://www.it-robotics.it/products/real-time-systems/cros/?lang=en) and the ROS [news announcement](http://www.ros.org/news/2014/12/robox-and-itrobotics-present-a-fully-ros-compliant-motion-controller.html).

Another alternative not in your OP could be [simple_message](http://wiki.ros.org/simple_message). Not native -- so it would fit in the *bridge* category -- and perhaps not as full featured, but used in almost all ROS-Industrial robot drivers.

mhaberler, 2015-06-17:
@gavanderhoorn - thanks for the hint to cros, this looks in fact like a viable option technically

unfortunately, due to [legal incompatibility](https://www.gnu.org/licenses/gpl-faq.html#AllCompatibility) (box col1 row 3) between GPL2 and (L)GPL3 we are unable to consider this option (and with us, just about every other GPL2(+) project for that matter). 

The only realistic option to consider this if it-robotics came around and changed licensing to (L)GPL(+) or a more liberal license like MIT or BSD. Given the number of copyright owners on LinuxCNC and machinekit, and the fact that some are deceased, it is impossible to follow the FSF recommendation to relicense the whole project as GPL3, and it would be undesirable for other reasons.

We had many such license clashes in the past and it is a bloody nuisance. Often licensors are not fully aware of the implications of their choice with respect to legal and hence economic consequences (it is safe to say this situation is a complete adoption killer). In some cases we managed to get stacks or libraries relicensed after some explaining. In all other cases we did not pursue any further.

gavanderhoorn, 2015-06-17:
Hm, I'd not considered the licensing issue. I'm happy most pkgs in ROS are BSD, which makes things a lot easier.

As to cros: the readme states:

> Licensing

> cROS is licensed under the GNU Lesser General Public License v3.0. For more licensing options and for paid support please write an email to cros@it-robotics.it.

So it might be an idea to contact them and see if things can be worked out.

mhaberler, 2015-06-17:
@gavanderhoorn - the [simple_message](http://wiki.ros.org/simple_message) scheme indeed seems to be a good start for trajectory playout, and that is likely a common use case.

If I understand this right, this could handle pre-planned trajectories at the joint position level assuming isochronous samples (type 1 message), and a more general arbitrary joint segment list of arbitrary segment duration (type 2).

it might well be that we wind up with several specialized API's and over time learn to do this in a more general way. 

This looks like a great start, thanks!

If I may add an idiot question:
- let's assume machinekit drives a robot of some kind and exposes the input side with simple_message. So far so clear - that's the command side.
- what about the status reporting side (eg current position)? that would be a separate publisher cyclically or on-change update a joint position?

gavanderhoorn, 2015-06-17:
> If I understand this right, this could handle pre-planned trajectories at the joint position level assuming isochronous samples (type 1 message), and a more general arbitrary joint segment list of arbitrary segment duration (type 2).

I don't understand to which 'type 1' and 'type 2' messages you are referring, but using [REP-I0004](https://github.com/ros-industrial/rep/blob/master/rep-I0004.rst):

 - `JOINT_TRAJ_PT` msgs are used to stream individual points of a joint space trajectory (so ROS->controller)
 - `JOINT_POSITION` or `JOINT_FEEDBACK` msgs are used to encode *current state* (controller->ROS)
 - `STATUS` is used to broadcast 'robot status' (ie: `in_motion`, `in_error`, etc).

> If I may add an idiot question:

> - let's assume machinekit drives a robot of some kind and exposes the input side with simple_message. So far so clear - that's the command side.
> - what about the status reporting side (eg current position)? that would be a separate publisher cyclically or on-change update a joint position?

See my comment above, but to clarify: all current implementations use cyclic publications.

mhaberler, 2015-06-17:
@gavanderhoorn - get it now, thanks.

gavanderhoorn, 2015-06-17:
Additional comment: to implement support for new robots/controllers, you typically only need to implement the 'server side'. The client side (bridging ROS msgs to the controller) is taken care of by the [industrial_robot_client](http://wiki.ros.org/industrial_robot_client).

luminize, 2015-06-18:
since Machinekit is mainly on debian the link above might be of interest.

I've just finished an installation of ROS on my debian Jessie VM as per:
https://wiki.debian.org/DebianScience/Robotics/ROS

when there are download errord like
```
ERROR [vcstools] Tarball download unpack failed: <urlopen error [Errno -2] Name or service not known>[/vcstools]
```
 during the step
```
wstool init -j8 src indigo-desktop-full-wet.rosinstall
```
then do `wstool update -j 4 -t src` as described here http://wiki.ros.org/jade/Installation/Source#Installation

mhaberler, 2015-06-19:
@luminize  - do I get this right this repo is amd64 only, no packages for i386? if so - here comes the next VM ;)

gavanderhoorn, 2015-06-19:
@mhaberler: building from source is also an option and (on a relatively modern PC) shouldn't take too long.

luminize, 2015-06-19:
@mhaberler I have no way of seeing. I've put the question on the Gazebo list (where I got pointed to the debian repo) here https://groups.google.com/a/osrfoundation.org/forum/#!topic/gazebo/YcXCRYqkcGs i'll keep you updated

mhaberler, 2015-06-19:
@gavanderhoorn - eventually I will - after having immersed in the _out-of-box experience_ ;)

ArcEye, 2015-06-19:
@luminize "I've just finished an installation of ROS on my debian Jessie VM "

Be very careful to only try installing the Jessie Virtualbox packages on a stock Jessie kernel.

I found to my cost that at some point they are hard coded against 3.16.0-4-amd64 and in other places use ${uname -r} so the wrong things get built against
This will result in a failure and lock up during the DKMS build and an unbootable kernel, because initrd-img has been screwed.

Thankfully I have several kernels on that partition so was able to boot another one, but still had all sorts of fun and games getting packaging sorted, because `dpkg --configure -a` just tries to build the modules again unless you manually delete everything, so that it fails and then use apt-get to 'delete' them and clear the package logs.

Wasn't very keen on VM's on Monday morning when all this occurred ;)

gavanderhoorn, 2015-06-19:
Instead of a full VM, I've had good success running things from a chroot (see [wiki/ROS/Tutorials/InstallingIndigoInChroot](http://wiki.ros.org/ROS/Tutorials/InstallingIndigoInChroot) for some info. Replace *Indigo* with any release obviously), or even docker (see [wiki/docker](http://wiki.ros.org/docker)).

zultron, 2015-06-20:
The link to Dr. Hamann's slides is broken.  Are those available somewhere?

mhaberler, 2015-06-20:
@zultron fixed

zultron, 2015-06-20:
When I was exploring this almost a year ago, I didn't find nearly such an abundance of possibilities.  Nice research!

At the time, I had settled on `ros_control` [1].  That project benefits from having the same API as the PR2 controller.  Many existing ROS robots use that API, probably against customized versions of the PR2 controller code, and notably Baxter uses `ros_control`, so real investment there.  It also works with ROS-i's 'MoveIt!', so it would plug right into industrial robot arm ROS-i stacks.

In the diagram on that page, Machinekit joints would be exposed through a Machinekit controller interface, a subclass of any of the `joint_{trajectory,position,velocity}` classes.  I envisioned the controller manager setting up a new Machinekit controller, which would interface to a local or remote Machinekit instance through Machinetalk or HALtalk, as appropriate.

The controller architecture seems to have clear parallels in the Machinetalk architecture.  Except for the actual `joint_trajectory` etc. ROS interfaces, the right-hand side of that diagram looks pretty much done by Machinekit.

[1]: http://wiki.ros.org/ros_control

****

Michael's question of where it connects to Machinekit is still unanswered in this `ros_control` picture.  Looking at this diagram from the 'Code Notes' [2]:

[image]

The controller inputs to a `ros_control` controller are commonly joint positions or velocities (see full list on ROS wiki), so TP and kins have already been done.  In the above diagram, controller inputs would most directly correspond to the `AXIS N` boxes inside the `EMCMOT` box, expanded further down in the 'Code Notes':

[image]

@mhaberler points out this division within the `motion` component in off-list correspondence:

> - motion has an "upper half" preparing segments, handling blending etc (command.c calling upon emc/tp/*), and a "lower half" (control.c) which interpolates along segments.

My best idea a year ago was to copy and paste the 'lower half', joint/axis controller code from `motion` into a new, separate `joint` component with ring buffers between it and user space.  Referring back to the `ros_control` diagram, the `hardware_interface::Machinekit` class would simply pass input joint commands with minimal modification through the ring buffer to this `joint` comp, which would pass joint states back out through a separate ring buffer back to `hardware_interface::Machinekit`.

[2]: https://github.com/machinekit/machinekit-docs/blob/master/src/code/Code_Notes.asciidoc

mhaberler, 2015-06-20:
@zultron - my take on this legacy drawings is: we should forget about that and turn the page.

There is a ton of convoluted crap there, and I think trying to force the ROS tp playout use case through using motion _again_ is going to lead nowhere. I think we should leave motion stand where it is for the CNC use case and be done with it.

For trajectory playout we can assume offline planning which hugely simplifies things. The way Bas and myself have envisaged this is through the (as of yet unmerged) halpipes branch which will enable components to connect through ringbuffers and strongly typed messages, just as like pins.

It boils down to resurrecting some parts of motion and placing them into separate comps with such a pipe API. And those would be _very_ simple in function: kins, interpolate, sample segments at servo rate, maybe blend, limit vel/acc/jerk. instantiate, connect through ringbuffers, run like a shell pipe arrangement.

And: most of this will be done in userland. The legacy architecture does _way too much_ in RT, simply because they never thought through the best design for the problem at hand - which is why there is like 90% of the motion code in a place where it doesnt belong. And motion doesnt even have a documented API about what, 20 years after they designed it?

The industrial_robot_client is a good example what all can be done in userland in this context.

zultron, 2015-06-20:
I was also proposing to leave `motion` as is, and crib the `joint` controller code into a new comp, only those small parts needed.  Just like you say?  In the interface I point to, kins should have already been done on the ROS side, since input data is in joint-space coords/waypoints.  (This is also true of `simple_message` that @gavanderhoorn points to.)  Not sure whether vel/acc/jerk limits should be left to ROS's offline planning, also.  Again, see the second diagram in my previous post.

It sounds like 'halpipes' would be perfect for this or any other interface we eventually go with.  Nice work!

Glad to have as much as possible out of RT, and `motion` out of the way!

As for [Baxter][1], that's a very complete, if also overwhelming, example of a robot interface (like I said, there's VC funding behind it ;).  It not only implements a `ros_control` interface, but also has a gazebo interface for @luminize.  It would be supremely cool to have a simulated robot connected to a 'sim' Machinekit instance with simultaneous display of the 3D robot in gazebo and a halscope window next to it!

[1]: https://github.com/RethinkRobotics/baxter

luminize, 2015-06-20:
I think that machinekit has a unique position in the scheme of things.
Machinekit can fulfil the bridge between use cases where RT is needed (like balancing, flight, high path accuracy etc) where a quick reaction time and stability of the calculations is needed. 

The use cases I see:
- configurable "firmware". Not as dumb as uC's, but very flexible in brewing unique layouts of hardware and sensor. Thus a command "go to (x,y,z)" by a ROS message can result in a movement to a certain position by MK kinematics. Of course this command could also be a pose (joints) thus no use kinematics.
- translating physical hardware to messages. Nothing more, nothing less. Interpreting of the messages can be done by ROS.

For other things like robots that interact with humans, collision detection etc,where a lot of external events happen (humans) this is way more difficult to compute, and I don't see a need for RT for that, because via CANOpen in ROS the "slow" motion is fast enough for most cases.

Did I forget anything of make a wrong assumption?

mhaberler, 2015-06-27:
@gavanderhoorn - let's talk message type ID's for a moment: I'm perusing https://github.com/ros-industrial/rep/blob/master/rep-I0004.rst#standard-messages and see you're establishing a type registry for ROS messages - which makes a lot of sense

this seems to assume a 32bit space for message type id's. Do you think this space could be reduced to say 16bits or maybe a bit more, but less than 24? Realistically much of this space will remain unused anyway, so it boils down to a change of the assignment procedure

you have

````
 65536-2147483647  -                            Reserved for future use
````
meaning the end of the space is currently at 0x7ffffff

reason why I'm asking: we have the means to transport a 32bit tag with each message all the way from say UI's/driving code through ringbuffers into RT space and back; I am hesitant to enlarge the size of this tag as it would be a quite invasive change, alignment issues come into play, and it'd be in the most time critical path of the code; using an extension bit to signify a larger tag does not help because the API is currently able to handle only 32bits

The current disposition of this type tag is as follows:

````
typedef enum {
    MF_STRING         =  0,    // payload is a printable string
    MF_PROTOBUF       =  1,    // payload is in protobuf wire format
    MF_NPB_CSTRUCT    =  2,    // payload is in nanopb C struct format
    MF_JSON           =  3,    // payload is a JSON object (string)
    MF_GPB_TEXTFORMAT =  4,    // payload is google::protobuf::TextFormat (string)
    MF_XML            =  5,    // payload is XML format (string)
    MF_LEGACY_MOTCMD  =  6,    // motion command C structs, with cmd_code_t type tag
    MF_LEGACY_MOTSTAT =  7,    // motion status C structs, with cmd_status_t type tag
    MF_BLOB =  8,              // payload is an opaque blob

    // add here as needed and change the #define to point to the last encoding
    // used in the base code
#define MF_LAST  MF_BLOB
    MF_UNUSED1        =  9,    // unused in code base  - user extensions
    MF_UNUSED2        =  10,    //
    // ...
} mf_encoding_t;

typedef struct {
    __u32 msgid     : 12; // must hold all proto msgid values (!)
    __u32 format    : 4;  // an mf_encoding_t - how to interpret the message
    __u32 more      : 1;  // zeroMQ marker for multiframe messages
    __u32 eor       : 1;  // zeroMQ marker end-of-route, next frame is payload
                          // zeroMQ route tags are marked by msgid == MSGID_HOP
    __u32 unused    : 14; // spare
} mfields_t;
```

To encompass ROS message id's into this scheme, we could:
- extend the msgid range from 12 to whatever is the maximum ROS message type ID needs, by reassigning spare bits
- include a typespace tag (machinetalk=0, ROS=1 etc) so the typespace/msgid tuple would be unique
- define an encoding_t enum MF_ROS_SERIAL to cover the ROS serial wire format

with this reduced type ID range machinetalk would perfectly be able to transport and runtime-identify ROS serial messages, which would ease integration a lot!

gavanderhoorn, 2015-06-27:
If I understand you correctly you'd like to use the ROS-I assigned ID as an identifier in your own messaging system, and inject ROS-I msgs 1-on-1 into that? 

We typically only use Simple Message *between* the ROS client nodes and the server parts running on the controller. Server programs mostly translate the contents into some internal representation, independent of the original msg. MachineKit seems to be a bit special: it uses an internal communication system that could potentially accommodate the actual Simple Message structures and is completely 'under our control' (with that I mean that you have complete control over the internals of the system, not something that we typically see in ROS-Industrial).

If I understood you correctly, I'd suggest creating a `MF_ROS_INDUSTRIAL` encoding id, and then encapsulating the entire ros-i msg (so header+payload) in a 'machine kit msg' (forgive me: I'm not too familiar with the terminology here). The IDs and msg contents are completely ROS-I specific, so I would even suggest translating msg contents into MachineKit-specific (but corresponding) msgs, which would completely avoid the issue afaics.

A bit of rationale for the `msg_type` field: a lot of industrial (robot) controllers have a very limited type system (only a single `int` type fi, with a fixed size and always signed), which we had to accommodate for. Hence the 32bit ID field (of which only the positive side is used). Restricting the ID space would then be a convention only, as we cannot change the size of the field itself.

gavanderhoorn, 2015-06-27:
@gavanderhoorn wrote:
> [..] The IDs and msg contents are completely ROS-I specific, [..]

To clarify: Simple Message isn't like `rosserial` in that it is a wire format for which we could write / generate automated (de)serialisation infrastructure. Similar to `rosserial`, we obviously must convert ROS msgs into a bytestream. Unlike `rosserial`, that is not a 1-to-1 conversion: fields are left out, added or changed (the `duration`/`velocity` fields in `JOINT_TRAJ_PT` fi: those don't exist in [trajectory_msgs/JointTrajectoryPoint](http://docs.ros.org/indigo/api/trajectory_msgs/html/msg/JointTrajectoryPoint.html)), and we only support a very limited subset of all the ROS msgs that are in existence.

mhaberler, 2015-07-02:
@gavanderhoorn - yes, it is about transporting ROS messages in the machinetalk transport, and have them runtime type/encoding tagged so they coexist peacefully with existing native messages

I might have jumped to conclusions too early wrt the applicable ROS message formats, and how we might come up with the most general way of interfacing machinekit to ROS

looking at it again your suggestion to tag the whole class of ROS messages with an encoding tag, instead  of trying to merge the message ID spaces without collisions makes a lot of sense (and is more general anyway)

so that means: a ROS-message-wrapped-in-machinetalk container. Fine with me. Translating into native is another exercise I'm not sure makes sense. What I could think of (far out) is the following: translate the msg format descriptors into .proto (similar like what is happening with the msg->DDS IDL transition), and write an encoder/decoder which uses protobuf introspection but generates/parses msg wireformat. That is something which for instance has been done with the [goby project](http://gobysoft.org/index.wt/software/goby) and would open a second route for ROS to migrate to protobuf in case the ROS2/DDS effort goes nowhere (which I rate has a certain chance of happening)

In terms of message type universes - I guess we should look at the range of messages used by rosserial then?

luminize, 2015-10-09:
this might be an interesting announcement for this issue (since we are Debian minded). Running ROS _and_ Machinekit from the same box doesn't seem strange to me at least:
https://lists.debian.org/debian-science/2015/10/msg00024.html

cdsteinkuehler, 2015-12-11:
I am beginning to look seriously at adding ROS support to talk to Machinekit's HAL layer.  There are many possible ways this can be done, but my initial thought is to craft a ROS package that receives and transmits appropriate ROS messages, tying the contents more or less directly into HAL (perhaps with a component similar to the existing motion if needed to handle the real-time side of path planning and closing feedback loops).  As I have very little experience with ROS, I would appreciate if anyone who does have some experience weighs in on the following questions:

* Should the Machinekit/ROS real-time code run on the same machine as
the "main" ROS controller, or would it be better to have two systems
(presumably communicating via Ethernet?)?

* Should I try to write the "glue" code for ROS messaging in Python or
C++ ?  Both are easily possible and can directly talk to HAL, virtually
any language is an option if communicating to HAL via protobuf messages.

* When I look at "Supported Hardware" on the ROS-i wiki:
    http://wiki.ros.org/Industrial/supported_hardware
  there are several communications formats listed (position streaming,
trajectory download, etc).  What would be the preferred method to
implement first?

* Which release of ROS-i should I begin working with?  Are there any
"official" (or recommended) pre-built images or virtual machines available?

* Do you think a fully x86 based system (likely using a Mesanet FPGA
card for machine I/O) would be a better starting platform, or something
like the BeagleBone Black would make a better "first example"?  This
relates somewhat to the first question (one or two systems), since I'm
not sure how well the "Desktop-Full" target would run on a 'Bone (it
seems like it could be a bit under-powered).

evanmj, 2015-12-11:
I wish I could help more, but I'm not into these sorts of details of ROS-I just yet... I'm very interested in what this bridge could do for the open robotics / machine control community.

The ROS-I training course materials are available online [here](http://aeswiki.datasys.swri.edu/rositraining/hydro/Exercises/).

The training VM is availalbe with setup instructions [here](http://aeswiki.datasys.swri.edu/rositraining/Hydro%20PC%20Setup).  Using this VM should save you many hours of install time and configuration.

It might be worth looking at (ros_control)[http://wiki.ros.org/ros_control] as well, as I think it could sit well in the "Real Robot" section of their data flow flowchart.

If you get stuck, hit up @shaun-edwards, he is very helpful.

gavanderhoorn, 2015-12-11:
My n cents:

> - Should the Machinekit/ROS real-time code run on the same machine as the "main" ROS controller, or would it be better to have two systems (presumably communicating via Ethernet?)?

This would seem to depend highly on where (at which level) the integration is done.

> - Should I try to write the "glue" code for ROS messaging in Python or C++ ? Both are easily possible and can directly talk to HAL, virtually any language is an option if communicating to HAL via protobuf messages.

For prototyping, I'd go with Python. Setting up the required interfaces (ROS side) in Python is probably 30 mins of work (or something in that order).

If, after profiling, the Python implementing proves a bottleneck, you can always try a C++ migration. Semantically, the APIs are equivalent, so that should be almost trivial.

Personally I find ROS & Python very nice to work with in early stages, for rapid prototyping and such.

> - When I look at "Supported Hardware" on the ROS-i wiki: http://wiki.ros.org/Industrial/supported_hardware there are several communications formats listed (position streaming, trajectory download, etc). What would be the preferred method to implement first?

In the past we've done some downloading drivers, mostly because of platform limitations (some controllers can only deal with complete motion *jobs* fi), but almost all new drivers are streaming based. If performance is an issue (due to latency of the network fi), intermediary (ring)buffers can be used.

> - Which release of ROS-i should I begin working with? Are there any "official" (or recommended) pre-built images or virtual machines available?

Installation of ROS on a desktop machine is (almost) a single line, 30 mins affair (provided you are on Ubuntu and have a decent machine). A Docker infrastructure has recently been setup ([docs](http://wiki.ros.org/docker)) as well, and should work nicely on other platforms (for which no binaries are available). Maybe relevant: [wiki.debian.org/DebianScience/Robotics/ROS](https://wiki.debian.org/DebianScience/Robotics/ROS).

There are some VMs (and the mentioned ROS-I training VM is one), but personally I'd never work in a VM.

Building from source is always an option, but obviously more involved.

> - Do you think a fully x86 based system (likely using a Mesanet FPGA card for machine I/O) would be a better starting platform, or something like the BeagleBone Black would make a better "first [...]

A desktop machine (with Ubuntu) is always easiest.

BeagleBone is pretty well supported too, see [wiki.ros.org/BeagleBone](http://wiki.ros.org/BeagleBone) fi. I don't think a *desktop-full* install is even possible on those systems, but for your application, I think a *core* or *base* install (see [REP-142](http://www.ros.org/reps/rep-0142.html#metapackages)) should be enough, and relatively lightweight.

mhaberler, 2015-12-11:
let's get going first and get smart later ;)

so I'd say - Python, same host for now, use the cython HAL bindings for wiggling some pins to work out the kinks in the interfacing

btw running on wheezy and doing ROS in a ubuntu container is a neat way of working around until we have the Ubuntu packages again for co-hosting ROS and machinekit; @gavanderhoorn - are there any specific LTS Ubuntu releases we should focus on ?

Eventually we'd want to go distributed - have a network-controlled realtime appliance playing out trajectories and operating sensors/switches etc

I note @strahlex did  remote Haltalk Python bindings - this might be a start for remote scalar I/O

for the 'lets move some joints in a coordinated way to play out a trajectory" exercise:

the way I understand it we could assume ROS passing a offline-planned joint trajectory (see further up), so a large part of the functionality of motion is taken care of upstream - what we need is a segment interpolator - and if we go fully smart and smooth we tack on a cubic interpolator stage; so if we get a joint trajectory at say 100Hz we could interpolate and play out smoothly at 1kHz or above

this segment interpolator and separate cubic does not exist as halcomps yet, but I feel confident of being able to do them in time for mid-Jan

what I have started on, but not yet finished is extending haltalk to serve ringbuffers to/from RT via zeroMQ sockets - that would be the vehicle eventually; but we do not need that in round 1, see the micromot example in src/machinetalk/tutorial and the ringbuffer/zeromq examples: all Python bindings in place to whip up a simple segment feeder are in place even if not totally elegant yet

(that is part of the multicore branch)

gavanderhoorn, 2015-12-11:
@mhaberler wrote:
> [..] @gavanderhoorn - are there any specific LTS Ubuntu releases we should focus on ?

ROS Indigo (LTS) currently supports Ubuntu Trusty (14.04, LTS). I'd go for those.

gavanderhoorn, 2015-12-11:
@mhaberler wrote:
> [..] so if we get a joint trajectory at say 100Hz we could interpolate and play out smoothly at 1kHz or above.

Note that typical ROS trajectories (ie: those from MoveIt) are rather sparse, but can be resampled using something like [industrial_trajectory_filters](http://wiki.ros.org/industrial_trajectory_filters). MoveIt can also be configured to generate denser trajectories by itself.

Trajectories generated by [descartes](http://wiki.ros.org/descartes) are typically much denser, but that is often a result of the input as much as the sampling.

cdsteinkuehler, 2015-12-11:
@gavanderhoorn The tutorial ROS-i install uses Hydro running on Ubuntu 12 (Precise Pangolin).  Is there an important reason to try and use Indigo/Trusty instead?  My current plan is to install Ubuntu on a spare system and try to work through a few of the ROS-i tutorials.  Then I'll try getting Machinekit-HAL running on Ubuntu, falling back to Debian (with ROS-i/Ubuntu running in a VM on the same host) if HAL on Ubuntu is too problematic.

...but this plan is still pretty fluid...and as they say: "No battle plan ever survives first contact with the enemy."  ;-)

gavanderhoorn, 2015-12-12:
@cdsteinkuehler wrote:
> The tutorial ROS-i install uses Hydro running on Ubuntu 12 (Precise Pangolin). Is there an important reason to try and use Indigo/Trusty instead?

That's just the tutorial @evanmj linked you to. That is for Hydro, so it will tell you to install Hydro.

You can find the Indigo version at [rositraining/indigo/Exercises](http://aeswiki.datasys.swri.edu/rositraining/indigo/Exercises/). I'd really go with Indigo and Trusty, as Hydro is already EOL.

luminize, 2015-12-12:
@cdsteinkuehler @mhaberler @gavanderhoorn 
I just got a signal from a python HAL component into a ROS publisher.

I installed MK and ROS on Debian Wheezy
I followed these instructions for ROS
http://wiki.ros.org/indigo/Installation/Debian
§2.1 did the ROS-Comm installation
§2.2.1 dependencies for ROS-Comm

When set up:
```
cd ~/ros_catkin_ws/src
git clone https://github.com/luminize/ros_hello_machinekit.git
cd ~/ros_catkin_ws
catkin make
```
start realtime stuff in terminal 1
```
halrun
```
in terminal 2 start ROS
```
roscore
```
in terminal 3 start the ros_hello_machinekit node
```
rosrun ros_hello_machinekit hello_world_publisher.py
```
change the pin value of the component created by the publisher node
```
show pin
setp rospub-inside-HAL.value-in 456
```
and see the value of the node change. We can then attach a subscriber to this publisher.
... fun, aint it ?...

picture or it didn't happen:
[image]

gavanderhoorn, 2015-12-12:
Nice. Adding a subscriber to your `hello_pub` node to go the other way should not be too difficult.

Side note: this is why I like Python for this kind of work: your `talker` is essentially 11 lines ..

luminize, 2015-12-12:
done :)

halfile and subscriber added.

- the halfile starts a thread and a siggen component
- start ROS
- start publisher
- start subscriber
- the publisher makes a component with an pin when it starts
- in HAL make a signal, connecting the cosine with the publisher pin

more detailed instructions:
https://github.com/luminize/ros_hello_machinekit/tree/master

picture:
[image]

gavanderhoorn, 2015-12-12:
Also nice.

But what I meant was to create a subscriber on the ROS side that proxies messages *into* machinekit (so setting pin values based on ROS msgs).

Also: if the `hal` bindings allow any kind of event driven callback system, you could use that instead of polling at a certain rate. That is a typical pattern with ROS 'drivers', and should avoid all kind of potentially nasty under/over sampling issues.

luminize, 2015-12-12:
@gavanderhoorn thanks,
I needed to do the subscriber to publisher example (part of my RTFM)

Next I'll get a pin in HAL to change by a subscriber. So in the same way the publisher now reads a pin from HAL, the subscriber is going to change a pin in HAL.

for that HAL pin, the subscriber must use a float, u32, s32 or bit. So I'll look up the books/docs on the types.

Are there any out of the box "debug" topics that change for the subscriber to connect to? if not then I'll update a counter at the frequency rate and put that in HAL. But that's not as sexy as a cosine :)

gavanderhoorn, 2015-12-12:
> I needed to do the subscriber to publisher example (part of my RTFM)

I figured :). Just clarifying my earlier comment, no criticism. Nice progress so far.

> for that HAL pin, the subscriber must use a float, u32, s32 or bit. So I'll look up the books/docs on the types.

You'll want to use something from [std_msgs](http://wiki.ros.org/std_msgs). Everything you list maps directly, apart from `bit`. I guess we treat `Bool` like `bit`.

> Are there any out of the box "debug" topics that change for the subscriber to connect to? if not then I'll update a counter at the frequency rate and put that in HAL. But that's not as sexy as a cosine :)

No, ROS out-of-the-box does 'nothing'. You'll have to make it do things :).

I don't know of any nodes that generate signals like that for you, but that shouldn't be too hard to create: [math.cos](https://docs.python.org/2/library/math.html#math.cos) :).

luminize, 2015-12-12:
> On 12 Dec 2015, at 18:08, G.A. vd. Hoorn <notifications@github.com> wrote:
> 
> I needed to do the subscriber to publisher example (part of my RTFM)
> 
> I figured :). Just clarifying my earlier comment, no criticism. Nice progress so far.
> 
No worries :)
> for that HAL pin, the subscriber must use a float, u32, s32 or bit. So I'll look up the books/docs on the types.
> 
> You'll want to use something from std_msgs. Everything you list maps directly, apart from bit. I guess we treat Bool like bit.
> 
Yes, I'll use book. I've looked thru these types and I was surprised about the arrays. Never thought about these, but could be useful gathering HAL pin values into arrays.
> Are there any out of the box "debug" topics that change for the subscriber to connect to? if not then I'll update a counter at the frequency rate and put that in HAL. But that's not as sexy as a [...]
Yeah, or I'll use a 2nd subscriber to get the value back into HAL again

Thanks for the pointers Gijs

strahlex, 2015-12-13:
looks very promising, looking forward to see the demo

luminize, 2015-12-13:
@gavanderhoorn @cdsteinkuehler 

With 2 small nudges from Alexander and Michael this afternoon I managed to also get data back into HAL.
I've made a 2nd subscriber, who gets the data from the publisher and puts that back into HAL
https://github.com/luminize/ros_hello_machinekit/blob/master/scripts/machinekit_subscriber.py

[image]

luminize, 2015-12-13:
@mhaberler, @cdsteinkuehler, @gavanderhoorn commented this in https://github.com/machinekit/machinekit/issues/689#issuecomment-164167532 above
> Also: if the hal bindings allow any kind of event driven callback system, you could use that instead of polling at a certain rate. That is a typical pattern with ROS 'drivers', and should avoid all kind of potentially nasty under/over sampling issues.

How should we go about this? because HAL is very fast, would not the number of messages in ROS get overflooded? Or make the event driven call back from a slower thread?

luminize, 2015-12-13:
Alexander pointed to `asciinema`. I've fallen in love with that one (asciinema). Record a Terminal session. Pauze the record, even copy the text right out of the session.
here's an embedded image link of a terminal showing the HAL. Awesome find Alexander!
[image]

mhaberler, 2015-12-14:
@luminize - step 1 is to become smarter about pumping data out of HAL, that is: dont report the full monty every cycle but only what has changed 

1. Drop the legacy hal bindings and use the cython bindings. The legacy bindings are useless for change detection, which is what you want for effective change reporting. Read through nosetests/*.py to get the hang how they work.

2. I think for _reporting_ HAL->ROS the better device is a HAL group - a set of signals; those live independent of a component and signals can be linked to pins as needed. For injecting HAL pin values from ROS you do need a component as you have already used, out pins only.

See [here how a HAL group can be defined at the halcmd level](https://github.com/machinekit/machinekit/blob/master/nosetests/test_groups.py).

See [here how it can be done in Python](https://github.com/machinekit/machinekit/blob/master/nosetests/test_groups.py).

3. For both comps and groups, the `changed()` method will retrieve the list of names of pins/signals which have changed since the last call to `changed()`. While I am not yet sure if the ROS publishing mechanism supports partial updates (i.e. only the values which changed, or the full list of values must be pushed), [this](https://github.com/machinekit/machinekit/blob/master/nosetests/test_groups.py#L66-L68) will yield the changed values only; [this](https://github.com/machinekit/machinekit/blob/master/nosetests/test_groups.py#L78-L82) yields all of them; if you want to do the latter, just test for `changed()` not being the empty list, then send an update.

4. If HAL is fast or slow is irrelevant - in the HAL->ROS direction  the reporting frequency counts, which is completely independent of any thread operation.

5. For injecting scalars ROS->HAL, the rate of updates by the ROS upper level counts, so not a HAL issue.

6. This pin wiggling is all fine for conveying discrete scalars.

7. For conveying a sequence of [JointTrajectoryPoints](http://docs.ros.org/indigo/api/trajectory_msgs/html/msg/JointTrajectoryPoint.html) we need to cook up a  minimal motion chain: a sequence of these JointTrajectoryPoints is essentially a seqment list (note that seqments are not isochronous but can have an arbitrary length, as specified by (`time_from_start`). It looks to me the `uniform_sample_filter`  from http://wiki.ros.org/industrial_trajectory_filters as suggested by @gavanderhoorn is a great start, the output of this should already by an isochronous sequence of pos/vel/acc samples per joint and could go directly into HAL pins to drive servo loops or steppers.

What I would do is push these samples into a ringbuffer and play them out to HAL pins at a constant rate - post sampling, you would run the servo loop and drive the motors.

Actually the delayline component should do for this purpose - use only the playout part, all we need to do is a bit of API to push samples into the input side via some basic Python glue - see [the ring-related examples](https://github.com/machinekit/machinekit/tree/master/nosetests). Happy to help with that.

strahlex, 2015-12-16:
@luminize did not know the `watch` command, very useful

mhaberler, 2015-12-16:
update on installing machinekit master on Ubuntu 14.04 as present on the [ros-industrial vbox image](http://aeswiki.datasys.swri.edu/rositraining/Indigo%20PC%20Setup), section 2.1:

good news - almost no issues with mk per se. Minor nits:

- #832 must be applied
- all incantations as per http://www.machinekit.io/docs/building-from-source/ work, with the following changes (I just used the wheezy packages)
-- no need to define wheezy-backports and pull cython from same; cython as per U14.04 is 0.20x which is good enough
- rsyslog setup has a wart I cant figure out why - local1 logs go to /var/log/syslog instead of /var/log/linuxcnc.log
- all runtests pass! so HAL/RT seems healthy.

on Ubuntu, as run on vbox on OSX:
- I had to install the VM three times - it would hang on resizing a window, after reset would refuse to start X properly. And no way out. I am using snapshots now to restore. The last crash was caused post-build when running axis_mm .
- unity is a complete joke, and seems to become worse by release.
- [compiz has become worse as a cpu hog, whatever that process does](https://bugs.launchpad.net/compiz/+bug/1293384) - I am about to dump unity and go for Lubuntu
- I tried to ` apt-get install compizconfig-settings-manager` and tweak "enable LAZY window positioning" as per above link, and also uncheck the "Sync to VBlank" option. That resulted in another "low res graphics mode startup" - trying to recover with a snapshot now. Complete mess.

luminize, 2015-12-16:
> On 16 Dec 2015, at 19:54, Michael Haberler <notifications@github.com> wrote:
> 
> unity is a complete joke, and seems to become worse by release.

Install gnome flashback session

http://www.omgubuntu.co.uk/2014/04/ubuntu-14-04-classic-gnome-flashback-session

mhaberler, 2015-12-16:
hallelujah! that seems to have made the trick (so far)

luminize, 2015-12-16:
Installed MK on the image. no problems except the logging issue you mention (got error after `sudo make setuid`)

I experienced the same crash you mention. The VM is now lost. Also running the image from VirtualBox. Dit you perhaps got the freeze when you dragged the machinekit configuration picker window?

mhaberler, 2015-12-16:
well, chose flashback-metacity, sort of usable

to get axis to run (really OpenGL) one needs to turn off 3D acceleration for that VM in VBox Settings!

yes moving a window or resizing triggers the bug - sometimes the VM comes back after several minutes

mhaberler, 2015-12-16:
well as far as HAL is concerned with Ubuntu 14, mission accomplished, but it looks like a tenuous relation ahead ;)

cdsteinkuehler, 2015-12-16:
Great news!

I'm hoping to encounter less hassle than the two of you since I'm not
actually running a VM, but we'll see (fingers crossed).  I also am not
particularly interested in running any GUIs, but it would helpful if
they worked (particularly things like halscope).

So if I'm following along correctly, you built the entire machinekit
source tree (with minor tweaks as listed), and didn't have to chop out
the CNC stack to get things to compile happily under Ubuntu?

-- 
Charles Steinkuehler
charles@steinkuehler.net

luminize, 2015-12-16:
I just reproduced my initial stuff like above https://github.com/machinekit/machinekit/issues/689#issuecomment-164287673 on the ubuntu ROS-i image.

I too installed the gnome flashback session like here https://github.com/machinekit/machinekit/issues/689#issuecomment-165212586

Now I sometimes noticed lag when dragging a window... however... I noticed that also on Debian VM's run under Virtualbox. But I didn't got the mushroom freeze.

to reproduce running ROS and MK: when you have the image installed and MK built from source open a terminal and type:
```
training_unit 1.4
```
this will get you into a catkin workspace
```
ros-industrial@ROS:~/industrial_training/training/work/1.4$
cd src
git clone http://github.com/luminze/ros_hello_machinekit.git
cd ..
catkin_make
```
open terminals and follow [these directions](https://github.com/luminize/ros_hello_machinekit#usage) except for the first line, use:
```
$ halrun -I ~/industrial_training/training/work/1.4/src/ros_hello_machinekit/halfile.hal
```

luminize, 2015-12-16:
On 16 Dec 2015, at 21:38, cdsteinkuehler <notifications@github.com> wrote:

> but it would helpful if
> they worked (particularly things like halscope).

halscope works :)

luminize, 2015-12-16:
So I guess now the ROS folks @gavanderhoorn and @shaun-edwards can have a stab at Machinekit when they have nothing to do in the holidays ;)

RunningLight, 2015-12-16:
From @mhaberler 

```rsyslog setup has a wart I cant figure out why - local1 logs go to /var/log/syslog instead of /var/log/linuxcnc.log```

Could this behavior be related to the changes in file/directory ownership and permissions between Debian and Ubuntu? Looking at my Ubuntu 14.04.3, 

in /var...
```
drwxrwxr-x 13 root syslog   4096 Dec 16 15:41 log
```
and in /var/log
```
-rw-r----- 1 syslog            adm   171701 Dec 16 15:56 syslog
```
Whereas, looking at my machinekit image from RCNelson (just happens to be the machine at my fingertips)

in /var...
```
drwxr-xr-x  9 root root  4096 Dec 16 06:25 log
```
and in /var/log...
```
-rw-r----- 1 root adm    2149 Dec 16 20:17 syslog
```

luminize, 2015-12-16:
@cdsteinkuehler :smile: 
[image]

evanmj, 2015-12-16:
For those with 'dragging' issues in the VM, have you installed the [virtualbox guest additions](https://www.virtualbox.org/manual/ch04.html#idp46854856355616)?  It helps with dynamically changing resolutions seen in the VM window and makes ubuntu in a VM work better from my experience.  It may not fix the dragging bug, but worth a shot if you have not tried installing the additions.

[136 more replies]

closed 2016-03-13 by luminize