machinekit/machinekit #693: Jessie: machinekit shutdown gives dl_close() error message

ArcEye, 2015-06-21:
Just started noticing this message in the terminal when closing machinekit.
Using a recent pull from same date as my last commit ( June 17th )
Also occurs on a rip cloned on May 25th
Can't recall if occurred prior to this date.
It could just be a Jessie thing, I will try booting Wheezy with a fresh clone and see what happens.

Jessie on amd64 with rt10 preempt kernel
The Jessie glibc version is 2.19-18
```
Shutting down and cleaning up Machinekit...
Inconsistency detected by ld.so: dl-close.c: 762: _dl_close: Assertion `map->l_init_called' failed!
Cleanup done
```
Nothing unusual in linuxcnc.log
```
Jun 21 09:48:50 INTEL-i7 msgd:0: startup pid=3709 flavor=rt-preempt rtlevel=1 usrlevel=1 halsize=524288 shm=Posix gcc=4.9.2 version=unknown
Jun 21 09:48:50 INTEL-i7 msgd:0: ØMQ=4.0.5 czmq=2.2.0 protobuf=2.6.1
Jun 21 09:48:50 INTEL-i7 msgd:0: configured: sha=53b3226
Jun 21 09:48:50 INTEL-i7 msgd:0: built:      Jun 20 2015 14:36:30 sha=53b3226
Jun 21 09:48:50 INTEL-i7 msgd:0: register_stuff: actual hostname as announced by avahi='INTEL-i7.local'
Jun 21 09:48:50 INTEL-i7 msgd:0: zeroconf: registering: 'Log service on INTEL-i7.local pid 3709'
Jun 21 09:48:50 INTEL-i7 msgd:0: rtapi_app:3714:user accepting commands at ipc:///tmp/0.rtapi.a42c8c6b-4025-4f83-ba28-dad21114744a
Jun 21 09:48:51 INTEL-i7 msgd:0: zeroconf: registered 'Log service on INTEL-i7.local pid 3709' _machinekit._tcp 0 TXT "uuid=a42c8c6b-4025-4f83-ba28-dad21114744a" "instance=566d361a-17f2-11e5-8991-c04a0002bfd1" "
Jun 21 09:49:03 INTEL-i7 rtapi:0: unload: 'trivkins' not loaded
Jun 21 09:49:03 INTEL-i7 rtapi:0: unload: 'tp' not loaded
Jun 21 09:49:03 INTEL-i7 rtapi:0: unload: 'motmod' not loaded
Jun 21 09:49:03 INTEL-i7 rtapi:0: unload: 'ddt' not loaded
Jun 21 09:49:03 INTEL-i7 rtapi:0: unload: 'hypot' not loaded
Jun 21 09:49:03 INTEL-i7 rtapi:0: unload: 'sim_spindle' not loaded
Jun 21 09:49:03 INTEL-i7 rtapi:0: unload: 'limit2' not loaded
Jun 21 09:49:03 INTEL-i7 rtapi:0: unload: 'lowpass' not loaded
Jun 21 09:49:03 INTEL-i7 rtapi:0: unload: 'near' not loaded
Jun 21 09:49:03 INTEL-i7 rtapi:0: unload: 'or2' not loaded
Jun 21 09:49:03 INTEL-i7 rtapi:0: unload: 'comp' not loaded
Jun 21 09:49:03 INTEL-i7 msgd:0: rtapi_app exit detected - scheduled shutdown
Jun 21 09:49:05 INTEL-i7 msgd:0: msgd shutting down
Jun 21 09:49:05 INTEL-i7 msgd:0: zeroconf: unregistering 'Log service on INTEL-i7.local pid 3709'
Jun 21 09:49:05 INTEL-i7 msgd:0: log buffer hwm: 0% (1 msgs, 119 bytes out of 524288)
Jun 21 09:49:05 INTEL-i7 msgd:0: normal shutdown - global segment detached
```

Nothing whatsoever in dmesg

Appears to be the same issue as per this bug report, albeit non-fatal in this case,
https://sourceware.org/bugzilla/show_bug.cgi?id=11941

Suggests the cause is mismatched dl_open / dl_close pairs and dl_close is being called after
a shared library has been closed by another part of the code, or similar.

Don't have a clue where that might be however.

ArcEye, 2015-06-21:
Can confirm that this solely affects Jessie.
Clone of today (21st June) built on wheezy and does not have same error message upon closing.

Probably just something to be aware of for now unless anything comes to mind as a cause.

cdsteinkuehler, 2015-06-21:
On 6/21/2015 4:24 AM, ArcEye wrote:
> Can confirm that this solely affects Jessie.
> Clone of today (21st June) built on wheezy and does not have same error message upon closing.
> 
> Probably just something to be aware of for now unless anything comes to mind as a cause.

This is likely benign, but could indicate lower level problems.

I just dug into this message since I get it when running the Altera
FPGA Linux development system on an unsupported platform (I use
Debian, they support Redhat).  The issue seems to indicate a
disagreement between the dll instances the program thinks are open vs.
the linker/ld.so.  In several cases, this seems to have been caused by
non-thread safe libraries, in others, it was just bad programming (not
properly keeping track of all opened libraries in all the various code
paths and trying to close a library that wasn't open or had already
been closed).

Regardless, it likely indicates an error somewhere in the Machinekit
code that should eventually be tracked down.

-- 
Charles Steinkuehler
charles@steinkuehler.net

mhaberler, 2015-06-21:
I guess I know who is the culprit. Patch follows. Other than that - harmless

mhaberler, 2015-06-22:
cannot reproduce yet - which config did you run?

ldd (Debian GLIBC 2.19-18) 2.19

gcc version 4.9.2 (Debian 4.9.2-10)

mah@amd64:~/machinekit/src$ uname -a
Linux amd64 3.18.13-rt10mah+ #2 SMP PREEMPT RT Thu Jun 18 17:36:28 CEST 2015 x86_64 GNU/Linux

currently at
commit b3d36af6bc6fa9b23c15a1e2bd8f0bdab2fca09b
Merge: ccc2e80 0d4d3dd
Author: John Morris <john@zultron.com>
Date:   Fri Jun 19 16:24:35 2015 -0500

    Merge pull request #691 from mhaberler/master

    rtapi_math: build userland shared library regardless of build flavor

ArcEye, 2015-06-22:
I have the same

uname -a
Linux INTEL-i7 3.18.13-rt10 #1 SMP PREEMPT RT Tue May 19 13:08:12 BST 2015 x86_64 GNU/Linux

gcc version 4.9.2 (Debian 4.9.2-10)

ldd (Debian GLIBC 2.19-18) 2.19

I was using /configs/sim/axis/axis_mm.ini  and lathe.ini

mhaberler, 2015-06-22:
hm, still no

wild guess - could you try this:

make exit_actions return immediately (before the loop) here:
https://github.com/machinekit/machinekit/blob/master/src/rtapi/rtapi_app.cc#L625

and retry

message still there?

mhaberler, 2015-06-22:
only difference I can see is i7 vs amd64

ArcEye, 2015-06-22:
That is just the name of the computer within the LAN, don't know why it is inserted there.

I will be able to try after lunch.

ArcEye, 2015-06-22:
That works, no message:)

Just to ensure I understood you correctly, this is what I edited the function to
```
// shut down the stack in reverse loading order
static void exit_actions(int instance)
{
    return;
    pb::Container reply;
    size_t index = loading_order.size() - 1;
    for(std::vector<std::string>::reverse_iterator rit = loading_order.rbegin();
	rit != loading_order.rend(); ++rit, --index) {
	do_unload_cmd(instance, *rit, reply);
    }
}
```

ArcEye, 2015-06-22:
There is something screwy going on here.

I removed the return, rebuilt and this time don't get the message.

I will reboot, rebuild from scratch and try again.

mhaberler, 2015-06-22:
hm, sure rtapi_app.cc was recompiled? could be that looming dependency problem

ArcEye, 2015-06-22:
Don't know what is going on.
It happened each time I ran MK on Jessie yesterday, irrespective of what pull I had built.

I have just done a fresh clone of the main MK repo, built it and no message

Done a fresh clone of my own repo, built and again no message.

Don't waste any more time on it, it must be something external to MK, but what it could be I have no idea.

If / when it occurs again I will try to see what could be different to today.

Trouble is MK is about the only program I run routinely from a terminal, so would be the only one I was aware of any ld error messages from.

If I get it again, I will launch a whole bunch of other programs on the command line and see what messages occur at the exit from those.

Sorry to have wasted your time, nothing worse than an intermitent problem

closed 2015-06-22 by ArcEye