machinekit/machinekit #706: Wheezy rt-preempt~amd64.deb has incorrect kernel dependency

ArcEye, 2015-07-09:
The wheezy kernel image package for rt is designated generically as `linux-image-rt-amd64`

The dependency in machinekit-rt-preempt_0.1.1392-1da.bbot.pr704.gitd4c7fb4~1wheezy_amd64.deb
is set at
`linux-image-rt.x86-amd64`

This matches the designations of our other amd64 kernels but not unfortunately Debian's.

I hacked the package to amend the dep and the user who found the problem confirmed that it was resolved with that change

I would submit a PR, if I knew anything about how the buildbot was set up :)

zultron, 2015-07-10:
Ah, already an issue for this.  See #709.

My copy+paste fault:  Xenomai, RTAI and RT_PREEMPT kernel 'featureset' names are `xenomai.x86`, `xenomai.beaglebone`, `rtai.x86`, and `rt`, not `rt.x86`.  A 'featureset' is basically a different build of the same kernel but with different patch sets and kernel configs.

ArcEye, 2015-07-10:
Fixed thank you

closed 2015-07-10 by ArcEye