machinekit/machinekit #715: invalid service name

RunningLight, 2015-07-17:
As reported to the mail list (https://groups.google.com/forum/#!topic/machinekit/TLYZzRSS0o8):

Yesterday, I pulled and built Machinekit on Debian 7.8 amd64 VM and pulled Machineface and Cetus. I downloaded and untarred `machinekit-client_v0.9.2-51_linux_x64.tar.gz` (had a library dependency 
problem with v0.9.3-53, more later). 

Ignoring for a minute its note about the new mklauncher service, I followed @strahlex 's "Testing mkwrapper" wiki page as I had before. Invoking my run script, I see healthy messages relating to configserver and machinekit starting, followed by: 

```ERROR: cannot register DNS 
serviceorg.freedesktop.Avahi.InvalidServiceNameError: Invalid service name ```

Huh? That didn't used to happen. I don't see anything useful in linuxcnc.log. I do see expected Avahi services advertised 

```
$ avahi-browse -a 
+   eth0 IPv6 Log service on debian78-amd64-lxde-virgin.local pid 25777 _machinekit._tcp     local 
+   eth0 IPv4 Preview service on debian78-amd64-lxde-virgin.local pid 25835 _machinekit._tcp     local 
+   eth0 IPv4 Command service on debian78-amd64-lxde-virgin.local pid 25835 _machinekit._tcp     local 
+   eth0 IPv4 Error service on debian78-amd64-lxde-virgin.local pid  25835 _machinekit._tcp     local 
+   eth0 IPv4 Status service on debian78-amd64-lxde-virgin.local pid  25835 _machinekit._tcp     local 
+   eth0 IPv4 File service on debian78-amd64-lxde-virgin.local pid  25835 _machinekit._tcp     local 
+   eth0 IPv4 Log service on debian78-amd64-lxde-virgin.local pid  25777 _machinekit._tcp     local 
+   eth0 IPv4 Machinekit on debian78-amd64-lxde-virgin.local  _machinekit._tcp     local 
```
OK, so I try invoking mklauncher. It also fails but with the -d option I see the same error message as before preceded by lines like: 

```
service: dsname = tcp://%fqdn)s:54443 port = 54443 txtrec =  ['dsn=tcp://%(fqdn)s:54443',...] name = Machinekit Launcher on %(fqdn)s 
```

Hmmm. Why is "%(fqdn)s" unresolved (or "replaced by service announcement" according to a comment in mklauncher)? 

Looks like this is related to #707 and #708

strahlex, 2015-07-18:
See also https://github.com/strahlex/QtQuickVcp/issues/34

The debug output is correct. `%(fqdn)s` is resolved in the Python module https://github.com/machinekit/machinekit/blob/master/lib/python/machinekit/service.py#L39

However, this line here looks weird: `tcp://%fqdn)s:54443` can you verify that the first bracket is missing?

strahlex, 2015-07-18:
Correct output should look like `service: dsname = tcp://%(fqdn)s:50770 port = 50770 txtrec = ['dsn=tcp://%(fqdn)s:50770', 'uuid=a42c8c6b-4025-4f83-ba28-dad21114744a', 'instance=9f87259e-2d29-11e5-8a61-247703279414', 'service=launcher'] name = Machinekit Launcher on %(fqdn)s
`

RunningLight, 2015-07-18:
Sorry, the missing left paren was my fault. I was in a hurry and tried to type on one machine while reading on another. You'd think I'd know better by now. In fact, I get "correct" output.

Once I work through your hints (and those from Yishin on the list) and make everything copacetic, I'll have to think about how we might create a FAQ for this service discovery stuff. Having to search issues logs for two different repos isn't for the faint of heart  :smirk:

RunningLight, 2015-07-18:
I confess I'm getting annoyed. I just spawned a fresh VM of Debian 7.8 (Wheezy) [hostname 'debian78-amd64-pkg-test'], installed MK from packages, and...wait for it...mklauncher *just works*. All the expected services are being published without error messages and I can read all about them in various avahi/bonjour browsers on my LAN. Don't know yet what's going on with the build-from-sources VM I was  working with.

I've been reading the source and I see how I misinterpreted the appearance of '%(fqdn)s' in the diagnostic messages. Sorry for this red herring. 

Reading strahlex/QtQuickVCP#34 was interesting but hasn't illuminated my particular problem.

I've gone back to the avahi website to find out what constitutes a "valid service name": it must be a valid UTF-8 string less than 64 bytes long, and it must not be NULL. Can't see yet how I, as an enduser, can be violating these requirements but there you are. On the other hand, the error message emitted by mklauncher doesn't come right out and say what the invalid service name is. Maybe some concatenation of things causes me to exceed the byte limit.

RunningLight, 2015-07-19:
Yes, it's the concatenation of strings which is doing me in. 

I use descriptive hostnames for my VMs. I had given a 26-character hostname to the host of the built-from-source MK which was failing. That turns out to be one character too many. 

With a Machinekit installed from packages on one Debian 7.8 (Wheezy) amd64 VM and a Machinekit built from source on another, mklauncher throws an "invalid service name" error when it tries to publish the launchercmd service *iff* the hostname is 26 or more characters long. For hostnames fewer than 26 characters in length, mklauncher succeeds on both Machinekits.

I'm supposing one can't independently establish a limit on the hostname length because the length of the concatenated service name also depends on the specific service being published (e.g., `launchercmd` adds more characters than `log`, etc. ), but it's interesting that a 26-character hostname plus the 6-character domain `.local` turns out to be a 32-character string (one of those pesky powers of 2 which used to be meaningful to us minicomputer programmers counting bits).

So, keep hostnames as short as possible. (Shades of HP-UX and its insistence that hostnames should not exceed 8 characters!)

strahlex, 2015-07-20:
Or maybe we can just shorten the service name. `Machinekit Launcher on xyz` is unnecessary long. I originally named it `Machinekit Launcher` to prohibit conflicts with the configserver naming services `Machinekit on xyz` per default. Since the Mklauncher is now the entry point that users will see we could just rename the launcher service to "Machinekit on xyz" and the config service "config on xyz".

strahlex, 2015-08-03:
Taking a closer look at how other devices announce itself following naming scheme could be adapted:
`xyz @ hostname`. This conform with several network printers and Apple devices I see in my network.

mhaberler, 2015-11-25:
ist this still an issue ?

strahlex, 2015-11-25:
At least did not pop up anywhere since then.

ArcEye, 2018-08-01:
Looks like fixed since

closed 2018-08-01 by ArcEye