machinekit/machinekit #727: rtapi_app_main(hal_parport): -19 No such device

rosaldorosa, 2015-07-24:
Linux debian-cnc 3.8-1-xenomai.x86-amd64 #1 SMP Debian 3.8.13-9 x86_64 GNU/Linux
debian jessie
--------------------------------------------------------------------------
MACHINEKIT - 0.1
Machine configuration directory is '/home/cnc/machinekit/configs/by_interface/parport/gantry'
Machine configuration file is 'gantry-hs.ini'
Starting Machinekit...
io started
halcmd loadusr io started
gantry.hal:14: insmod failed, returned -19:
rtapi_app_main(hal_parport): -19 No such device

See /var/log/linuxcnc.log for more information.
Shutting down and cleaning up Machinekit...
--------------------------------------------------------------------------
/dev/parport exists
dmesg:
[    5.551065] ppdev: user-space parallel port driver
[    5.595380] parport_pc 00:08: reported by Plug and Play ACPI
[    5.595441] parport0: PC-style at 0x378, irq 5 [PCSPP,TRISTATE]
[    5.691916] lp0: using parport0 (interrupt-driven).
[    5.997819] systemd-udevd[188]: starting version 215
[    6.884438] shpchp: Standard Hot Plug PCI Controller Driver version: 0.4
--
[   27.881832] cgroup: systemd (1) created nested cgroup for controller "devices" which has incomplete hierarchy support. Nested cgroups may change behavior in the future.
[ 1855.293554] parport0: cannot grant exclusive access for device ppdev0
[ 1855.293560] ppdev0: failed to register device!

lsmod
parport_pc             28152  1 
parport                42214  3 lp,ppdev,parport_pc

mhaberler, 2015-07-26:
see this: https://groups.google.com/forum/#!topic/machinekit/oJvb2mLNQl4
also google for 'linuxcnc parport blacklist' for hints

mhaberler, 2015-07-27:
@rosaldorosa please report success - then document what you did here - so the issue can be closed

mhaberler, 2015-11-25:
closing due to inactivity

closed 2015-11-25 by mhaberler