machinekit/machinekit #733: FDM should be renamed to FFF

strahlex, 2015-08-05:
Because of trademark and patent issues the therm FDM (fused deposition modeling) should be renamed to FFF (fused filament fabrication). See http://reprap.org/wiki/Fused_filament_fabrication

I did not know that the term FDM is trademarked when I used it name 3D printing related stuff in the Machinekit repo.

luminize, 2015-08-05:
Any place specific?

strahlex, 2015-08-05:
A few places: https://github.com/machinekit/machinekit/search?p=2&q=fdm&utf8=%E2%9C%93
Changing it will definitely break compatibility for configs not contained within Machinekit

luminize, 2015-08-05:
did a search on using trademarked names. http://www.lisashea.com/lisabase/writing/gettingyourbookpublished/trademarks.html Since it's not used for advertising a product or similar I would be hesitant to change this. There are more occurences for branch names, like probotix, or intel. So that would be a difficult undertaking. In short, I don't know, I'm not much use here.

strahlex, 2015-08-06:
The question is if I advertise a product that uses Machinekit could that lead to a problem? FDM is obviously used to describe the technology.

I have just seen that some hal remote components are also named `fdm-xy`

dlarue, 2015-08-18:
if it's a problem then what are the options/steps?
To not change the option might be to license the use, ask permission, etc by contacting the owner.
Otherwise a change must be made( I like FFF better ).

No changes are easier so what would be the harm asking the Trademark owner? We see Trademarks disclaimers all the time on various products, documents, etc. If they say no or want money then the answer is defined and the work migrating must be done.

strahlex, 2015-08-19:
Steps for renaming FDM to FFF would include:
- announcing the change on the mailing list so users can prepare their configuration
- renaming subroutine folders 
- renaming some HAL remote components
- renaming INI file sections
- renaming Python module names
- finding all other places where FDM was used (e.g. documentation)

RunningLight, 2015-08-21:
http://lawitm.com/3d-printing-law-trademarks-why-fdm-isnt-for-everybody/

Reading this leaves me thinking we're in a gray area. The Machinekit project is not trying to sell an "FDM Printer", it's trying to provide software which can be used to control FDM printers among many other types of machines. FFF seems safer but I'm not a lawyer nor do I play one on TV.

strahlex, 2015-08-22:
Interesting article. Since it seems we are in some gray area I would recommend proceeding with the renaming. It's not that much work. We just have to make sure that the fallout is small.

ArcEye, 2018-08-01:
Seem to have got by for 3 yrs without being sued and nothing done to change name.

closed 2018-08-01 by ArcEye