machinekit/machinekit #734: discriminant <negative value> < 0 in velocity calculation!

strahlex, 2015-08-05:
The error `discriminant <negative value> < 0 in velocity calculation!` keeps appearing when printing object at high speeds. The error seems to be by no way harmful or disturbing the execution of the program it is just annoying and confusing for the user.

@robEllenberg This error message seems to come from the TP. Do you have an idea what could cause it?

strahlex, 2015-08-05:
See https://github.com/machinekit/machinekit/blob/5a1a3241aa643053bb2275a3d97581fd38eb9979/src/emc/tp/tp.c#L2161

robEllenberg, 2015-08-05:
Hi Alexander,

That condition is not necessarily an error,  but in the past it has been
correlated with trajectory planner issues, so it used to be useful when I
was doing heavy development. Now, it can be suppressed without issue.

Also, thanks to a bunch of testing by Sam, Seb, and others working on
Linuxcnc 2.7, we've tracked down and fixed a few more tp edge cases.
Rebasing my most recent feature branch onto machinekit should cure this
particular issue. I'll take a stab at it this week.

Rob

See
https://github.com/machinekit/machinekit/blob/5a1a3241aa643053bb2275a3d97581fd38eb9979/src/emc/tp/tp.c#L2161

strahlex, 2015-08-12:
@robEllenberg Any progress on the rebase?

I have noticed that this error message appears most frequently when I use feed override.

robEllenberg, 2015-08-13:
It's about 95% there, I'm fixing some artifacts from the travel.  Once I
can run a batch test to make sure I didn't break anything, it will be
merge-ready.

Rob

ArcEye, 2018-08-01:
Think this has been overtaken by events

closed 2018-08-01 by ArcEye