machinekit/machinekit #741: Tn M6 REMAP issue in MDI mode

f1oat, 2015-08-30:
A simple Machinekit configuration file showing the issue is available at [https://github.com/f1oat/files/tree/master/sim.axis_remap_mdi_issue](https://github.com/f1oat/files/tree/master/sim.axis_remap_mdi_issue)

* Calling the following subroutine in MDI mode ( `o<totos> call`)  prints only "toto1" message.
```javascript
o<totos> sub
(MSG, toto1)
T1 M6
(MSG, toto2)
T2 M6
o<totos> endsub
m2
```
* Calling it from a toto.ngc file launched in AUTO mode prints both "toto1" and "toto2" messages.
* If REMAP statement is removed from the INI file, the problem disappears.

mhaberler, 2015-08-30:
as a workaround, try to use a Python remap procedure instead of NGC

this is more likely to work

f1oat, 2015-08-31:
Thanks. I have implemented the same test in Python, and that works!
I have updated [https://github.com/f1oat/files/tree/master/sim.axis_remap_mdi_issue](https://github.com/f1oat/files/tree/master/sim.axis_remap_mdi_issue)
Use  "linuxcnc axis_remap_python.ini" for this variant. 
Queuebuster seems working as I can see console debug messages from the Python code printed according to real machine motion.

However, I had to comment out "self.toolchange_flag = True" in my [remap.py](https://github.com/f1oat/files/blob/master/sim.axis_remap_mdi_issue/python/remap.py#L19) to avoid SIGSEV in milltask
I do not know if there is any drawback without this line coming from [stdglue.py](https://github.com/machinekit/machinekit/blob/5a1a3241aa643053bb2275a3d97581fd38eb9979/nc_files/remap_lib/python-stdglue/stdglue.py#L197)

mhaberler, 2015-08-31:
good to hear you solved it

re segfault: this should be fixed, I would like to reproduce
please indicate procedure

f1oat, 2015-08-31:
I have rewritten all my tool change code in Python and it works in simu. Test on real machine to be done tomorrow.

For the segfault, get last version of [https://github.com/f1oat/files/tree/master/sim.axis_remap_mdi_issue](https://github.com/f1oat/files/tree/master/sim.axis_remap_mdi_issue)
Then, remove the # at beginning of [remap.py line 19](https://github.com/f1oat/files/blob/master/sim.axis_remap_mdi_issue/python/remap.py#L19)
Launch  "linuxcnc axis_remap_python.ini".
After homing, in MDI issue "o\<totos\> call" command.
The segfault should appear.

mhaberler, 2015-09-01:
I've failed to reproduce this segfault

please do the following:

- start the config
- find out the pid of milltask
- in a separate window, gdb -p <pid of milltask>
- hit 'c' for continue
- reproduce the segfault
- do a backtrace in the gdb window which should have stopped on the SIGSEGV
- post it here

just in case - do a make clean and make before retrying

thanks!

f1oat, 2015-09-01:
Strange. Are you sure you removed the # in remap.py line 19?
I have updated machinekit to last commit in github + make clean & make.
The segfault is still here.
```javascript
Program received signal SIGSEGV, Segmentation fault.
Interp::execute_call (this=this@entry=0x1334340, settings=settings@entry=0x1335350, current_frame=0x13473b0, 
    call_type=<optimized out>) at emc/rs274ngc/interp_o_word.cc:278
278		    if (remap->remap_py || remap->prolog_func || remap->epilog_func) {
(gdb) bt
#0  Interp::execute_call (this=this@entry=0x1334340, settings=settings@entry=0x1335350, current_frame=0x13473b0, 
    call_type=<optimized out>) at emc/rs274ngc/interp_o_word.cc:278
#1  0x00007f0c4f0dd9d2 in Interp::convert_control_functions (this=0x1334340, block=0x13354c0, settings=0x1335350)
    at emc/rs274ngc/interp_o_word.cc:740
#2  0x00007f0c4f13e6d3 in Interp::_execute (this=this@entry=0x1334340, command=<optimized out>)
    at emc/rs274ngc/rs274ngc_pre.cc:294
#3  0x00007f0c4f13ef16 in Interp::execute (this=this@entry=0x1334340, command=<optimized out>)
    at emc/rs274ngc/rs274ngc_pre.cc:511
#4  0x00007f0c4f13ef6c in Interp::execute (this=0x1334340, command=<optimized out>, line_number=<optimized out>)
    at emc/rs274ngc/rs274ngc_pre.cc:527
#5  0x0000000000412deb in emcTaskPlanExecute (command=0x0, line_number=<optimized out>) at emc/task/emctask.cc:593
#6  0x0000000000420361 in emcTaskIssueCommand (cmd=0x12a12a0) at emc/task/emctaskmain.cc:2180
#7  0x000000000040ffbb in emcTaskExecute () at emc/task/emctaskmain.cc:2563
#8  main (argc=<optimized out>, argv=<optimized out>) at emc/task/emctaskmain.cc:3323
(gdb) list
273		block_pointer cblock = &CONTROLLING_BLOCK(*settings);
274		remap_pointer remap = cblock->executing_remap;
275	
276		switch (settings->call_state) {
277		case CS_NORMAL:
278		    if (remap->remap_py || remap->prolog_func || remap->epilog_func) {
279			CHKS(!PYUSABLE, "%s (remapped) uses Python functions, but the Python plugin is not available", 
280			     remap->name);
281			plist.append(settings->pythis);   //self
282			current_frame->tupleargs = bp::tuple(plist);
(gdb) p remap
$1 = (remap_pointer) 0x0

```