machinekit/machinekit #742: Synchronized trajectories across HAL instances are not possible

luminize, 2015-08-31:
When multiple machines need to execute a synchronized path (like 2 robots which must work together) it is not possible to start and synchronize movement at the same time.

Since at this time joint constraints for non-trivial kinematics are not obeyed https://github.com/machinekit/machinekit/issues/438 plus it would be difficult to have a robot constraining its peers the logical thing would be to off-line plan the motion and feed the segments to the interpolator.

However, there then needs to be a common starting point of the motion

mhaberler, 2015-08-31:
To distribute such a job over several HAL instances on different hosts, each driving a machine, for coordinated motion to happen I think the following prerequisites are needed:

- the trajectories of each machine should be planned offline such that all can be played out at a given rate without further acc/vel limit application which would affect relative speeds
- the playout of the trajectories must start synchronized at all HAL instances.

The latter requirement mandates a synchronized clock on all HAL instances, and possibly a motion command to start a playout at  a given point in time. As the trajectories and their time/vel/acc properties are precomputed the motion of the machines should reflect the planning result.

A suitable vehicle for time-syncing HAL instances is the [IEEE1588 protocol (aka Precision Time Protocol)](https://www.belden.com/docs/upload/Precision_Clock_Synchronization_WP.pdf)  which was specifically designed for this purpose - syncing hosts over Ethernet-type media down to the sub-microsecond range.

Fortunately a PTP implementation is available in the debian package stream (`apt-get install ptpd`).

mhaberler, 2015-08-31:
To make an IEEE1588 synchronized clock usable in HAL we need to
- synchronize the clocks of HAL hosts to a master (as above)
- provide a means within HAL to refer to this clock
- provide an error metric which can be used as an estop source in case synchronization fails or is not achieved within a given error band

the clock is simply accessed by the clock_gettime() family of calls; this is in fact the call supporting `rtapi_get_time()` with RT-PREEMPT, so on this RT kernel no extra facility is needed (Xenomai is TBD).

The estimated error of the synchronisation can be retrieved by the `adjtimex()` system call (`esterror` field of the timex structure ) and also displayed in the shell with `adjtimex -p` (`apt-get install adjtimex`).

One way to go about this would be a userland component which retrieves the `esterror` value periodically (eg a few times per second) and mirrors this value in a HAL pin. Everything else like thresholding and estop can go from there.

Coordination between all HAL instances will be needed to establish a global `HAL nodes sychronized` - coordinated trajectory playout can begin - property.

slickqt, 2015-11-06:
we use M code for sync the action,that is one sync position is a M code, for example with 10 sync position, we set M160-M170 as sync point, when the first machine reach that point it wait until the other machine get to that point, when them get to same point, it will run continue.
example:
machine 1:
G0 X100
M160;(machine 1 is ready,sync with machine 2)
G0 X200;
.....
M161;(another sync point with machine 2)

machine 2:
G0 Y300;
M160;(sync with machine1)
...
M161;(sync with machine 1)

luminize, 2015-11-06:
I think that is not a full synchronisation between machines.
If you use it like this, then you can do waiting between machines.
But what i don’t see in your solution is how you will make sure that both machines are synchronised on the entire path

dbskccc, 2015-11-07:
yes ,it is not full synchronisation, total synchronisation is complicated even in the same machine, it related to more factors such as motor/servo driver and the mechanics.

closed 2018-08-04 by ArcEye