machinekit/machinekit #749: Missing Debian package dependency python-avahi

strahlex, 2015-09-10:
The package python-avahi is required to use the Machintalk applications mklaucher, mkwrapper and configserver. So far the machinekit Debian package does not require this dependency. I think it would be a good idea to add it.

strahlex, 2015-09-15:
However, `python-avahi` is in the dependency list of the `machinekit` package. I have no idea why fetching the dependency of the `machinekit` package using `apt-get` does not install `python-avahi`.

strahlex, 2015-09-15:
Turns out I installed the `build-deps` only. However, it may be a good idea to find a way to install the Machinekit runtime dependencies for RIP installs too.

mhaberler, 2015-09-15:
yes, that is a bummer. 

Maybe the method with the pseudo-package just having dependencies as created by mk-build-deps could be reused for that? that is very elegant.

strahlex, 2015-09-16:
Thats a good idea. I wonder if other projects use the same method?

kinsamanka, 2015-12-18:
Can you try:
```
apt-get install `apt-cache depends machinekit | awk '/Depends:/{print$2}'`
```

mhaberler, 2016-02-25:
is this still pending?

strahlex, 2016-02-29:
Yes. @kinsamanka I get
```
Reading state information... Done
E: Unable to locate package <libgl1>
E: Unable to locate package <libglu1>
E: Unable to locate package <machinekit-rt-threads>
E: Unable to locate package <python
E: Unable to locate package <tclreadline>
```

strahlex, 2016-03-08:
This command works:
```
sudo apt-get --ignore-missing install $(apt-cache depends machinekit | awk '$1~/Depends/{/g' | tr "<>" " " | sed 's/machinekit-rt-threads//g')
```

luminize, 2017-07-27:
a year of no activity, closing, please re-open if wrongly closed

closed 2017-07-27 by luminize