machinekit/machinekit #750: Build system not exiting on errors

zultron, 2015-09-10:
This error found in a recent build (both RIP and package) broke userland (only) module builds badly enough that they didn't complete; yet, `make` continued building, finally exiting with a success code.
```
make[1]: *** No rule to make target `objects/posix/halcomp-srcs/hal/i_components/modulus.o', needed by `../rtlib/posix/modulus.so'.  Stop.
make[1]: *** Waiting for unfinished jobs....
```

This is very confusing, since the Buildbot continued on to package builds and unit tests, which failed in bizarre ways.

http://buildbot.dovetail-automata.com/builders/d7-32-bld/builds/964
http://buildbot.dovetail-automata.com/builders/d7-32-pkg/builds/1156
http://buildbot.dovetail-automata.com/builders/d7-32-pos-tst/builds/835

bobvanderlinden, 2015-12-17:
Is this solved by #791 ?

ArcEye, 2018-08-01:
Historical and solved by subsequent developments

closed 2018-08-01 by ArcEye