machinekit/machinekit #753: Cannot easily use Machinetalk in other projects

bobvanderlinden, 2015-09-12:
At the moment using Machinetalk in other projects is not a smooth process. Machinetalk-protobuf needs to be cloned (using git clone or git subtree) and the .proto files need to be compiled into .h, .c, .py, .js files. After that user-projects need to copy those files, reference the machinetalk-protobuf directory or use git subtree and integrate the Makefile into their build process.

This is a general problem that (I think) needs to be solved per language/framework.

For Python projects it's standard to have a package in pypi, so that people can install it using `pip install machinetalk-protobuf` or reference machinetalk-protobuf in their setup.py. This is covered in issue https://github.com/machinekit/machinetalk-protobuf/issues/26

For NodeJS projects it's standard to have a package in npmjs, so that people can install it using `npm install machinetalk-protobuf` or reference the machinetalk-protobuf in their setup.py. This is covered in issue https://github.com/machinekit/machinetalk-protobuf/issues/25

For C/C++ projects it's standard to have include files and libraries available through pkg-config, installable through `make install`. Additionally it is helpful if the .proto files are installed in a standard location, also using make install. This is covered in issue https://github.com/machinekit/machinetalk-protobuf/issues/24

strahlex, 2015-09-15:
Btw. I have create a Qt pro und pri file to use in Qt projects some time ago: https://github.com/strahlex/QtQuickVcp/tree/master/3rdparty/machinetalk-protobuf-qt
It basically creates a machinekit-protobuf shared library. The reason I created this was because the build process on Mac, Windows, Android and Linux is very different and Qt provides a good abstraction for that. Using the Makefile provided with machinetalk-protobuf was not a viable solution for Windows and Android.

bobvanderlinden, 2015-09-15:
Ah! That works very nicely. Proper install rule and everything. Do you suggest other applications to use this same library to link to machinetalk-protobuf or should they build the protobuf files into their own library/code?

strahlex, 2016-07-11:
close this issue here since duplicated in machinetalk-protobuf repo

closed 2016-07-11 by strahlex