machinekit/machinekit #754: INI file loading always creates tmp file

strahlex, 2015-09-15:
Some replacement magic is used to handle includes and environment variables when loading INI files with the linuxcnc script. However, it looks like replacement also happens when no includes and environment variables are used. This can be verified by echoing the INIFILE variable after the replacement operations. It is always located in the `tmp` folder.

The culprit of this problem is somewhere here in the linuxcnc script:
https://github.com/machinekit/machinekit/blob/master/scripts/linuxcnc.in#L349

strahlex, 2015-09-16:
Looks like this was added by mistake: https://github.com/machinekit/machinekit/commit/34c00e082c6446dfbc541de6163a4382e8473dfe#diff-d91a2e9cb50a65ff52946ba9a6546f38R335

ArcEye, 2015-09-17:
"Looks like this was added by mistake: 34c00e0#diff-d91a2e9cb50a65ff52946ba9a6546f38R335"

The git 'blame game' must be played with care, git is a flawed piece of software, nothing more so than the alleged history that blame displays.

Immediately before the code in question is this 
```
@@ -329,6 +332,26 @@ if [ "$PREVINIFILE" != "$INIFILE" ]; then
     trap "rm -f $INIFILE" 0 1 2 3 15; # Clean up the temporary file
```
which shows the code pre-existed and was moved

I inserted code at the beginning of the script which shifted all the line numbers, one of my gripes with git is its inability to differentiate between real change and line numbers, spacing etc. and its general inconsistency.
In some instances if you move a file it will show it as a move, in others it will show a deletion and a new file being created, depends entirely on the work sequence.

Returning to this issue, if you have a fast computer and time on your hands run
```
git grep 'trap "rm -f $INIFILE" 0 1 2 3 15; # Clean up the temporary file' $(git rev-list --all)
```

When I did this I returned hundreds of hits right back to this one
```
commit d337feba8e2a2d2f8143fa0b6f7fefa92634c2c4
Author: Alexander Rössler <mail.aroessler@gmail.com>
Date:   Sat Jan 10 12:34:47 2015 +0100

    linuxcnc: added environment variable replacement for ini files
```
Linuxcnc has always used a tmp copy of the inifile, but not with this code

Is it possible you are the author of your own problems?

ArcEye, 2015-09-17:
Found a far better search method, not sure the previous one had quite finished when I looked at it
```
git log -S 'trap "rm -f $INIFILE" 0 1 2 3 15; # Clean up the temporary file' --source --all
```
```
commit e8428fac81c057095956a3529c93cd63b1e40250 refs/heads/master
Author: Alexander Rössler <mail.aroessler@gmail.com>
Date:   Mon Jul 20 12:00:42 2015 +0200

    Squashed 'src/machinetalk/proto/' changes from 9378252..152d2a2
    
    152d2a2 Merge pull request #18 from strahlex/machinekit3
    e86a66e machinetalk-protobuf: added default PKG-CONFIG variable to make upstream merge possible
    e350cc0 machinetalk-protobuf: cleaning up duplicate comments
    46c9536 Merge pull request #16 from strahlex/shutdown
    6536eae Merge pull request #17 from strahlex/increments
    3860b1b status: adding per axis increments value to config message
    cdedb9c enabled shutdown and confirm shutdown message types
    bece831 machinetalk/mkwrapper: added per axis max-acceleration and max-velocity and simplified protobuf message update of config channel
    
    git-subtree-dir: src/machinetalk/proto
    git-subtree-split: 152d2a27a326fd1dc3f90a33792c82d9c5538ba6

commit 34c00e082c6446dfbc541de6163a4382e8473dfe refs/remotes/origin/hal-triggers
Author: Mick <arceye@mgware.co.uk>
Date:   Wed Feb 25 16:20:02 2015 +0000

    Cosmetic rebranding to Machinekit
    
    Graphics and text changes to show Machinekit running
    Symlinks to original user dirs and new ones created if not present
    Text changes to configs etc
    
    Changes just skin deep, underlying libs, logs etc same
    
    Signed-off-by: Mick Grant <arceye@mgware.co.uk>

commit 07e96d5df4631e6157f010110880431e09c36b7b refs/remotes/origin/hal-triggers
Author: Alexander Rössler <mail.aroessler@gmail.com>
Date:   Thu Jan 15 22:12:05 2015 +0100

    Revert "linuxcnc: added environment variable replacement for ini files"
    
    This reverts commit d337feba8e2a2d2f8143fa0b6f7fefa92634c2c4.

commit d337feba8e2a2d2f8143fa0b6f7fefa92634c2c4 refs/remotes/origin/hal-triggers
Author: Alexander Rössler <mail.aroessler@gmail.com>
Date:   Sat Jan 10 12:34:47 2015 +0100

    linuxcnc: added environment variable replacement for ini files

commit ed3090e7c8ebb904117e6b9f7975e282e7aaaeb8 refs/remotes/origin/hal-triggers
Author: Alexander Rössler <mail.aroessler@gmail.com>
Date:   Sat Jan 10 12:44:04 2015 +0100

    linuxcnc: fix temporary file creation when using include
```

Looks like ed3090e7 refers specifically to what you are talking about?

strahlex, 2015-09-17:
Ah, yes I remember. We removed this patch after a few days. Interesting how this was added again.

Nevertheless, the faulty code is now removed.

closed 2015-09-17 by strahlex