machinekit/machinekit #757: Accessing HAL using Machinetalk

strahlex, 2015-09-16:
The current approach of accessing HAL using Machinekit work by creating HAL remote components. This is fine if you want to create custom user interfaces. However, this approach feels unnatural when one only want to access a specific HAL pin/signal. 

For example a user wants to create a LED to visualize the value of the `axis.0.homed` pin of the motion component. With HAL remote components you would create a new HAL remote component in HAL, connect it to the `axis.0.homed` pin using a HAL signal and create a HAL remote component and button on the client. This method seems to be way to complicated for the given scenery. It would be great to have a way to access HAL pins and HAL signals from a Machinetalk client without creating HAL remote components. This significantly increase the user experience when using HAL with Machinetalk.

mhaberler, 2015-09-16:
initial remarks:

- the way pin ownership works a pin **must** be owned by a component, or an instance. You cannot create a pin which is unowned.  That said, it is in principle conceivable to add or delete pins from a (possibly remote) component on the fly; the legacy code restriction - pins cannot be added or removed once a component is ready'ied - is history.

- currently the subscription mechanism, as well as change detection, works on the remote component or group level. It is conceivable to add a pin or signal property meaning 'individually report changes at the pin/signal level in addition/instead of the group level'. However, one needs to be aware that currently HAL names are not unique across objects so code would have to check that a group/rcomp name does not equals some other HAL object to avoid aliasing. But that is solvable. The zeromq subscription matching code is certainly strong enough to cope with the traffic.

- the primary reason why aggregates like rcomps or groups are reported is matching and network interaction performance. Once a group or rcomp is readied for change detection a compiled form is created (compiled_group/compiled component) which trades off matching speed for memory. The way things are implemented now is - once a group or rcomp is adopted by haltalk, it is transformed into the compiled form. But there is nothing to be said in principle against recompiling if a change is detected (signal or pin added/deleted).

- the network traffic will increase significantly as individual pin/signal reports translate into individual update messages. That is a key factor I would keep in mind, especially for slow links and dozens of UI objects. I do see a way around this though (in principle): replicate state of groups and rcomps **to the client** as a whole, and do the individual subscription client-side; essentiall mirror some subset of HAL state and tack onto that state locally. This would also have benefits for general state inspection by clients (we are lacking something like the HAL explorer in Axis). There is a lot to be said for aggregated updates - much of the overhead in networking applications is per transaction and not so much per item in a transaction, and that is true for haltalk as well.

- there is one semantic issue I would keep in mind. If pins/signals are reported individually the temporal relation of change reports may change. I think there is a change serial which enables correlating detection runs, but it is something to keep in mind.

I will consider the idea - I need to sleep over this. I see your point, but if that feature comes in it better not crater haltalk performance and prop up network traffic an order of magnitude.

mhaberler, 2015-09-17:
having thought over it a bit. It helps to stand back a bit and look at the bigger picture.

The pattern we use with machinetalk is by selective and mutual _state replication_ . State driven by a client is replicated into HAL state, and HAL state is replicated into (possibly many) clients.

A haltalk client obtains a view (a subset if you will) of the overall HAL state (currently by subscribing to aggregates like group or a rcomp). For client-originated state modification the interaction currently is on a per-scalar level (addressed by name or faster by HAL object id)

If I understand your proposal correctly the issue you have is with the handling of aggregates on the HAL-to-client side - so the requirements:
- to create a remote component
- to create a group
- to subscribe to the rcomp or group names
- while - on the UI element level - you are really interested in the per-scalar interaction, not those aggregates, and an API which focuses on the per-scalar interaction.

Does that describe your issue? If it does, it might help to separate the issues:
1. replicating state as efficiently as possible, in terms of HAL-side and network overhead - which is why aggregates are used
2. how to access that replicated state (on an aggregate or per-scalar level)

So the task would be to enable (2) without killing (1).

mhaberler, 2015-09-17:
sorry, your `axis.0.home` example is just convoluted.

You link that pin to a signal, add the signal to a group and you get the value.

strahlex, 2015-09-17:
What I am looking for is a way for a haltalk client to watch a given pin without touching the HAL configuration. If a HAL group could be used for this purpose there needs to be a way to add a pin using e.g. its name or handle to this group from the client side. There could be for example a "master" or "general" group one could subscribe to.

The end user should be able to select a HAL pin or signal from a list (like the Axis feature) and link it to the value of user interface component. Usually this operation will not happen very often so compiling the watch-list may work. The HAL describe operation of halrcmd may give us the list of HAL pins and signals to select from.

closed 2018-08-04 by ArcEye