machinekit/machinekit #765: Create Vagrant VMs for Machinekit

strahlex, 2015-10-07:
Today I stumbled upon  [Vagrant](https://www.vagrantup.com/). This tool claims to simplify setting development environment using VirtualBox and other VMs. The functionality is very simple but powerful:

**To create a Vagrant configuration:**
1. Choose a base image (e.g. Debian Jessie)
2. Create a provisioning script that install software (in our case Machinekit, Qt, etc...)

**To use it:**
1. Install VirtualBox and Vagrant
2. Clone a git repo containing the Vagrant config
3. Start vagrant
4. Use the VM
5. Reset the when you broke it

The advantage over the naked VirtualBox VM is that one can fully automate the setup process and a working development environment can be set up in a few keystrokes.

I can imagine creating 3 different configurations:
- a Machinekit test VM -> package install to try out Machinekit on the Desktop
- a Machinekit developer VM with RIP install and development tools
- a Machinekit SDK config with all the Qt UI tools

strahlex, 2015-10-07:
The best thing about Vagrant is that all configs could share the same base provisioning script setting up the base Debian environment. Mixing the provisioning scripts together should also be possible.

luminize, 2015-10-07:
That sounds very usefull!

luminize, 2015-10-07:
I've installed vagrant and I already had virtual box.
`$ vagrant box add debian/jessie64` installs a debian jessie box.
will try this out further, because this could be the "easy install" regarding a VM

strahlex, 2015-10-07:
Yes, I already crated a basic Machinekit package provisioner. Will upload the repo when I come home.

strahlex, 2015-10-08:
Here we go: https://github.com/strahlex/machinekit-vagrant
vagrant up sets up a a Jessie VM with Machinekit packages

luminize, 2015-10-08:
I'm just off to my project but will check this evening for sure. Did a bare Jessie setup yesterday evening.

luminize, 2015-10-08:
this could solve https://github.com/machinekit/machinekit/issues/229

luminize, 2015-10-08:
@strahlex I try your vagrant file, and the line https://github.com/strahlex/machinekit-vagrant/blob/master/Vagrantfile#L40 is problematic since different users have different locations.
I'll make a PR for that.

so, here are my results.
after `vagrant up` letting the entire terminal run to the the prompt returns.
```
==> default: Setting up machinekit (0.1.1479-1da.bbot.pr764.gitce22b09~1jessie) ...
==> default: Restarting udev (via systemctl): udev.service
==> default: .
==> default: Setting up machinekit-posix (0.1.1479-1da.bbot.pr764.gitce22b09~1jessie) ...
==> default: Processing triggers for libc-bin (2.19-18+deb8u1) ...
==> default: Processing triggers for python-support (1.0.15) ...
Bass-MacBook-Pro-3:machinekit-vagrant basdebruijn$
```

The VM gives me the "oh no" feeling
[image]

I sent the shutdown signal to the VM.
then `vagrant up` again in the terminal and voila, the LXDE starts.

@mhaberler @strahlex We might make a machinekit organisation repo with different folders for different setups, like packages for trying out, a git based development image, Alex' QT work.

I'm very excited about this :)

strahlex, 2015-10-10:
@luminize I have to find a way to start LXDE right away.

luminize, 2015-10-11:
I just do `vagrant up` and that starts the VM and that just starts LXDE. I experience no troubles.

luminize, 2017-07-23:
@machinekoder This should be in the documentation, right? If still needed please open a new issue in machinekit-docs.

closed 2017-07-23 by luminize