machinekit/machinekit #771: machinekit/lib/python/_hal.so: undefined symbol: _Py_ZeroStruct

luminize, 2015-10-17:
I try to make a userland python component and took the example from the docs
```python
#!/usr/bin/python
import hal, time

h = hal.component("can-device")
h.newpin("in", hal.HAL_FLOAT, hal.HAL_IN)
h.newpin("position", hal.HAL_FLOAT, hal.HAL_OUT)
h.ready()
try:
    while 1:
        time.sleep(0.01)
        pass
        h['out'] = h['in']
except KeyboardInterrupt:
    raise SystemExit
```
this gets me the error message:
```bash
bas@debian:~/canopen-machinekit$ python create-can-device.py 
Traceback (most recent call last):
  File "create-can-device.py", line 2, in <module>
    import hal, time
  File "/home/bas/machinekit/lib/python/hal.py", line 30, in <module>
    import _hal
ImportError: /home/bas/machinekit/lib/python/_hal.so: undefined symbol: _Py_ZeroStruct
```
I'm using python3
```bash
bas@debian:~/canopen-machinekit$ python --version
Python 3.4.2
```
Is this a known thing?

luminize, 2015-10-17:
addendum:
I compiled hal when I was using python 2.7

luminize, 2015-10-17:
I found the `--with-python=<path>` compile option in `configure`, but I can't figure out how to use it.

luminize, 2015-10-17:
Giving a try but my experience stops here I'm afraid:
`./configure --with-python=python3.4`
which gives me
```bash
checking python version...   File "<string>", line 1
    import sys;raise SystemExit,sys.hexversion<0x2040000
                               ^
SyntaxError: invalid syntax
configure: error: Python version too old (2.4 or newer required)
bas@debian:~/machinekit/src$ python
Python 3.4.2 (default, Oct  8 2014, 10:45:20) 
[GCC 4.9.1] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import sys
>>> sys.hexversion
50594544
```

mhaberler, 2015-10-17:
the Python API changed between v2 and v3 so those are not compatible
no point in defibrillizing the legacy hal bindings
the cython bindings might actually work with reworked build instructions though, I guess nobody tried yet

mhaberler, 2015-10-17:
gave it a try - a mk Python3 migration looks like a major project; tk, tcl, hal bindings - the whole legacy 

sorry, this looks infeasible short term

mhaberler, 2015-11-25:
is this still pending?

luminize, 2015-11-25:
that depends if we want to connect the HAL for use canopen with python 3. Which is we're this came to light. This looks like the same problem: https://github.com/machinekit/machinekit/issues/563#issuecomment-94250235
I've taken the liberty (and added) to put this on a long term category?

luminize, 2016-03-13:
don't think this still adds value. closing

closed 2016-03-13 by luminize