machinekit/machinekit #781: Fix Cython dependencies

strahlex, 2015-11-02:
And it hit me https://github.com/machinekit/machinekit/blob/master/src/hal/cython/Submakefile#L23

On Debian stretch I get:

     hal/cython/machinekit/hal.c:1177:59: error: unknown type name ‘comp_type’

Any ideas how to resolve the issue? The cython version is the same as in jessie-backports and it works fine with that.

ArcEye, 2015-11-02:
https://groups.google.com/forum/#!topic/machinekit/HCpxjxZukf8

This relates to the same errors, with a mix of stretch and jessie packages

Again, the cython version was 0.21.1 (ie the same as Jessie)

The 'unknown types' are defined in src/hal/lib/hal.h and hal_group.h and those files are
included in src/hal/cython/machinekit/hal.c

On a normal C compile I would start looking at the $INCLUDEPATH, but I know diddly about cython

strahlex, 2015-11-03:
Looks like Cython is not working with gcc5 and Protobuf on stretch is compiled against gcc5 and not working with gcc-4.9. Kind of a locked in situation. Probably have to use a VM for now.

mhaberler, 2015-11-03:
see issue #740 

gcc5 managed to break the ABI wholesale and repair on a per-package level is in progress

avoid gcc5 for now - wast of time

strahlex, 2015-11-03:
@mhaberler gcc5 is only part of the deal

I tracked this issue down to Cython 0.23 from either stretch or the jessie-backports. The same error happens when using the package from jessie-backports on a pure jessie installation with gcc-4.9.

I cannot avoid gcc5 at the moment since all libraries on stretch are built against it. (I have to use stretch because of the Broadwell CPU) I hope that the rest of the dependencies work out.

bobvanderlinden, 2015-11-12:
The errors were fixed for me with:
https://github.com/bobvanderlinden/machinekit/commit/75bac131ede7458b07aab95e7bc377da0960ee55

Compiling it with clang resulted in more intuitive output suggesting to use `enum comp_type` instead of `comp_type`.

I wasn't sure this was also happening on master, but if it is a problem there too, it might be a good idea to do a PR?

ArcEye, 2015-11-12:
I don't pretend to understand cython, but comp_type is already an enum defined in hal.h

Are you saying you built MK to a running completion just changing this?

I got around the cython issue on stretch by using gcc-4.9 and the jessie packages, but then hit more problems with building the machinetalk section of the code.

bobvanderlinden, 2015-11-12:
@ArcEye The problem is that `comp_type` is not a typedefed enum (so you need to refer to it as `enum comp_type`), however in `hal_const.pxd` it is referred to as an typedefed enum (so it refers to it as `comp_type`).

I tried this with a newer cython version and a newer gcc and clang version, so I'm not 100% sure whether this is a problem with the compiler or cython. However if @strahlex has tried it with just a newer gcc compiler, I think the problem pops up because gcc now does not accept referring to an enum without mentioning whereas it would ignore or just warn about it earlier.

closed 2015-12-03 by luminize