machinekit/machinekit #784: lincurve component bad math

presslab-us, 2015-11-13:
The lincurve component is acting weird, and it looks like some refactoring created a bug.  The order of operations was changed in a formula which screwed up the interpolation.

Here's the new code
https://github.com/machinekit/machinekit/blob/master/src/hal/i_components/lincurve.icomp#L60
And the old code
https://github.com/jepler/linuxcnc-mirror/blob/master/src/hal/components/lincurve.comp#L61

You can see the added parenthesis are in the wrong spot.

luminize, 2015-11-14:
do you have a development setup where you can make a PR?

presslab-us, 2015-11-14:
No, sorry.  I could try to get one set up, but I remember last time I compiled it I struggled with the dependencies a bit.

luminize, 2015-11-14:
ok, I’ll pick this one up later
Bas

ArcEye, 2015-11-14:
Looks like an auto-completion of brackets typo.

I have junked Geany because of that and other problems.
Just trying a re-build now

ArcEye, 2015-11-14:
Fix added to PR at
https://github.com/machinekit/machinekit/pull/783

presslab-us, 2015-11-14:
Thanks guys!!

luminize, 2015-11-14:
@presslab-us there's a problem regarding the buildbot and generating the package for the ARM platforms. The ARM packages will not be built, so at this time, unless you're not running from ARM packages you cannot upgrade yet. Sorry :(

presslab-us, 2015-11-14:
Okay, yes I'm using a BBB.  I'm using the lincurve to correct spindle speed on a cheap spindle controller.  So for now I will just fudge it.  But I am happy to know that it will be fixed eventually.  :)

luminize, 2015-12-17:
@presslab-us now that there should be packages again, can you update and see if the problem is still around?

presslab-us, 2015-12-19:
Great!  Yes I will test it soon, should be in the next couple days.

presslab-us, 2015-12-23:
I have confirmed that this is working properly now.  Thanks guys!

closed 2015-12-23 by presslab-us

luminize, 2015-12-23:
@presslab-us Thanks for the feedback!