machinekit/machinekit #785: password fore RPi2-machinekit incorrect

hrehaustechnik, 2015-11-14:
Dear ladies and gentlemen

we tried any machinekit fore the Raspberry PI 2. The only who booted is the RPi2-machinekit-1.0.img.

But when we type in the password raspberry the prompt tells us the password is incorrect.
We tried it dozen time it will not work.
Send us the path of the file were the password is located that we can change it.

Please help us

Thank you very much in response and fore your good websites

Paul Bruenen

luminize, 2015-11-14:
Have you tried googling:
machinekit raspberry pi 2 password

Which leads me to here:
https://www.raspberrypi.org/forums/viewtopic.php?f=37&t=33809&start=775

If you search these forums there are a lot of folks mentioning user mk with password machinekit

hrehaustechnik, 2015-11-14:
Hello Bas Bruijn

Thank you fore your quick answer.

We figured out that to things to be changed. First you have to change 
the cmdline.txt in Boot. How you do this is descript in Change ... .pdf. 
If you put the SD card in the Raspberry and the raspberry stops booting 
and is waiting fore your respond refer to adduser.gif and type the new 
user and passwd in. Then exit and the Machinekit starts and the new user 
and passwd works fine. After that make the cmdline.txt original. That is it.

Best Regards

Paul Bruenen

luminize, 2015-11-14:
Hi Paul,

I don’t really understand your answer in relation to your question.
I thought you didn’t know the password.

But it seems that you have solved your issue. if so, would you then please close the issue?

Cheers,
Bas

hrehaustechnik, 2015-11-14:
Hello Bas

sorry that we answered so short.

1. Insert Machinecit SD card in your Computer sd slot.
2. In Boot change command line. text. At the end insert: init=/bin/sh.
3. Insert the sd card in Raspberry
4. Raspberry stops boot (wait patient that it really stops). Then type 
in: sudo adduser {user name } then press Enter and in the following step 
by step session give new password. At the point new user hit Enter that's it
5. Exit : sync
              exec /sbin/init
6. Erase on your Computer the init=/bin/sh in your Boot comman line.text

That is all. Please contact again if things unclear.

Best regards

Paul

luminize, 2015-11-14:
Paul,

do you, or do you not still have a problem with your original problem that you don’t know the password ?

Bas

hrehaustechnik, 2015-11-14:
Hello Bas

what we wrote you is a description how to solve that problem. One we forget.

7. it will not be able to shut down. Open a Terminal and type halt ore 
sudo halt

Yes! Sir! we know now the password and the user name.

Please fill the steps in the issue number xxyyyzz we forget and end this 
issue. You will do us a great favour.

Thanks a lot fore your help

Have a nice weekend

Paul

luminize, 2015-11-14:
ok, archiving this issue for the offspring

closed 2015-11-14 by luminize