machinekit/machinekit #786: useful and/or supported architectures unclear (in particular armv6/rpi1)

mhaberler, 2015-11-16:
This issue is about clarification, and suggestions to fix the website purporting otherwise.

- HAL does assume 64bit atomic operations, even if the current codebase does not use atomics.
- armv6 (as present in the Raspberry Pi) cannot do 64bit atomics. The armv7l and onwards architectures (eg rpi2) can do 64bit atomics.
- while it is possible to compile the current codebase on a armv6 platform, there is a chance that pin operations on HAL float values are incorrect, in particular in the userland -> RT component direction. I therefore do not recommend using armv6 as a platform for machinekit. I also advise against suggestions on the website it there is a supported build or way to use armv6l. These MUST be fixed.
- forthcoming work (see "HAL multiKore": http://blog.machinekit.io/2015/11/machinekit-meetup-summary-of-talks.html) WILL require the use of atomic operations. Post merge, the codebase will NOT compile on armv6 any more.

In summary:

**There is no support for armv6 and other platforms without 64bit atomics even if you are able build and install machinekit on such a platform.**

RunningLight, 2015-12-14:
So, is it time yet to eradicate the arm6 material in machinekit.io/docs/packages-debian? It points to a different distro on 0ptr.link but who knows how out of date it is or will become.

kinsamanka, 2015-12-15:
Not yet, I'll move the RPi repo to the packagecloud.io too.  ;-)

koppi, 2016-06-02:
The Raspberry Pi 3 comes with ARMv8.

Work is done to get a 64 bit Raspbian, see:
* http://raspberrypi.stackexchange.com/questions/43921/raspbian-moving-to-64-bit-mode

Patches to U-Boot for running the Pi 3 AArch64 code by Stephen Warren:
* http://lists.denx.de/pipermail/u-boot/2016-April/251376.html
* https://github.com/swarren/rpi-3-aarch64-demo

ArcEye, 2018-08-01:
Think this has been overtaken by events and hardware releases

closed 2018-08-01 by ArcEye