machinekit/machinekit #787: ARM Port Image Building

the-snowwhite, 2015-11-16:
I put up this subject here in an effort to gather Documentation on all the other aspects of creating  a MK  ARM port that not are related to building MK itself. 
And Unify this process over the various supportable arm platforms as much as possible.

Initially I narrow down the targets to BBB, BBx15 and NANO / Atlas. (which are the current "hot potatoes" know to me).

Boot:

U-boot v. 2015.10 
This should be trivial as it is supported in mainline for all involved platforms:

    http://www.denx.de/wiki/U-Boot

Device tree (dts)

individual for all boards.

Pin muxing support (capemgr ?) :

only for the Beagles.

and then boot env's

    boot.scr
    boot.txt
   can also be stored in u-boot var storage mem pointed to place(nand / flash / sd-card, etc).

Kernel:

I would think the current 1 month release window for the 4.1-ltsi-rt ARM kernel would be nice to target unified across all these platforms:

     http://ltsi.linuxfoundation.org/ltsi-4.1-development-schedule

Rootfs:

I have found following cross runnable recipe as substitute for the current image build scripts
that need to run natively on Arm:

   http://ms-cheminfo.com/?q=node/93

THis could serve as a new beginning point for scripting the whole process ?

the-snowwhite, 2015-11-23:
Would it be possible to create a hook / spinoff from the new build process that could autobuild Arm Sd-card images ?

mhaberler, 2015-11-23:
I think you should consult with Robert Nelson - he currently does that for us

the-snowwhite, 2015-12-11:
Thanks
I have managed to collect my notes from the first mksoc jessie test build image I released in september in 1 text file.
and also the arm rip jessie script that worked at that time.
+ layout an allmost finshed template for running the image building on non arm hardware.

all gathered here:
https://github.com/the-snowwhite/mksoc/issues

the-snowwhite, 2015-12-11:
I expect delaying contacting Robert N. on this issue until I get the initial generic socfpga debian image script debugged and working.

After that it would in theory be possible to run the provided rip script on the socfpga board and have a working Machinekit installation.

Then only missing:
ability to connect to the hardware pins (thru the fpga partition)
armhf jessie packages ?
Spare time...

the-snowwhite, 2015-12-19:
Ok Robert N has been working here:
https://eewiki.net/display/linuxonarm/DE0-Nano-SoC+Kit

-----

In parallel I have a full x86_64 cross sd-card image build script(s) in this folder:

https://github.com/the-snowwhite/mksoc/tree/master/scripts/Divided_scripts_and-notes

u-boot 2016.x.x
3.10-rt-ltsi rt-preempt as before
qemu-debootstrap chroot rootfs build.

------
+ v2 of the Machinekit Jessie rip build script here:

https://github.com/the-snowwhite/mksoc/blob/master/scripts/mksoc-jessie-mk_rip_build-instal-v2.sh

the-snowwhite, 2015-12-19:
So high time for:
[creating Machinekit soc-fpga driver and finishing Quartus config](https://github.com/machinekit/machinekit/issues/843)

the-snowwhite, 2015-12-19:
New documentation has been added here:
Robert Nelson. -->  https://eewiki.net/display/linuxonarm/DE0-Nano-SoC+Kit

And I have placed a more simple and conservative(fully scripted) method for creating a bootable sd card and installing machinekit here: 

Michael B. --> https://github.com/the-snowwhite/mksoc/blob/master/scripts/Divided_scripts/socfpga-debian-jessie--sd-image-gen.sh

https://github.com/the-snowwhite/mksoc/blob/master/scripts/mksoc-jessie-mk_rip_build-instal-v2.sh

I have tried to keep the scripts as simple and self documenting as possible.

last there is the initial Quartus project template(s) for generating files to configure the fpga part of the soc.

https://github.com/the-snowwhite/mksoc/blob/master/QuartusProjects/sockit_ghrd/Makefile

the-snowwhite, 2016-02-14:
This issue will also needs future documentation:
Or a link ?
https://github.com/machinekit/machinekit/issues/869

How to specify sources only for x86 or armfh ?

mhaberler, 2016-03-29:
I only now discovered this page by @robertcnelson on the DE0 effort: https://www.eewiki.net/display/linuxonarm/DE0-Nano-SoC+Kit

closed 2016-04-16 by the-snowwhite