machinekit/machinekit #788: "Code Safari on Hangout" topics undefined

mhaberler, 2015-11-17:
this issue is about covering a specific topic in Machinekit Virtual Meetups as mentioned on the list

I'm starting with the topics mentioned so far 

Please add your topic of interest as a followup here, and propose a date using the (yet to be defined) G+ online calendar!

- what happens on "realtime start/stop", on component load/unload, thread start/stop, various halcmd commands like link/unlink addf/delf
- what happens after a 'machinekit foo.ini' command
- how do interpreter, task, io, motion play together - following say an MDI command from start to finish
- how the machinetalk stack works: discovery, connections, HAL interaction, Alex's mkwrapper
- How and Why of Google protobuf
- Exploring the G-code Interpreter (just came to my mind)

luminize, 2015-11-17:
I have made a google calendar (not sure if this is what you mention with G+) with the following urls:
[ical link](https://calendar.google.com/calendar/ical/nl40eeqfok4g8fj17nor2soth0%40group.calendar.google.com/public/basic.ics)
[html link](https://calendar.google.com/calendar/embed?src=nl40eeqfok4g8fj17nor2soth0%40group.calendar.google.com&ctz=Europe/Amsterdam)
[xml link](https://calendar.google.com/calendar/feeds/nl40eeqfok4g8fj17nor2soth0%40group.calendar.google.com/public/basic)

ArcEye, 2015-11-18:
All valid topics.

I would suggest an opener based upon the 2nd and 3rd items, basically tracing execution from invocation through to execution of a gcode.

That would give a basis upon which to build later, with the specifics of internals in parts of that process.

luminize, 2015-11-18:
Count me in. I don't have a preference yet wrt day/time. Very busy.
Do you have any preferences ?

bobvanderlinden, 2015-11-18:
That's a really good idea! The topics (1,2,3) are quite interesting. It's still a bit of a mystery how all components interact exactly. Knowing the exact protocols between them would be useful as well (what parts talk protobuf, which ones use nml, etc).

Also, what are the minimal steps required to get a HAL working manually without any of the shell scripts (like `machinekit` or `realtime`). Your topic already cover these questions, so I'm interested what you're going to tell ;)

sirop, 2015-11-19:
My question may seem to be far fetched or too general, but I'll put it here as otherwise I'll forget it.

I think there is  a good reason for libwesockets offering the option to use https://www.wolfssl.com/wolfSSL/Products-wolfssl.html instead of OpenSSL as
OpenSSL is not so light weight.

But there may be a license issue with *wolfSSL*  as you can find at https://www.wolfssl.com/wolfSSL/License.html in case of Commercial Licensing .

So my general questions are:  
 
- Is there any OpenSSL lightweight substitute that has a more liberal license than *wolfSSL* ?  

- Does Machinekit need any tailor-made encryption library?

the-snowwhite, 2015-11-19:
I'm certainly very interested in all mentioned subjects and my hesitation is only due to, " the do not have too many balls in the air at one time syndrome", having to keep my stress level low.

I noticed I was missing a subject, and found out i may have interpreted this original wording:
""if I hit a button here, what happens" "

As:
" if this pin goes high, what happens"

Reason is that the hal can be used for tasks other than executing g-code (an example: like for a remote (autonomous) flight controller). , temperature regulation, pids etc.

darenschwenke, 2015-11-19:
http://www.matrixssl.org/index.html
GPL, as long as your project is GPL too.
Commercial license/support available if not.

mhaberler, 2015-11-19:
@sirop Sorry, hijacking of an off-topic issue is frowned upon. It does not work that way, even if you are in danger of "forgetting" it.

Please create an on-topic issue, or post to the list, and move the discussion there. Thanks.

mhaberler, 2015-11-23:
I have thought through the flow of themes a bit and now plan to start the series with the theme **the G-code interpreter**. 

This might be a bit surprising but it is not only very useful to understand its workings and API (gcode execution and the canon layer, both of which I'll cover),  it is also helps when we discuss structural problems in the code later on, like #106. And both API's are very helpful to understand the next chunk: **how do interpreter, task, io, motion play together - following say an MDI command from start to finish**

Everything else is on a lower conceptual level which can be detailed in followups. I'll reedit the theme sequence in the first post according to this flow.

As a date for the first chunk - The G-code Interpreter - **I propose Dec 1st, 2015, 20:00 UTC** but I am flexible. I think we should be done with the hard material in 30-45 mins and wrap up after an hour. 

I hope this will be two-way with Q & A capability, so get your microphones plugged and tested - happy to act as a test site (haberlerm@gmail.com).

As a format, I'll use a recorded hangout, and I need to get fluent with that method, screencast, camera, and slides all used together - I have never done that before. I will prepare the interpreter talk and do a short dry run on Nov 30 20:00 UTC with @luminize - just enough to get the technical details sorted. I'd really appreciate if somebody fully clued up on hangout could brief me a bit what is required and how to avoid major goofs, and maybe tune in on Nov 30 just to help with shaking things out - wont take long!

After the first run, lets discuss here the good, the bad and the ugly, and how we can improve things!

unseenlaser, 2015-11-25:
count me in

zhivko, 2015-11-25:
For me I would love to see quickstart for making 6 servo motor (3 translation servo motors, 3 rotational one) operational, with machine kit.
Also I am interested in websocket (laser measurement sensor) integration in machine kit to allow on the fly Z axis correction.
Dec 1st, 2015, 20:00 UTC is OK with me, count me in.

mhaberler, 2015-11-25:
@zhivko - I understand your problem - but which part of that you cannot solve with example configs and the manual?

The intent for this series is to make as many people understand the structure and workings of machinekit. There is practically no internals documentation, and the code base is rather complex. And parts of it need refactoring. I think I can bring some explaining to the table here, since I've been working on this for a long time.

So the plan is to get as many folks up to speed on internals - it is not intended as a 'machinekit 101' bootcamp.

unseenlaser, 2015-11-25:
@zhivko - look in the examples a typical servo configs is boss & Masak
a laser measuring z could be done , look at thc examples
the best place for advice is post in the main forum .

-- 

The information contained in this message is confidential and is intended
for the addressee only. If you have received this message in error or there
are any problems please notify the originator immediately. The unauthorised
use, disclosure, copying or alteration of this message is strictly
forbidden. This mail and any attachments have been scanned for viruses
prior to leaving the RcTechnix network. RcTechnix will not be liable for
direct, special, indirect or consequential damages arising from alteration
of the contents of this message by a third party or as a result of any
virus being passed on.

RcTechnix reserves the right to monitor and record e-mail messages being
sent to and from this address for the purposes of investigating or
detecting any unauthorised use of its system and ensuring effective
operation.

(c) RcTechnix

zhivko, 2015-11-25:
With manual you think "man machinekit" and with the samples: https://github.com/machinekit/machinekit/tree/master/configs?
Sorry I think this is off topic - but is somebody interested in more interactive way to share knowledge around machinekit using gitter?

zhivko, 2015-11-25:
@unseenlaser Thank you I found:
https://github.com/machinekit/machinekit/tree/9e5dedd360fddd31cb8963a714f7297519c6bd4e/configs/by_machine/boss
But where is Masak ?

unseenlaser, 2015-11-25:
contact me off list if you need guidance
/attic/demo_masak

-- 

The information contained in this message is confidential and is intended
for the addressee only. If you have received this message in error or there
are any problems please notify the originator immediately. The unauthorised
use, disclosure, copying or alteration of this message is strictly
forbidden. This mail and any attachments have been scanned for viruses
prior to leaving the RcTechnix network. RcTechnix will not be liable for
direct, special, indirect or consequential damages arising from alteration
of the contents of this message by a third party or as a result of any
virus being passed on.

RcTechnix reserves the right to monitor and record e-mail messages being
sent to and from this address for the purposes of investigating or
detecting any unauthorised use of its system and ensuring effective
operation.

(c) RcTechnix

zhivko, 2015-11-25:
In boss.readme you refer to:
http://wiki.machinekit.io/
But I could not open that page - Like server is down?

mhaberler, 2015-11-25:
no, consequence of a mass rename. Refers to wiki.linuxcnc.org most likely. PR welcome!

ArcEye, 2015-11-25:
@zhivko  the best place for documentation generally - is the documentation
https://github.com/machinekit/machinekit-docs

The configs in the source tree are partially legacy ones
Stuff like the mazak and boss have holes in them and should be rough guidance only, not a cut and paste job.  You probably do not have a motenc board anyway.

I will fix the link and another reference to obsolete component loading syntax

mhaberler, 2015-11-25:
@zhivko - sorry, this is off-topic. No issue hijacking please - move this to the mailing list. Thanks.

ThomSmithUK, 2015-11-25:
I have followed this here from the google groups list.  A linuxCNC user, future machinekit user, and interested in the internal workings of the software.  I would probably view the youtube vid after the event.

luminize, 2015-11-25:
hi all,

I just added links to the calendar event (easy for copying into your own calendar) [in the comment above](https://github.com/machinekit/machinekit/issues/788#issue-117259757)

mhaberler, 2015-12-01:
I created a [G+ event  for the G-code interpreter hangout](https://plus.google.com/events/ci1o4m17i977tv6e5rucks6nhp0) today

My first real use of G+ and Hangouts, so please bear with me

please watch the G+ event notification, I am not sure if I need to post a link to the actual Hangout there or not - I'll be online a bit earlier to resolve any issues

so - get your microphones plugged in!

luminize, 2015-12-02:
Here's the link to the recorded hangout of the first part
https://plus.google.com/events/c35prtt5ptrbmcc292bmhiketuc

mhaberler, 2015-12-04:
ok, the first one is behind us - my 0.1€:

on the format: in theory the hangouts on air route looks fine (feedback, Q&A capability and all), but my impression is - this does not scale. The audience would have been much larger if the video/audio would have been oneway only and not limited like hoa is - this seems fine for a small group but it looks 10 is really the limit

so maybe next time around we just use a plain hangout, and use gitter as feedback channel

I must also note keeping the flow of the presentation going and peeking into the chat channel continuously is a bit hard, so maybe we can do something like gitter as feedback and maybe Bas can act like a moderator there, help organize the feedback and Q&A sessions a bit?

on the length: I think it was a bit too much in one go, I went on for 90mins or so, so I'll try to cut down the next unit a bit so to stay within the hour frame

let's just give this a try and see how it works out, we'll learn along the ride

mhaberler, 2015-12-04:
**how do interpreter, task, io, motion play together - following say MDI command from start to finish** - I am currently thinking through how to structure this unit

we have a very good basis with the motion pipeline roughly understood

I would concentrate on the milltask coordinator (aka 'task' or 'emctask') - kind of the 800pound gorilla in the room nobody wants to mess with and barely anybody dives into

Again I'd break it down to the core jobs and flows, and then turn to examples 

as for examples, I think the best approach is to conceptually cut out any GUI and investigate what certain GUI interactions cause at the [linuxcnc module](https://github.com/machinekit/machinekit-docs/blob/master/src/common/python-interface.asciidoc)  API level, as well as actions downstream

this will exercise a bit of HAL pin wiggling/feedback (iocontrol and motion), and also show how the motion pipeline is tied into task

that naturally leaves open a third safari range, the motion/tp/kinematics complex - maybe we do this in two shots; first me giving a rough overview on structure/flows/states/API's, and - in early January - Rob Ellenberg weighing in on a more intimate view of the tp.

after that - a bit of HAL POOPS ("prinicples of operations" ;-) 

I am open to suggestions - let me know; barring any that is what I'll go for

ArcEye, 2015-12-04:
Re format: exactly as per my comments at the time, the normal hangouts is fine for the delivery and questions can be channelled via email thread or gitter - spot on with my take on it.
I am happy to go onto the actual live video meet and monitor Q's too if required, albeit probably only needs one person.

Content, a good candidate to build on the last talk.
Go for it :+1:

luminize, 2015-12-04:
@ArcEye,
that would be a welcome addition. Please do!
Bas

strahlex, 2015-12-04:
Please don't confuse Hangouts and Hangouts on Air.

*Hangouts on Air* is the broadcasting thing that streams to YouTube
*Hangouts* is the Skype-like conference call limited to 10 people

My recommendation: Make the *Hangouts on Air* public and use a Hangout or Skype with a second person to monitor the feedback and questions.

strahlex, 2015-12-04:
We could also use a TitanPad for Feedback and QA: https://titanpad.com/
Let people just write the questions down in a structured way.

strahlex, 2015-12-04:
Lets try this: add some stuff:
https://titanpad.com/QuqVl7r88u

luminize, 2015-12-04:
Hi Alexander,

Ill check your link this weekend.
One of the benefits of gitter is that there's already communication going on so low barrier for further engagement.

Cheers
Bas

strahlex, 2015-12-04:
@luminize  Just click on it. No login required.

strahlex, 2015-12-04:
The problem of gitter and forum-like dialog is that the text is not in a structured form. The TitanPad  thing is wiki-like and allows other people to answer the questions as well or to write down the answers. The resulting document could be stored together with the talk afterwards.

bobvanderlinden, 2015-12-04:
Were we doing Hangouts last time? It was [recorded](https://plus.google.com/events/c35prtt5ptrbmcc292bmhiketuc) somehow, so I'd think it is Hangouts on Air... on the other hand I didn't see any of the thumbsup/thumbsdown buttons nor chat, so it might've been Hangouts... I'm confused.

@strahlex I like the idea of having a structure for the presentation up-front that people can comment and edit on. Titanpad seems like a good tool for that.

strahlex, 2015-12-04:
@bobvanderlinden It was *Hangouts* screen-shared over *Hangouts on Air* indeed very confusing.

bobvanderlinden, 2015-12-04:
Ok wat. That explains a thing or two. If Hangouts on Air properly supports chat and having more than 10 people, I'd say that would be a good choice.  Does it matter whether feedback is given through chat or video/audio? I'm alright with giving feedback as chat. Having a presentation split up into multiple sections so that, after each section, the chat can be checked for questions might a good option.

strahlex, 2015-12-04:
*Hangouts on Air* has no instant feedback. People can comment on the Google+ page, but this requires the talker to monitor the page. Furthermore, the talk is slightly time delayed. Therefore, the idea of having one person that monitors the the reception and Google+ page is a good. In general I think its better to answer questions at the end of the talk.

bobvanderlinden, 2015-12-04:
Ah ok, the delay seems indeed quite significant and might be annoying for instant feedback. Yes, a 20m-30m talk seems like a good length to me.

mhaberler, 2015-12-08:
outline for today's installment (Tue Dec 8 8PM UTC) - the [G+ event should be coming up shortly here](https://plus.google.com/+MichaelHaberler/posts):

**From Command to Pin and Back:
The Task CNC Stack Coordinator**

This talk gives an overview of the task coordinator, the key component managing states and modes and responsible for the flows between UI and the motion and iocontrol components. We'll review the core functionality, and go through working examples using the Python linuxcnc module, tracing the sequence of events and their causal relation.

This talk is targeted at people who want to understand what is going on, and why - and maybe improving code or fixing bugs. To follow, you should have used machinekit previously, and some programming experience. Since we'll be referring to motion pipeline: if you've missed the first talk, [the recording is here](https://www.youtube.com/watch?v=Bg38894LPPg)

For feedback, you will need a github account since we will use gitter.im as chat channel.

I will put up a link to slides and example references later today here, as well as the link to the hangout event.
A [recording will be available on youtube.](https://www.youtube.com/channel/UCRIv6RgL37sGQhp7T3zWKTw)

We will try a different format this time since voice feedback was problematic given the number of attendees: please post feedback and questions to https://gitter.im/machinekit/events - a moderator will help managing the feedback channel.

--------------------

What will different from round 1:

- I will try to keep it a tad shorter
- I will clearly separate _what the current code does_, as opposed to musing what be _nice to have_ or _needs fixing_.
- let's experiment with the new feedback format - and let's review how that worked after the talk in this issue

mhaberler, 2016-01-07:
I am planning to do a talk on the motion component, tentatively Mon Jan 11 8PM UTC

**What Motion does:
From Canon Commands to Joint Movements**

motion is a key component and the bridge between the non-RT canon command layer, and the HAL space. It is also a rather complex piece which has become the kitchen sink of "everying moving" wrt to the CNC stack above. 

This talk will focus on an overview, major functions, modes and API's. It is intended as a primer - I am working on the assumption that Rob Ellenberg will cover details of trajectory planning like blending more in-depth (no date set yet).

This will build on the previous talk, in particular assuming a basic understanding of Canon. Beyond coordinated moves, we'll look into homing, jogging and mode switching.

---

Since this would conclude the "CNC stack intro" (interpreter, task, motion), as a sequel I'd like to turn to giving an inside view of the realtime stack - HAL, the associated demons, component builds and loading, drivers etc. I will be a tad busy through mid-february [with non-machinekit activities](http://www.ballonfestival-tannheimertal.de/), I could make Feb 15 for sure; potentially before that time permitting

luminize, 2016-01-07:
@mhaberler ping when you have the google-hangout-on-air scheduled so the machinekit google account can add the event in the calender.

For all others:
the calender links in comment https://github.com/machinekit/machinekit/issues/788#issuecomment-157478028 have been edited to the right urls

ArcEye, 2018-08-01:
Historical and discontinued

closed 2018-08-01 by ArcEye