machinekit/machinekit #789: Setting up Google Hangout meetups and Machinekit Calendar

luminize, 2015-11-17:
These meetups will be open to people interested in the subject of the meetup and are meant to improve communication by discussing topics, share knowledge and insights in a face-to-face situation.

These links will be the links you can use for your all time favourite Machinekit calendar:

* [ical link](https://calendar.google.com/calendar/ical/nl40eeqfok4g8fj17nor2soth0%40group.calendar.google.com/public/basic.ics)
* [html link](https://calendar.google.com/calendar/embed?src=nl40eeqfok4g8fj17nor2soth0%40group.calendar.google.com&ctz=Europe/Amsterdam)
* [xml link](https://calendar.google.com/calendar/feeds/nl40eeqfok4g8fj17nor2soth0%40group.calendar.google.com/public/basic)

Until we have some experience and can fine-tune we'll start like this:
* People who have an idea/proposal/wish for a meetup discuss this on the machinekit list or in an issue (new or existing one) and decide together when that's to take place. The first one (at the time of writing this its schedule is yet unknown) is mentioned here: https://github.com/machinekit/machinekit/issues/788
* When the time has been decided somebody needs to ping me (let's start with me and expand when this proves to work) to add an event in the Machinekit calendar.
* This event will show up in your calendar which you have added from the above links.
* I'll add as only invite the machinekit list, so the invite will show up as a post. Exposing the event.

Logically the ones having the meeting can put links to the event in the places they deem necessary.

All other stuff regarding the setting up of these events, calendar and workflow please comment in this thread.

mhaberler, 2015-11-25:
could you add the Dec 1 and Dec 8 hangouts to the calendar?

luminize, 2016-01-07:
I've just set the Calender of the machinekit.io@gmail.com account and updated the links in comment https://github.com/machinekit/machinekit/issues/789#issue-117438841

closed 2016-03-13 by luminize