machinekit/machinekit #791: Build and CI infrastructure needs replacement

zultron, 2015-11-19:
This issue is to track technical issues surrounding the replacement of the Buildbot.

bobvanderlinden, 2015-11-19:
What do you have in mind for requirements?

To give an initial suggestion for an alternative:
If Machinekit can be built on a simple Ubuntu/Debian system and can run without needing root, then you can probably use Travis CI. I use it for [node-machinetalk](https://travis-ci.org/bobvanderlinden/node-machinetalk) to run unit tests on every push and it does automatically does releases when I push tags. It also builds all PRs and gives feedback in Githubs PR pages automatically. See https://github.com/NixOS/nixpkgs/pull/11150 for for an example of an PR with feedback.

The drawback is that it only works for x86_64, not ARM/Beaglebone like your Buildbot does now.

ArcEye, 2015-11-19:
I suspect that the main problem will be the environment and all the dependencies.
In that sense it is not different to a normal build.

This will be more easily managed inside a docker image.  Once that image is made it can be distributed for use in various locations.
If the dependencies change, it only needs one person to alter the image and test it and then the new image can be distributed again.

Keeps the bot fu to a minimum and should produce consistent results whoever is running it.

zultron, 2015-11-19:
Hi fellas, I updated the first comment here to give an overview of what this is all about.  Sorry for the length; hopefully it's broken up well enough that you can just pick out the sections of interest to you.

@bobvanderlinden:  I wish we could use Travis CI!  But, it's too good to be true.  The Ubuntu-only and x86-only limitations are prohibitive, though:  Machinekit won't build on Ubuntu at the moment (something about default linker options?), and more importantly, we need to keep the Wheezy/Jessie and amd64/i686/armv7l package streams going.

@ArcEye:  You're sounding suspiciously like the next Buildbot maintainer!  ;)

Let me know what the updated description is missing (or whether it should be aggressively chopped down).

Next step for me is to finish a minimal configuration (almost done) and share.

Again, appreciate the support!  It's great to see folks rolling up their sleeves.  Docker is a lot of fun, partly because so much is possible with just a small learning curve.

bobvanderlinden, 2015-11-19:
@zultron Thanks for the write-up. The docker files will be most important. I wasn't aware Docker also ran on ARM (it doesn't on 32bit x86 hosts :(). That's good news! I'm interested what you come up as the docker file.

Not sure if this is still of use after the buildbots work, but when you have a dockerifle, it should also be possible to [use the docker file inside TravisCI](http://docs.travis-ci.com/user/docker/). You could still make use of their Github integration or as a backup.

zultron, 2015-11-19:
@bobvanderlinden, Oh, wow, that could be a game changer, esp. if ARM can be emulated (partly depends on whether TravisCI has `binfmt_misc` configured properly).

Docker isn't officially supported on i686 or ARM, but the distros do package it for those architectures.  Then you have to work around the Docker 'official' distro images (e.g. `debian:jessie`) all being x86_64 arch; there are unofficial images with other architectures like https://hub.docker.com/r/armv7/armhf-debian/

Would you be willing to do a proof-of-concept with a simple example for x86, and even ARM if you're feeling adventurous?

zultron, 2015-11-20:
@bobvanderlinden, the possibility of building on Travis CI with Docker could be an even easier way through this problem than the current plan.

It would be a big help if we knew that Travis CI could build for Jessie; here's a minimal test:

- Whip up a quick Dockerfile for amd64-arch debian:jessie that installs gcc and compiles a stub `foo.c` program, and run the build on Travis CI
- See if the build succeeds, and see if we can get the resulting `a.out` file back
If we knew this worked and if we had simple instructions to repeat in the next day, I would drop the current effort to investigate Travis CI further.

It would be a HUGE help if we knew that Travis CI could build for Jessie, armv7l architecture; here are two possible feasibility tests:

- In an armv7 Docker image, test qemu and `binfmt_misc` by copying `qemu-arm-static` into the Docker image under `/usr/bin`
- In an armv7 Docker image, test qemu with proot (@kinsamanka had this figured out earlier this year)

RobertCNelson, 2015-11-20:
@zultron Found myself playing with docker.io, it's now enabled in the beagleboard.org's jesse repo:

```
debian@BeagleBoard-X15:~$ uname -r
4.1.13-ti-r33
debian@BeagleBoard-X15:~$ sudo docker info
Containers: 1
Images: 1
Storage Driver: aufs
 Root Dir: /var/lib/docker/aufs
 Backing Filesystem: extfs
 Dirs: 3
 Dirperm1 Supported: true
Execution Driver: native-0.2
Kernel Version: 4.1.13-ti-r33
Operating System: Debian GNU/Linux 8 (jessie)
CPUs: 2
Total Memory: 1.808 GiB
Name: BeagleBoard-X15
ID: JNYS:RPDI:AQ6E:PYGR:OOEB:IHHF:G5FV:XX2P:GR4C:IEUN:RA4F:TH5Z
debian@BeagleBoard-X15:~$ sudo docker run armv7/armhf-ubuntu_core /bin/echo 'Hello world'
Hello world
```

zultron, 2015-11-20:
@RobertCNelson, great!  Are your `docker.io` packages different from the ones in `jessie-backports`?

The 'armv7' repo doesn't provide a Wheezy base image; can you recommend one?

bobvanderlinden, 2015-11-20:
@zultron Yay, got it working. Debian ARM on Docker on TravisCI. Travis has binfmt_misc enabled and mounted, so I only needed to register the right qemu-user-static binaries and copy those to the docker container.

Travis file:
https://github.com/bobvanderlinden/travis-playground/blob/42bce26232e488726cbbe63fbd219be498262bb3/.travis.yml

Setting up `binfmt_misc`:
https://github.com/bobvanderlinden/travis-playground/blob/42bce26232e488726cbbe63fbd219be498262bb3/.travis.before_install.sh

Docker file:
https://github.com/bobvanderlinden/travis-playground/blob/42bce26232e488726cbbe63fbd219be498262bb3/Dockerfile

You can see the resulting build here:
https://travis-ci.org/bobvanderlinden/travis-playground/builds/92335616

You'll see `/bin/echo It works!`. `/bin/echo` is an ARM binary.

I had placed the qemu-user-static binaries into the git repo, so that I could easily copy it from the Dockerfile (I couldn't directly copy from `/usr/bin/qemu-*-static` from a `Dockerfile`). It's a bit ugly, but it can be cleaned up.

I'm at the moment trying your build steps from machinekit-docker in the ARM-based Dockerfile on Travis: https://travis-ci.org/bobvanderlinden/travis-playground/builds/92339882
It takes a while... From what I read, the maximum build time is 120 minutes, so we'll so how it goes.

the-snowwhite, 2015-11-20:
@zultron thank you for the very informative and valuable information dump.

I may be late to the party but I found here:
https://github.com/kavolorn/Docker-ARM-OpenCV/blob/master/Dockerfile

Way to another official arm branch on Docker hub:

https://hub.docker.com/search/?q=armbuild&page=1&isAutomated=0&isOfficial=0&starCount=0&pullCount=0

Having an Arm Jessie build:
https://hub.docker.com/r/armbuild/debian/

In this repository:
https://github.com/armbuild
https://github.com/armbuild/docker-brew-debian/tree/dist

! unrelated, this repository also has a docker-armhf-rootbuilder:

https://github.com/armbuild/docker-armhf-rootbuilder
I added this in #787

RobertCNelson, 2015-11-20:
@zultron it's just the jessie-backport package rebuilt for jessie (so beagleboard.org users don't need to enable the backport repo) and i added the kernel config changes along with aufs so docker.io is nice and happy. ;)

bobvanderlinden, 2015-11-20:
@the-snowwhite https://hub.docker.com/r/armbuild/debian/~/dockerfile/ That looks a lot like what I used!

RobertCNelson, 2015-11-20:
@zultron this might work... (first time ever doing a docker setup)

Dockerfile: https://github.com/rcn-ee/machinekit/blob/master/wheezy/Dockerfile

I've got an apt repo that is only "debian" + "machinekit" dependency:

```
deb [arch=armhf] http://repos.rcn-ee.com/debian-machinekit/ wheezy main
deb-src [arch=armhf] http://repos.rcn-ee.com/debian-machinekit/ wheezy main
```

bobvanderlinden, 2015-11-20:
@RobertCNelson The `FROM` line should have the base image. The image where your commands are executed on. For me `robertcnelson/machinekit-wheezy` isn't available, because it isn't on hub.docker.com. Try using `FROM debian:wheezy`.

@the-snowwhite I used the docker image you suggested (as it did almost the same as what I was doing). I published a new image with the necessary repository fiddling that machinekits build process depends on. It can be found [here](https://hub.docker.com/r/bobvanderlinden/machinekit-build-debian-armhf/). It can be pulled with `docker pull bobvanderlinden/machinekit-build-debian-armhf`.

@zultron I've got the right build rules in Travis with the image mentioned above. You can see the current build process [here](https://travis-ci.org/bobvanderlinden/travis-playground/builds/92357838#L1160).
However, it fails on some strange file conflict between libzmq3 and libzmq4. It seems libzmq4 contains the file `/usr/lib/arm-linux-gnueabihf/libzmq.so.3.1.0`, which seems strange to me. Have you seen something like this before?

RobertCNelson, 2015-11-20:
@bobvanderlinden odd, it should be public

https://hub.docker.com/r/robertcnelson/machinekit-wheezy/

Updating it with your change..

bobvanderlinden, 2015-11-20:
Ah, it isn't pushed yet (no "latest" tag yet here:
https://hub.docker.com/r/robertcnelson/machinekit-wheezy/tags/)

Use `docker login` and then `docker push robertcnelson/machinekit-wheezy`.

the-snowwhite, 2015-11-21:
Hmmm
I think I will be needing some human instructions:
I am able to run the hello-world and whale examples.
However:
Trying to do the same thing with any of your docker files Including armbuild/debian
I get stuck at the file not found error message:

    mib@Debian-8:~/mydockerbuild$ docker run bobvanderlinden/machinekit-build-debian-armhf
    Unable to find image 'bobvanderlinden/machinekit-build-debian-armhf:latest' locally
    latest: Pulling from bobvanderlinden/machinekit-build-debian-armhf
    097ffd1b0a22: Pull complete 
    67c4ee1dc6e6: Pull complete 
    106eecddfea1: Pull complete 
    8caa4511cdf8: Pull complete 
    01af1736f15f: Pull complete 
    Digest: sha256:3d0eddcc5a312507e45fc47290b52cb3f70b40a32b67bfa6d14eeae5d1deec6d
    Status: Downloaded newer image for bobvanderlinden/machinekit-build-debian-armhf:latest
    no such file or directory
    Error response from daemon: Cannot start container 5828234d5c1dfb7095276b3cb8a58f153f76469aa1a8ba4930a70315ed8b550a: [8] System error: no such file or directory

    mib@Debian-8:~/mydockerbuild$ sudo ls -l /var/lib/docker/aufs/diff/
    total 44
    drwxr-xr-x  7 root root 4096 Nov 21 02:50    01af1736f15fa74ebd7d17ac8c89a6b0fa9a6fad1092a6158020640d613f7f7f
    drwxr-xr-x 17 root root 4096 Nov 21 02:50 097ffd1b0a22bf8e78599c70fd6f9481b7cc4483807c097c0e752a9058be4d4b
    drwxr-xr-x  2 root root 4096 Nov 21 02:49 0a6ba66e537a53a5ea94f7c6a99c534c6adb12e3ed09326d4bf3b38f7c3ba4e7
    drwxr-xr-x  3 root root 4096 Nov 21 02:50 106eecddfea1dbb2fb589044999a7b1c42f6e7ad21903cdfa7d77093e1bf12de
    drwxr-xr-x  4 root root 4096 Nov 21 02:50 5828234d5c1dfb7095276b3cb8a58f153f76469aa1a8ba4930a70315ed8b550a
    drwxr-xr-x  8 root root 4096 Nov 21 02:50 5828234d5c1dfb7095276b3cb8a58f153f76469aa1a8ba4930a70315ed8b550a-init
    drwxr-xr-x  4 root root 4096 Nov 21 02:49 641c11ef78c18e2babe50fd6eb4f49189ff89af2f5af4b3a2439e24066629c1c
    drwxr-xr-x  8 root root 4096 Nov 21 02:49 641c11ef78c18e2babe50fd6eb4f49189ff89af2f5af4b3a2439e24066629c1c-init
    drwxr-xr-x  2 root root 4096 Nov 21 02:50 67c4ee1dc6e6d06131447c0d19ad566783afe3e017cfedf4603236f3e67d06e2
    drwxr-xr-x  2 root root 4096 Nov 21 02:50 8caa4511cdf8c2b11b69e3e1b254bb7bfd673648e76753e2d1a7ef99da9e7e85
    drwxr-xr-x  2 root root 4096 Nov 21 02:49 b901d36b6f2fd759c362819c595301ca981622654e7ea8a1aac893fbd2740b4c

the-snowwhite, 2015-11-21:
rebooted
Deleted all images
restarted Docker service
And ran as Superuser just to be sure (I have added user to docker group)

Seems like the folder complained about in error message contains only empty folders / files ?

    mib@Debian-8:~/mydockerbuild/ARM$ docker rm $(docker ps -a -q)    
    2a1268a057c2
    mib@Debian-8:~/mydockerbuild/ARM$ docker rmi $(docker images -q)
    Deleted: 01af1736f15fa74ebd7d17ac8c89a6b0fa9a6fad1092a6158020640d613f7f7f
    Deleted: 8caa4511cdf8c2b11b69e3e1b254bb7bfd673648e76753e2d1a7ef99da9e7e85
    Deleted: 106eecddfea1dbb2fb589044999a7b1c42f6e7ad21903cdfa7d77093e1bf12de
    Deleted: 67c4ee1dc6e6d06131447c0d19ad566783afe3e017cfedf4603236f3e67d06e2
    Deleted: 097ffd1b0a22bf8e78599c70fd6f9481b7cc4483807c097c0e752a9058be4d4b

    mib@Debian-8:~/mydockerbuild/ARM$ sudo service docker restart       

    mib@Debian-8:~/mydockerbuild/ARM$ sudo docker run bobvanderlinden/machinekit-build-debian-armhf           
    Unable to find image 'bobvanderlinden/machinekit-build-debian-armhf:latest' locally
    latest: Pulling from bobvanderlinden/machinekit-build-debian-armhf
    097ffd1b0a22: Pull complete 
    67c4ee1dc6e6: Pull complete 
    106eecddfea1: Pull complete 
    8caa4511cdf8: Pull complete 
    01af1736f15f: Pull complete 
    Digest: sha256:3d0eddcc5a312507e45fc47290b52cb3f70b40a32b67bfa6d14eeae5d1deec6d
    Status: Downloaded newer image for bobvanderlinden/machinekit-build-debian-armhf:latest
    no such file or directory
    Error response from daemon: Cannot start container cad8dfb50047a70b16d90cf85314a270019ea4e2a83fc12c30c33412783e5d60: [8] System error: no such file or directory
    mib@Debian-8:~/mydockerbuild/ARM$ sudo cp -R /var/lib/docker/aufs/diff/ /home/mib/mydockerbuild/test 
    mib@Debian-8:~/mydockerbuild/ARM$ sudo chown -R mib:mib /home/mib/mydockerbuild/test                   
    mib@Debian-8:~/mydockerbuild/ARM$ ls /var/lib/docker/aufs/diff/           
    ls: cannot access /var/lib/docker/aufs/diff/: Permission denied
    mib@Debian-8:~/mydockerbuild/ARM$ sudo ls /var/lib/docker/aufs/diff/ 
    01af1736f15fa74ebd7d17ac8c89a6b0fa9a6fad1092a6158020640d613f7f7f  8caa4511cdf8c2b11b69e3e1b254bb7bfd673648e76753e2d1a7ef99da9e7e85
    097ffd1b0a22bf8e78599c70fd6f9481b7cc4483807c097c0e752a9058be4d4b  cad8dfb50047a70b16d90cf85314a270019ea4e2a83fc12c30c33412783e5d60
    106eecddfea1dbb2fb589044999a7b1c42f6e7ad21903cdfa7d77093e1bf12de  cad8dfb50047a70b16d90cf85314a270019ea4e2a83fc12c30c33412783e5d60-init
    67c4ee1dc6e6d06131447c0d19ad566783afe3e017cfedf4603236f3e67d06e2

    mib@Debian-8:~/mydockerbuild/ARM$ sudo ls -la -R /var/lib/docker/aufs/diff/cad8dfb50047a70b16d90cf85314a270019ea4e2a83fc12c30c33412783e5d60/
    /var/lib/docker/aufs/diff/cad8dfb50047a70b16d90cf85314a270019ea4e2a83fc12c30c33412783e5d60/:
    total 16
    drwxr-xr-x 4 root root 4096 Nov 21 04:32 .
    drwxr-xr-x 9 root root 4096 Nov 21 04:32 ..
    -r--r--r-- 1 root root    0 Nov 21 04:32 .wh..wh.aufs
    drwx------ 2 root root 4096 Nov 21 04:32 .wh..wh.orph
    drwx------ 2 root root 4096 Nov 21 04:32 .wh..wh.plnk

    /var/lib/docker/aufs/diff/cad8dfb50047a70b16d90cf85314a270019ea4e2a83fc12c30c33412783e5d60/.wh..wh.orph:
    total 8
    drwx------ 2 root root 4096 Nov 21 04:32 .
    drwxr-xr-x 4 root root 4096 Nov 21 04:32 ..
    
    /var/lib/docker/aufs/diff/cad8dfb50047a70b16d90cf85314a270019ea4e2a83fc12c30c33412783e5d60/.wh..wh.plnk:
    total 8
    drwx------ 2 root root 4096 Nov 21 04:32 .
    drwxr-xr-x 4 root root 4096 Nov 21 04:32 ..
    mib@Debian-8:~/mydockerbuild/ARM$ 

    mib@Debian-8:~/mydockerbuild/ARM$ docker version
    Client:
     Version:      1.9.1
     API version:  1.21
     Go version:   go1.4.2
     Git commit:   a34a1d5
     Built:        Fri Nov 20 12:59:02 UTC 2015
     OS/Arch:      linux/amd64

    Server:
     Version:      1.9.1
     API version:  1.21
     Go version:   go1.4.2
     Git commit:   a34a1d5
     Built:        Fri Nov 20 12:59:02 UTC 2015
     OS/Arch:      linux/amd64
    mib@Debian-8:~/mydockerbuild/ARM$ docker info
    Containers: 1
    Images: 5
    Server Version: 1.9.1
    Storage Driver: aufs
     Root Dir: /var/lib/docker/aufs
     Backing Filesystem: extfs
     Dirs: 7
     Dirperm1 Supported: true
    Execution Driver: native-0.2
    Logging Driver: json-file
    Kernel Version: 3.16.0-4-amd64
    Operating System: Debian GNU/Linux 8 (jessie)
    CPUs: 16
    Total Memory: 15.61 GiB
    Name: Debian-8
    ID: SZYU:RJSB:2E27:A7DP:UXD2:QOJU:XJTX:326J:OHCP:OLQL:DIMI:4DIS
    WARNING: No memory limit support
    WARNING: No swap limit support

zultron, 2015-11-21:
@bobvanderlinden, WOW, you've shown that we can build for Jessie armv7l architecture in Travis CI, and this could be a replacement for the Buildbot.  Now I'm debating whether to stop with the Buildbot project & go drink martinis while you knock this one out.  ;)  Really nice work!

The Travis CI 120 minute build time might be enough even with no optimizations.  It's been a while, but I think that a fully-emulated qemu build on a single CPU took over an hour but under two.  (That doesn't include unit tests.)

I can't explain the libzmq3/4 problem yet, but yes it's all too familiar.  Where are the sources for your `bobvanderlinden/machinekit-build-debian-armhf` image?  I'd start by checking how the APT repos are configured.  Did you get a Wheezy repo in there, somehow?

zultron, 2015-11-21:
I'm having trouble finding an armhf Wheezy base image where it's transparent how it was built.  It looks like many images are built on someone's box and then uploaded as a `.tar.gz`, which raises issues about security and reproducibility.

@RobertCNelson What base image do you use for your `robertcnelson/machinekit-wheezy` Docker image?

@the-snowwhite The `armbuild/debian` images look reproducible from the github repo you linked to, but I don't understand the exact process of how the image is built from those scripts and finally lands on the Docker hub.  Guessing, the `armbuild` user runs the `update.sh` script on private hardware to generate a `rootfs.tar.xz`.  Then somewhere (where?) the image is generated from that tarball and the [`Dockerfile`][1].  How does the Docker hub 'automated build' fit into the picture?  If we use that for a base image, are we trusting the `armbuild` user and his `rootfs.tar.xz` file?  (That seems reasonable, since the original scripts are transparent and Online.net Labs runs the Scaleway ARM cloud; just doing my due diligence. ;)

[1]: https://hub.docker.com/r/armbuild/debian/~/dockerfile/

mhaberler, 2015-11-21:
@zultron - the build time limit - even if it bites us now - may not be an issue once we split the build into three packages (cnc stack, rt, machinetalk-protobuf), so no reason to view as showstopper for now

bobvanderlinden, 2015-11-21:
@the-snowwhite Ah sorry. I should've been more clear. The image is ARM-based, which means that you need some kind of emulation to execute the binaries. This is where `binfmt_misc` comes into play. It's a feature in the kernel that recognizes the binaries as being ARM and executes them using `qemu-arm-static`. You still need to 'register' `qemu-arm-static` as the executable that runs the ARM-based binaries. I used [this file](https://github.com/bobvanderlinden/travis-playground/blob/master/.travis.setup_binfmt_misc.sh) to do that. Notice the `/usr/local/bin/qemu-arm-static` path in there. That same filepath is available inside the `armbuild/debian` Docker image (and as I'm using that as the base image, it's also in `bobvanderlinden/machinekit-build-debian-armhf`.

@zultron Thanks!
> Now I'm debating whether to stop with the Buildbot project & go drink martinis while you knock this one out.

The build system is still falling apart :P No time for martinis yet!

> Where are the sources for your bobvanderlinden/machinekit-build-debian-armhf image?

https://github.com/bobvanderlinden/docker-machinekit-build-debian-armhf
Automated builds do not work, as I still require qemu-arm-static to be configured. That cannot be done on Docker Hubs, but is possible with TravisCI. At the moment I have manually pushed the image to Docker Hubs.

> Did you get a Wheezy repo in there, somehow?

You're right. I used the [Wheezy dovetail and backports repositories](https://github.com/bobvanderlinden/docker-machinekit-build-debian-armhf/blob/master/Dockerfile#L5-L8), so that is the problem.

> It looks like many images are built on someone's box and then uploaded as a .tar.gz, which raises issues about security and reproducibility.

Yes, that's how docker works for the base images apparently. [This is how they create the wheezy image](https://github.com/tianon/docker-brew-debian/blob/d8d023e100f72982b31a8c932570080c3101b7e7/wheezy/Dockerfile).

Also, if you're concerned about reproducability, check out NixOS and its package manager Nix. One of its goals is reproducability. It comes closest to atomic upgrades and deterministic builds than any other distro/package manager. It can build its own reproducable live CD image or VirtualBox image (). It's also the only distro (that I know) that does proper automated tests for their OS (that are again reproducible for others). This probably isn't the right place to convince you, so: http://nixos.org/nix/
You can install the Nix package manager alongside apt (or even on Mac OSX or Windows), as it does not alter existing system configuration or install conflicting files.

@mhaberler That's indeed good to know. I'll try to get the Travis stuff in a somewhat workable state, so we can see how much of a problem it is now.

mhaberler, 2015-11-21:
@bobvanderlinden - whacky idea re qemu and armv7 docker image, since I'm dabbling with docker on my cubox-i (armv7 quad-core): 

would it be possible to setup things such that this docker image runs both on Intel (switching to qemu) and armv7 (native, sans qemu) platforms? if so, it'd be a single repo to maintain, rather than two

bobvanderlinden, 2015-11-21:
@mhaberler I think that's possible. Would be cool if you gave it a try. The only extra dependency is having the kernel properly instructed to use `/usr/local/bin/qemu-arm-static`. If that's there, yes I think that's possible to use it for both. You can try running the current Machinekit build on your cubox-i:
```
docker pull bobvanderlinden/machinekit-build-debian-armhf:wheezy
git clone --depth 1 git://github.com/machinekit/machinekit.git
docker run -v $(pwd)/machinekit:/usr/src/machinekit bobvanderlinden/machinekit-build-debian-armhf:wheezy /bin/bash -c "cd /usr/src/machinekit && debian/configure -p && yes | mk-build-deps -i -r && debuild"
```
The result should be similar to [the output of travis](https://travis-ci.org/bobvanderlinden/travis-playground/builds/92420972#L1158-L1162). It still results in an error though.
@zultron Could you have a look at the error of the build --^? I'm not that familiar with Debian's packaging tools.

the-snowwhite, 2015-11-21:
There is something called github service hooks, where you can link 
dockerhub directly to All your github repos or just specific ones:
https://docs.docker.com/docker-hub/github/

Then there is the build on you own pc process:
http://docs.docker.com/linux/step_four/

or manually fro a github repo:
https://docs.docker.com/v1.8/reference/commandline/build/

mhaberler, 2015-11-21:
@bobvanderlinden well armv7h just seems to be the baseline arm version we can support

working on reproducing, need to move from SD to esata/ssd setup but looking good for the moment

ArcEye, 2015-11-21:
@the-snowwhite 
>> There is something called github service hooks, where you can link dockerhub directly to All your github repos or just specific ones: https://docs.docker.com/docker-hub/github/

I may have completely mis-understood (quite likely) but if docker can use github hooks, what is the point of using Travis?

I'll be on the sidelines waiting for the target to stop moving, before doing much more:)

bobvanderlinden, 2015-11-21:
@the-snowwhite @ArcEye Yes, that is indeed useful to create Docker images on Docker Hubs. This is what the Automated Builds seem to be doing: adding a hook on Github to Docker Hubs.
However, I haven't found a way to configure the kernel to support `binfmt_misc` on docker hubs. We need that to build the ARM-based images. That's why I needed to manually build and push the images for now on my own machine (this could potentially be done by Travis as well). The official images also seem to avoid Docker Hubs somehow as I can't find any logs on how those images were built. It seems those are pushed by some other method.

Apart from building Docker images, we also want to run tests. Maybe we want to deploy resulting binaries somewhere to download. For PRs we want feedback on whether the build/tests succeeded or not. I don't know Docker Hubs that well, but it seems it is only doing `docker build` (to build images) and not `docker run` (to execute configure/make/tests inside a container).

darenschwenke, 2015-11-21:
@bobvanderlinden this help?  https://groups.google.com/forum/#!msg/coreos-dev/YC-G_rVFnI4/ncS5bjxYWdcJ

bobvanderlinden, 2015-11-21:
@darenschwenke Yes, that's basically what Travis is doing atm. Those commands cannot run on Docker Hubs, as it only executes your Dockerfile. You cannot define commands that need to run *outside* of the container in the Dockerfile. The line line that is important (`echo ':arm:M::\x7fELF\x01\x01\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x02\x00\x28\x00:\xff\xff\xff\xff\xff\xff\xff\x00\xff\xff\xff\xff\xff\xff\xff\xff\xfe\xff\xff\xff:/usr/bin/qemu-arm-static:' > /proc/sys/fs/binfmt_misc/register`) is ran outside of the container, after that they use `docker run` to execute stuff inside the container.
Docker Hubs just doesn't support this kind of thing yet, so we need to look at other build-services. Travis is used by many other projects for a wide variety of builds, so that's why I wanted to give that a try.

That said, Travis is doing fine atm. It's now building armhf and x86_64 as separate build jobs: https://travis-ci.org/bobvanderlinden/travis-playground/builds/92442071
Doing `debuild` didn't work previously, so it's now only compiling using `make`.

kinsamanka, 2015-11-21:
On 21 Nov 2015 8:49 pm, "Bob van der Linden" <notifications@github.com>
wrote:
>
> @the-snowwhite @ArcEye Yes, that is indeed useful to create Docker images
on Docker Hubs. This is what the Automated Builds seem to be doing: adding
a hook on Github to Docker Hubs.
> However, I haven't found a way to configure the kernel to support
binfmt_misc on docker hubs. We need that to build the ARM-based images.
That's why I needed to manually build and push the images for now on my own
machine (this could potentially be done by Travis as well). The official
images also seem to avoid Docker Hubs somehow as I can't find any logs on
how those images were built. It seems those are pushed by some other method.
>
> Apart from building Docker images, we also want to run tests. Maybe we
want to deploy resulting binaries somewhere to download. For PRs we want
feedback on whether the build/tests succeeded or not. I don't know Docker
Hubs that well, but it seems it is only doing docker build (to build
images) and not docker run (to execute configure/make/tests inside a
container).
>

Docker Hubs can actually build and test the resulting binaries. We were
able to cross compile mk with qemu using the automated builds.

The only issue is that there's no elegant way of extracting the binaries -
the resulting container is way bigger than the compiled binaries (although
you can delete the sources inside the container to reduce the size, you
still need to download the container and extract the results).

I think we can use Docker Hub to verify that the PR builds and passes the
nosetests and let the packaging be handled separately.

bobvanderlinden, 2015-11-21:
> Docker Hubs can actually build and test the resulting binaries. We were
able to cross compile mk with qemu using the automated builds.

Sounds interesting. Do you still have the dockerfile somewhere?

> The only issue is that there's no elegant way of extracting the binaries -
the resulting container is way bigger than the compiled binaries (although
you can delete the sources inside the container to reduce the size, you
still need to download the container and extract the results).
> I think we can use Docker Hub to verify that the PR builds and passes the
nosetests and let the packaging be handled separately.

I might be misunderstanding, but why would you go the route of extracting things from a docker image? You can access everything from within the container, so you can upload the resulting files to somewhere else within the container.

Also, another update on the progress:
https://github.com/bobvanderlinden/machinekit/commit/6031d91c9bd04b7b970cf5b599da3c4a10eb1869
https://travis-ci.org/bobvanderlinden/machinekit
It's now integrated into the a fork of the machinekit repository. The travis-playground isn't needed anymore.

bobvanderlinden, 2015-11-21:
Hmm, the x86_64 builds are finishing succesfully, but the armhf builds hit the timelimit of 50 minutes. Apparently 120 minutes is only for travis-ci.com, which is their paid service: http://docs.travis-ci.com/user/customizing-the-build/#Build-Timeouts
It's now building with `-j2`, we'll see whether that helps.

the-snowwhite, 2015-11-22:
It seems like Docker recently added to something called Machine that enables driver support:
Here:
https://github.com/docker/machine/pull/650

    Implement libvirt/KVM support #692    Closed
    Proposal: KVM/Qemu driver #504    Closed

    Docker images to enable qemu binfmt support.

    Run docker run --rm --privileged vicamo/binfmt-qemu:latest to enable qemu binfmt support.

    FROM debian:stable

    MAINTAINER vicamo@gmail.com
 
    RUN apt-get update \
	    && apt-get install -y --no-install-recommends \
		    binfmt-support \
		    qemu-user-static \
    	    && apt-get clean \
	    && rm -rf /var/lib/apt/lists/*_dists_*

    ENTRYPOINT ["dpkg-reconfigure", "qemu-user-static"]

Sorry for just dumping the tab links:

https://hub.docker.com/search/?q=qemu%20machine%20arm&page=1&isAutomated=1&isOfficial=0&starCount=0&pullCount=0

https://github.com/Ulexus/docker-qemu
https://github.com/docker/machine/issues/504
https://github.com/docker/machine/issues/1626
https://github.com/docker/machine/pull/650
https://github.com/fventuri/machine/tree/504-kvm-qemu-driver
https://docs.docker.com/machine/
https://github.com/nathanleclaire
https://hub.docker.com/search/?q=qemu&page=1&isAutomated=1&isOfficial=0&starCount=0&pullCount=0
https://www.google.dk/search?q=docker+pull+ulexus%2Fqemu-bootstrap&oq=docker+pull+ulexus%2Fqemu-bootstrap&aqs=chrome..69i57&sourceid=chrome&es_sm=119&ie=UTF-8
https://github.com/Ulexus/docker-qemu-bootstrap
https://github.com/vicamo/docker-binfmt-qemu
https://hub.docker.com/r/tianon/qemu/
https://github.com/tianon/dockerfiles/tree/master/qemu
https://hub.docker.com/r/ulexus/qemu-bootstrap/
https://hub.docker.com/r/vicamo/binfmt-qemu/
https://github.com/vicamo/docker-binfmt-qemu/blob/master/Dockerfile

zultron, 2015-11-22:
@bobvanderlinden, amd64 builds working probably means i686 will, too.  Wow, congrats!

Your work is taking a lot of pressure off me and really the whole project, since we got into this mess by having only one irreplaceable person and one unreproducible system.  By now you've gone way beyond the simple proof-of-concept I asked for, and aren't far from replacing the core functionality of the current Buildbot with something that any (replaceable) person can trivially reproduce.  And those are exactly the important goals of this ticket.  Thank you!

If you want to make me completely redundant, get those armhf builds to succeed.  You can crib from a (hopefully) working example:  my [`dxsbuild`][1] scripts contain some tricks for optimizing x86-cross-arm builds, maybe getting us below the 50 minute bar.

Can Travis CI store state between builds?  If so, `ccache` helps a lot; just install [2], set up [3] and be sure `$PATH` is set up to pick up the links in `/usr/lib/ccache` [4].

`distcc` should work with Travis CI, and save time by farming compilation out to a (non-`qemu`lated) cross-compile toolchain running outside (or inside?) Docker.  Just install and configure [5], start the daemon [6], and configure the build environment [7], [8].

I'm pretty sure that with some of these optimizations, the build time can be reduced to under 50 minutes.  IIRC, I had the build time down to ~10 minutes with four CPUs.  And as @mhaberler says, once the CNC application is split off, this time will be reduced drastically.

[1]:  https://github.com/zultron/dxsbuild/
[2]: https://github.com/zultron/dxsbuild/blob/master/share/Dockerfile#L48
[3]: https://github.com/zultron/dxsbuild/blob/master/scripts/debian-binary-package.sh#L46-L59
[4]: https://github.com/zultron/dxsbuild/blob/master/share/Dockerfile#L133
[5]: https://github.com/zultron/dxsbuild/blob/master/share/Dockerfile#L48-L53
[6]: https://github.com/zultron/dxsbuild/blob/master/scripts/distcc.sh#L35-L46
[7]:  https://github.com/zultron/dxsbuild/blob/master/scripts/distcc.sh#L17-L33
[8]: https://github.com/zultron/dxsbuild/blob/master/scripts/distcc.sh#L31

mhaberler, 2015-11-22:
@cdsteinkuehler  - arm7 docker status update, cubox-i: finally got this to run on Jessie - with the docker.io package by @RobertCNelson (big thanks!)

key step is running a kernel with various docker-related options in place. I finally went for [3.14.57 precompiled from here](http://forum.solid-run.com/linux-kernel-and-bootloaders-on-cubox-i-and-hummin-f10/kernel-3-14-57-4-2-6-available-t1156.html) - the process and documentation is bizarre but the result works:

`````
mah@cubox-i:~$ uname -a
Linux cubox-i 3.14.57 #1 SMP PREEMPT Mon Nov 9 15:21:39 MST 2015 armv7l GNU/Linux
mah@cubox-i:~$ docker run armv7/armhf-ubuntu_core /bin/echo 'Hello world'
Hello world
`````

root on SSD now, so now onto bigger challenges

mhaberler, 2015-11-22:
@bobvanderlinden - running [your test from above](https://github.com/machinekit/machinekit/issues/791#issuecomment-158629439)
starts fine (pulls in zmq4 :-/ ), then stalls on 

````
setting up libboost-thread-dev (1.49.0.1) ...
Setting up machinekit-build-deps (1.0) ...
Processing triggers for python-support ...
This package has a Debian revision number but there does not seem to be
an appropriate original tar file or .orig directory in the parent directory;
(expected one of machinekit_0.1.1.orig.tar.gz, machinekit_0.1.1.orig.tar.bz2,
machinekit_0.1.1.orig.tar.lzma,  machinekit_0.1.1.orig.tar.xz or machinekit.orig)
continue anyway? (y/n) Use of uninitialized value $ans in pattern match (m//) at /usr/bin/debuild line 1002.
mah@cubox-i:~$
````

but that isnt a docker problem anymore but our build. so optimistic nevertheless ;)

bobvanderlinden, 2015-11-22:
@the-snowwhite Can you explain what we're solving? I'm a bit confused.
The Dockerfile installs `qemu-user-static` inside the docker instance. From within the container it sets up `qemu-user-static`. That's the same I'm doing from the host (related to `/proc/sys/fs/binfmt_misc/register`). To do that from the container, you indeed need `--privileged` because `/proc/` of the host needs to be exposed to the container. I think `--privileged` is not something you can use on Docker Hubs.

@zultron 
Caching is only available for paid services. But we *could* compile clone and compile machinekit once in a Docker container, upload the resulting image and use that on the CI. However, it's more stable to always start with a clean repo. `make clean` is sometimes needed. In addition, the Docker container image needs to be updated every once in a while to keep the 'caches' up-to-date.

Apart from that, `distcc` indeed sounds like a good solution. I'm not sure how well that performs over the internet, but it is worth a shot. We only need hardware... preferably a quad-core ARM... with enough ram... so...

@mhaberler That's great! Try to run:
```sh
git clone https://github.com/bobvanderlinden/machinekit
cd machinekit
docker run -v $(pwd):/usr/src/machinekit bobvanderlinden/machinekit-build:wheezy-armhf /bin/bash /usr/src/machinekit/.travis/build.sh
```
That should be exactly what Travis is doing, apart from the `binfmt_misc` stuff.

... so... to get `distcc` running like @zultron is suggesting, that cubox-i sounds like it fits the job. Is there some way to expose its `distccd` service so that Travis can access it? The distcc FAQ is suggesting [to do this over ssh](https://distcc.googlecode.com/svn/trunk/doc/web/faq.html#Internet-wide distributed compilation). This should be possible in Travis as well, but [the private key of Travis can only be used on builds for the master branch, not PR builds](http://docs.travis-ci.com/user/encryption-keys/), which makes sense.

We could say that x86 builds for PR are sufficient and ARM builds only run when a PR is pulled. After the project is split, this could *maybe* eventually be lifted. (I'm still a bit wary on the build times, as it would seem from comparing the log-size between x86 and qemu-arm that it would need at least twice as much time: cutting the project in half might not be enough for the 50m timelimit).

kinsamanka, 2015-11-22:
@bobvanderlinden Travis is very cool! 

Here's my ["hello world"](https://travis-ci.org/kinsamanka/travis-test/builds/92553055) creating armhf binaries.

The cool thing is that we can deploy the results using ftp or ssh :smiley: 

I think we should start splitting up the codebase so we stay within the 50 min limit.

bobvanderlinden, 2015-11-22:
@kinsamanka
Oh awesome! Cross compilation works correctly? I was thinking that wasn't possible due to the build system using binaries that it previously built, but with emulation *and* cross-compilation it indeed works!
So... that means no more need for external ARM-based systems, just purely running on Travis?

Also, how did you built the `kinsamanka/docker-qemu-chroot` image? The `FROM` statement in [that is on docker hubs](https://hub.docker.com/r/kinsamanka/docker-qemu-chroot/~/dockerfile/) refers it itself, not `scratch` or `debian` or anything. How does that work?

Indeed, you can generate a private key for Travis, so that pushes to master (or tags) can also deploy to ssh with that key.

Yes, splitting up the project is still very preferable. However your test already showed we can already compile armhf within the timelimit right?

kinsamanka, 2015-11-22:
It's still Qemu emulation but the speed up is due to using native cross compiler instead of the emulated compiler. All the magic is done behind the scenes by proot.

Note that this is just compiling, there are no nosetests or regression tests being done. I think it's still better to use real hardware for the tests.

I'm making use of docker tags:
[image]

Here are the dockerfiles that I've used:
https://github.com/kinsamanka/docker-qemu-chroot
https://github.com/kinsamanka/dockerfiles/tree/master/mk-qemu-cross

BTW, it's now down to 18 minutes by using ```make -j2``` and removing the ```setup_binfmt_misc``` step: https://travis-ci.org/kinsamanka/travis-test/builds/92557316

cdsteinkuehler, 2015-11-22:
On 11/22/2015 5:10 AM, Michael Haberler wrote:
> @cdsteinkuehler  - arm7 docker status update, cubox-i: finally got this to run on Jessie - with the docker.io package by @RobertCNelson (big thanks!)

Great news all around!

I've been working on my own VM stack here (for running my personal
servers).  The upgrades are just about finished, at which point I'll
have a docker-aware VM server available for x86 containers that can be
put to use, and some time to migrate some of my ARM systems to run
docker as well.

It's exciting to see such quick progress!

-- 
Charles Steinkuehler
charles@steinkuehler.net

the-snowwhite, 2015-11-22:
@bobvanderlinden If we are talking about the ability to run qemu emulation on a hub service like Dockerhub.
Work was started but stranded on the way to to getting upgraded to an plugin extension:
https://docs.docker.com/engine/extend/plugins/

As far as I can find out the code enabling the ability to run qemu on a hub server started here in a separate branch (504-kvm-qemu-driver) :
https://github.com/fventuri/machine/commit/872782272fa16017b58569a12f6662f17e3c997e

moving to here:
https://github.com/SvenDowideit/machine/commit/cc451afd5da96fa38a655ab78ec5aee3bdfb5da3
as a driver PR
https://github.com/docker/machine/compare/master...SvenDowideit:add-qemu-driver

That then got closed:
https://github.com/docker/machine/pull/650

Refering to implemention of plugins instead of drivers.
https://github.com/docker/machine/issues/1626

And that abandoned code has not been transformed into a plugin extension:

https://github.com/docker/machine/pull/1902

zultron, 2015-11-22:
@bobvanderlinden `distcc` runs locally within Travis CI, not over the Internet.  The idea is to run an amd64 binary version of `gcc-arm-none-eabi` to compile for ARM.  Then the compiler doesn't need to be qemu-lated, and runs at native speed; this speeds up `gcc` and `g++` by 500% (though the linking step is still in qemu).  Am I making sense?

[Example][1] for configuration on Jessie; IIRC, Ubuntu already has the cross-build tools, so just `dpkg --add-architecture armhf` and configure the APT repos.  Then [install][2] the tools.

Anyway, with @kinsamanka helping out (his `proot` stuff is magic), you two heavyweights don't need my help.  Very pleased!

[1]: https://github.com/zultron/dxsbuild/blob/master/share/Dockerfile#L19-L30
[2]: https://github.com/zultron/dxsbuild/blob/master/share/Dockerfile#L37-L38

luminize, 2015-11-22:
Hi Guys,
I can't help you _now_ with the technical stuff, because you are going so fast that that's just impossible for me :)
But good work going on, and I really like how you work together!

But if in the future it's more in line of using a set of instructions for setting up, running this docker image to get a set of packages, or a successful build, all in a 1 hour time window (then I personally can make some time in an evening and succeed that same evening) then let me know of I can test the workflow!

zultron, 2015-11-23:
I discussed with Michael and we thought it important to spell out the top-level goals without implying a solution.

1a) MUST build packages for all architectures and supported debian versions
1b) MUST having packages numbered such that a newer package will be detected by apt-get update/upgrade
1c) SHOULD upload successful builds automatically to deb.dovetail-automata.com (can be manually for a start)
1d) MUST keep apt-get from deb.dovetail-automata.com running unchanged (many people depend on it).

Important but NOT urgent goals:
2a) There SHOULD be a simple way to run regression tests for all arches and deb versions without per-arch/per deb version OS.
2b) The results of regression tests SHOULD be made available eventually online in some form.
2c) Regression tests MUST run x86/posix regression results. They MAY run other arches and deb versions.

mhaberler, 2015-11-23:
John will be busy for a day or two (a platoon's worth of family forming on the horizon :-)  but he'll surely elaborate on where not-so-easy problems in the above are and how he plans to address them

I just comment on requirement 2c) because that is one I understand in full and discussed at length:

we observed that:
- work will likely show up in non-flavor-dependent code, or if so, in code which includes posix and rt-preempt builds
- if the regression passes chances are VERY high this passes on all flavors
- changes to xenomai ONLY are extremely rare
- since kernel threads builds are history, a major source of regression issues is cut out

so 2c) is about applying 80/20 - get most of the bang with minimal effort.

In fact I'll modify this to state amd64/posix because of i386 and amd64, the latter is the 'more prickly' build since it has more rigid demands on alignment, int size, shared library issues and position-independent code so we better use amd64 as a yardstick

mhaberler, 2015-11-23:
@bobvanderlinden, @kinsamanka - help me across the street: if I get this right, what you are up to addresses 1a and/or 2a, correct?

kinsamanka, 2015-11-23:
> 
> I discussed with Michael and we thought it important to spell out the top-level goals without implying a solution.
> 
> 1a) MUST build packages for all architectures and supported debian versions

I think this is doable using Travis CI, I’m currently working on i386 builds.

> 1b) MUST having packages numbered such that a newer package will be detected by apt-get update/upgrade

We need to see your code on how you are currently doing this.

> 1c) SHOULD upload successful builds automatically to deb.dovetail-automata.com (can be manually for a start)

Doable. Must provide ssh access with only scp/sftp enabled.

> 1d) MUST keep apt-get from deb.dovetail-automata.com running unchanged (many people depend on it).

I think deb signing should still be under your control?

> 
> Important but NOT urgent goals:
> 2a) There SHOULD be a simple way to run regression tests for all arches and deb versions without per-arch/per deb version OS.

I’m trying this on Travis. Issue #794 must be addressed first.

> 2b) The results of regression tests SHOULD be made available eventually online in some form.

Travis can handle this.

> 2c) Regression tests MUST run x86/posix regression results. They MAY run other arches and deb versions.

My docker container currently passes the regression tests. No nose tests yet.

mhaberler, 2015-11-23:
@kinsamanka - only commenting on 1d), I think of it like so:

- the overall goal is to have deliverables which enable _everybody_ to run _any_ build and/or regression test without special hardware/special os installs/maybe even without a decent Intel platform (just on some cloud service), and maybe even feed their own apt repos
- if we get the buildbot 'green button' grid up again: bonus value, but this is lower priority and we might tradeoff several links against a consolidated view of a single bot (which again might be a SPOF)
- that said, we cannot let the deb.dovetail-automata.com apt repo keel over because so many people rely on it and have it in their sources.list - plus at least @RobertCNelson 's SD image build relies on it (right?) and then some other user bases (or even products) do have it engraved

mhaberler, 2015-11-23:
ok, #794 is understood, I will look into it

kinsamanka, 2015-11-23:
> On 23 Nov 2015, at 11:05 PM, Michael Haberler <notifications@github.com> wrote:
> 
> @kinsamanka - only commenting on 1d), I think of it like so:
> 
> 	• the overall goal is to have deliverables which enable everybody to run any build and/or regression test without special hardware/special os installs/maybe even without a decent Intel platform (just on some cloud service), and maybe even feed their own apt repos
> 	• if we get the buildbot 'green button' grid up again: bonus value, but this is lower priority and we might tradeoff several links against a consolidated view of a single bot (which again might be a SPOF)
> 	• that said, we cannot let the deb.dovetail-automata.com apt repo keel over because so many people rely on it and have it in their sources.list - plus at least @RobertCNelson 's SD image build relies on it (right?) and then some other user bases (or even products) do have it engraved

The debs on deb.dovetail-automata.com  are signed using John’s GPG keys.

I think whoever provides a repo should do the signing themselves, otherwise we’ll just provide an unsigned repository.

mhaberler, 2015-11-23:
@kinsamanka - right: everybody running an apt repo does his own signing 

that is also in line with the stated policy: the machinekit project per se does not distribute packages, but individuals are encouraged to do so and there is nothing wrong with having more than one apt repo  (or regression test runs for that matter)

zultron, 2015-11-23:
@kinsamanka, @mhaberler, agreed, the owner of the APT repo signs the packages.  In the Dovetail Automata repo's case, packages may need to be gathered from multiple builders (Travis CI or single-configuration Buildbots) into a single APT repo; they will only be signed after that.

zultron, 2015-11-23:
@kinsamanka, @bobvanderlinden How can Travis CI make build results available to satisfy 1c?  Is pushing with scp/ftp/etc. the only way, or can Travis CI make files available on the web for pulling?

bobvanderlinden, 2015-11-23:
@zultron Yes, pushing to ssh. Travis cannot host files AFAIK.

@kinsamanka I'd like to merge your work into the Machinekit repository (first on my fork of course). Is that a good idea? Should we do some coordination? Maybe on Gitter perhaps?

mhaberler, 2015-11-23:
@kinsamanka we (bob bas zultron me) hang out on https://gitter.im/machinekit/machinekit - in case you're only, tune in - great quick turnaround coordination

zultron, 2015-11-23:
# Package versioning requirements for 1d and 1b

To enable the result of this effort to feed deb.dovetail-automata.com,
numbering needs to follow certain conventions in order to fit the
structure of the APT repo and not break existing configurations
(requirement 1d), as well as provide smooth version incrementing
so that `apt-get` can correctly upgrade newer packages (requirement 1b).

Package version:
- The package version must increment with successive builds
  - This ensures `apt-get` correctly tracks the package stream
- The `machinekit_VERSION.orig.tar.bz2` source tarball, and thus the
  version, is shared by all source and binary packages
  - Debian packages are built on releases from the upstream
    project with incrementing version numbers; we emulate this by
    generating an incrementing minor version

Package release:
- The source package release must be unique for each OS (esp. Wheezy
  and Jessie)
  - Source packages are configured differently for each OS (e.g. tcl
    versions), and thus cannot be shared
  - Packages must have non-conflicting file names to sit in the same
    pool directory with the same package built for another distro
- Arch-specific binary builds (amd64, i386, armhf) for a single OS
  share a source package, and thus share the release

Since the current monolithic Buildbot will be replaced with many
independent builders, whether Travis CI or single-configuration
Buildbots, the challenge is how to generate identical versions and
releases with no central coordination.

The current Buildbot generates unique, continually incrementing
version numbers by generating a release number from the Buildbot build
number.  For example, see the `ant` builder build [#1517][1]: in the
`configure-source-package` step, this builder generates one source
package per OS with the following versions:

	0.1.1517-1da.bbot.pr795.git8a6a0e1~1wheezy
	0.1.1517-1da.bbot.pr795.git8a6a0e1~1jessie

- Version `0.1.1517`:
  - `0.1`:  Manual, arbitrary choice
  - `1517`:  The `ant` builder build number
- Release `1da.bbot.pr795.git8a6a0e1~1wheezy`:
  - `1da.bbot`:  Arbitrary choice representing package origin
  - `pr795`:  GitHub PR
  - `git8a6a0e1`:  Git repo short sha1
  - `~1wheezy`:  OS

This scheme complies with the above requirements:

- The `ant` builder build number increments after every succesive
  build; it is combined with `0.1` to produce a version number for the
  'release' tarball
- The release is mostly data to help humans identify the build, but by
  adding `~1wheezy` or `~1jessie` it is unique for each OS

With this version naming scheme, packages for multiple distros can sit
in a single APT repo `pool` directory; see [this example][2].

Since the new CI system's only central coordination is GitHub,
especially the `ant` builder build number needs to be replaced by
another number that can be generated from the GitHub merge commit.
That probably means the GitHub merge commit timestamp, expressed as
`date +%s`, seconds since epoch.  All other version and release
components can be established by convention (e.g. `0.1`), extracted
from the merge commit (e.g. `pr795`), or based on build parameters
(e.g. `~1wheezy`).

[1]: http://buildbot.dovetail-automata.com/builders/ant/builds/1517
[2]: http://deb.dovetail-automata.com/pool/main/m/machinekit/

bobvanderlinden, 2015-11-23:
@zultron I'm not sure whether increments on successful builds are as trivial. Travis does supply us with the build number, but it always increments even on failures. It is available as an environment variable: https://travis-ci.org/bobvanderlinden/machinekit/jobs/92786096#L159 If Travis indeed does not supply us with the number of succesful builds, then I'm thinking of retrieving the last build from dovetail... but that's a bit hacky.
That said, I do like that we can reference it back to one particular build/test/deployment, instead of having multiple failures for that same number. Is it a problem to use the build number of travis?

zultron, 2015-11-23:
@bobvanderlinden Basing the version number on the GitHub merge commit timestamp is sufficient.  They don't need to be incremented by just one, as long as the next build version is always higher than the previous.

By using the timestamp and adopting other conventions,  we could even mix packages from Travis CI, Buildbot and other systems into a single repo!

Did that make sense, and did it answer your question?

[184 more replies]

closed 2015-12-17 by kinsamanka