machinekit/machinekit #792: improving latency results: not every tweak is known - let's collect them here

mhaberler, 2015-11-20:
this acts as a collector issue for tweaks related to improving latency of all platforms and relevant kernels (rt-preempt, xenomai)

it might eventually be made a manual section after which this can be closed and maintenance happens in the manual

mhaberler, 2015-11-20:
rt-preempt/any architecture:

/dev/cpu_dma_latency looks promising: see https://access.redhat.com/articles/65410

mhaberler, 2015-11-20:
rt-preempt/measuring latency/any architecture:

[cyclictest](https://rt.wiki.kernel.org/index.php/Cyclictest) is the way to do it IMO - other than our latency_test, this code is maintained and used by the rt-preempt developers 

on the rpi2 I needed a minor tweak to get cyclictest to work:

mhaberler, 2015-11-20:
rpi2/armv7k/[3.18.9-rt5-v7+](http://docs.emlid.com/navio/Downloads/Real-time-Linux-RPi2/) results:

````
mah@rpi2:/next/home/mah/src/rt-tests$ sudo  cyclictest -t1 -p 80 -n -i 10000 -l 10000
# /dev/cpu_dma_latency set to 0us
policy: fifo: loadavg: 0.67 0.42 0.36 1/142 27899

T: 0 (27877) P:80 I:10000 C:  10000 Min:     12 Act:   13 Avg:   16 Max:      66
````

mhaberler, 2015-11-20:
i386/j1900 mobo/4.1.10-rt10mah rt-preempt results:

`````
mah@j1900:/next/home/mah/src/rt-tests-i386$ sudo  cyclictest -t1 -p 80 -n -i 10000 -l 10000
# /dev/cpu_dma_latency set to 0us
policy: fifo: loadavg: 0.00 0.01 0.05 1/284 7160

T: 0 ( 7155) P:80 I:10000 C:  10000 Min:      9 Act:   10 Avg:   10 Max:      21
``````

luminize, 2015-11-20:
This is a welcome thread!
I'll read up and post my results

claudiolorini, 2015-11-20:
... my 0,5 cents:
a fine-tuning trick that can give some results is moving IRQs along processors(i've seen slight
improvment on Zynq platforms but it should work also on other multiprocessor architectures)

- the IRQ mapping can be seen in:
  > cat /proc/interrupts 
  
for example if the mmcard irq index is 56 on the CPU 1 , is possible to move it on the CPU2 
writing in smp_affinity with this command: 
  >  sudo echo 2 | sudo tee /proc/irq/56/smp_affinity

the effect of moving around the IRQs can be seen here:
https://github.com/machinekit/machinekit/issues/554

C.

luminize, 2015-11-21:
I got 3 tests to add
all tests were done with cyclictest running for approx 3 hours

1. workstation 2x quad core **without** kernel boot options `processor.max_cstate=1 idle=poll` CPU (one of 8) info below
2. same as above, but **with** `processor.max_cstate=1 idle=poll` boot option
3. J1900 motherboard, **with** `processor.max_cstate=1 idle=poll` boot option

the difference between 1 and 2 are visible.
the 3rd test does not get good latencies at all. I guess I must dig into the bios further.

### 1
```
bas@debian:~$ uname -a
Linux debian 4.1.10-rt11bdb #2 SMP PREEMPT RT Sat Oct 31 18:12:01 CET 2015 x86_64 GNU/Linux
bas@debian:~$ sudo  cyclictest -t -p 80 -n -i 10000
[sudo] password for bas: 
# /dev/cpu_dma_latency set to 0us
policy: fifo: loadavg: 1.07 0.69 0.69 1/391 6826           

T: 0 ( 4556) P:80 I:10000 C: 876622 Min:      3 Act:    6 Avg:    4 Max:      36
T: 1 ( 4557) P:80 I:10500 C: 834878 Min:      2 Act:    3 Avg:    4 Max:      20
T: 2 ( 4558) P:80 I:11000 C: 796929 Min:      2 Act:    3 Avg:    4 Max:      46
T: 3 ( 4559) P:80 I:11500 C: 762280 Min:      2 Act:    6 Avg:    4 Max:      34
T: 4 ( 4560) P:80 I:12000 C: 730518 Min:      2 Act:    5 Avg:    4 Max:      37
T: 5 ( 4561) P:80 I:12500 C: 701297 Min:      2 Act:    3 Avg:    4 Max:      26
T: 6 ( 4562) P:80 I:13000 C: 674324 Min:      2 Act:    5 Avg:    4 Max:      30
T: 7 ( 4563) P:80 I:13500 C: 649349 Min:      2 Act:    4 Avg:    4 Max:      33
```

### 2
```
bas@debian:~$ sudo  cyclictest -t -p 80 -n -i 10000
[sudo] password for bas: 
# /dev/cpu_dma_latency set to 0us
policy: fifo: loadavg: 0.63 0.79 0.83 1/316 3658           

T: 0 ( 1299) P:80 I:10000 C: 906899 Min:      2 Act:    4 Avg:    3 Max:      18
T: 1 ( 1300) P:80 I:10500 C: 863713 Min:      2 Act:    4 Avg:    4 Max:      37
T: 2 ( 1301) P:80 I:11000 C: 824454 Min:      2 Act:    4 Avg:    3 Max:      17
T: 3 ( 1302) P:80 I:11500 C: 788608 Min:      2 Act:    4 Avg:    4 Max:      38
T: 4 ( 1303) P:80 I:12000 C: 755749 Min:      2 Act:    4 Avg:    3 Max:      18
T: 5 ( 1304) P:80 I:12500 C: 725519 Min:      2 Act:    4 Avg:    3 Max:      18
T: 6 ( 1305) P:80 I:13000 C: 697615 Min:      2 Act:    4 Avg:    3 Max:      15
T: 7 ( 1306) P:80 I:13500 C: 671777 Min:      2 Act:    3 Avg:    3 Max:      22
```

### 3
```
bas@debian:~$ uname -a
Linux debian 4.1.10-rt11mah #1 SMP PREEMPT RT Sun Nov 1 11:47:31 CET 2015 i686 GNU/Linux
bas@debian:~$ sudo  cyclictest -t -p 80 -n -i 10000
# /dev/cpu_dma_latency set to 0us
policy: fifo: loadavg: 0.67 0.83 0.77 1/276 4784           

T: 0 ( 1081) P:80 I:10000 C:1541158 Min:      4 Act:   10 Avg:   10 Max:      67
T: 1 ( 1082) P:80 I:10500 C:1467770 Min:      4 Act:   10 Avg:   10 Max:      71
T: 2 ( 1083) P:80 I:11000 C:1401053 Min:      4 Act:   10 Avg:   10 Max:      68
T: 3 ( 1084) P:80 I:11500 C:1340138 Min:      4 Act:   11 Avg:   10 Max:      63
```

### cpu info of tests 1 and 2
```
less /proc/cpuinfo
processor       : 0
vendor_id       : GenuineIntel
cpu family      : 6
model           : 23
model name      : Intel(R) Xeon(R) CPU           E5420  @ 2.50GHz
stepping        : 6
microcode       : 0x60b
cpu MHz         : 1998.000
cache size      : 6144 KB
physical id     : 0
siblings        : 4
core id         : 0
cpu cores       : 4
apicid          : 0
initial apicid  : 0
fpu             : yes
fpu_exception   : yes
cpuid level     : 10
wp              : yes
flags           : fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx lm constant_tsc arch_perfmon pebs bts rep_good nopl aperfmperf pni dtes64 monitor ds_cpl vmx est tm2 ssse3 cx16 xtpr pdcm dca sse4_1 lahf_lm dtherm tpr_shadow vnmi flexpriority
bugs            :
bogomips        : 5000.67
clflush size    : 64
cache_alignment : 64
address sizes   : 38 bits physical, 48 bits virtual
power management:
```

luminize, 2015-12-28:
With a current newer kernel the latency got improved w.r.t nr **1** here https://github.com/machinekit/machinekit/issues/792#issuecomment-158676527
```
bas@debian:~$ sudo cyclictest -a -t -n -p99
[sudo] password for bas: 
# /dev/cpu_dma_latency set to 0us
policy: fifo: loadavg: 0.72 0.71 0.77 1/348 4197           

T: 0 ( 1904) P:99 I:1000 C:8819268 Min:      1 Act:    3 Avg:    3 Max:      22
T: 1 ( 1905) P:99 I:1500 C:5879512 Min:      1 Act:    4 Avg:    3 Max:      22
T: 2 ( 1906) P:99 I:2000 C:4409634 Min:      2 Act:    3 Avg:    3 Max:      31
T: 3 ( 1907) P:99 I:2500 C:3527707 Min:      2 Act:    3 Avg:    3 Max:      29
T: 4 ( 1908) P:99 I:3000 C:2939756 Min:      2 Act:    4 Avg:    3 Max:      25
T: 5 ( 1909) P:99 I:3500 C:2519790 Min:      2 Act:    3 Avg:    3 Max:      28
T: 6 ( 1910) P:99 I:4000 C:2204817 Min:      2 Act:    3 Avg:    3 Max:      18
T: 7 ( 1911) P:99 I:4500 C:1959837 Min:      2 Act:    3 Avg:    3 Max:      28
^Cbas@debian:~$ uname -a
Linux debian 4.1.15-rt17mah #6 SMP PREEMPT RT Wed Dec 23 20:12:08 CET 2015 x86_64 GNU/Linux
```

gmoccapy, 2016-01-06:
Here are my results without any optimisatiions

Laptop ACER E5-771G-7169

'''
root@machinekit:~# cyclictest -t -p 80 -n -i 10000
# /dev/cpu_dma_latency set to 0us
policy: fifo: loadavg: 0.58 0.83 0.91 1/316 6121           

T: 0 ( 5621) P:80 I:10000 C: 131440 Min:      1 Act:    4 Avg:    4 Max:      38
T: 1 ( 5622) P:80 I:10500 C: 125181 Min:      1 Act:    3 Avg:    3 Max:      29
T: 2 ( 5623) P:80 I:11000 C: 119491 Min:      1 Act:    3 Avg:    3 Max:      36
T: 3 ( 5624) P:80 I:11500 C: 114296 Min:      1 Act:    4 Avg:    3 Max:      25
'''

Processor Intel I7 ; 8 GB Ram
'''
processor       : 0
vendor_id       : GenuineIntel
cpu family      : 6
model           : 61
model name      : Intel(R) Core(TM) i7-5500U CPU @ 2.40GHz
stepping        : 4
microcode       : 0x12
cpu MHz         : 2401.406
cache size      : 4096 KB
physical id     : 0
siblings        : 4
core id         : 0
cpu cores       : 2
apicid          : 0
initial apicid  : 0
fpu             : yes
fpu_exception   : yes
cpuid level     : 20
wp              : yes
flags           : fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc aperfmperf eagerfpu pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch ida arat epb pln pts dtherm intel_pt tpr_shadow vnmi flexpriority ept vpid fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid rdseed adx smap xsaveopt
bugs            :
bogomips        : 4797.07
clflush size    : 64
cache_alignment : 64
address sizes   : 39 bits physical, 48 bits virtual
power management:
'''
OS Info (Debian Jessie 8.2)
'''
root@machinekit:~# uname -rvmpo
4.1.0-0.bpo.1-rt-amd64 #1 SMP PREEMPT RT Debian 4.1.3-1~bpo8+1 (2015-08-12) x86_64 unknown GNU/Linux

'''

I think to use MESA 7i76E quiete ok, In the background was 2 x glxgears, 1 x latency test and surfing in the internet and getting linuxcnc

Norbert
(gmoccapy)

Norbert

mhaberler, 2016-01-06:
interesting article: https://lttng.org/blog/2016/01/06/monitoring-realtime-latencies/

btw we're on good terms with the LTTNG folk

sirop, 2016-01-18:
I have "stolen" the BIOS settings from https://github.com/sirop/mk/blob/master/Machinekit-Xenomai-Thinkpad-X200.md#konfiguration-linux--xenomai 
```
GRUB_CMDLINE_LINUX_DEFAULT="quiet  lapic=notscdeadline hpet=disable i915.i915_enable_rc6=0 i915.powersave=0 intel_idle.max_cstate=1 processor.max_cstate=1 isolcpus=1 idle=poll"
```
Set them all except *xeno_hal.smi=1* .  
My hardware: https://gist.github.com/sirop/47d19d9e2da3039e93cb

```
slave@debian-slave:~/rt-tests$ sudo  ./cyclictest -t -p 80 -n -i 10000
# /dev/cpu_dma_latency set to 0us
policy: fifo: loadavg: 0.05 0.06 0.05 1/232 3988

T: 0 ( 3980) P:80 I:10000 C: 160885 Min:      4 Act:    7 Avg:    7 Max:      20
T: 1 ( 3981) P:80 I:10500 C: 153224 Min:      5 Act:    8 Avg:    7 Max:      21
T: 2 ( 3982) P:80 I:11000 C: 146259 Min:      4 Act:   10 Avg:    7 Max:      27
T: 3 ( 3983) P:80 I:11500 C: 139900 Min:      5 Act:    8 Avg:    7 Max:      21
slave@debian-slave:~/rt-tests$ uname -a
Linux debian-slave 3.2.0-4-rt-amd64 #1 SMP PREEMPT RT Debian 3.2.73-2+deb7u2 x86_64 GNU/Linux
```

Although this test was not written for xenomai, I ran it on  identical hardware with xenomai kernel:
```
cml@debian-cml2:~/rt-tests$ sudo  ./cyclictest -t -p 80 -n -i 10000
# /dev/cpu_dma_latency set to 0us
policy: fifo: loadavg: 0.25 0.06 0.06 1/223 7077
T: 0 ( 7069) P:80 I:10000 C:  36773 Min:      5 Act:   11 Avg:    9 Max:      27
T: 1 ( 7070) P:80 I:10500 C:  35022 Min:      6 Act:   10 Avg:   11 Max:      26
T: 2 ( 7071) P:80 I:11000 C:  33430 Min:      7 Act:    9 Avg:    9 Max:      27
T: 3 ( 7072) P:80 I:11500 C:  31976 Min:      6 Act:   12 Avg:    9 Max:     135
cml@debian-cml2:~/rt-tests$ uname -a
Linux debian-cml2 3.8-1-xenomai.x86-amd64 #1 SMP Debian 3.8.13-9 x86_64 GNU/Linux
```

mhaberler, 2016-03-05:
Gemi  @kinsamanka built an RT-PREEMPT kernel for the raspberry2 today, it's already in the deb.machinekit.io apt repo:

`````
mah@raspberrypi:~/rt-tests $ apt-cache search 4.1.18-rt17-v7+
linux-firmware-image-rt-4.1.18-rt17-v7+ - Linux kernel firmware, version 4.1.18-rt17-v7+
linux-headers-rt-4.1.18-rt17-v7+ - Linux kernel headers for 4.1.18-rt17-v7+ on armhf
linux-image-rt-4.1.18-rt17-v7+ - Linux kernel, version 4.1.18-rt17-v7+
`````

pretty good performer:
````
mah@raspberrypi:~/rt-tests $ sudo  cyclictest -t1 -p 80 -n -i 10000 -l 10000
# /dev/cpu_dma_latency set to 0us
policy: fifo: loadavg: 0.33 0.25 0.15 1/179 1465

T: 0 ( 1462) P:80 I:10000 C:  10000 Min:     11 Act:   15 Avg:   14 Max:      42
````

congratulations!

kinsamanka, 2016-03-05:
That kernel is not yet ready, there's still some issues when all cores are
used (try the --smp flag on cyclictest) :-/

I think it's missing some rt patches or kernel config.

mhaberler, 2016-03-15:
updated rt-preempt kernel for jessie in deb.machinekit.io to  4.1.19-rt22mah for i386 and amd64:

i386, j1900 motherboard:

````
mah@j1900:~/rt-tests$ sudo  cyclictest -smp -p 80 -n -i 10000 -l 10000
defaulting realtime priority to 2
# /dev/cpu_dma_latency set to 0us
policy: fifo: loadavg: 0.23 0.24 0.11 1/220 1636

T: 0 ( 1636) P: 2 I:10000 C:  10000 Min:      9 Act:   18 Avg:   15 Max:      34
````

amd64 Brazos N450 motherboard:
````
root@amd64:~# sudo  cyclictest -smp -p 80 -n -i 10000 -l 10000
defaulting realtime priority to 2
# /dev/cpu_dma_latency set to 0us
policy: fifo: loadavg: 0.31 0.31 0.19 1/358 2236

T: 0 ( 2234) P: 2 I:10000 C:  10000 Min:      9 Act:   30 Avg:   21 Max:     129
````

mhaberler, 2016-03-19:
@the-snowwhite: latest mksocfpga test img with 4.4.4 rt-preempt kernel:

machinekit@mksocfpga:~/rt-tests$ sudo  ./cyclictest -smp -p 80 -n -i 10000 -l 10000
defaulting realtime priority to 2
# /dev/cpu_dma_latency set to 0us
policy: fifo: loadavg: 0.83 1.17 0.59 1/81 4641

T: 0 ( 4639) P: 2 I:10000 C:  10000 Min:     18 Act:   37 Avg:   28 Max:     211

the-snowwhite, 2016-03-19:
@mhaberler 

Seems like there is room for significant improvement compared to these other Cyclone V HPS soc test slides:

http://events.linuxfoundation.org/sites/events/files/slides/toyooka_LCE2014_v4_0.pdf

Official rocketboards current old 3.10 kernel results:

https://rocketboards.org/foswiki/view/Documentation/AlteraSoCLTSIRTKernel

claudiolorini, 2016-04-07:
...just jumped on top of a 4.4.6-rt13 on Zynq MYIR-Zturn and the results seem to be quite encouraging:

`machinekit@machinekit:~$` sudo  cyclictest -t1 -p 80 -n -i 10000 -l 10000
# /dev/cpu_dma_latency set to 0us
policy: fifo: loadavg: 0.89 0.33 0.13 1/106 1017          

T: 0 (  998) P:80 I:10000 C:  10000 Min:      0 Act:   18 Avg:   23 Max:      64
`

gmoccapy, 2016-04-07:
The problem is on this test, that it depends very strongly on the time you start the test after booting the PC.

On my "work machine" I started cyclictest after installing the kernel and got a value around 1200, then I went away, leaving the machine doing nothing, except waiting. After about half an hour a come back and started the cyclictest again from the same terminal and the value went up to about 7500.

So I started playing around with gmoccapy, chnaged some code, compiled some stuff etc.
After one Hour I closed all open programms, exept the "cycletest terminal". Now the values went up to 13000.

Reboot:

start cyclictest and I got again values around 1200.

So IMHO we need to set up a "virtual" usage of the PC / Device for certain time and then start the test. IMHO the values here are not comparable.

Please correct me if I am wrong!
(All values from memory, If needed, I can repeat the test and document in detail)

Norbert

claudiolorini, 2016-04-07:
I've tried a just a couple of times with short (10000) and longer (100000) duration and different CPU
loads obtaining 'reasonable' results around 60 max. Tomorrow i'll try the scenario you suggest to 
see what happens... maybe is something related to the architecture ARM vs. x86. 

The real problem is that i wasn't able to test with the machinekit 'latency-histogram' application, 
obtained just a couple of 'lines' then 100%100% CPUs and sistem stuck. I suspect i'm missing
some library or something similar... I think that i'll wait @mhaberler to have a functional system 
similar to mine and see if it is the same to him (i'm such a lazy boy... ;-)

dkhughes, 2016-04-07:
I think gmoccapy has a good point, but I'm not seeing the effects as heavily with the microzed with kernel 4.4.0 and rt patch. Runs after boot up and a long delay of idleness are giving about the same results, but this is with low background CPU load.

Results for smp:
```
administrator@mctrl:/$ sudo cyclictest --smp -p 80 -n -i 10000 -l 10000
# /dev/cpu_dma_latency set to 0us
WARN: Running on unknown kernel version...YMMV
policy: fifo: loadavg: 0.31 0.35 0.38 3/174 17905          

T: 0 (17883) P:80 I:10000 C:  10000 Min:     14 Act:   18 Avg:   20 Max:      48
T: 1 (17884) P:80 I:10500 C:   9524 Min:     16 Act:   23 Avg:   22 Max:      58
```
And a single thread run:
```
administrator@mctrl:/$ sudo cyclictest -t1 -p 80 -n -i 10000 -l 10000
# /dev/cpu_dma_latency set to 0us
WARN: Running on unknown kernel version...YMMV
policy: fifo: loadavg: 0.33 0.35 0.38 1/173 17944          

T: 0 (17924) P:80 I:10000 C:  10000 Min:     13 Act:   20 Avg:   19 Max:      53
```

claudiolorini, 2016-04-08:
...i've done some repeated tests, and i can confirm Norbert doubts about 
the variability of the cyclictest (Max) results, anyway Avg readings seem to give
an overall idea of what is happening:

machinekit@machinekit:~$ sudo cyclictest -t1 -p 80 -n -i 10000 -l 10000
T: 0 ( 1006) P:80 I:10000 C:  10000 Min:      0 Act:   18 Avg:   23 Max:      52
ven  8 apr 2016, 08.32.47, CEST
T: 0 ( 1038) P:80 I:10000 C:  10000 Min:      0 Act:   18 Avg:   23 Max:      66
ven  8 apr 2016, 08.44.08, CEST
T: 0 ( 1104) P:80 I:10000 C:  10000 Min:      0 Act:   18 Avg:   20 Max:      42
ven  8 apr 2016, 09.14.34, CEST
T: 0 ( 1142) P:80 I:10000 C:  10000 Min:      0 Act:   18 Avg:   23 Max:      73
ven  8 apr 2016, 09.41.15, CEST
T: 0 ( 1173) P:80 I:10000 C:  10000 Min:      0 Act:   36 Avg:   22 Max:      54
ven  8 apr 2016, 09.43.41, CEST
T: 0 ( 1210) P:80 I:10000 C:  10000 Min:      0 Act:   18 Avg:   20 Max:      47
ven  8 apr 2016, 09.49.21, CEST
T: 0 ( 1221) P:80 I:10000 C:  10000 Min:      0 Act:   18 Avg:   21 Max:      42
ven  8 apr 2016, 09.54.31, CEST

C.

claudiolorini, 2016-04-08:
just a couple of pictures, wiggling an IO with 4.4.6-RT

[image]

and a plain 4.4.6

[image]

the numbers shown by cyclictest seem to make sense.
C.

mhaberler, 2016-04-08:
I think latency-test predates cyclictest, and it worked on RTAI is well, so made sense back then

mhaberler, 2016-04-08:
heads up on [stap](https://en.wikipedia.org/wiki/SystemTap): I stumbled across this interesting tool on [HN](https://blog.cloudflare.com/revenge-listening-sockets/?p=0), was not aware of this

It allows ad-hoc probes and histograms of kernel functions
see [debian instructions](https://sourceware.org/systemtap/wiki/SystemtapOnDebian) - needs a package and the `-dbg` version of the kernel image

to those building kernels (@cdsteinkuehler @claudiolorini @kinsamanka @zultron @the-snowwhite @RobertCNelson) - it might make sense to add these config options to our kernels in the future: https://sourceware.org/systemtap/wiki/SystemTapWithSelfBuiltKernel

talking of which: anyone aware of a Travis/Dockerfile combo for cross-building an ARM kernel?

RobertCNelson, 2016-04-08:
Normally, CONFIG_DEBUG_INFO made things just too massive to ship, but there's a new option: CONFIG_DEBUG_INFO_SPLIT which keeps the vmlinuz/*.ko smaller..

I'll enable this on 4.6.0-rc3 and see what happens for a release..
```
-# CONFIG_DEBUG_INFO is not set
+CONFIG_DEBUG_INFO=y
+# CONFIG_DEBUG_INFO_REDUCED is not set
+CONFIG_DEBUG_INFO_SPLIT=y
+CONFIG_DEBUG_INFO_DWARF4=y
+# CONFIG_GDB_SCRIPTS is not set
```

Regards,

RobertCNelson, 2016-04-08:
CONFIG_DEBUG_INFO_SPLIT makes things nice..
```
voodoo@hades:/opt/github/linux-dev/deploy$ du -sh ./*
219M	./dbg
172M	./org
voodoo@hades:/opt/github/linux-dev/deploy$ du -sh ./*/*
860K	./dbg/linux-firmware-image-4.6.0-rc2-armv7-devel-r35_1cross_armhf.deb
7.2M	./dbg/linux-headers-4.6.0-rc2-armv7-devel-r35_1cross_armhf.deb
29M	./dbg/linux-image-4.6.0-rc2-armv7-devel-r35_1cross_armhf.deb
47M	./dbg/linux-image-4.6.0-rc2-armv7-devel-r35-dbg_1cross_armhf.deb
800K	./dbg/linux-libc-dev_1cross_armhf.deb
4.0K	./dbg/linux-upstream_1cross_armhf.changes
4.0K	./dbg/linux-upstream_1cross.debian.tar.gz
4.0K	./dbg/linux-upstream_1cross.dsc
135M	./dbg/linux-upstream_4.6.0-rc2-armv7-devel-r35.orig.tar.gz
860K	./org/linux-firmware-image-4.6.0-rc2-armv7-devel-r35_1cross_armhf.deb
7.2M	./org/linux-headers-4.6.0-rc2-armv7-devel-r35_1cross_armhf.deb
29M	./org/linux-image-4.6.0-rc2-armv7-devel-r35_1cross_armhf.deb
800K	./org/linux-libc-dev_1cross_armhf.deb
4.0K	./org/linux-upstream_1cross_armhf.changes
4.0K	./org/linux-upstream_1cross.debian.tar.gz
4.0K	./org/linux-upstream_1cross.dsc
135M	./org/linux-upstream_4.6.0-rc2-armv7-devel-r35.orig.tar.gz
```

RobertCNelson, 2016-04-09:
@mhaberler 4.4.6-ti-rt-r16 in the apt repo has then enabled for you. ;)

ArcEye, 2016-07-17:
4.6.4-rt8 builds and runs fine 64bit on Jessie

Here is an extreme example of the caching effect on an Intel i7 quad core with 8 threads

Normal latency-test

[image]

latency-test with fast dummy base thread, 450% lower

[image]

mhaberler, 2016-07-17:
@RobertCNelson sorry - completely slept through this; thanks!  we need to see if we can use this -rt kernel and still not exceed the RT cycle budget, it is a tad close on the BB cpu

mhaberler, 2016-07-17:
@ArcEye it would be interesting to see what happens if you bind base and servo to different cores

I guess this is a case where the base thread prevents cache eviction of the servo thread somehow

ArcEye, 2016-07-18:
> @ArcEye it would be interesting to see what happens if you bind base and servo to different cores

What method do you recommend for binding threads as opposed to processes?

mhaberler, 2016-07-19:
use the `cpu=<number>` param, e.g.:
````
newthread servo 1000000 cpu=2
`````
halcmd currently does not display the CPU; linuxcnc.log does

ArcEye, 2016-07-19:
OK, I hacked latency-test to accept arguments $1 and $2, which were the cpu numbers for base and servo thread respectively.

From various permutations, it appears that only assigning both to the same CPU will get close to the result obtained allowing the default cpu affinity to operate.

I assume(?) from that, the default affinity makes no distinction between threads from the same process and puts them on the same CPU, hence the cache filling effect works.

mhaberler, 2016-07-19:
yes, I think that's in line with my guess - the base thread keeps the cache hot for the servo thread; if the base thread is on a different core, it keeps the 'wrong' cache hot and the servo thread's cache lines get evicted anyway

eventually Intel's [Cache Allocation Technology](http://www.intel.com/content/www/us/en/communications/cache-allocation-technology-white-paper.html) will have a similar effect, but that is Intel-only and only on very recent Xeon CPU's

it might be worth an [experiment](https://indico.cern.ch/event/497446/attachments/1229239/1801121/A_few_experiments_with_the_Cache_Allocation_Technology.pdf) 

not on my box here, though :-/
````
root@next:/home/mah# apt install intel-cmt-cat
root@next:/home/mah# pqos
NOTE:  Mixed use of MSR and kernel interfaces to manage
       CAT or CMT & MBM may lead to unexpected behavior.
WARN: Cache monitoring capability not supported!
WARN: Cache allocation not supported on model name '      Intel(R) Celeron(R) CPU  J1900  @ 1.99GHz'!
ERROR: No Platform QoS capability discovered
ERROR: discover_capabilities() error 1
Error initializing PQoS library!
````

mhaberler, 2016-07-19:
as for the 'tools in the bag' theme, I think we should give [perf](https://perf.wiki.kernel.org/index.php/Tutorial) a closer look - the list of pre-defined events looks interesting (cache-misses etc)

ArcEye, 2016-07-19:
>  not on my box here, though :-/

Nor on mine
`WARN: Cache allocation not supported on model name '       Intel(R) Core(TM) i7-3770S CPU @ 3.10GHz'!`

Interestingly, being able to limit both threads to just one CPU, gets  better results than before.
The` isolcpus 1,2,3,4,5,6,7`  parameter, with a fast `cpu-hog` program running on CPU 0  too effectively.

Might not be too good for any userspace programs trying to get a look in though!

[image]

This is probably because the default affinity was not fixed, as could be seen by some movement when looking at *top* with latency-test running.
So there was some overlap and hopping between caches.

However if different CPUs are set, the results are marginally even worse than just running a servo thread, presumably because they NEVER share the same cache and have increased overhead.

[image]

ArcEye, 2016-07-19:
Does not like 6th gen i7 either

`WARN: Cache allocation not supported on model name 'Intel(R) Core(TM) i7-6700HQ CPU @ 2.60GHz'!`

luminize, 2017-01-02:
I'm setting up a new j1900 PC, so I'm looking into performance. As of yet I got sorta good results when I use an i386 installation, with a 4.1.36-rt42 kernel. The 4.4.38-rt49 kernel I made has (looking at max latency) 50% poorer performance (just compiled the kernel, no tweaking).

To make things easy I've made 2 scripts so one can plot a nice histogram, as found on the OSADL website. info here: https://github.com/luminize/realtime-tools

to give 2 examples: the j1900 as well as a system with 2 quad core E5420 xeon processors

j1900
[image]

2 quad core E5420 xeon
[image]

nsicko42, 2017-10-19:
Raspberry Pi 3B PREEMPT RT Kernel

```
$ uname -a
Linux raspberrypi 4.9.56-rt37-v7+ #1 SMP PREEMPT RT Thu Oct 19 10:03:25 KST 2017 armv7l
```
```
$ sudo cyclictest -a -t -n -p99
# /dev/cpu_dma_latency set to 0us
policy: fifo: loadavg: 3.51 1.79 1.05 5/240 1556

T: 0 ( 1552) P:99 I:1000 C: 110063 Min:      7 Act:    9 Avg:    8 Max:      42
T: 1 ( 1553) P:99 I:1500 C:  73375 Min:      7 Act:   10 Avg:    8 Max:      30
T: 2 ( 1554) P:99 I:2000 C:  55032 Min:      7 Act:   10 Avg:    9 Max:      33
T: 3 ( 1555) P:99 I:2500 C:  44025 Min:      7 Act:   10 Avg:    8 Max:      28
```

luminize, 2018-01-04:
2 more results,

**Beaglebone X15**
```
debian@BeagleBoard-X15:~/rt-tests$ uname -a
Linux BeagleBoard-X15 4.9.69-ti-rt-r88 #1 SMP PREEMPT RT Tue Jan 2 19:19:06 UTC 2018 armv7l GNU/Linux
```

```
debian@BeagleBoard-X15:~/rt-tests$ sudo cyclictest -t1 -p 80 -n -i 10000 -l 10000
# /dev/cpu_dma_latency set to 0us
policy: fifo: loadavg: 0.15 0.19 0.15 1/327 592            

T: 0 (32179) P:80 I:10000 C:  10000 Min:     14 Act:   17 Avg:   19 Max:      52
```

**am437xIDK**
```
machinekit@am437xIDK:~/rt-tests$ uname -a
Linux arm 4.9.41-rt23-gc038d21a22 #1 PREEMPT RT Tue Sep 26 18:48:13 EDT 2017 armv7l GNU/Linux
```

```
machinekit@am437xIDK:~/rt-tests$ sudo cyclictest -t1 -p 80 -n -i 10000 -l 10000
# /dev/cpu_dma_latency set to 0us
policy: fifo: loadavg: 0.11 0.11 0.04 1/174 2687          

T: 0 ( 2686) P:80 I:10000 C:  10000 Min:      9 Act:   19 Avg:   19 Max:      37
```

cdsteinkuehler, 2018-01-04:
For the PREEMPT_RT kernels, this is a great reference with lots of
detail on what to tweak:

http://linuxrealtime.org/index.php/Improving_the_Real-Time_Properties

-- 
Charles Steinkuehler
charles@steinkuehler.net

closed 2018-08-04 by ArcEye