machinekit/machinekit #793: libwesockets: SSL library license

sirop, 2015-11-20:
I think there is  a good reason for libwesockets offering the option to use https://www.wolfssl.com/wolfSSL/Products-wolfssl.html instead of OpenSSL as
OpenSSL is not so light weight.

But there may be a license issue with *wolfSSL*  as you can find at https://www.wolfssl.com/wolfSSL/License.html in case of Commercial Licensing .

So my general questions are:  
 
- Is there any OpenSSL lightweight substitute that has a more liberal license than *wolfSSL* ?  

- Does Machinekit need any tailor-made encryption library?

mhaberler, 2015-11-21:
curious - could you outline what the intended setup looks like and how the licensing issue comes in?

the current situation is like so:
- the whole codebase is anywhere from GPL-only, LGPL, LGPL2 and some MIT & BSD parts thrown in
- I do not think there is a realistic chance of relicensing the whole stack to say MIT or BSD - that would need consent from over hundred contributors (some of them deceased) 
- the machinetalk middleware (zeromq, protobuf, mDNS) acts as a licensing firewall between UI's or other parts talking to the HAL/rt or CNC stacks (for instance clients using @bobvanderlinden and @strahlex 's remote API's could be done completely license-unencumbered as the network API is a legal boundary)
- the only link between remote UI's or other agents based on the above is the [protobuf definitions](https://github.com/machinekit/machinetalk-protobuf) which are MIT-licensed and will remain so no matter what
- as for webtalk (the websockets/machinetalk translator and minimal http server), this can run either on the rt host or remotely; either way, the websockets/JSON API it provides is a network interface which again is a license boundary; AFAICT this is the only place where openssl is used (could be Sergey's web ui does so as well, need to check)
- webtalk was a start, but it never caught on big time, and I envisage @bobvanderlinden 's work to  effectively replace it eventually; it might make sense to make the webtalk part optional to reduce build requirements

It helps to view this a bit analogous to the Linux kernel: the kernel code is GPL, but the system call API is a licensing boundary. You can run pretty much anything _ontop_ of the system call API and essentally ignore the licensing situation of the kernel code. Adding a change to the kernel, or say  a driver, is a different matter - this must fit the Linux kernel rules (GPL). In the machinekit case contributions need to fit in the above list of licensing options.

So I'm unsure what value add a openSSL replacement will add in value on the licensing issue

sirop, 2015-11-21:
Yes, your answer is quite detailed. Thanks.
>curious - could you outline what the intended setup looks like and how the licensing issue comes in?

http://git.libwebsockets.org/cgi-bin/cgit/libwebsockets/tree/README.build.md#n279
says in *wolfSSL/CyaSSL replacement for OpenSSL*:
>It contains a OpenSSL compatibility layer which makes it possible to pretty
much link to it instead of OpenSSL, giving a much smaller footprint.

So we looked at wolfSSL license:
>Businesses and enterprises who wish to incorporate wolfSSL products into proprietary appliances or other commercial software products for re-distribution must license commercial versions.

Finally we refused to use any encryption with libwesockets as 
- both OpenSSL and wolfSSL would mean  deceleration and a bigger footprint,
- our WiFi connection between server and client is anyway encrypted.  

So encryption or encryption licenses are not an urgent problem for us now,
but I was merely curious...

mhaberler, 2015-11-21:
@sirop - call me a bonehad, I still do not get it. What is the purpose of the exercise? get code with a smaller footprint? resolve some licensing question ?

Using or not using https is a matter of configuration, both can be even had in parallel

sirop, 2015-11-21:
@mhaberler - I beg your pardon if I confused you.  
Yes, we wanted a smaller footprint. That's why we looked at wolfSSL as wolfSSL praises itself to be suited for embedded application. But for commercial products they - wolfSSL - have a commercial license.
That is where my general licensing question arose from.  
You answered this question in detail.  

*As for smaller footprint: trying to send out websocket messages at about 1 kHz  we ran into the problem of memory pressure. But that's another issue...*

mhaberler, 2015-11-21:
so you are actually _using_ webtalk.. right? or Sergey's webui?

mhaberler, 2015-11-21:
_ websocket messages at about 1 kHz_ - what are you doing?

well no wonder. Please describe what you want to do, in *total*, I am not a CSI and I have no plans to play ping-pong for piecemeal bits and pieces

sirop, 2015-11-21:
That experiment with websocket messages  at 1 kHz originates from the time as we did not use yet Machinekit and therefore no webtalk or Sergey's UI, but solely an EtherCAT master stack with a libwebsocket server. We thought to offload some of calculation from the master machine to some other machine by using websockets.  

But as I now got acquainted with Machinekit, we'll sure find some other way to distribute the load of numeric calculations. Thus the memory pressure in case of libwebsockets is no longer an urgent problem for us.  
I  just wanted to mention that piece of experience with libwebsockets.

mhaberler, 2015-11-23:
Since you are asking about optimizing the memory footprint of software which is not even part of machinekit I am out of my depth to comment

what exactly is "offloading some calculations"? which bottleneck are you talking about?

I observe with that idea you are introducing a network stream plus websockets encoding in servo loop control values (or even feedback - I cannot tell since you do not tell), which will add significant delay and jitter along the commanded position/velocity path 

if it is _in the servo loop_ you stand a good chance of building an oscillator - delay in the loop is one way to build a system with a second order differential equation transfer characteristic

mhaberler, 2015-11-25:
closing for now, can be reopened as new information surfaces

closed 2015-11-25 by mhaberler