machinekit/machinekit #794: Run-time option to disable zeroconf announcement

zultron, 2015-11-23:
By default, Zeroconf announcement requires `avahi-daemon` at run-time, undesirable especially in containerized build environments without well-developed infrastructure to manage daemon life cycles.  Zeroconf announcement is optional, and can be disabled in `machinekit.ini`, but when enabled and `avahi-daemon` is not running, RTAPI refuses to initialize.

I have a [hacked patch][1] to disable Zeroconf announcement from `configure.ac` when an environment variable is set at build-time.  However, this is the wrong place to put this logic, since the resulting `.ini` file goes into packages with unintended defaults.

@mhaberler, could you please look at this issue and propose a better solution?  A simple setting tested at run-time to override the `machinekit.ini` setting, such as a command-line option or environment variable, would be much easier than code to munge the `.ini` file.  A solution mainlined in the next few days would smooth the transition to a Buildbot replacement.  (PRs welcome! ;)

[1]: https://github.com/zultron/machinekit/commit/e884169487435243969b270e0adb8bca0cbd8434

strahlex, 2015-11-23:
@zultron Can you please further explain why running `avahi-daemon` is undesirable in the docker container? Testing service announcements might be crucial for future integration tests.

I am not in favor of any hacks to pass configurations settings into Machinekit as long as https://github.com/machinekit/machinekit/issues/104 is not fixed. Any additional command line argument makes migration harder. The hacked patch to disable it at configure time should be good enough for the moment.

zultron, 2015-11-23:
@strahlex, it's just about the extra machinery to start and stop `avahi-daemon`.  Especially stopping the daemon may be less of an issue in Docker; my past pain was from having the daemon persist in a normal `chroot`.

Good point about regression tests.

Thanks for looking at this.  Let's put it aside for now until our new build containers are working.  You may well be right that this is a non-requirement.

mhaberler, 2015-11-23:
it looks to me a an environment variable overriding machinekit.ini[MACHINEKIT]REMOTE=0/1 does the trick - no big deal of a change. 

In case in machinekit.ini REMOTE=1 is set, and the override environment variable is set, the machinekit.ini file will be treated like it has REMOTE=0.

sounds about right?

zultron, 2015-11-23:
To be clear, you'll introduce an environment variable, say `$MACHINEKIT_REMOTE`, and if it's set to `0` or `1`, it will override any setting in `machinekit.ini`?

That sounds perfect.

mhaberler, 2015-11-23:
yes, tenatively MACHINETALK_LOCAL MACHINETALK_NO_MDNS or somesuch 

no, as I said: it JUST does change behavior as REMOTE=0 were specified in machinekit ini

if REMOTE=0 is in machinekit.ini, all the noise about IPv4/6 etc is ignored anyway, and Unix domain sockets are used instead - so no need to overspecify with  the whole yadayada

strahlex, 2015-11-23:
Another environment variable in the wind that nobody knows what it is used for...

The code becomes less and less maintainable the more sources we add. Thats why I want to use elektra, a simple get('machinekit/machinetalk/remote') and where done.

mhaberler, 2015-11-23:
@strahlex Absolutely. Then, what is the alternative to get to container-based builds quickly? 

Q: what is the reason to not run avahi-demon in a container? I do not understand that part quite yet. Any showstoppers here?

zultron, 2015-11-23:
@mhaberler Sounds like @strahlex has a valid aversion to this; let's pause on this issue for now, for the reasons stated [above](https://github.com/machinekit/machinekit/issues/794#issuecomment-159003093); no confirmed showstoppers here.

mhaberler, 2015-11-24:
I do not see a reason why avahi-daemon cannot run in a container, a virtual eth interface which groks multicast is present in a container:

`````
mah@mah2:~$ docker run -a stdin -a stdout -i -t  ubuntu:14.04 /bin/bash
root@a1828c666e8a:/# ifconfig
eth0      Link encap:Ethernet  HWaddr 02:42:ac:11:00:02
          inet addr:172.17.0.2  Bcast:0.0.0.0  Mask:255.255.0.0
          inet6 addr: fe80::42:acff:fe11:2/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1

`````

this image does start avahi-daemon: https://github.com/ianblenke/docker-avahi

other than what we need, this talks to the host dbus-daemon, which is not what we want (it might not sport a dbus-daemon afterall), so installing dbus-daemon in the container probably does the trick

closed 2015-12-08 by zultron