machinekit/machinekit #800: What's the reason for restricting multiple signals to one output pin?

luminize, 2015-11-27:
I'm trying to understand why an output pin can not be connected to multiple signals.
A signal can have only 1 source and multiple outputs. I would like to understand the reason for the restriction that output pins can only have signal attached.

```
halcmd: newinst lutn or-1 pincount=2 functn=0xe
halcmd: newinst lutn or-2 pincount=2 functn=0xe
halcmd: newinst lutn or-3 pincount=2 functn=0xe
halcmd: show pin
Component Pins:
  Comp   Inst Type  Dir         Value  Name                             Epsilon         Flags
    73     74 bit   IN          FALSE  or-1.in-00                    			0
    73     74 bit   IN          FALSE  or-1.in-01                    			0
    73     74 bit   OUT         FALSE  or-1.out                      			0
    73     74 s32   OUT             0  or-1.time                     			0
    73     82 bit   IN          FALSE  or-2.in-00                    			0
    73     82 bit   IN          FALSE  or-2.in-01                    			0
    73     82 bit   OUT         FALSE  or-2.out                      			0
    73     82 s32   OUT             0  or-2.time                     			0
    73     90 bit   IN          FALSE  or-3.in-00                    			0
    73     90 bit   IN          FALSE  or-3.in-01                    			0
    73     90 bit   OUT         FALSE  or-3.out                      			0
    73     90 s32   OUT             0  or-3.time                     			0

halcmd: net sig-or1-out-to-or2-in or-1.out or-2.in-00
halcmd: net sig-or1-out-to-or3-in or-1.out or-3.in-00
<stdin>:9: Pin 'or-1.out' was already linked to signal 'sig-or1-out-to-or2-in'
```

cdsteinkuehler, 2015-11-27:
On 11/27/2015 12:42 PM, Bas de Bruijn wrote:
> 
> ```
> halcmd: net sig-or1-out-to-or2-in or-1.out or-2.in-00
> halcmd: net sig-or1-out-to-or3-in or-1.out or-3.in-00
> <stdin>:9: Pin 'or-1.out' was already linked to signal 'sig-or1-out-to-or2-in'
> ```

I think the problem is you are trying to give the same signal multiple
names.  This causes problems in most traditional schematic packages
(where all nodes that are electrically connected are expected to have
the same net name), and it doesn't surprise me that this also causes
problems in HAL.

Rather than trying to name each node of your net, you need a
higher-level net name that is used for each node, like so:

```
halcmd: net meaningful-name or-1.out
halcmd: net meaningful-name or-2.in-00
halcmd: net meaningful-name or-3.in-00
```

...and yes, manually typing or otherwise crafting netlists is a
special kind of painful, which is why every few months someone
proposes some alternate way of crafting HAL files, but so far nothing
has really taken off.

-- 
Charles Steinkuehler
charles@steinkuehler.net

strahlex, 2015-11-27:
I think the problem is that you have to name the signal. Most PCB EDA
packages do that for you. The HAL Python bindings can do that too. Just
try to net 2 pins.

cdsteinkuehler writes:

> On 11/27/2015 12:42 PM, Bas de Bruijn wrote:
>
> I think the problem is you are trying to give the same signal multiple
> names.  This causes problems in most traditional schematic packages
> (where all [...]

-- 
Alexander

luminize, 2015-11-28:
@strahlex @cdsteinkuehler The thing I don't understand is why there needs to be a "higher-level" net name to get data from 1 output to multiple inputs.
if we were to do this on the breadboard it in the current way it would be like this:
```
component0.out──higher level net name──┬──component1.in1
                                       ├──component2.in1
                                       └──component3.in2
```
where if we could do it like this:
```
component0.out──net name 1──component1.in1
component0.out──net name 2──component2.in1
component0.out──net name 3──component3.in2
```

it would:
* mean that configurations are better readable and understandable, because now to see where a value comes from one has to first find the original signal where the "source" is used. 
* be easier to automate with block like editors, because once you know the source and the target you just add a wire between that.

some background for my question is this post on the ros-users list.
http://lists.ros.org/pipermail/ros-users/2015-November/069769.html
http://wiki.ros.org/rosimple
http://erlerobotics.com/docs/ROS/ROSimple/Intro.html

some further digging led me to here https://blockly-demo.appspot.com/static/demos/code/index.html
so you can generate python (and other languages for that matter) code from blocks.

mhaberler, 2015-11-28:
> 
> I'm trying to understand why an output pin can not be connected to multiple signals.
> A signal can have only 1 source and multiple outputs. I would like to understand the reason for the restriction that output pins can only have signal attached.

Signals are typed, named memory locations - or what is called a variable in programming languages.

two signals with different names have different memory locations.

Pins are pointers to a signal. They can point to some explicitly named signal, or they point to their default signal if not linked.

At any time, a pointer can only point to a single location. Hence a pin can be linked to a single signal only.

luminize, 2015-11-28:
@mhaberler thanks, but that raises another question for me:
[in the example above](https://github.com/machinekit/machinekit/issues/800#issuecomment-160276110) why not let a `component1.in1` as well as `component2.in1` both point directly to the value of 'component0.out`? why the need for the variable?

mhaberler, 2015-11-28:
that is conceivable but not a good idea because of the default pin value semantics

NB a pin can have a default value, which is stored in its default signal

at first link to an unlinked signal, the signal "inherits" the default value of that pin

at unlink time, the current value of the signal is copied to the pin's default signal and becomes the - possibly modified - default value; on next link to an unlinked signal, the game repeats

if you abuse the default signal as a net name (btw which default signal, the in or the out default signal?) you mess up the default value semantics

I do not fully understand what the problem actually is - do you take issue that for linking foo.out to bar.in you need a named signal to do so? is it the extra signal name which you think is unnecessary? Is this a  convenience or a functional argument?

luminize, 2015-11-28:
I guess my problem I have (maybe personal opinion) is that on the abstraction level
- a pin is like a piece of metal coming out of an IC
- when I connect it to a pin of another IC, I solder a wire between those 2 pins.
- If i want to connect a pin of third component to the first pin, i don't cut the wire in the middle, strip the ends and make a 3-way joint, instead I solder 1 wire directly between the pins of the IC.
- the value resides in the output pin
- the wire is just an extension of input pins

so what happens in my head when I want to connect these pins is that I think "these 2 pins need to be connected",  I find the 2 pins on the IC's, solder the wire, done. I miss that analogy in HAL. When I'm looking into a HAL setup I'm not interested in the signal (wire colour) but in its source.

If you want to take advantage of the flexible HAL and wiring up a machine configuration on-the-fly then it's hard to see which output is the source of the input. Like debugging a HAL configuration. I don't think i'm wrong saying that everybody agrees that this is a difficult thing to read.

Hence the question about why (answered: pointer) an output pin can't have multiple signals (wires) connected (solder) to it.
To my simple view of the world I thought it was the other way around: a signal (pointer) looking to a pin (variable), instead of a pin (pointer) updating (writing) a value into the signal (variable) where reading pins point to.
I see (after pondering later could become "I saw") a wire as a link between pins (pointer from target to source)

Need to ponder this a bit. To take the analogy of copper:
The way HAL works now is that the signal is the full trace, including all branches and vias on the PCB.

mhaberler, 2015-11-28:
yes, that is what a 'named net' or signal is about - it is not a single solder joint, it's the name for the whole trace. The name of the trace is independent of the pin names it is/is not soldered to.

luminize, 2015-11-28:
ok, thanks for explaining. question answered, not sure I like it, but you can't argue facts :)

closed 2015-11-28 by luminize

cdsteinkuehler, 2015-11-28:
On 11/28/2015 7:31 AM, Bas de Bruijn wrote:
> Hence the question about why (answered: pointer) an output pin can't have multiple signals (wires) connected (solder) to it.
> To my simple view of the world I thought it was the other way around: a [...]

There are two features in most schematic design packages to deal with
the problems you're describing, and I think it would be possible to do
them in HAL as well.

The first is automatic generation of net names.  In almost every
schematic, there are some small nets that typically connect just two
or there signals and are not considered important enough to have their
own name.  When created, these signals are assigned an automatically
generated name (like NET0001).

The second feature is when connecting anything to an existing net, the
name is automatically inherited, so there is no need to repeatedly
specify the net name.

Using your example, if you don't care about the net name, you would
then be able to do something like this:

halcmd: net or-1.out or-2.in-00
halcmd: net or-1.out or-3.in-00

...or if you want to name the net:

halcmd: net my-net-name or-1.out or-2.in-00
halcmd: net or-1.out or-3.in-00

....or maybe:
halcmd: net my-net-name or-1.out
halcmd: net or-1.out or-2.in-00
halcmd: net or-1.out or-3.in-00

I think some bits of this might even work already, and it doesn't seem
like it would be that hard to implement any missing pieces.  There
might need to be a new halcmd (connect/conn?) to avoid confusion with
the old-style net semantics, but the HAL API to add a signal/net
should be able to have a single interface.

-- 
Charles Steinkuehler
charles@steinkuehler.net