machinekit/machinekit #818: lincurve HAL component is broken

strahlex, 2015-12-05:
The lincurve HAL component seems to be broken:
```
2134   2189 float IN           20.1  lincurve.test2.in                0.000010        0
  2134   2189 float OUT      0.150625  lincurve.test2.out               0.000010        0
  2134   2189 float I/O      0.150625  lincurve.test2.out-io            0.000010        0
  2134   2189 s32   OUT          2417  lincurve.test2.time                              0
  2134   2189 float I/O            20  lincurve.test2.x-val-00          0.000010        0
  2134   2189 float I/O           100  lincurve.test2.x-val-01          0.000010        0
  2134   2189 float I/O           0.3  lincurve.test2.y-val-00          0.000010        0
  2134   2189 float I/O           0.8  lincurve.test2.y-val-01          0.000010        0
```
The expected output should be somewhere between 0.3 and 0.8.

Furthermore, these two lines are extremely dangerous:
https://github.com/machinekit/machinekit/blob/master/src/hal/i_components/lincurve.icomp#L56-L57
If the user does not configure the component correctly the pointer can go anywhere.

strahlex, 2015-12-05:
This looks like the half of the `lincurve.test2.y-val-00` is added to the correct proportion of change. The code however looks correct.

strahlex, 2015-12-05:
The generated code does not look weird either. Can someone please try to reproduce this problem.

luminize, 2015-12-17:
hi @strahlex is this a duplicate of #784 ?

luminize, 2015-12-30:
@strahlex since #784 was solved, is this still a problem?

ArcEye, 2016-04-20:
I think we can take it the PR #784 solved it as someone managed to use it recently in a 
`PRU software encoder enhancement`
( https://groups.google.com/forum/#!topicsearchin/machinekitPRU$20software$20encoder$20enhancements/machinekit/G1jSKfiE6jU )

closed 2016-04-20 by ArcEye