machinekit/machinekit #823: Compilation on ODROID-

kaklik, 2015-12-09:


mhaberler, 2015-12-09:
we do not see your configure args - please pastebin config.log

what RT kernel are you going to use on that platform?

kaklik, 2015-12-09:
Thanks for fast response.  Here is my config.log http://pastebin.com/7HEzJYXG
I am going to use the ODROID's factory preempt kernel:
Linux odroid 3.10.80-135 #1 SMP PREEMPT Sat Nov 28 13:10:52 BRST 2015 armv7l armv7l armv7l GNU/Linux
I do not need hard real-time in my application because i will control the machine trough SPI interface.

kinsamanka, 2015-12-09:
You're using Ubuntu?

Only debian wheezy is supported for armhf architectures at the moment.

kaklik, 2015-12-09:
Yes, I am using Ubuntu 14.04, but according to documentation at: http://www.machinekit.io/docs/building-from-source/
Use of Ubuntu should not be the main problem.  
I have tried to compile original LinuxCNC on the same machine without a problem.

mhaberler, 2015-12-09:
we need to have a chat with our website editor who went over the top on the issue - this blatant blurb has to go ;)

sorry for that. No - we will not support Ubuntu for now. 

We might have to do this eventually for ROS integration but it is too much of a chore to be warranted for the drive-by users.

So: please use wheezy.

bobvanderlinden, 2015-12-10:
Having support for building on Ubuntu is indirectly related to #336.

There is a docker container with Debian wheezy and all dependencies to build Machinekit here:
https://hub.docker.com/r/kinsamanka/mkdocker/
Tag name: `wheezy-64`

mhaberler, 2015-12-10:
I think the question was about running on ubuntu, not about building for wheezy on ubuntu?

RunningLight, 2015-12-10:
>we need to have a chat with our website editor who went over the top on the issue - this blatant blurb has to go ;)

>sorry for that. No - we will not support Ubuntu for now.

Mea culpa. I didn't know we had a website editor, anyone can post a PR to the website just as any other part of our github distro, but I did write the "over the top" wording. I was hoping to keep the door open for more developers to increase our reach. In my defense, I note I was hardly being definitive, using phrases like "likely will work", "little alteration", "nuances of working in distributions other than Debian....":

```
They likely will work with little alteration on recent Debian-based distributions 
such as Ubuntu 14.04LTS and Linux Mint 17 but please, let’s address the 
nuances of working in distributions other than Debian or releases other than 
Wheezy on supplemental pages so this page can remain focused on its primary 
goal of getting you to a working machinekit environment.
```
The sharp response to my proposal to rework the website was dispiriting, came at a time when I had a serious health scare, and to be frank, left me thinking "the website is dead, long live machinekit-docs".

I will post a PR to neuter the tone of "building from source" but I have no stomach for generating equivalent material on machinekit-docs when I don't have any feel for its organization or flow.

mhaberler, 2016-01-08:
@kaklik note by now machinekit builds from source just fine - see #689 

we do not have Ubuntu packages but the source builds work fine

kaklik, 2016-01-08:
@mhaberler  If I understand #689 correctly it is about compiling and running machinekit on desktop Ubuntu on Intel based computer. But I can not confirm that.  I have tried to compile  machinekit on my Ubuntu 15.10 laptop.  But without success 

kaklik@kaklik-ultrabook:~/programy/machinekit/src$ make
Reading 54/320 dependency files
Done reading dependencies

Reading 0/0 realtime dependency files
Done reading realtime dependencies
make: Entering directory '/home/kaklik/programy/machinekit/src'
Compiling machinetalk/webtalk/webtalk_wsproxy.cc
machinetalk/webtalk/webtalk_wsproxy.cc: In function ‘int serve_http(libwebsocket_context*, wtself_t*, libwebsocket*, void*, size_t)’:
machinetalk/webtalk/webtalk_wsproxy.cc:184:66: error: too many arguments to function ‘int libwebsockets_serve_http_file(libwebsocket_context*, libwebsocket*, const char*, const char*)’
     if (libwebsockets_serve_http_file(context, wsi, buf, mt, NULL))
                                                                  ^
In file included from machinetalk/webtalk/webtalk.hh:37:0,
                 from machinetalk/webtalk/webtalk_wsproxy.cc:13:
/usr/include/libwebsockets.h:861:1: note: declared here
 libwebsockets_serve_http_file(struct libwebsocket_context *context,
 ^
machinetalk/webtalk/webtalk_wsproxy.cc: In function ‘int callback_http(libwebsocket_context*, libwebsocket*, libwebsocket_callback_reasons, void*, void*, size_t)’:
machinetalk/webtalk/webtalk_wsproxy.cc:247:62: error: ‘WSI_TOKEN_HTTP_URI_ARGS’ was not declared in this scope
      int arglen = lws_hdr_copy(wsi, uriargs, sizeof uriargs, WSI_TOKEN_HTTP_URI_ARGS);
                                                              ^
machinetalk/webtalk/webtalk_wsproxy.cc:424:33: error: invalid use of incomplete type ‘struct callback_http(libwebsocket_context*, libwebsocket*, libwebsocket_callback_reasons, void*, void*, size_t)::libwebsocket_pollargs’
      short zevents = poll2zmq(pa->events);
                                 ^
machinetalk/webtalk/webtalk_wsproxy.cc:201:12: note: forward declaration of ‘struct callback_http(libwebsocket_context*, libwebsocket*, libwebsocket_callback_reasons, void*, void*, size_t)::libwebsocket_pollargs’
     struct libwebsocket_pollargs *pa = (struct libwebsocket_pollargs *)in;
            ^
machinetalk/webtalk/webtalk_wsproxy.cc:425:38: error: invalid use of incomplete type ‘struct callback_http(libwebsocket_context*, libwebsocket*, libwebsocket_callback_reasons, void*, void*, size_t)::libwebsocket_pollargs’
      zmq_pollitem_t additem = { 0, pa->fd, zevents };
                                      ^
machinetalk/webtalk/webtalk_wsproxy.cc:201:12: note: forward declaration of ‘struct callback_http(libwebsocket_context*, libwebsocket*, libwebsocket_callback_reasons, void*, void*, size_t)::libwebsocket_pollargs’
     struct libwebsocket_pollargs *pa = (struct libwebsocket_pollargs *)in;
            ^
machinetalk/webtalk/webtalk_wsproxy.cc:436:38: error: invalid use of incomplete type ‘struct callback_http(libwebsocket_context*, libwebsocket*, libwebsocket_callback_reasons, void*, void*, size_t)::libwebsocket_pollargs’
      zmq_pollitem_t delitem = { 0, pa->fd, 0 };
                                      ^
machinetalk/webtalk/webtalk_wsproxy.cc:201:12: note: forward declaration of ‘struct callback_http(libwebsocket_context*, libwebsocket*, libwebsocket_callback_reasons, void*, void*, size_t)::libwebsocket_pollargs’
     struct libwebsocket_pollargs *pa = (struct libwebsocket_pollargs *)in;
            ^
machinetalk/webtalk/webtalk_wsproxy.cc:441:10: error: ‘LWS_CALLBACK_CHANGE_MODE_POLL_FD’ was not declared in this scope
     case LWS_CALLBACK_CHANGE_MODE_POLL_FD:
          ^
machinetalk/webtalk/webtalk_wsproxy.cc:443:12: error: invalid use of incomplete type ‘struct callback_http(libwebsocket_context*, libwebsocket*, libwebsocket_callback_reasons, void*, void*, size_t)::libwebsocket_pollargs’
      if (pa->prev_events == pa->events) // nothing to do
            ^
machinetalk/webtalk/webtalk_wsproxy.cc:201:12: note: forward declaration of ‘struct callback_http(libwebsocket_context*, libwebsocket*, libwebsocket_callback_reasons, void*, void*, size_t)::libwebsocket_pollargs’
     struct libwebsocket_pollargs *pa = (struct libwebsocket_pollargs *)in;
            ^
machinetalk/webtalk/webtalk_wsproxy.cc:443:31: error: invalid use of incomplete type ‘struct callback_http(libwebsocket_context*, libwebsocket*, libwebsocket_callback_reasons, void*, void*, size_t)::libwebsocket_pollargs’
      if (pa->prev_events == pa->events) // nothing to do
                               ^
machinetalk/webtalk/webtalk_wsproxy.cc:201:12: note: forward declaration of ‘struct callback_http(libwebsocket_context*, libwebsocket*, libwebsocket_callback_reasons, void*, void*, size_t)::libwebsocket_pollargs’
     struct libwebsocket_pollargs *pa = (struct libwebsocket_pollargs *)in;
            ^
machinetalk/webtalk/webtalk_wsproxy.cc:447:35: error: invalid use of incomplete type ‘struct callback_http(libwebsocket_context*, libwebsocket*, libwebsocket_callback_reasons, void*, void*, size_t)::libwebsocket_pollargs’
      zmq_pollitem_t item = { 0, pa->fd, 0 };
                                   ^
machinetalk/webtalk/webtalk_wsproxy.cc:201:12: note: forward declaration of ‘struct callback_http(libwebsocket_context*, libwebsocket*, libwebsocket_callback_reasons, void*, void*, size_t)::libwebsocket_pollargs’
[... cut ...]
machinetalk/webtalk/webtalk_wsproxy.cc:201:12: note: forward declaration of ‘struct callback_http(libwebsocket_context*, libwebsocket*, libwebsocket_callback_reasons, void*, void*, size_t)::libwebsocket_pollargs’
     struct libwebsocket_pollargs *pa = (struct libwebsocket_pollargs *)in;
            ^
Makefile:397: recipe for target 'objects/machinetalk/webtalk/webtalk_wsproxy.o' failed
make: *** [objects/machinetalk/webtalk/webtalk_wsproxy.o] Error 1
make: Leaving directory '/home/kaklik/programy/machinekit/src'
kaklik@kaklik-ultrabook:~/programy/machinekit/src$ 
 
Here is my new config.log http://pastebin.com/kuTk2CGS

mhaberler, 2016-01-08:
I have tested only with Ubuntu 14, because that is what we need for ROS integration since it is a LTS release. There is no way we can support every kink and turn Ubuntu takes. 

Maybe just use Ubuntu trusty?

That said: evidently you have the wrong czmq/zeromq libraries. See the versions which we provide for building, and get those. Maybe it works.

closed 2018-08-04 by ArcEye