machinekit/machinekit #824: Building from source fails; configure.ac errors

luminize, 2015-12-09:
Hi
I try to build the latest of MK on a VirtualBox VM, Jessie.
I've done this before with success and now it fails.

short:
followed these instructions to the letter:
https://github.com/luminize/machinekit-docs/blob/5dcf4865142a5eda5b953765d9e7ab41e11e957d/machinekit-documentation/getting-started/APT-packages-jessie.asciidoc

then:
```
sudo apt-get install libjansson-dev \
  libwebsockets-dev libxenomai-dev python-pyftpdlib
```
and from machinekit.io/docs/building-from-source/
```
sudo apt-get install git dpkg-dev
sudo apt-get install --no-install-recommends devscripts equivs
git clone https://github.com/machinekit/machinekit.git
cd machinekit
debian/configure -px
sudo mk-build-deps -ir
cd src
./autogen.sh
```
the build fails with:
```
machinekit@vb-jessie:~/machinekit/src$ ./autogen.sh
configure.ac:30: error: possibly undefined macro: AC_MSG_ERROR
      If this token and others are legitimate, please use m4_pattern_allow.
      See the Autoconf documentation.
configure.ac:92: error: possibly undefined macro: AC_DEFINE
configure.ac:95: error: possibly undefined macro: AC_SUBST
autoreconf: /usr/bin/autoconf failed with exit status: 1
```
git log:
```
machinekit@vb-jessie:~/machinekit/src$ git log
commit b896bcf5ed3f858b3285c81abd060d957cd32d60
Merge: 93c5463 3d3466b
Author: John Morris <john@zultron.com>
Date:   Tue Dec 8 01:55:33 2015 -0600

    Merge pull request #821 from kinsamanka/travis-ssh
    
    Create temporary files out of the build tree
```

full output of the entire endeavor:
http://pastebin.com/3r0pyUfp

am I doing something wrong?

luminize, 2015-12-09:
I cloned an older working VM and pulled the latest master and the build succeeds. So I guess this is not an issue but setup error.

closed 2015-12-09 by luminize

ArcEye, 2015-12-10:
I have had something like that before.  Think it was down to automake / autoconf versions

luminize, 2015-12-10:
so, I checked the versions of automake / autoconf and they are the same.

what's strange is that the messages below about `Attention, the package has been created in the current directory,
not in ".." as indicated by the message above!`

```
machinekit@vb-jessie:~/machinekit$ sudo mk-build-deps -ir
dh_testdir
dh_testroot
dh_prep
dh_testdir
dh_testroot
dh_install
dh_installdocs
dh_installchangelogs
dh_compress
dh_fixperms
dh_installdeb
dh_gencontrol
dh_md5sums
dh_builddeb
dpkg-deb: building package `machinekit-build-deps' in `../machinekit-build-deps_0.1.1-1_all.deb'.

The package has been created.
Attention, the package has been created in the current directory,
not in ".." as indicated by the message above!
Selecting previously unselected package machinekit-build-deps.
(Reading database ... 112273 files and directories currently installed.)
Preparing to unpack machinekit-build-deps_0.1.1-1_all.deb ...
Unpacking machinekit-build-deps (0.1.1-1) ...
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Correcting dependencies... Done
The following packages will be REMOVED:
  machinekit-build-deps
0 upgraded, 0 newly installed, 1 to remove and 0 not upgraded.
1 not fully installed or removed.
After this operation, 27.6 kB disk space will be freed.
Do you want to continue? [Y/n] n
Abort.
(Reading database ... 112276 files and directories currently installed.)
Removing machinekit-build-deps (0.1.1-1) ...
mk-build-deps: apt-get install call failed
```
so when I try to install the generated deb manually i get a lot of missing dependencies. And on the other VM this just works.
Will drill down into what I did different.
```
machinekit@vb-jessie:~/machinekit$ ls *.deb
machinekit-build-deps_0.1.1-1_all.deb
machinekit@vb-jessie:~/machinekit$ sudo dpkg -i machinekit-build-deps_0.1.1-1_all.deb 
Selecting previously unselected package machinekit-build-deps.
(Reading database ... 112273 files and directories currently installed.)
Preparing to unpack machinekit-build-deps_0.1.1-1_all.deb ...
Unpacking machinekit-build-deps (0.1.1-1) ...
dpkg: dependency problems prevent configuration of machinekit-build-deps:
 machinekit-build-deps depends on libboost-python-dev; however:
  Package libboost-python-dev is not installed.
 machinekit-build-deps depends on libgl1-mesa-dev; however:
  Package libgl1-mesa-dev is not installed.
 machinekit-build-deps depends on libglu1-mesa-dev; however:
  Package libglu1-mesa-dev is not installed.
 machinekit-build-deps depends on libgtk2.0-dev; however:
  Package libgtk2.0-dev is not installed.
 machinekit-build-deps depends on libmodbus-dev (>= 3.0); however:
  Package libmodbus-dev is not installed.
 machinekit-build-deps depends on libncurses-dev; however:
  Package libncurses-dev is not installed.
 machinekit-build-deps depends on libreadline-dev; however:
  Package libreadline-dev is not installed.
 machinekit-build-deps depends on libusb-1.0-0-dev; however:
  Package libusb-1.0-0-dev is not installed.
 machinekit-build-deps depends on libxmu-dev; however:
  Package libxmu-dev is not installed.
 machinekit-build-
dpkg: error processing package machinekit-build-deps (--install):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 machinekit-build-deps
```