machinekit/machinekit #829: rtapi_sem_take() :  rtapi.h commentary typo?

sirop, 2015-12-15:
>   2) It calls a blocking function, like **rtapi_sem_take()**.  It will
>        resume when the function unblocks.

https://github.com/machinekit/machinekit/blob/9e5dedd360fddd31cb8963a714f7297519c6bd4e/src/rtapi/rtapi.h#L687-L688  

There is no function `rtapi_sem_take()` in machinekit repo:
https://github.com/machinekit/machinekit/search?utf8=✓&q=rtapi_sem_take

ArcEye, 2015-12-15:
That is a legacy comment from linuxcnc.

If you look in the linuxcnc code base, it exists at rtai_rtapi.c : 1255

```
int rtapi_sem_take(int sem_id)
{   
    sem_data *sem;

    /* validate sem ID */
    if ((sem_id < 1) || (sem_id > RTAPI_MAX_SEMS)) {
	return -EINVAL;
    }
    /* point to the semaphores's data */
    sem = &(sem_array[sem_id]);
    /* is the semaphore valid? */
    if (sem->users == 0) {
	return -EINVAL;
    }
    /* get the semaphore */
    rt_sem_wait(&(ossem_array[sem_id]));
    return 0;
}
```

sirop, 2015-12-15:
Yes, thought something like that.

May I ask you a question about this comment on rtapi mutexes?
> They do _not_ replace semaphores, and can result in significant slowdowns if contention
> is severe. However, unlike semaphores they can be used from both
> user and kernel space.  

rtapi.h L376-L379

Is it also  linuxcnc legacy?
As at least Xenomai semaphores can be set in userland.

ArcEye, 2015-12-15:
The comments probably do date from when all kernels were rtai.  
What is userland has blurred considerably since then, now that xenomai and rt-preempt kernels are running .so not .ko modules

If you grep the code you will see very few instances of rtapi_mutex_try() being used (4 files), right in init stages mostly.

However in rtapi_common.c and rtapi_msgd.c for instance, it is used with no check on the return value, so the call might as well have been rtapi_mutex_get() anyway.

Only rtapi_support.c and ringbench.c use it 'properly'

The mutex_get / mutex_give mechanism is used extensively in the HAL pin code in the get / set values ops etc

For a much better explanation - @mhaberler   It's for you;)

sirop, 2015-12-18:
Do not know if I am hijacking my own issue or going astray.
Just want to share some experience with RTAPI mutexes. 
Compared cycle times of the update function
in case of
https://github.com/sirop/EtherCAT_HAL/blob/master/pdortapi2dc.comp ( RTAPI mutexes)
and
https://github.com/sirop/EtherCAT_HAL/blob/master/pdort2dc.comp (Xenomai Native Semaphores).

With RTAPI mutexes I get cycle times of about 23-25 us,
with Xenomai semaphores it is about 43-45 us.

sirop, 2016-01-18:
All cleared. LinuxCNC legacy is anyway a bigger subject.
Issue closed.

closed 2016-01-18 by sirop