machinekit/machinekit #834: Compile error on wheezy amd64 using gcc version 4.7.2 (Debian 4.7.2-5)

kinsamanka, 2015-12-17:
```

Compiling realtime hal/drivers/mesa-hostmot2/pwmgen.c
In file included from hal/drivers/mesa-hostmot2/abs_encoder.c:13:0:
./rtapi/rtapi_compat.h: In function 'kernel_threads':
./rtapi/rtapi_compat.h:127:5: error: implicit declaration of function 'RTAPI_BIT' [-Werror=implicit-function-declaration]
Compiling realtime hal/drivers/mesa-hostmot2/tp_pwmgen.c
cc1: some warnings being treated as errors
make[2]: *** [objects/xenomai/hal/drivers/mesa-hostmot2/abs_encoder.o] Error 1
```

kinsamanka, 2015-12-17:
There's no error on both armhf and i386 builds.

ArcEye, 2015-12-17:
Does it happen with 4.6?

Even Debian did not use 4.7, they used 4.6 to build all their Wheezy kernels.
After all sorts of bugs when 4.7 first came out, I used 4.6 all the time I was on Wheezy

kinsamanka, 2015-12-17:
4.6 is what the current travis-ci uses.

4.7 is in preparation for Michael's future work ;-)

ArcEye, 2015-12-17:
> 4.7 is in preparation for Michael's future work ;-)

If this is the ROS stuff, I can't help feeling that having anything to do with Ubuntu is a retrograde step.

But one for them :zap:

mhaberler, 2015-12-17:
@ArcEye nope, it is about the atomics support which changed from 4.6 to 4.7, and the C11 atomics are assumed for the multicore HAL branch

luminize, 2015-12-17:
Just got a compile error too when building the latest on my Wheezy VM.
```
checking whether the C compiler works... no
configure: error: in `/home/bas/machinekit/src':
configure: error: C compiler cannot create executables
See `config.log' for more details
```
config.log (which it refers to)
http://pastebin.com/RYq1yUS1

ArcEye, 2015-12-17:
I see, I only ever used 4.9 on Jessie for building that AFAIR.

I would suspect that trying to pull 4.9, for complete C11 compliance onto Wheezy, is likely to break libstdc++ and libc6 versions and everything built against them. :(

kinsamanka, 2015-12-17:
4.7 is available from wheezy-backports no need to import from jessie.

If you're using docker,  use kinsamanka/mkdocker:wheezy-64

ArcEye, 2015-12-17:
>  4.7 is available from wheezy-backports no need to import from jessie.

gcc-4.7 is available from the main Wheezy repo, gcc-4.9 is not available at all for Wheezy AFAIK
https://packages.debian.org/search?keywords=gcc-4.9

kinsamanka, 2015-12-17:
Steps to reproduce the error using docker containers:
```
docker pull kinsamanka/mkdocker:wheezy-64
docker run -it --rm=true kinsamanka/mkdocker:wheezy-64
proot-helper bash
cd /usr/src
git clone --depth=1 git://github.com/machinekit/machinekit
cd machinekit/src
./autogen.sh
./configure --with-posix --with-rt-preempt --with-xenomai \
     --without-xenomai-kernel \
     --without-rtai-kernel
make -j4
```

kinsamanka, 2015-12-17:
I've checked the (travis-ci logs)[https://s3.amazonaws.com/archive.travis-ci.org/jobs/97402328/log.txt] and this is just a warning:
```
Compiling realtime hal/drivers/mesa-hostmot2/abs_encoder.c
In file included from hal/drivers/mesa-hostmot2/abs_encoder.c:13:0:
./rtapi/rtapi_compat.h: In function 'kernel_threads':
./rtapi/rtapi_compat.h:127:5: warning: implicit declaration of function 'RTAPI_BIT' [-Wimplicit-function-declaration]
```

kinsamanka, 2015-12-17:
A new one after the the fix:
```
Compiling realtime hal/drivers/mesa-hostmot2/dpll.c
hal/drivers/mesa-hostmot2/pins.c: In function 'hm2_get_pin_secondary_name':
hal/drivers/mesa-hostmot2/pins.c:245:21: error: implicit declaration of function 'sprintf' [-Werror=implicit-function-declaration]
hal/drivers/mesa-hostmot2/pins.c:245:21: warning: incompatible implicit declaration of built-in function 'sprintf' [enabled by default]
hal/drivers/mesa-hostmot2/pins.c:251:14: warning: incompatible implicit declaration of built-in function 'sprintf' [enabled by default]
cc1: some warnings being treated as errors
```

mhaberler, 2015-12-18:
@kinsamanka - just tried your docker run on current master

this confuses me: http://pastebin.com/6ybwJg5A - OSError: [Errno 38] Function not implemented ?

kinsamanka, 2015-12-18:
Docker hub is having issues again with the autobuilder. :-/

Use "kinsamanka/machinekit_builder"  instead of "kinsamanka/mkdocker".

This is an issue with the makefile as posix and rt-preempt generates a
warning instead of an error.

mhaberler, 2015-12-18:
reproduced

mhaberler, 2015-12-18:
curious - do we have a gcc/gcc setting now where warnings are treated as errors?

mhaberler, 2015-12-18:
see #837

closed 2015-12-18 by kinsamanka