machinekit/machinekit #840: website out of date post build/CI infrastructure change

mhaberler, 2015-12-18:
#791  is now closed. Unfortunately, the associated work is not. The website needs severe work:

The reference to the buildbot is completely obsolete because there is no such animal anymore.

My suggestion: 
- Replace "Buildbot" by "The build process" outlining how the current pieces fit together (travis, packagecloud, docker containers, cross building - maybe with a diagram showing the flow)
- add a section which outlines how one can set up package builds on her own - after all that was a key goal of the effort.

The reference to RTAI on the home page has to go. 

Debian packages: 

- there are lots of kernels "provided". Which ones of those actually are still present, work and can be installed? are they moved to the new apt repo? if not, how will these be hosted?
- if there are other sources of RT kernels (debian, some other APT-repo) list them here.
- for RT-PREEMPT it might make sense to add a recipe page because it is becoming so simple (including a kernel config)
- delete RTAI here for good.

Arm6 - Rpi1: I am _very_ wary on conveying "is supported" for an architecture which has no 64bit atomics. And then, probably nobody buys an Rpi1 anymore. Can we narrow this down to "ARM7 and onwards"?

"Building From Source" needs work - too verbose and fluffy stuff like "The details we didn’t talk about" referring to material which is forthcoming since years but never comes - this has to go; let's pare this to the minimum and take out the funny prose which is just TL;DR 

My pet peeve - while we are at it: Finding the "How to report an error" page is a serious click safari - this should be under the righthandside menu under right under "Getting started"

I am happy to engage in rewriting and drawing a diagram, but not alone.

Who chips in with what? show of hands please!

mhaberler, 2015-12-18:
http://www.machinekit.io/docs/packages-debian/ 

 I think the header is a misnomer - it is about "installing and running of pre-built packages"

"The RT-PREEMPT realtime kernel comes from the stock wheezy package stream. " - really? inconsistent with what follows below. What do we recommend here?

The section "Install development environment packages" duplicates (badly) the information on http://www.machinekit.io/docs/building-from-source/ - cut here and focus on latter since has  nothing to do with running packages

luminize, 2015-12-18:
I've given my opinion in the past.
Instructions should be in machinekit-docs. In the github repo.
With on the website references to there

mhaberler, 2015-12-18:
fine, but that does not suggest where to make the cut given the issues at hand?

what will be the website then - just a trampoline with some links?

RunningLight, 2015-12-18:
On this subject, sometime last summer I found myself visualizing a favorite Far Side cartoon:  https://s-media-cache-ak0.pinimg.com/736x/67/66/1b/67661bbc78344ddcbd321a835c81ddb9.jpg

I still lack a clear vision of how machinekit-docs can be effectively organized for those readers not already in the know; i.e., those readers most likely to turn to them for help. Hate the use of Google Pages for machinekit.io if you like, but they gave us control over the look and feel with a professional looking result. [please note that Google Pages are processed and displayed through a different toolchain from what creates what you see in the GitHub Web Browser, as I have yammered about in the past, but they are also just another GitHub repo: machinekit/machinekit-github.io]

As I see it, the Google-Pages-based website will indeed become a trampoline, much like the original Yahoo was.

I have no problem working on documentation, remember the reworking I proposed before. However, since I can't contribute to the description of the current state of affairs w.r.t. the build process, since I am by nature verbose and fluffy in my writing, and since many of the issues noted here crept is as various contributors added bits and pieces wherever they thought best, I think I'll wait until you guys agree to a new organizational split before I take a crack at any of this.

mhaberler, 2015-12-18:
any PR is better than no PR.

sirop, 2015-12-18:
BTW, could there be something like https://en.wikipedia.org/wiki/LXR_Cross_Referencer
or http://woboq.com/codebrowser.html  on that new machinekit page?
For a more comfortable way of browsing the code and of great help for a newbie?

mhaberler, 2015-12-18:
there can be something - provided a) somebody stands up to do it b) gives a maintenance promise for a low-digit number of years c) shows on which servers this will run during this period (a,b and c not being me)

the task currently at hand is: get the most blatant errors/misinformation out of the website, which requires people to review and improve the web content and post PR's.

kinsamanka, 2015-12-18:
let me try out if RPi1 still works or not after the holidays

bobvanderlinden, 2015-12-19:
Is there a way to do PRs for the website? If not, there is an additional issue. I agree with @luminize in the sense that there should be one source of documentation. Whatever that source should be, it should be editable by any of the maintainers (and preferably forkable). If machinekit.io is only maintainable by a few people, then something needs to change. Move things to machinekit-docs on Github? Generate website pages from machinekit-docs?

If we want to take a similar approach to what we used for the build system of Machinekit (more decentralization and offloading hosting-work), I'd suggest looking at services like https://readthedocs.org/ or https://www.gitbook.com/, which build documentation and hosts the result as a website. readthedocs provides links back to Github to edit each of the generated pages. An example of such hosted documentation can be found here: http://vagrant-django.readthedocs.org/en/latest/, which is generated from https://github.com/oogles/vagrant-django/tree/develop/docs. It provides a way for anyone to work on documentation, while also removing clutter from Github around the content for reading.

The discussion is somewhat derailing from the original issue. Is this something that needs a different issue?

ArcEye, 2015-12-19:
> Is there a way to do PRs for the website? 

https://github.com/machinekit/machinekit.github.io

mhaberler, 2015-12-19:
@bobvanderlinden - see also http://www.machinekit.io/docs/website/ 
I installed jekyll locally, clone the website repo, then
````
BigMac:machinekit.github.io mah$ jekyll serve --watch
....
    Server address: http://127.0.0.1:4000/    # <-- local preview URI
`````
edit until happy with preview, commit, send PR

bobvanderlinden, 2015-12-19:
Oh sorry, I wasn't aware that was machinekit.io, but it makes sense. Still
though, is that where we are documenting all of this? Shouldn't it be a
landing page that directs users to machinekit-docs where needed? The topic
(how the build process works) isn't exactly for people who are starting out
with machinekit.

luminize, 2015-12-19:
@bobvanderlinden I just has a Skype chat with @mhaberler and we will write a list post re documentation. It's safe to say the current situation is not 2016 proof and we propose to do an experiment. Included in that experiment is writing the ci documentation. Bottom line communal effort and not a 1 person sacrifice. Writing on passenger seat in car so more detail/proposition later on the day

luminize, 2015-12-20:
@bobvanderlinden the promised post [is here](https://groups.google.com/forum/#!topic/machinekit/bWa5mwg8a9w) and #844 as a point for gathering w.r.t. finding out what we need to do.

zultron, 2015-12-21:
The litmus test for what should be in the machinekit.io docs and what should be in the distributed docs is whether a vendor would need to chop it out before redistributing Machinekit.

For example, instructions for downloading the bleeding-edge packages from `deb.machinekit.io` should stay on `www.machinekit.io`.  A vendor should have a different way of distributing binaries, and quite possibly a completely different set of instructions for users to update their software.

gmoccapy, 2016-01-04:
Happy new Year!

On machinekit.io I found a quick start part, explaining how to set up apt:

sudo sh -c \
    "echo 'deb http://deb.dovetail-automata.com wheezy main' > \
    /etc/apt/sources.list.d/machinekit.list; \
    apt-get update ; \
    apt-get install dovetail-automata-keyring"
sudo apt-get update

Should not 
http://deb.dovetail-automata.com

be
http://deb.dovetail-automata.com/dists

I have not tested it, because I am fighting with my laptop and installing WHEEZY, unfortunately the amd-64 UEFI bootmanager of Wheezy is known to have a bug

strahlex, 2016-01-04:
Its correct, but that the instructions refer to wheezy might be outdated.

sirop, 2016-01-06:
*@ArcEye  asked me to move the issue to this thread.*

Referrering to http://www.machinekit.io/docs/packages-debian/ .

Just tried to install machinekit-xenomai on an identical machine as I did it two months ago:
```
~$ uname -a
Linux debian-slave 3.8-1-xenomai.x86-amd64 #1 SMP Debian 3.8.13-12~1wheezy~1da x86_64 GNU/Linux
~$ halrun
msgd:0 stopped
rtapi:0 stopped
halcmd: hal_init() failed: -22
NOTE: 'rtapi' module must be loaded
halcmd: hal_init() failed: -22
NOTE: 'rtapi' module must be loaded
halcmd: hal_init() failed: -22
NOTE: 'rtapi' module must be loaded
halcmd: hal_init() failed: -22
NOTE: 'rtapi' module must be loaded
```
So `rtapi.so` cannot be found somehow...  

```
~$ sudo find / -iname rtapi.so
/usr/lib/pyshared/python2.7/machinekit/rtapi.so
/usr/lib/linuxcnc/posix/rtapi.so
/usr/lib/linuxcnc/xenomai/rtapi.so
/usr/lib/python2.7/dist-packages/machinekit/rtapi.so
```

But the same on the other machine (same hardware) yields:
```
~$  sudo find / -iname rtapi.so
/usr/lib/rtapi.so
/usr/lib/linuxcnc/xenomai/rtapi.so
/usr/lib/linuxcnc/posix/rtapi.so
/usr/lib/pyshared/python2.7/machinekit/rtapi.so
/usr/lib/python2.7/dist-packages/machinekit/rtapi.so
```  
So the older installtion has `/usr/lib/rtapi.so` .
This older machinekit-xenomai installation is of 1st Nov 2015.

ArcEye, 2016-01-06:
First Q.
What apt stanza do you have in /etc/apt/sources.list.d/machinekit.list ?
(Not sure where all the DNS entries are pointing to currently)
Does removing all the packages, setting machinekit.list to 
`deb http://deb.machinekit.io/ wheezy main`
and running `apt-get update`, then reinstalling the packages, cure it?

Second Q.
Does `/usr/lib/rtapi.so` actually exist on your other machine, 
or is it just a symlink to `/usr/lib/linuxcnc/xenomai/rtapi.so` or similar ?

I don't have a xenomai installation on a x86 currently, will try one if Q1 fails to make a difference

sirop, 2016-01-06:
>Second Q.
>Does /usr/lib/rtapi.so actually exist on your other machine, 
>or is it just a symlink to /usr/lib/linuxcnc/xenomai/rtapi.so or similar ?
>`/usr/lib/rtapi.so` is not a symlink on the older installation.  

```
$ ls -ls /usr/lib/rtapi.so
160 -rw-r--r-- 1 root root 160008 Nov  1 05:41 /usr/lib/rtapi.so  
```
```
$ ldd /usr/lib/rtapi.so
        linux-vdso.so.1 =>  (0x00007fffb8fbe000)
        liblinuxcnchal.so.0 => /usr/lib/liblinuxcnchal.so.0 (0x00007fd4574f8000)
        liblinuxcnc-pb2++.so.0 => /usr/lib/liblinuxcnc-pb2++.so.0 (0x00007fd4571d8000)
        libmtalk.so.0 => /usr/lib/libmtalk.so.0 (0x00007fd456fc0000)
        libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fd456daa000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fd456a1f000)
        libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007fd456717000)
        libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007fd456513000)
        librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007fd45630b000)
        libprotobuf.so.7 => /usr/lib/libprotobuf.so.7 (0x00007fd45600a000)
        libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007fd455df3000)
        libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fd455bd7000)
        libczmq.so.1 => /usr/lib/x86_64-linux-gnu/libczmq.so.1 (0x00007fd45598c000)
        libzmq.so.3 => /usr/lib/x86_64-linux-gnu/libzmq.so.3 (0x00007fd4556e8000)
        libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fd455466000)
        liblinuxcncini.so.0 => /usr/lib/liblinuxcncini.so.0 (0x00007fd455261000)
        libavahi-common.so.3 => /usr/lib/x86_64-linux-gnu/libavahi-common.so.3 (0x00007fd455054000)
        libavahi-client.so.3 => /usr/lib/x86_64-linux-gnu/libavahi-client.so.3 (0x00007fd454e43000)
        libuuid.so.1 => /lib/x86_64-linux-gnu/libuuid.so.1 (0x00007fd454c3d000)
        libjansson.so.4 => /usr/lib/x86_64-linux-gnu/libjansson.so.4 (0x00007fd454a30000)
        /lib64/ld-linux-x86-64.so.2 (0x00007fd45795f000)
        libsodium.so.10 => /usr/lib/libsodium.so.10 (0x00007fd4547d3000)
        libdbus-1.so.3 => /lib/x86_64-linux-gnu/libdbus-1.so.3 (0x00007fd45458d000)
```

```
$ ldd /usr/lib/linuxcnc/xenomai/rtapi.so
        linux-vdso.so.1 =>  (0x00007fff2b4c8000)
        libudev.so.0 => /lib/x86_64-linux-gnu/libudev.so.0 (0x00007fcdc2eb2000)
        libnative.so.3 => /usr/lib/libnative.so.3 (0x00007fcdc2cab000)
        libxenomai.so.0 => /usr/lib/libxenomai.so.0 (0x00007fcdc2aa3000)
        libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fcdc2887000)
        librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007fcdc267f000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fcdc22f3000)
        /lib64/ld-linux-x86-64.so.2 (0x00007fcdc32f2000)
```

> First Q.
> What apt stanza do you have in /etc/apt/sources.list.d/machinekit.list ?
> (Not sure where all the DNS entries are pointing to currently)
> Does removing all the packages, setting [...]

I inserted `deb http://deb.machinekit.io/ wheezy main` into */etc/apt/sources.list.d/machinekit.list* .
I purged and reinstalled only *machinekit-xenomai* and *machinekit-posix* .  
That did not help.  

If you advise also to reinstall *linux-image-xenomai.x86-amd64* , then I can do it only tonight.  

BTW, if you do not `apt-get install  libxenomai-dev` , then *xeno-config* would be missing.

ArcEye, 2016-01-06:
I am in the process of a x86 Wheezy install to a spare machine.

Will try to duplicate by following the machinekit.io instructions and see where it gets.

ArcEye, 2016-01-06:
Sorry, cannot reproduce your errors.

Here is my install sequence (all bar one are direct lifts from the machinekit.io site)

```
Install Wheezy i386 from net-install iso

From a root shell:

echo 'deb http://deb.dovetail-automata.com wheezy main' > /etc/apt/sources.list.d/machinekit.list
echo 'deb http://ftp.us.debian.org/debian wheezy-backports main' > /etc/apt/sources.list.d/wheezy-backports.list"
apt-get update

apt-get install dovetail-automata-keyring

apt-get update

apt-get install linux-image-xenomai.x86-686-pae

apt-get install machinekit-xenomai

apt-get install machinekit-posix

apt-get install libxenomai-dev

(I want a development install so I add)
apt-get install libczmq-dev python-zmq libjansson-dev libwebsockets-dev libxenomai-dev python-pyftpdlib
apt-get install -t wheezy binutils gcc gcc-4.7 libc-dev-bin libc6-dev libexpat1-dev libitm1 libssl-dev    libssl-doc linux-libc-dev manpages-dev python-dev python2.7-dev zlib1g-dev
apt-get install -t wheezy-backports cython
```

Find shows
```
find / -iname rtapi.so
/usr/lib/pyshared/python2.7/machinekit/rtapi.so
/usr/lib/linuxcnc/posix/rtapi.so
/usr/lib/linuxcnc/xenomai/rtapi.so
/usr/lib/python2.7/dist-packages/machinekit/rtapi.so
```
Running a sim on posix threads results in:

![Posix] (https://github.com/ArcEye/misc-images/blob/gh-pages/Selection_001.png)

Rebooting into the xenomai kernel and setting user to group with
`usermod -a -G xenomai mick`

results in:

![Xenomai] (https://github.com/ArcEye/misc-images/blob/gh-pages/Selection_002.png)

ArcEye, 2016-01-06:
If this does not help, I will be able to repeat the process with x86_amd64 later on another machine

My kernel version is exactly the same as yours and the `~1da` section of it indicates it is from the new set of builds AFAIK

sirop, 2016-01-06:
@ArcEye 

I carefully deinstalled any remnants of *linux-image-xenomai.x86-amd64*
and then followed your https://github.com/machinekit/machinekit/issues/840#issuecomment-169318259  till the development install.  

It is still:
```
$ halrun
msgd:0 stopped
rtapi:0 stopped
halcmd: hal_init() failed: -22
NOTE: 'rtapi' module must be loaded
halcmd: hal_init() failed: -22
NOTE: 'rtapi' module must be loaded
halcmd: hal_init() failed: -22
NOTE: 'rtapi' module must be loaded
halcmd: hal_init() failed: -22
NOTE: 'rtapi' module must be loaded
halcmd: hal_init() failed: -22
NOTE: 'rtapi' module must be loaded
halcmd: hal_init() failed: -22
NOTE: 'rtapi' module must be loaded
halcmd: hal_init() failed: -22
NOTE: 'rtapi' module must be loaded
```

As I installed *linux-image-xenomai.x86-amd64*  on Dec 7, I thought that something has changed inbetween.  
But maybe it is just the *halrun* script mixing up some paths?

ArcEye, 2016-01-07:
>  But maybe it is just the halrun script mixing up some paths? 

Not in x86
```
mick@Sempron:~$ halrun
msgd:0 stopped
rtapi:0 stopped
halcmd: loadrt debounce
halcmd: show pin
Component Pins:
  Comp   Inst Type  Dir         Value  Name                             Epsilon         Flags
    73     74 bit   IN          FALSE  debounce.0.0.in               			0
    73     74 bit   OUT         FALSE  debounce.0.0.out              			0
    73     74 s32   I/O             0  debounce.0.0.state            			0
    73     74 bit   IN          FALSE  debounce.0.1.in               			0
    73     74 bit   OUT         FALSE  debounce.0.1.out              			0
    73     74 s32   I/O             0  debounce.0.1.state            			0
    73     74 bit   IN          FALSE  debounce.0.2.in               			0
    73     74 bit   OUT         FALSE  debounce.0.2.out              			0
    73     74 s32   I/O             0  debounce.0.2.state            			0
    73     74 bit   IN          FALSE  debounce.0.3.in               			0
    73     74 bit   OUT         FALSE  debounce.0.3.out              			0
    73     74 s32   I/O             0  debounce.0.3.state            			0
    73     74 bit   IN          FALSE  debounce.0.4.in               			0
    73     74 bit   OUT         FALSE  debounce.0.4.out              			0
    73     74 s32   I/O             0  debounce.0.4.state            			0
    73     74 bit   IN          FALSE  debounce.0.5.in               			0
    73     74 bit   OUT         FALSE  debounce.0.5.out              			0
    73     74 s32   I/O             0  debounce.0.5.state            			0
    73     74 bit   IN          FALSE  debounce.0.6.in               			0
    73     74 bit   OUT         FALSE  debounce.0.6.out              			0
    73     74 s32   I/O             0  debounce.0.6.state            			0
    73     74 bit   IN          FALSE  debounce.0.7.in               			0
    73     74 bit   OUT         FALSE  debounce.0.7.out              			0
    73     74 s32   I/O             0  debounce.0.7.state            			0
    73     74 s32   I/O             5  debounce.0.delay              			0
    73     74 s32   OUT             0  debounce.0.time               			0

halcmd:
```
This was over a ssh connection.

I will try to create a x86_amd64 partition of Wheezy and run again today, but not sure if I will have time or not.
Either way will come back when done.

mhaberler, 2016-01-07:
@sirop - sorry, with this kind of reporting it is impossible to help. " till the development install." is roundabout meaningless.

Again: [you want time and help: please follow this and do your part of the job](https://github.com/mhaberler/asciidoc-sandbox/wiki/Reporting-an-error-with-Machinekit). All steps, no exceptions.

Thanks.

ArcEye, 2016-01-07:
I am assuming he meant until this section of my post

> (I want a development install so I add)
apt-get install libczmq-dev python-zmq libjansson-dev libwebsockets-dev libxenomai-dev python-pyftpdlib
apt-get install -t wheezy binutils gcc gcc-4.7 libc-dev-bin libc6-dev libexpat1-dev libitm1 libssl-dev    libssl-doc linux-libc-dev manpages-dev python-dev python2.7-dev zlib1g-dev
apt-get install -t wheezy-backports cython

and proceeding on that basis ;)

ArcEye, 2016-01-07:
Ok, there is a problem with the x86_amd64 versions of the xenomai packages.

I get the same output as you.
```
$ halrun
msgd:0 stopped
rtapi:0 stopped
halcmd: hal_init() failed: -22
NOTE: 'rtapi' module must be loaded
```

Have to go out, will have a better look later

sirop, 2016-01-07:
@mhaberler 

Indeed @ArcEye  guessed it right in his comment https://github.com/machinekit/machinekit/issues/840#issuecomment-169638480  
and showed that I am not alone :) .

ArcEye, 2016-01-07:
I don't have any more time until the weekend, but there is definitely something amiss with the amd64 packages.
There is nothing leaping out as missing, rtapi.so exists.

@sirop do you actually need xenomai on a x86_amd64?

I always use rt-preempt now, haven't used xenomai for a couple of years and it will be discontinued before too long,  rt-preempt now seems to give as good or better latency.

Doesn't answer the problem with the packages but may give a way to get running.

mhaberler, 2016-01-07:
@sirop now please follow the instructions here to create a core dump and backtrace of rtapi_app:

https://github.com/mhaberler/asciidoc-sandbox/wiki/Debugging-RT-components

and post the backtrace

mhaberler, 2016-01-07:
I cannot reproduce any error.

Just build master on an amd64 with Xenomai. Works perfect for me, including regression runtests.

`````
mah@amd64:~/machinekit-test/src$ uname -a
Linux amd64 3.8-1-xenomai.x86-amd64 #1 SMP Debian 3.8.13-12~1jessie~1da x86_64 GNU/Linux
mah@amd64:~/machinekit-test/src$ git log|head -5
commit 84c8addfcba0b0583604e8f103b6c5d7ff8a8072
Merge: 6c4e9c2 5e2a604
Author: Alexander Rössler <mail@roessler.systems>
Date:   Mon Jan 4 10:53:54 2016 -0600mah@amd64:~/machinekit-test/src$ uname -a

mah@amd64:~/machinekit/src$ lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 8.1 (jessie)
Release:	8.1
Codename:	jessie
````

mhaberler, 2016-01-07:
@sirop @ArcEye I have doubts this issue is the right place.  If this goes on for much longer, please move to a separate issue, provided there is one.

sirop, 2016-01-07:
@mhaberler 
You build from source whereas ArcEye and me are using the debian packages. 
And both  ArcEye and me are still on **wheezy**.  
If you would advise to switch to jessy from now on, then it is even better for me.
I cannot produce any core dump following your instructions as as error occurs before the PID of *rtapi_app* can be known to me.  

@ArcEye  As for rt-preemt instead of xenomai: I do not mind using rt-preemt. However, I remember to have read a two years old discussion about porting xenomai RTDM to rt-preempt. However, I do not remember the outcome of this discussion. Maybe, you know more.  
RTDM can get interesting in connection with IgH EtherCAT-Master as in https://github.com/sittner/ec-debianize .

P.S. @mhaberler , ok, I'll set up a separate issue for this from now on.

mhaberler, 2016-01-07:
we are still in the process of moving the apt repo, see #841 and #842 

please build from source until this is resolved

ArcEye, 2018-08-01:
website builder has been up and running for 2 yrs now.

Any specific issues can go to relevant repo

closed 2018-08-01 by ArcEye