machinekit/machinekit #841: build automation of required packages missing: czmq pyzmq libwebsockets..

mhaberler, 2015-12-18:
we have some supplementary packages which we build since either the debian versions are outdated or buggy or both

we have the toolchain explored for the build of machinekit proper - travis, docker etc

what is missing is to funnel these supplementary packages through the same build/upload chain so that is a) fully automatic b) can be arbitrarily reproduced by other users

I would also suggest to move these packages to the machinekit organisation on github

I am out of my depth doing that automating on my own (too much rocket science and voodoo) but willing to try on a particular package with coaching by someone in the know

let's start with a simple one, like libwebsockets - who guides me?

kinsamanka, 2015-12-18:
I think we need a repo that contains all the suplementary packages.

From there we can add the needed travis-ci code.

mhaberler, 2015-12-18:
@kinsamanka make me understand why an uber repo - what does it gain you? run just once for all packages?

bobvanderlinden, 2015-12-19:
Are we talking about 1. a Debian package repository containing all dependencies?
Or are we talking about 2. including the source of all non-standard dependencies into the Git repository of Machinekit?

For 1 we need to build and package all non-standard dependencies. It would also be *very* preferable (IMO) to have the process of building and packaging standardized and simplified, so that multiple dependencies can be built by anyone this way.

For 2 we still need a way to integrate the dependencies into the build system. This does not sound preferable to me.

> what is missing is to funnel these supplementary packages through the same build/upload chain so that is a) fully automatic b) can be arbitrarily reproduced by other users

Since you mention automatic and reproducible builds, I need to mention Nix again. At the moment it's one of the few options where you're guaranteed to get functionally the same output in a build each and every time from whichever system you (or anyone else) build. This isn't a solution for building Debian packages though.

>we have some supplementary packages which we build since either the debian versions are outdated or buggy or both

Which dependencies specifically do we have that need patching, updating or packaging on Debian? Can we start a list which ones need work and what need to be done?

>let's start with a simple one, like libwebsockets - who guides me?

For libwebsockets, Debian jessie already has an package for this: https://packages.debian.org/jessie/libwebsockets3
libwebsockets in jessie is v1.2.2. The latest version is v1.6.0. Does this need updating? If not, does the Debian package need patching?

sirop, 2015-12-19:
@bobvanderlinden
> Debian package need patching?  

https://packages.debian.org/jessie/libwebsockets3 seems to be built with these options:
https://sources.debian.net/src/libwebsockets/1.2.2-1/CMakeLists.txt/#L30-L43

Changing some of these options maybe?

mhaberler, 2015-12-19:
@bobvanderlinden  we have a few prerequisite packages which are:

- libwebsockets - too old in debian, incompatible API changes - so 
@zultron packaged a working version here https://github.com/zultron/libwebsockets-deb
- czmq libzmq libsodium pyzmq - same thing, see also @zultron's github repo
- cython: the version in wheezy is too old, wheezy-backports version is ok, as is jessie and ubuntu trusty
- kernels - a range of RT-PREEMPT, Xenomai and RTAI kernels packaged by @zultron and uploaded manually do deb.dovetail-automata.com (I _think_ - maybe @zultron could comment?)

now as time progresses these packages will become less as newer debian releases will contain working versions; for instance jessie covers zmq completely in usable (for now) form. But for the time being, these packages need to be built by us, and added to the apt repos.

There is another angle for automating the build and upload of these packages: it allows us decouple from the "lets hope the debian maintainer gets his butt off the ground" stalemate when adopting new features/API's - zmq in particular, as we still have to add encryption/authentication and zmq is moving ground here. With sticking with whatever a random debian maintainer happens to upload eventually we are stuck here.

So they are "standard" dependencies - they are just too old. The packaging is already covered by @zultron - what is missing is tacking on a travis build for all arches and distros and uploading the resulting packages to an apt repo.

We are positively NOT talking about including the source of these packages in the mk repo.

zultron, 2015-12-21:
Once Wheezy is out of the picture (see #842), the question of package deps will be much simplified.  I thought about how to automate these builds in Travis as well, but ultimately decided it's too much trouble to be worth it.  These things rarely change, if ever, and we already have scripts that can build these things, and anyway the Debian package tools aren't rocket science.

The kernels can't build in Travis anyway.  The x86 kernels take 45 minutes with `make -j8`, and would go way over the 50 minute Travis limit with the max `make -j2`.  The Beaglebone kernel might be buildable if a real Canadian cross-build setup were worked out, but there's no need, since @RobertCNelson provides those.

The way I've been handling these is simply to build them into their own APT repo, and set up a `reprepro` config that pulls those deps and the latest MK packages into the production repo, and signs them.  That also takes care of the Jessie RT_PREEMPT kernel, which must be grabbed from a `jessie-backports` snapshot, and will also handle the `hostmot2` packages, of which I've never reproduced a build.  Very simple stuff.  See https://github.com/zultron/mkdeploy for the system I'm using now, which runs on a $5/month VPS.

mhaberler, 2015-12-21:
@zultron - I fundamentally disagree

One of the problems we have  is the dependency on a single person. The undocumented process of manual build and upload of packages (kernel or otherwise) adds to that, and we need to make it less, not more. 

I also note it was you who insisted on removing that particular dependency, and - thankfully - set in motion the process we are in, but frankly we would be very ill advised to stop here.

As for the hope with getting away without _any_ private package builds: let us put it rest, because it will not happen - rather, it will happen again. Examples:

- zmq/czmq: there are very strong arguments to use the v5 API as soon as this stabilizes because some features will significantly simplify the code, for example non-lossy (blocking) publisher channels and improvements in encryption/authentication
- libwebsockets is just now going through a major API revision and stabilisation phase at 1.6

A docker setup packaging all the steps involved is a recipe which everybody can reproduce. Wether that is run on travis or some other machine like mine because of time limits is a secondary question; the primary issue is reproducability by any non-author/non-guru. That we simply do not have, and I think we must strive for having it.

zultron, 2015-12-21:
Well, then by all means, do it!  I didn't mean to stand in the way.  There's no rocket science involved in reproducing these packages (unlike the Machinekit CI system!), so this will be much easier to accomplish.

zultron, 2015-12-21:
Here's a list of package sources.

Xenomai kernel:
(See [these scripts][1] for instructions on configuring kernel packages)
- https://github.com/zultron/xenomai-deb
- https://github.com/zultron/linux-ipipe-deb
- https://github.com/zultron/linux-latest-deb
- https://github.com/zultron/linux-tools-deb
[1]: https://github.com/zultron/dxsbuild/tree/master/scripts/package

Wheezy only:
- https://github.com/zultron/libsodium-deb
- https://github.com/zultron/zeromq3-deb
  - Depends on `libsodium`
- https://github.com/zultron/pyzmq-deb
  - Depends on `zeromq3`
- https://github.com/zultron/python-pyftpdlib-deb
- https://github.com/zultron/jansson-deb

Jessie + Wheezy:
- https://github.com/zultron/czmq-deb
  - Depends on `zeromq3`
- https://github.com/zultron/libwebsockets-deb

Retired:
- https://github.com/zultron/rtai-deb

bobvanderlinden, 2016-01-07:
I don't really understand how to build those -deb repositories. I usually don't use debian/ubuntu for packaging, so I probably don't know how things should work. I tried using debbuild, but it looks for debian/changelog. I tried renaming xenomai-deb to debian and build from its parent directory, but now it looks for `./configure`. Should I clone the Xenomai repository, move xenomai-deb into that and rename it to debian?

zultron, 2016-01-07:
@bobvanderlinden Sorry, my git repos are organized slightly differently from the usual upstream repo.  Try something like this (untested):

    wget http://download.gna.org/xenomai/stable/xenomai-2.6.4.tar.bz2
    tar xf xenomai-2.6.4.tar.bz2
    cd xenomai-2.6.4
    git clone https://github.com/zultron/xenomai-deb.git debian
    debbuild [your debbuild options]

bobvanderlinden, 2016-01-07:
It seems xenomai has their own debian directory. `debuild -uc -us` worked.
Is this the one we should use or would it be better to use yours? What
changes have been made to the debian files?

zultron, 2016-01-12:
@bobvanderlinden If the Xenomai release works out of the box, I think it's fine to use it.  Last time I checked, they hadn't updated the Debian packaging.

Because of this, there may be an issue with upgrading from the deb.da.com packages.  Check that the upstream Xenomai release `debian/changelog` version-release is greater than `2.6.4-1`.  (My error:  I should've set it to `2.6.4-0.~1` or such for smooth upgrade.)

ArcEye, 2018-08-01:
This has been resolved now.

API changes forced wheezy to be frozen and from Jessie and above the packages
for the new v3 czmq / zmq API are straight from Debian.

closed 2018-08-01 by ArcEye