machinekit/machinekit #842: Post-CI rebuild:  long-term package repo big picture

zultron, 2015-12-18:
Now that the former Buildbot functions have been migrated onto Travis CI etc. (see #791), we need a strategy for long-term maintenance of a package repository, and a short-term plan to migrate users toward it and away from `deb.dovetail-automata.com`.

This issue is to track that problem.

zultron, 2015-12-18:
There's a potential opportunity for a low-effort migration of many users to a new repo in parallel with upgrading to Jessie.  There are currently few users running Jessie, and as users upgrade, APT configurations can be updated.  Over time, Wheezy users will dwindle, at which point it will cause little disruption to turn off the old `deb.dovetail-automata.com` repo.  If we want to go this way, the time is now, before the Jessie user base grows.  Does this sound reasonable?

zultron, 2015-12-18:
What is the consensus about the DNS name to use for the new repository?  If packagecloud.io is not a viable option, let's pick one, such as `deb.machinekit.io`.

At the same time, if the project uses a `machinekit.io` domain name on the repository, it makes sense to generate new package signing keys for a 'Machinekit' organization.

bobvanderlinden, 2015-12-19:
> If packagecloud.io is not a viable option, let's pick one, such as `deb.machinekit.io`.

Another vote for `deb.machinekit.io`.

> it makes sense to generate new package signing keys for a 'Machinekit' organization

I agree.

mhaberler, 2015-12-19:
re deb.machinekit.io - I manage the DNS for machinekit.io so I added 

```
$ host deb.machinekit.io
deb.machinekit.io has address 188.166.26.158
````
which is where deb.dovetail-automata.com currently points to.

Anything else?

zultron, 2015-12-21:
`deb.machinekit.io` sounds good.  It needs to be pointed at a repo signed with some sort of Machinekit organization's keys, though, and I recommend an APT configuration like this, with the trailing `debian/` in the URL:

    deb http://deb.machinekit.io/debian/ jessie main

I remember @kinsamanka had a Docker container that Travis CI could upload packages to via sftp.  This sounds like a great option for a long-term package deploy/distribution system.

zultron, 2015-12-21:
What's the timeline for pushing Jessie as a supported distro?  If it's earlier rather than later, it would help with this migration issue, and is related to #841 (which, if any, packages must be rebuilt) and #840 (how machinekit.io docs should be updated).

mhaberler, 2015-12-21:
I am sure we will have to live with one or the other package being custom-built for a long time, even post-jessie 

for a start, it would be great to have a single example package (like the ones packaged in your repo - for instance libwebsockets) extended to go through a travis build and upload to an apt repo like machinekit proper; then other packages could clone that recipe

re #840 - this is about the website, not docs, and there is nothing holding us back from starting with removing outdated or wrong information, and starting on documenting the travis build

for instance the buildbot page can be replaced right now
the question of kernels is open - since you provided them, I think nobody else but you can correct that information?

kinsamanka, 2015-12-21:
The builds are now currently being pushed to deb.mah.priv.at using sftp.

zultron, 2015-12-21:
@mhaberler IMO it's really not worth the time to do that.  If you want to, though, I suggest starting by reproducing the Machinekit package builds.  It took me all of ten minutes the first time (quite brilliant!).  It would be easy enough to extend that to the building of MK deps.

zultron, 2015-12-21:
If @kinsamanka is already pushing debs to `deb.mah.priv.at`, why not point `deb.machinekit.io` there and just set up some signing keys?  That would tick a big box with little effort.

mhaberler, 2016-01-05:
deb.mah.priv.at is not being updated, that was just a fallback in case packagecloud did not work out. Rather it is the one at packagecloud.io which deb.machinekit.io points to. 

If there are signing keys missing, that is the one which needs attention. But it seems it worked - maybe @kinsamanka already did that?

we still need to reconcile the website contents with kernels actually being present there (http://deb.machinekit.io/pool/main/l/linux/), also the hostmot2 firmware packages are missing

zultron, 2016-01-05:
### Bringing up the deb.machinekit.io repo

@kinsamanka has already done the hardest work to bring up the deb.machinekit.io repo, and signed packages are already being automatically deployed into deb.mah.priv.at.  Really great work!  All that's left is minor work fixing signatures and packages.

Here are lists of what's done, and what's left to do:

####  Completed steps

The basic build/deploy/sign/distribute mechanisms are all in place:

- [x] Push packages to an `incoming` directory on deb.mah.priv.at
  - The [`send_binaries.sh`][1] script pushes build results with `sftp` to `$SFTP_DEPLOY_ADDR`
  - (`$SFTP_DEPLOY_ADDR` is set in the [Travis CI settings][2])
- [x] Set up crontab with `reprepro` to pull from `incoming` directory (guessing); see:
    - Pool files [here][3]
    - Dists [here][4]
- [x] Verify updated packages in `pool` directory
  - The [last Travis CI build of master][5] was PR #853, rev. 84c8add; sure enough:
    - http://deb.mah.priv.at/debian/pool/main/m/machinekit/machinekit_0.1.1451926434-1mk.travis.master.git84c8addf~1jessie.dsc
- [x] Verify signature mechanism works
  - These are being signed by @kinsamanka's keys, perhaps in a cronjob on deb.mah.priv.at:

          $ wget -q -O - http://deb.mah.priv.at/debian/dists/jessie/InRelease \
          >    | gpg --verify --keyserver-options auto-key-retrieve=1
          gpg: Signature made Mon 04 Jan 2016 11:59:54 AM CST using RSA key ID 6B26D27A
          gpg: Good signature from "GP Orcullo <kinsamanka@gmail.com>"
          gpg: WARNING: This key is not certified with a trusted signature!
          gpg:          There is no indication that the signature belongs to the owner.
          Primary key fingerprint: 04C4 E917 DE0F 0DE2 6F55  831A FDD4 CC1B 648D 4DEF
               Subkey fingerprint: AE27 0344 5C21 52F4 EB2B  E08B 046C C201 6B26 D27A

### Remaining steps

We're currently stuck without keys and DNS entries in place; @mhaberler, can you work with @kinsamanka on this?

- [ ] Generate new keys for the Machinekit project
- [ ] Put them in place on deb.mah.priv.at
- [ ] Build a `machinekit-keyring` Debian package, perhaps based off [mine][6]
  - [ ] Add `Conflicts: dovetail-automata-keyring`, according to Debian [package transition guidelines][7]
- [ ] Add `machinekit-keyring` package to the repo
  - [ ] Remove the `dovetail-automata-keyring` package
- [ ] Update Travis builds to point at new repo/new keys
- [ ] Update website install instructions

As @mhaberler points out, there will be some packages missing:
- [ ]  @kinsamanka will have to show us how to get `hostmot2` packages from linuxcnc.org into the repo.
- [ ] I sucked the old Jessie PREEMPT_RT kernel from a snapshot, as detailed [here][8].
- [ ] Beaglebone kernel?
- [ ] Other packages?

Once the above are done, we can just go into a test/find broken stuff/fix it/repeat loop until everything works.  

[1]: https://github.com/machinekit/machinekit/blob/master/.travis/send_binaries.sh
[2]: https://travis-ci.org/machinekit/machinekit/settings
[3]: http://deb.mah.priv.at/debian/pool/main/m/machinekit/
[4]: http://deb.mah.priv.at/debian/dists/
[5]: https://travis-ci.org/machinekit/machinekit
[6]: https://github.com/zultron/dovetail-automata-keyring-deb
[7]: https://wiki.debian.org/PackageTransition
[8]: https://github.com/zultron/mkdeploy/blob/master/reprepro-templates/tmpl.updates#L32-L42

mhaberler, 2016-01-06:
@zultron sorry @kinsamanka  did all this setup remotely and I am out of my depth at the steps you outlined except any DNS changes

I was not even sure if deb.mah.priv.at or packagecloud was the upload target, turns out this is a docker process serving it

I do think Gemi is busy right now, not sure when he will resurface 

you have an account on my machine - could you help in case Gemi is unavailable?

re Beaglebone kernel - I see no point in adding that since it is covered by Robert ?

kinsamanka, 2016-01-06:
I'm using docker-compose to run the http and ssh server at deb.mah.priv.at.

kinsamanka, 2016-01-06:
Sorry crappy phone ;-/

The one running at deb.mah.priv.at is using docker-compose to run both http
and ssh server. Once the incoming deb from travis-ci.org is received it is
signed and processed by reprepro.

We can run the container at deb.machinekit.io or point it to d.m.p.a

All that needs to be done is change the signing keys.

mhaberler, 2016-02-19:
I started a writeup on Gemi's setup on my host: https://github.com/mhaberler/asciidoc-sandbox/wiki/The-machinekit-package-build-and-distribution-process

I do not see much point in retaining the packagecloud service - replicating that looks easy enough and we are not dependent on packagecloud

mhaberler, 2016-02-25:
I am trying to get the deb.mah.priv.at repo ready for migrating deb.machinekit.io as Johns buildbot should be retired. Also, we should get ready for armhf jessie at last.

This means that the
- packages from the travis builds are uploaded - this works
- the few prerequisite packages we need must be in deb.mah.priv.at - see below
- for armhf, the packages in http://repos.rcn-ee.com/debian/ must match the ones being built against in travis so we can drop our own packages asap. IMO it would be best if the travis armhf builder would build against the packages in http://repos.rcn-ee.com/debian/ .

Status on deb.mah.priv.at:

- we STILL have this [broken zeromq4 package around](http://deb.mah.priv.at/debian/pool/main/z/)
- the [czmq2-dev package](http://deb.mah.priv.at/debian/pool/main/c/czmq/) is outdated relative to http://deb.dovetail-automata.com/pool/main/c/czmq/ used in the build (note version number), so an install from deb.mah.priv.at fails
- We cannot therefore currently install armhf with prereq's coming only from deb.mah.priv.at.
- it is IMPOSSIBLE for me to delete, or upload packages to @kinsamanka ' s battery of containers running on my machine thanks to complete lack of information as to how to do this

Status on deb.dovetail-automata.com:

- obviously the travis builder pulls the prereq packages from here -  however, the czmq2 version on http://repos.rcn-ee.com/debian/  is behind the one on http://deb.dovetail-automata.com. We cannot therefore currently install armhf with prereq's coming only from rcn-ee.com.
- since the arm package building has stopped, the packages are not updated.

Therefore, to install a current armhf machinekit package, the following insanity is needed:

- install a outdated machinekit armhf package to get all the dependencies in the right version.
- wipe the DA package source, and the machinekit package, and leave the prereq's lying around.
- switch to deb.mah.priv.at 
- install the machinekit package from there.

Let me be candid: this situation is ridiculous.

**Three months after Johns buildbot just had to be replaced right now we STILL are at square one, and the basics DO NOT WORK.** Everybody just had to scramble right now for the replacement, unfortunately after the first 90%, everybody seems to have dropped their tools, and seems not to care if the result actually works.

Frankly, I am getting tired of jumping up and down to get those basics ticked which I cannot do myself. These are:

@kinsamanka - I formally ask you to document the steps needed to manually upload packages to, and remove packages from the server on mah.priv.at, within a reasonable timeframe. It is impossible to serve a community this size when everything depends on you getting around to patch up things, and nobody else can, with weeks between time slots.

@zultron - I ask you to work with Robert Nelson to have any packages in rcn-ee upgraded to whatever the travis builder expects, and directly work with @kinsamanka to have the build requirements and what is actually available on rcn-ee and deb.mah.priv.at/eventually deb.machinekit.io to be in sync.

@zultron - also, please work with @kinsamanka directly to have the kernels uploaded which are worth retaining so at least SOMETHING is there - I suggest the i386 and amd64 rt-preempt and xenomai kernels and nothing more.

Please work together in good faith to actually get working results soon. 
And please actually **test if the replacement works. Or give instructions what to test.**

luminize, 2016-02-25:
@kinsamanka if you have any info on the documentation @mhaberler mentions, you can send it to me if you like. I'll help you out with getting the diagrams into shape. Pictures with phone of sketches or whatnot. I'll happily digitize them for your all to review and use. If you all do the thinking I'll do the grunt work.

Really: hand drawn sketches and scribbles are no problem.

the-snowwhite, 2016-02-26:
I was able to get past configure and almost  pull off a complete armhf sid cross build today by using the libczmq3 from stretch / sid.

May be worth to look into

https://packages.debian.org/stretch/libczmq3 <https://packages.debian.org/stretch/libczmq3>

Best wishes
Michael B.

kinsamanka, 2016-02-26:
I've updated the [README](https://github.com/kinsamanka/mkdebs/blob/master/README.md) to include the commands on how to manually add/remove packages.

I've also moved the docker directory to `/home/mk_repo`.

The packages are not properly signed at the moment as there was a bug on the signing script and I'm waiting for the updated signing keys.

mhaberler, 2016-02-26:
@the-snowhite we know since a while this works, see https://github.com/machinekit/machinekit/issues/598#issuecomment-103177469

the issue is inconsistency between what is in repos, and what is used for the build, rendering the result unusable

mhaberler, 2016-02-26:
@kinsamanka thanks for the update on install/remove - sorely needed. I sent instructions on the updated key and hope this works - we can very well go ahead on the contents in parallel

I will try to weed out the obvious problem packages in the afternoon, and try to copy over the xenomai and rt-preempt kernels from DA

however, we will not get results until the packages being used in the build match the ones being available for download in the new repo 

- should we copy over czmq2 from DA ?
- do we need rcn-ee to upgrade or not?
- should we change the prerequisites to czmq3?

please advise

mhaberler, 2016-02-26:
@kinsamanka - it would make sense to transfer all repos needed for build and serving packages to the machinekit organisation so we can apply the C4 process there

I assume that is https://github.com/kinsamanka/mkdebs and https://github.com/kinsamanka/machinekit_builder

any others?

kinsamanka, 2016-02-26:
Only those two.

I'll do it tomorrow.

mhaberler, 2016-02-26:
@kinsamanka - trying to reproduce your instructions in https://github.com/kinsamanka/mkdebs/blob/master/README.md

for a start, try to remove `libczmq-dbg_2.2.0-0.2~1da~wheezy1_amd64.deb`  which lives [here](http://deb.mah.priv.at/debian/pool/main/c/czmq/)

`````
mah@mah2:/home/mk_repo$ docker-compose ps
     Name                 Command            State                       Ports
-------------------------------------------------------------------------------------------------
mk_data          /true                       Exit 0
mk_reprepro      /bin/sh -c /run.sh          Up
mk_sftp_server   /bin/sh -c /run.sh travis   Up       0.0.0.0:9022->22/tcp
mk_web_server    nginx -g daemon off;        Up       0.0.0.0:9443->443/tcp, 0.0.0.0:9080->80/tcp
`````
so the reprepro container is named `mk_reprepro`.

Next step: try to remove that package:

`````
mah@mah2:/home/mk_repo/incoming$ docker exec mk_reprepro reprepro remove wheezy libczmq-dbg_2.2.0-0.2~1da~wheezy1_amd64.deb
Not removed as not found: libczmq-dbg_2.2.0-0.2~1da~wheezy1_amd64.deb
Exporting indices...
`````
Is the `<packagename>` supposed to be different? What would that be?

kinsamanka, 2016-02-26:
> Take note of the debian package naming scheme:
> `<packagename>_<VersionNumber>-<DebianRevisionNumber>_<DebianArchitecture>.deb`

It should be:
`docker exec mk_reprepro reprepro remove wheezy libczmq-dbg`

mhaberler, 2016-02-27:
ok, got it - thanks

sirop, 2016-02-29:
@zultron 
I do not know if the signature issue was finally solved within the above discussion,
but doing 
```
sudo sh -c "echo 'deb http://deb.machinekit.io wheezy main' > /etc/apt/sources.list.d/machinekit.list"
sudo apt-get update
```
on wheezy amd64  I get 
```
Reading package lists... Done
W: GPG error: http://deb.machinekit.io wheezy Release: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 7F32AE6B73571BB9
```
Tried it also with `http://deb.dovetail-automata.com` instead of `http://deb.machinekit.io` with the same outcome.

mhaberler, 2016-02-29:
@sirop - currently @kinsamanka is putting the finishing touches on moving debs to a new host - that is already in progress and this is the reason for what you are seeing

part of this will be new signing keys - hopefully temporary for a few days max

please use http://deb.dovetail-automata.com for now, deb.machinekit.io will be switched 

we'll announce when the new repo is online and fully usable

mhaberler, 2016-02-29:
@kinsamanka - I'm getting second thoughts on the signing strategy - it seems there is some confusion what signing actually means

If I understand correctly, [SecureApt](https://wiki.debian.org/SecureApt) does signing of the repository metadata **only**, i.e the Release file and the checksums therein, chained to checksums in subordinate Release files, and that is what reprepro seems to manage by itself just fine

I find no reference in this document on dpkg-sig or signing the package per se (signature contained in the _gpgbuilder ar member file)

Note also John's package are not signed, for example (on mah.priv.at:)
`````
mah@mah2:~$ ar tv /var/spool/apt-mirror/mirror/deb.dovetail-automata.com/pool/main/c/czmq/libczmq2_2.2.0-0.6~1jessie~1da_amd64.deb
rw-r--r-- 0/0      4 Dec 10 08:35 2015 debian-binary
rw-r--r-- 0/0   4079 Dec 10 08:35 2015 control.tar.gz
rw-r--r-- 0/0  80296 Dec 10 08:35 2015 data.tar.xz
````

it seems to me the dpkg-sig step mentioned in https://github.com/machinekit/mk_deb_server/blob/master/README.md is actually superfluous - we were using apt-get just fine from John's repo with just signed metadata but without signed packages?

[This article](http://blog.packagecloud.io/eng/2014/10/28/howto-gpg-sign-verify-deb-packages-apt-repositories/) also suggests repo metadata signing is the prevalent method:

> Further, most official Debian package files from the publicly accessible repositories do not have GPG signatures. The official repository metadata is GPG signed, as are the source packages, but the .deb packages themselves are not.

I'm not sure if signing the source packages makes a difference, but it could be we're doubling up on signing for low returns

I dont exclude I am overlooking something fundamental, but it seems the package signing could just be dropped

mhaberler, 2016-03-01:
the `deb.machinekit.io` DNS entry is now in transition - dont refer to it until victory is declared :-)

for the record, post switch this will be needed for the new signing key: 

   `apt-key adv --keyserver keyserver.ubuntu.com --recv 43DDF224`

ArcEye, 2016-03-01:
> I dont exclude I am overlooking something fundamental, but it seems the package signing could just be dropped

AFAIK, which may not be much,  you need to sign one or the other, not both.

On my private repo, initially I signed the packages, but that is difficult to automate securely and time consuming manually.

I discovered that reprepro actually signed the repository Release files etc when I was trying to find a way to automate it and ended up with an unsigned package but signed Release files etc and it seemed to work just the same.

So yes, I believe you are correct and it is certainly simpler that way.
Will be no less secure as the checksums for the packages still have to match.

mhaberler, 2016-03-01:
yeah, we dropped the package signing and it looks great so far

mhaberler, 2016-03-04:
I _think_ the migration to deb.machinekit.io is complete, and I would appreciate feedback if this works for you too

to try it out:

fetch the new signing key:  `sudo apt-key adv --keyserver keyserver.ubuntu.com --recv 43DDF224`

depending on your distro, set  `/etc/apt/sources.list.d/machinekit.list` to:
    `deb http://deb.machinekit.io/debian jessie main`
or 
   `deb http://deb.machinekit.io/debian wheezy main`

apt-get update
apt-get upgrade

probably will pull in an updated machinekit package.

BB package downloads are slightly tested and look ok. At some later point in time Gemi will add raspberry packages and a kernel but those are not present yet.

The content is essentially a snapshot (including kernels) of deb.dovetail-automata.com plus current machinekit builds uploaded from travis.

I have manually added a more recent RT-PREEMPT kernel which I had good success with on j1900 platforms and I thought it might be useful for other people: `linux-image-4.1.15-rt17mah` 

The setup is along the lines of https://github.com/mhaberler/debrepo which should give enough hints to reproduce on another host. This does not use [Gemi's docker-compose based setup](https://github.com/machinekit/mk_deb_server)  but rather uses sftp, reprepro and apache2 directly which turned out to be easier to manage and update.

ps: Extra thanks to @kinsamanka for chaperoning me through this and generally being a great sport!

sirop, 2016-03-04:
@mhaberler 
I installed `linux-image-xenomai.x86-amd64` on wheezy today in the morning using `http://deb.dovetail-automata.com`.  
Now I have `deb http://deb.machinekit.io/debian` instead in `/etc/apt/sources.list.d/machinekit.list`.  
Following your above instructions I did:
```
slave@debian-slave:~$ sudo apt-get upgrade
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following packages will be upgraded:
  linux-image-3.2.0-4-amd64
1 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 23.5 MB of archives.
After this operation, 2,048 B of additional disk space will be used.
```

But I am already on a xenomai kernel:
```
slave@debian-slave:~$ uname -a
Linux debian-slave 3.8-1-xenomai.x86-amd64 #1 SMP Debian 3.8.13-12~1wheezy~1da x86_64 GNU/Linux
```

So `apt-get upgrade` thinks that I do not have the kernel I need ...

ArcEye, 2016-03-04:
Well done @mhaberler and @kinsamanka 

@sirop You are already running the latest version of Xenomai kernel, so nothing to upgrade

http://deb.machinekit.io/debian/pool/main/l/linux/linux-image-3.8-1-xenomai.x86-amd64_3.8.13-12~1wheezy~1da_amd64.deb

Should really use `apt-get dist-upgrade` to make sure all the upgrades are pulled, package dependencies resolved and orphaned packages removed.

mhaberler, 2016-03-04:
@sirop - your perception of how kernel installs on Debian work is off - this is certainly **not an error in the apt repo**

just because `linux-image-xenomai.x86-amd64` is installed, this does not mean some other kernel can NOT be installed. I am positive you had `linux-image-3.2.0-4-amd64` installed before you installed the Xenomai kernel. Have a look under /boot, and read up on apt-cache policy.

mhaberler, 2016-03-04:
Dear Yankee brethren:

is there somebody around with a well-connected hosted machine/VM with a domain name who could act as a second machinekit apt repo? 

A few GB of space should suffice (currently it's around 250MB)

I am not looking at apt-mirroring past startup - I would rather have the second apt-repo to be directly fed by travis to avoid common-mode failures

luminize, 2016-03-04:
> I think the migration to deb.machinekit.io is complete, and I would appreciate feedback if this works for you too

I followed the instructions on a pristine VM and installed the posix packages. Didn't install kernel. Instructions all worked for me.

will update the instructions.

luminize, 2016-03-04:
> Dear Yankee brethren:

> is there somebody around with a well-connected hosted machine/VM with a domain name who could act as a second machinekit apt repo for redundancy ? 

Dear yankee brother :) I can provide a subdomain right away, I have a static IP but no continuous on-line machine (yet). I'm willing to donate a machine, but I need help/instructions if/when I set it up.

mhaberler, 2016-03-04:
@luminize I understand NL is a usual place for yankees to washed ashore in Europe next only to UK but I was looking for a US-based server specifically for geographic redundancy ;) thanks though!

a physical machine is overkill - some basic hosted VM does it as long it can be reached by `sftp <domainname>`

luminize, 2016-03-04:
I've updated the instructions.
http://www.machinekit.io/docs/packages-debian/
which will lead you to
https://github.com/machinekit/machinekit-docs/blob/master/machinekit-documentation/getting-started/installing-packages.asciidoc#configure-apt

mhaberler, 2016-03-05:
There are no packages for the raspberry, in particular not the rpi1 (arm6) which is not a useful architecture for running HAL

RunningLight, 2016-03-05:
> Dear Yankee brethren:

>is there somebody around with a well-connected hosted machine/VM with a domain name who could act as a second machinekit apt repo for redundancy ?

>A few GB of space should suffice (currently it's around 250MB)

>I am not looking at apt-mirroring past startup - I would rather have the second apt-repo to be directly fed by travis to avoid common-mode failures

option 1) Somebody else.

option 2) Me. I've hosted from home (Maryland) in the past. My Comcast Cable service provides asymmetric download/upload speeds. Haven't tested lately (and I'm away from home at the moment) but the upload speed is typically 5-7 Mbps; let's call it 3/4 MB/s. Is that "well connected" enough? Because it's residential service, my IPs are dynamic, but various services offer dynamic IP tracking. I own some domain names such as tarradiddle.org and tarradiddle.net that are not in use.  If no one else steps up, I can put one of my ex-test boxes online for you, like I did when John and I were first playing with BuildBot. My cable modem, router, and box would be on an UPS but I can't guarantee up-time reliability.

option 3) Use a suitable US-based hosted service. I'm willing to underwrite if the cost is modest, say a dollar a day or less (not the first time I've made the offer). Someone else will have to do the market research.

Regards,
Kent

mhaberler, 2016-03-05:
@RunningLight - I think some cheap hosted VM would be the more stable solution

I would prefer not to be involved in running this at all - just another dependency
setup help is fine

mhaberler, 2016-03-05:
as far as I'm concerned, this issue is solved (modulo secondary apt repo), closing

closed 2016-03-05 by mhaberler

luminize, 2016-03-05:
@kinsamanka @mhaberler well done! I think most people do not appreciate the cost of delivering the last pieces.

RunningLight, 2016-03-06:
Absolutely agree, Bas. Props to all involved. Wish I had been able to pitch
in to help make it happen.

Regards,
Kent

On Sat, Mar 5, 2016 at 6:39 PM, Bas de Bruijn <notifications@github.com>
wrote:

> @kinsamanka @mhaberler well done! I think most people do not appreciate
> the cost of delivering the last pieces.
>
>
>
> > On 05 Mar 2016, at 20:22, Michael Haberler <notifications@github.com>
> wrote:
> >
> > Closed #842.
>
>

the-snowwhite, 2016-03-06:
Some after thoughts:
1 I was able to update my build script, however not noticing the instructions were for arm xenomai kernel and not arm rt-preempt turned it into an unexpected hour long battle... because I copy and pasted mindlessly first.

2. I would recommend replacing Sid with next upcoming Stretch as this works just fine.

zultron, 2016-03-07:
On 03/05/2016 05:39 PM, Bas de Bruijn wrote:
> @kinsamanka @mhaberler well done! I think most people do not appreciate
> the cost of delivering the last pieces.

+1  Well done guys, as usual!