machinekit/machinekit #846: `loadrt [HOSTMOT2](DRIVER) config=[HOSTMOT2](CONFIG)` throws error

luminize, 2015-12-24:
I can't get my Mesa 5i20 card to run.
/var/log/linuxcnc.log stays empty so I can't supply a lot of info.

[pastebin of ini file](http://pastebin.com/Jr3NEAWF)
[pastebin of halfile](http://pastebin.com/0nVJJQJd)
[dmesg output](http://pastebin.com/k4RuTbA2)

```
bas@xw6600:~$ uname -a
Linux xw6600 3.19.0-25-generic #26~14.04.1-Ubuntu SMP Fri Jul 24 21:16:20 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux
```
I run an ini file and hal file copied from the configs.
The `SVST8_4.BIT` comes fresh from the mesa website, and the correct file location is in the halfile.
```
bas@xw6600:~/hal-configuration$ DEBUG=5
bas@xw6600:~/hal-configuration$ halrun -f hm2-stepper.hal -i 5i20.ini
msgd:0 stopped
rtapi:0 stopped
hm2-stepper.hal:43: insmod failed, returned -1:
rtapi_rpc(): reply timeout
```
hm2-stepper.hal:43 says : `loadrt [HOSTMOT2](DRIVER) config=[HOSTMOT2](CONFIG)`
after the recompile the result of `sudo find / -name hostmot2.so`
```
bas@xw6600:~/machinekit$ sudo find / -name hostmot2.so
[sudo] password for bas:
/home/bas/machinekit/rtlib/rt-preempt/hostmot2.so
/home/bas/machinekit/rtlib/xenomai/hostmot2.so
/home/bas/machinekit/rtlib/posix/hostmot2.so
```

the check-system-configuration script output (we knew logging on Ubuntu was not working):
```
bas@xw6600:~/machinekit$ scripts/check-system-configuration.sh
Warning:  Logging appears to be not working

          Found rsyslogd configuration and /var/log/linuxcnc.log
          exists, but a test did not appear in the log.
          Please investigate.
```

sirop, 2015-12-24:
@luminize  
I had a similar case with a component compiled by me on Debian Wheezy.
Try this:
```
loadrt  /home/bas/machinekit/rtlib/YourFlavour/hm2_pci
```
You may get now a false postitive about the component not loaded but the component will be loaded
as you can check with `show comp` .

mhaberler, 2015-12-25:
logging on ubuntu _is_ working, the log output just happens to go to /var/log/syslog

luminize, 2015-12-27:
I haven't got a coredump file yet, but the log shows:
```
14:47:13 xw6600 msgd:0: ØMQ=4.0.5 czmq=2.2.0 protobuf=2.5.0 libwebsockets=<no version symbol> 
Dec 27 14:47:13 xw6600 msgd:0: configured: sha=cfa203d
Dec 27 14:47:13 xw6600 msgd:0: built:      Dec 24 2015 22:22:06 sha=cfa203d
Dec 27 14:47:13 xw6600 msgd:0: register_stuff: actual hostname as announced by avahi='xw6600.local'
Dec 27 14:47:13 xw6600 msgd:0: zeroconf: registering: 'Log service on xw6600.local pid 4778'
Dec 27 14:47:13 xw6600 msgd:0: rtapi_app:4785:user accepting commands at ipc:///tmp/0.rtapi.a42c8c6b-4025-4f83-ba28-dad21114744a
Dec 27 14:47:13 xw6600 msgd:0: hal_lib:4785:rt hm2: loading Mesa HostMot2 driver version 0.15
Dec 27 14:47:13 xw6600 msgd:0: hal_lib:4785:rt hm2_pci: loading Mesa AnyIO HostMot2 driver version 0.7
Dec 27 14:47:13 xw6600 msgd:0: hal_lib:4785:rt hm2_pci: discovered 5i20 at 0000:01:09.0
Dec 27 14:47:13 xw6600 msgd:0: rtapi_app:4785:user signal 11 - 'Segmentation fault' received, dumping core (current dir=/home/bas/hal-configuration)
Dec 27 14:47:13 xw6600 msgd:0: rtapi_app:4785:user  --- rtapi_app backtrace: ---
Dec 27 14:47:13 xw6600 msgd:0: rtapi_app:4785:user 7f22e1679ee5 inl              (/usr/include/x86_64-linux-gnu/sys/io.h:83)
Dec 27 14:47:13 xw6600 msgd:0: rtapi_app:4785:user 7f22e167a4d3 hm2_plx9030_fixu (hal/drivers/mesa-hostmot2/hm2_pci.c:377)
Dec 27 14:47:13 xw6600 msgd:0: rtapi_app:4785:user 7f22e167b3c3 hm2_pci_probe    (hal/drivers/mesa-hostmot2/hm2_pci.c:678)
Dec 27 14:47:13 xw6600 msgd:0: rtapi_app:4785:user 7f22e40bc070 pci_register_dri (rtapi/rtapi_pci.c:551)
Dec 27 14:47:13 xw6600 msgd:0: rtapi_app:4785:user 7f22e167b847 rtapi_app_main   (hal/drivers/mesa-hostmot2/hm2_pci.c:806)
Dec 27 14:47:13 xw6600 msgd:0: rtapi_app:4785:user 004073a0 do_load_cmd      (rtapi/posix/rtapi_app.cc:571)
Dec 27 14:47:13 xw6600 msgd:0: rtapi_app:4785:user 00409a92 rtapi_request    (rtapi/posix/rtapi_app.cc:837)
Dec 27 14:47:13 xw6600 msgd:0: rtapi_app:4785:user 7f22e645f9cb ?? ??:0
Dec 27 14:47:13 xw6600 msgd:0: rtapi_app:4785:user 00408644 mainloop         (rtapi/posix/rtapi_app.cc:1286)
Dec 27 14:47:13 xw6600 msgd:0: rtapi_app:4785:user 0040411f main             (rtapi/posix/rtapi_app.cc:1643)
Dec 27 14:47:13 xw6600 msgd:0: rtapi_app:4785:user 7f22e4dfdec4 ?? ??:0
Dec 27 14:47:13 xw6600 msgd:0: rtapi_app:4785:user 404208 ?? ??:0
Dec 27 14:47:13 xw6600 msgd:0: rtapi_app:4785:user ffffffffffffffff ?? ??:0
Dec 27 14:47:13 xw6600 msgd:0: rtapi_app:4785:user  --------------------
Dec 27 14:47:13 xw6600 msgd:0: rtapi_app exit detected - scheduled shutdown
Dec 27 14:47:14 xw6600 msgd:0: zeroconf: registered 'Log service on xw6600.local pid 4778' _machinekit._tcp 0 TXT "uuid=a42c8c6b-4025-4f83-ba28-dad21114744a" "instance=555a5828-aca0-11e5-9f5f-001f2900ec6f" "service=log" "dsn=ipc:///tmp/0.log.a42c8c6b-4025-4f83-ba28-dad21114744a"
Dec 27 14:47:15 xw6600 msgd:0: msgd shutting down
Dec 27 14:47:15 xw6600 msgd:0: zeroconf: unregistering 'Log service on xw6600.local pid 4778'
Dec 27 14:47:15 xw6600 msgd:0: log buffer hwm: 0% (14 msgs, 1238 bytes out of 524288)
Dec 27 14:47:15 xw6600 msgd:0: normal shutdown - global segment detached
```
I did some searching and the current hmi_pci.c in Machinekit https://github.com/machinekit/machinekit/commit/89b393c585c9b78c95afddc42d4012213288fea9 differs from the one in the jepler/linuxcnc mirror https://github.com/jepler/linuxcnc-mirror/commit/67fe0c2c4a71d939a864fc14d649590aeeb0db26

mhaberler, 2015-12-27:
well if you look at the call stack you will find https://github.com/machinekit/machinekit/blob/master/src/hal/drivers/mesa-hostmot2/hm2_pci.c#L377 does have an IO operation, which is privileged

also, the pathname suggests this is flavor posix - arent you trying to run FLAVOR=rt-preempt?
sudo make setuid done?

a key line is missing of the log, like this one, which shows the flavor:
`Dec 26 13:32:38 jessie64 msgd:0: rtapi_app:12341:user pid=12341 flavor=posix gcc=4.9.2 git=v0.1~task-zmq-cmd-channel~0afdd90`

the code difference is a function rename

luminize, 2015-12-27:
i'm running this on Ubuntu because of ROS #689 
I'll reboot into Jessie where I have an RT-PREEMPT kernel and have a try

mhaberler, 2015-12-27:
it is not a kernel issue, it is a question if the rtapi_app_whatever is setuid root - it needs to be to do direct I/O to hardware

luminize, 2015-12-27:
yes I did `sudo make setuid`

luminize, 2015-12-27:
this is on jessie now, but the owner is not root, but bas
```
bas@debian:~$ ls -l machinekit/rtlib/rt-preempt/hm2_pci.so 
-rwxr-xr-x 1 bas bas 43760 Dec 27 15:36 machinekit/rtlib/rt-preempt/hm2_pci.so
```

luminize, 2015-12-28:
on jessie I've manually copied the `.BIT` file I downloaded to `/lib/firmware/hm2/5i20` as root
```
bas@debian:~$ ls -l /lib/firmware/hm2/5i20/
total 164
-rw-r--r-- 1 root root 167061 Dec 28 10:06 SVST8_4.BIT
```
and that results in correct startup.
However, doing the same in Ubuntu still does not work.

luminize, 2015-12-30:
this issue was solved by manually uploading the firmware with mesaflash. We couldn't figure out why the card worked on Debian Jessie and not on Ubuntu.
So closing the issue.

closed 2015-12-30 by luminize