machinekit/machinekit #847: mkwrapper generates SIGABRT randomly on Raspberry Pi2

yishinli, 2015-12-28:
Machinekit branch: master
MachikekitClient: machinekit-client_v0.9.6-112_linux_x64.tar.gz
Machineface branch: master
Config Settings: [qqvsim](https://github.com/araisrobo/qqvsim)
Machinekit running on RPi2 with minibian-jessie.

After machine-on with MachinekitClient, then start "Hal Configuration" to watch more than 20 hal signals, the mkwrapper.py is likely to get terminated in around 20 minutes. 

linuxcnc.log is attached; here's the core dump trace:
```
$ gdb `which rtapi_app_posix` core 
GNU gdb (Raspbian 7.7.1+dfsg-5) 7.7.1
This GDB was configured as "arm-linux-gnueabihf".
Reading symbols from /home/pi/proj/machinekit.master/libexec/rtapi_app_posix...done.
warning: core file may not match specified executable file.
[New LWP 2397]
[2015-12-28.001-linuxcnc.log.zip](https://github.com/machinekit/machinekit/files/73036/2015-12-28.001-linuxcnc.log.zip)

[New LWP 2398]
[New LWP 2386]
[New LWP 2390]
[New LWP 2391]
[New LWP 2393]
[New LWP 2389]
Core was generated by `/usr/bin/python /home/pi/proj/machinekit.master/bin/mkwrapper -ini /home/pi/pro'.
Program terminated with signal SIGABRT

, Aborted.
#0  0x76ce9f50 in ?? ()
```

Please suggest approaches about digging into this issue further.

mhaberler, 2015-12-28:
mkwrapper is a python script, and the core dump says so:
````
Core was generated by `/usr/bin/python /home/pi/proj/machinekit.master/bin/mkwrapper -ini /home/pi/pro'.
Program terminated with signal SIGABRT
````
so the proper gdb incantation is
`````
gdb /usr/bin/python <corefile>
`````

that said, such abort signals are typical for a mismatch between compiler and architecture, for instance running arm7 code on an arm6 platform

let's see what the backtrace says

yishinli, 2015-12-29:
The backtrace says:
```
(gdb) bt
#0  0x76ce9f50 in __GI_raise (sig=sig@entry=6) at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
#1  0x76ceb304 in __GI_abort () at abort.c:89
#2  0x769cac28 in ?? () from /usr/lib/arm-linux-gnueabihf/libzmq.so.3
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

Should I better look further into Thread-1 or Thread-2 ?

```
$ gdb /usr/bin/python 2015-12-28.001-core
GNU gdb (Raspbian 7.7.1+dfsg-5) 7.7.1
Reading symbols from /usr/bin/python...Reading symbols from /usr/lib/debug//usr/bin/python2.7...done.
done.
Illegal process-id: 2015-12-28.001-core.

warning: core file may not match specified executable file.
[New LWP 2397]
[New LWP 2398]
[New LWP 2386]
[New LWP 2390]
[New LWP 2391]
[New LWP 2393]
[New LWP 2389]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
Core was generated by `/usr/bin/python /home/pi/proj/machinekit.master/bin/mkwrapper -ini /home/pi/pro'.
Program terminated with signal SIGABRT, Aborted.
#0  0x76ce9f50 in __GI_raise (sig=sig@entry=6) at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
56	../nptl/sysdeps/unix/sysv/linux/raise.c: No such file or directory.
```

```
thread apply all where 4

Thread 7 (Thread 0x759c4460 (LWP 2389)):
#0  0x76d8a7a4 in epoll_wait () at ../sysdeps/unix/syscall-template.S:81
#1  0x001d91dc in pyepoll_poll.lto_priv.3053 (self=0x76cba580, args=<optimized out>, kwds=<optimized out>)
    at ../Modules/selectmodule.c:1034
#2  0x0008376c in call_function (oparg=<optimized out>, pp_stack=0x759c36b8) at ../Python/ceval.c:4033
#3  PyEval_EvalFrameEx (f=f@entry=
    Frame 0x75a1e030, for file /usr/lib/python2.7/dist-packages/pyftpdlib/ioloop.py, line 491, in poll (self=<Epoll(_poller=<select.epoll at remote 0x76cba580>, sched=<_Scheduler(_tasks=[], _cancellations=0) at remote 0x759f35f0>, socket_map={6: <FTPServer(ioloop=<...>, max_cons_per_ip=5, socket=<_socketobject at remote 0x75a0b260>, family_and_type=(2, 1), _map={}, _af=2, max_cons=256, handler=<classobj at remote 0x759ef1f0>, _fileno=6, accepting=True, ip_map=[], backlog=5, addr=('0.0.0.0', 46897)) at remote 0x759eebc0>}) at remote 0x759f31d0>, timeout=1), throwflag=throwflag@entry=0) at ../Python/ceval.c:2679
(More stack frames follow...)

Thread 6 (Thread 0x73dff460 (LWP 2393)):
#0  0x76ecea40 in do_futex_wait (isem=isem@entry=0x73f004a8) at ../nptl/sysdeps/unix/sysv/linux/sem_wait.c:48
#1  0x76eceaf4 in __new_sem_wait (sem=sem@entry=0x73f004a8) at ../nptl/sysdeps/unix/sysv/linux/sem_wait.c:69
#2  0x00151590 in PyThread_acquire_lock (waitflag=1, lock=0x73f004a8) at ../Python/thread_pthread.h:324
#3  lock_PyThread_acquire_lock.lto_priv.2556 (self=0x76cba6c0, args=<optimized out>) at ../Modules/threadmodule.c:52
(More stack frames follow...)

Thread 5 (Thread 0x747ff460 (LWP 2391)):
#0  0x76ed02f4 in send () at ../sysdeps/unix/syscall-template.S:81
#1  0x769f0a64 in ?? () from /usr/lib/arm-linux-gnueabihf/libzmq.so.3
#2  0x769f0cc8 in ?? () from /usr/lib/arm-linux-gnueabihf/libzmq.so.3
#3  0x769e70b4 in ?? () from /usr/lib/arm-linux-gnueabihf/libzmq.so.3
(More stack frames follow...)

Thread 4 (Thread 0x74fff460 (LWP 2390)):
#0  0x76d8a7a4 in epoll_wait () at ../sysdeps/unix/syscall-template.S:81
#1  0x769ca7bc in ?? () from /usr/lib/arm-linux-gnueabihf/libzmq.so.3
---Type <return> to continue, or q <return> to quit---
#2  0x769f66b8 in ?? () from /usr/lib/arm-linux-gnueabihf/libzmq.so.3
#3  0x76ec7e90 in start_thread (arg=0x74fff460) at pthread_create.c:311
(More stack frames follow...)

Thread 3 (Thread 0x76f24000 (LWP 2386)):
#0  0x76d83514 in select () at ../sysdeps/unix/syscall-template.S:81
#1  0x001a4bcc in floatsleep (secs=<optimized out>) at ../Modules/timemodule.c:948
#2  time_sleep.lto_priv.2652 (self=<optimized out>, args=<optimized out>) at ../Modules/timemodule.c:206
#3  0x0008376c in call_function (oparg=<optimized out>, pp_stack=0x7ebdb7b8) at ../Python/ceval.c:4033
(More stack frames follow...)

Thread 2 (Thread 0x72bff460 (LWP 2398)):
#0  convertsimple.lto_priv.1427 (arg='\x08', p_format=0x72bfdfd0, p_format@entry=0x72bfdfc8, p_va=0x72bfe180, 
    p_va@entry=0xbb004 <super_init.lto_priv.160+68>, flags=0, flags@entry=1964615000, msgbuf=msgbuf@entry=0x72bfe054 "\204\206\a", 
    bufsize=bufsize@entry=256, freelist=0x72bfdfcc, freelist@entry=0x72bfdfc4) at ../Python/getargs.c:583
#1  0x00078954 in convertitem (freelist=0x72bfdfc4, bufsize=256, msgbuf=0x72bfe054 "\204\206\a", levels=0x72bfdfcc, 
    flags=<optimized out>, p_va=<optimized out>, p_format=<synthetic pointer>, arg=<optimized out>) at ../Python/getargs.c:514
#2  vgetargs1.lto_priv.1814 (args=<optimized out>, format=<optimized out>, 
    format@entry=0x75195720 "T", <incomplete sequence \315>, p_va=0xbb004 <super_init.lto_priv.160+68>, p_va@entry=0x72bfe160, 
    flags=1964615000, flags@entry=0) at ../Python/getargs.c:345
#3  0x00078684 in PyArg_ParseTuple (args=<optimized out>, format=0x224888 "s*:write") at ../Python/getargs.c:85
(More stack frames follow...)

Thread 1 (Thread 0x735ff460 (LWP 2397)):
#0  0x76ce9f50 in __GI_raise (sig=sig@entry=6) at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
#1  0x76ceb304 in __GI_abort () at abort.c:89
#2  0x769cac28 in ?? () from /usr/lib/arm-linux-gnueabihf/libzmq.so.3
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

yishinli, 2015-12-29:
After recompile Machinekit with python-dbg and Boost.Python in debug mode, the core dump shows the error is related to zmq/backend/cython/socket.*.so. 
```
$ gdb /usr/bin/python-dbg core
GNU gdb (Raspbian 7.7.1+dfsg-5) 7.7.1
Reading symbols from /usr/bin/python-dbg...done.
warning: core file may not match specified executable file.
[New LWP 10993]
[New LWP 10981]
[New LWP 10984]
[New LWP 10988]
[New LWP 10986]
[New LWP 10992]
[New LWP 10985]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
Core was generated by `/usr/bin/python-dbg /home/pi/proj/machinekit/bin/mkwrapper -ini /home/pi/proj/k'.
Program terminated with signal SIGSEGV, Segmentation fault.
Cannot access memory at address 0x0
#0  0x7696c8cc in ?? () from /usr/lib/arm-linux-gnueabihf/libzmq.so.3
(gdb) info threads
  Id   Target Id         Frame 
  7    Thread 0x747ff460 (LWP 10985) 0x76d6f7a4 in epoll_wait () at ../sysdeps/unix/syscall-template.S:81
  6    Thread 0x72dff460 (LWP 10992) 0x76d65730 in poll () at ../sysdeps/unix/syscall-template.S:81
  5    Thread 0x73fff460 (LWP 10986) 0x76d6f7a4 in epoll_wait () at ../sysdeps/unix/syscall-template.S:81
  4    Thread 0x735ff460 (LWP 10988) 0x76eb3a40 in do_futex_wait (isem=isem@entry=0x73700480) at ../nptl/sysdeps/unix/sysv/linux/sem_wait.c:48
  3    Thread 0x751ca460 (LWP 10984) 0x76d6f7a4 in epoll_wait () at ../sysdeps/unix/syscall-template.S:81
  2    Thread 0x76f0b000 (LWP 10981) 0x76d68514 in select () at ../sysdeps/unix/syscall-template.S:81
* 1    Thread 0x723ff460 (LWP 10993) Cannot access memory at address 0x0
0x7696c8cc in ?? () from /usr/lib/arm-linux-gnueabihf/libzmq.so.3
(gdb) thread 1
[Switching to thread 1 (Thread 0x723ff460 (LWP 10993))]
#0  0x7696c8cc in ?? () from /usr/lib/arm-linux-gnueabihf/libzmq.so.3
(gdb) bt
#0  0x7696c8cc in ?? () from /usr/lib/arm-linux-gnueabihf/libzmq.so.3
#1  0x769755c4 in ?? () from /usr/lib/arm-linux-gnueabihf/libzmq.so.3
#2  0x7699aa5c in ?? () from /usr/lib/arm-linux-gnueabihf/libzmq.so.3
#3  0x7698bfb4 in ?? () from /usr/lib/arm-linux-gnueabihf/libzmq.so.3
#4  0x769a1740 in ?? () from /usr/lib/arm-linux-gnueabihf/libzmq.so.3
Cannot access memory at address 0x0
#5  0x766c3304 in ?? () from /usr/lib/python2.7/dist-packages/zmq/backend/cython/socket.arm-linux-gnueabihf_d.so
Cannot access memory at address 0x0
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```
I am going to recompile libzmq from source code, and do another test.
Advices are appreciated.

yishinli, 2015-12-29:
After installing libzmq3-dbg, there's more details:
```
(gdb) info threads
  Id   Target Id         Frame 
  7    Thread 0x747ff460 (LWP 10985) 0x76d6f7a4 in epoll_wait () at ../sysdeps/unix/syscall-template.S:81
  6    Thread 0x72dff460 (LWP 10992) 0x76d65730 in poll () at ../sysdeps/unix/syscall-template.S:81
  5    Thread 0x73fff460 (LWP 10986) 0x76d6f7a4 in epoll_wait () at ../sysdeps/unix/syscall-template.S:81
  4    Thread 0x735ff460 (LWP 10988) 0x76eb3a40 in do_futex_wait (isem=isem@entry=0x73700480) at ../nptl/sysdeps/unix/sysv/linux/sem_wait.c:48
  3    Thread 0x751ca460 (LWP 10984) 0x76d6f7a4 in epoll_wait () at ../sysdeps/unix/syscall-template.S:81
  2    Thread 0x76f0b000 (LWP 10981) 0x76d68514 in select () at ../sysdeps/unix/syscall-template.S:81
* 1    Thread 0x723ff460 (LWP 10993) 0x7696c8cc in index (this=0x21999dc, item_=0x1) at array.hpp:138
(gdb) thread 1
[Switching to thread 1 (Thread 0x723ff460 (LWP 10993))]
#0  0x7696c8cc in index (this=0x21999dc, item_=0x1) at array.hpp:138
138	in array.hpp
(gdb) bt
#0  0x7696c8cc in index (this=0x21999dc, item_=0x1) at array.hpp:138
#1  zmq::dist_t::match (this=0x21999dc, pipe_=0x1) at dist.cpp:60
#2  0x769755c4 in zmq::mtrie_t::match (this=this@entry=0x21999cc, data_=0x723fd9a6 "", data_@entry=0x723fd9a0 "interp", size_=<optimized out>, func_=
    0x7699a9c4 <zmq::xpub_t::mark_as_matching(zmq::pipe_t*, void*)>, arg_=arg_@entry=0x2199650) at mtrie.cpp:394
#3  0x7699aa5c in zmq::xpub_t::xsend (this=0x2199650, msg_=0x723fd9a0) at xpub.cpp:128
#4  0x7698bfb4 in zmq::socket_base_t::send (this=this@entry=0x2199650, msg_=msg_@entry=0x723fd9a0, flags_=flags_@entry=3) at socket_base.cpp:716
#5  0x769a1740 in s_sendmsg (Cannot access memory at address 0x0
s_=0x2199650, msg_=0x723fd9a0, flags_=3) at zmq.cpp:350
#6  0x766c3304 in ?? () from /usr/lib/python2.7/dist-packages/zmq/backend/cython/socket.arm-linux-gnueabihf_d.so
Cannot access memory at address 0x0
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

mhaberler, 2015-12-29:
well but that thread is not the one which received the signal? the symptom looks like an assert() in the zmq library

difficult to help - this is a lot of moving parts; any chance you can reproduce on a stock install (like from packages on a 386?)

yishinli, 2015-12-30:
The crash does not happen in i386 machines. We only observed it on RPi2.
Yes, we did see zmq Mailbox assertion messages. The following issue might be relevant:
[assertion failure in mailbox.cpp:82 #1108](https://github.com/zeromq/libzmq/issues/1108)

I doubt there was "disconnect event occurs on RPi2".

mhaberler, 2015-12-30:
This is a wild guess, but worth a try:

- atomics and cmpxch seems to be involved
- the rpi compilers default to arm6 code generation for rpi1 compatibility
- atomics on arm6 are seriously hampered

maybe you want to rebuild by forcing arm7 code generation? I had to do this for the disentangle branch to support 64bit atomics

see these flags: https://github.com/mhaberler/machinekit/blob/disentangle/src/configure.ac#L629-L631

NB the key flags are -march=armv7-a and -ffast-math, the others may not be valid for the rpi2

yishinli, 2015-12-31:
I tried compiling with "-march=armv7-a -mfpu=neon-vfpv4 -ffast-math".
Still, there's "Assertion failed: ok (mailbox.cpp:82)".

yishinli, 2016-01-01:
Issues seems resolved; has been running for more than 3 hours without ZMQ assertion.
[mkwrapper.py modifications:](https://github.com/araisrobo/machinekit/commit/cb388f5be13f044bf4bd17f638b33b08e7436e6e)
* use single thread to synchronize send() and recv() of the zmq socket
* merge LinuxCNCWrapper.poll() to LinuxCNCWrapper.process_sockets()
* resolved the "mkwrapper generates SIGABRT randomly on Raspberry Pi2"

yishinli, 2016-01-03:
@strahlex - I did a git-rebase on my branch, and accidentally over-wrote your comments about my previous commit. Here's the updated [commit](https://github.com/araisrobo/machinekit/commit/28c97e24d0b5f6387eb8cb8380b7fe743eed57c2). Please take a look.
* use single thread to synchronize send() and recv() of the zmq socket
* merge LinuxCNCWrapper.poll() to LinuxCNCWrapper.process_sockets()
* resolved the "mkwrapper generates SIGABRT randomly on Raspberry Pi2"
* prevent process_sockets() from polling earlier than poll-interval

strahlex, 2016-01-04:
Looks better now. Can you make a PR please.

yishinli, 2016-01-04:
PR submitted

closed 2016-01-04 by yishinli