machinekit/machinekit #851: M109 for new config files

electronlove, 2015-12-31:
Came across an issue with pin assignment for M109 that stalls a printer after setting the feed rate.  "halcmd getp Therm.temp0" should instead be "halcmd getp Therm.temp0" with the newer config files.

luminize, 2016-01-04:
@electronlove could you be a bit more specific? is this in the configuration files? All? Some?

electronlove, 2016-01-04:
I only came across this when implementing the older M109 code where it uses a slightly older pin assignment "Therm.temp0" insread of the newer "Therm.ch-04.value". It looked like it was in just the M109 file since it uses the PID setup in the hal file

luminize, 2016-01-04:
I was a bit curious so I did a search for ["Therm.temp0"](https://github.com/machinekit/machinekit/search?p=1&q=Therm.temp0&utf8=%E2%9C%93) and ["Therm.ch-04.value"](https://github.com/machinekit/machinekit/search?utf8=%E2%9C%93&q=Therm.ch-04.value) . The results shows some differences.

electronlove, 2016-01-04:
Hi.  It looks like the change was only implemented for the CRAMPS and Replicape board.  Is the goal to move towards the g-code remappings then ?

strahlex, 2016-01-04:
A common base for all 3D printer configs would be preferable. However, nobody has made an attempt to convert all existing configs so far.

mhaberler, 2016-01-07:
"Came across an issue" is not a problem description. Please specify or close the issue.

closed 2016-01-07 by electronlove