machinekit/machinekit #854: Galileo 2 and MachineKit

jpratt3000, 2016-01-04:
Has anyone tried using this on a Intel Galileo 2 board yet?

unseenlaser, 2016-01-05:
i'm not aware of one , and looking at the spec's of the board , i doubt it
ever will .

-- 

The information contained in this message is confidential and is intended
for the addressee only. If you have received this message in error or there
are any problems please notify the originator immediately. The unauthorised
use, disclosure, copying or alteration of this message is strictly
forbidden. This mail and any attachments have been scanned for viruses
prior to leaving the RcTechnix network. RcTechnix will not be liable for
direct, special, indirect or consequential damages arising from alteration
of the contents of this message by a third party or as a result of any
virus being passed on.

RcTechnix reserves the right to monitor and record e-mail messages being
sent to and from this address for the purposes of investigating or
detecting any unauthorised use of its system and ensuring effective
operation.

(c) RcTechnix

mhaberler, 2016-01-05:
before trying, have a look at http://www.intel.com/content/www/us/en/embedded/products/galileo/galileo-overview.html

Do you see the magic "i2c" word between PWM, GPIO and CPU? some parts of this might be a real dog when it comes to I/O latency - the early Intel embedded boards were next to ridiculous due to this feature (> 100mS GPIO latency reported), not sure how this panned out

jpratt3000, 2016-01-05:
wind river open sourced a subset of their vxworks and are calling is rocket, so i thought i'd give it a go to see if it works or not.

there was a few things on the machinekit blog that there is a lot of unneeded items in the repo, if i forked the current master is there a list of items that are not needed?

mhaberler, 2016-01-07:
closing since this is not a problem issue . Please stick to the mailing list for questions.

closed 2016-01-07 by mhaberler