machinekit/machinekit #856: Building from debian packages for wheezy 3.8-1-xenomai.x86-amd64

sirop, 2016-01-07:
Follow these build instructions https://github.com/machinekit/machinekit/issues/840#issuecomment-169318259  on debian wheezy with 3.8-1-xenomai.x86-amd64  in order to reproduce:
```
$ halrun
msgd:0 stopped
rtapi:0 stopped
halcmd: hal_init() failed: -22
NOTE: 'rtapi' module must be loaded
```
Consider the following comments till https://github.com/machinekit/machinekit/issues/840#issuecomment-169772741

ArcEye, 2016-01-08:
Can you try building a RIP install on the Wheezy partition?
I will be able to try also later this afternoon.

I suspect it will work, doubt the problem is the kernel, rather something amiss in the machinekit package.

If that works, you can simply remove all the machinekit packages, clean the RIP and do a user install
( ./configure --prefix=/usr ) if working from a RIP is inconvenient.

You will need the development packages at https://github.com/machinekit/machinekit/issues/840#issuecomment-169772741 to build the RIP

sirop, 2016-01-08:
@ArcEye 

I tried to build the RIP version.
```
~/machinekit$ which halrun
/home/slave/machinekit/scripts/halrun
~/machinekit$ /home/slave/machinekit/scripts/halrun
msgd:0 stopped
rtapi:0 stopped
halcmd: hal_init() failed: -22
NOTE: 'rtapi' module must be loaded
halcmd: hal_init() failed: -22
```

The same error.
```
~/machinekit$ uname -a
Linux debian-slave 3.8-1-xenomai.x86-amd64 #1 SMP Debian 3.8.13-12~1wheezy~1da x86_64 GNU/Linux
~/machinekit$ git log | head -5
commit 84c8addfcba0b0583604e8f103b6c5d7ff8a8072
Merge: 6c4e9c2 5e2a604
Author: Alexander Rössler <mail@roessler.systems>
Date:   Mon Jan 4 10:53:54 2016 -0600

ArcEye, 2016-01-08:
My RIP build works fine, are you sure you set the environment in the shell you were in?
If not, you will just use the runtime from the system installed packages.

```
mick@Sempron:/usr/src/machinekit$ uname -a
Linux Sempron 3.8-1-xenomai.x86-amd64 $1 SMP Debian 3.8.13-12~1wheezy~1da x86_64 GNU/Linux

mick@Sempron:/usr/src/machinekit$ git log | head -5
commit 84c8addfcba0b0583604e8f103b6c5d7ff8a8072
Merge: 6c4e9c2 5e2a604
Author: Alexander Rössler <mail@roessler.systems>
Date:   Mon Jan 4 10:53:54 2016 -0600

mick@Sempron:/usr/src/machinekit$ halrun
msgd:0 stopped
rtapi:0 stopped
halcmd: newinst debounce db1 pincount=4
halcmd: show pin
Component Pins:
  Comp   Inst Type  Dir         Value  Name                             Epsilon         Flags
    73     74 bit   IN          FALSE  db1.0.in                      			0
    73     74 bit   OUT         FALSE  db1.0.out                     			0
    73     74 s32   I/O             0  db1.0.state                   			0
    73     74 bit   IN          FALSE  db1.1.in                      			0
    73     74 bit   OUT         FALSE  db1.1.out                     			0
    73     74 s32   I/O             0  db1.1.state                   			0
    73     74 bit   IN          FALSE  db1.2.in                      			0
    73     74 bit   OUT         FALSE  db1.2.out                     			0
    73     74 s32   I/O             0  db1.2.state                   			0
    73     74 bit   IN          FALSE  db1.3.in                      			0
    73     74 bit   OUT         FALSE  db1.3.out                     			0
    73     74 s32   I/O             0  db1.3.state                   			0
    73     74 s32   I/O             5  db1.delay                     			0
    73     74 s32   OUT             0  db1.time                      			0
```
Original problem presumably is in the machinekit-xenomai_0.1 or machinekit_0.1 packages for x86_amd64

I will try the 
http://deb.machinekit.io/pool/main/l/linux/linux-image-4.1.0-0.bpo.1-rt-amd64_4.1.3-1~bpo8+1_amd64.deb
package tomorrow and build against that

If that is OK, will try purging all the packages and running it just from packages on rt-preempt

zultron, 2016-01-08:
@sirop, can you try uninstalling all the `machinekit-*` packages from your system to ensure there's no conflict with the RIP install?

sirop, 2016-01-08:
@ArcEye 
>My RIP build works fine, are you sure you set the environment in the shell you were in?  

Do you mean `~/machinekit$ . scripts/rip-environment` ? 
Yes, I did it.

@zultron 
I did 
```
~/machinekit$ sudo apt-get purge machinekit-xenomai machinekit-posix  machinekit-rt-preempt
Reading package lists... Done
Building dependency tree
Reading state information... Done
Package 'machinekit-posix' is not installed, so not removed
Package 'machinekit-rt-preempt' is not installed, so not removed
Package 'machinekit-xenomai' is not installed, so not removed
0 upgraded, 0 newly installed, 0 to remove and 7 not upgraded.
```
before the installation.

zultron, 2016-01-08:
How about 

    sudo apt-get purge machinekit machinekit-dev

sirop, 2016-01-08:
Neither there:
```
~/machinekit$ sudo apt-get purge machinekit machinekit-dev
[sudo] password for slave:
Reading package lists... Done
Building dependency tree
Reading state information... Done
Package 'machinekit' is not installed, so not removed
Package 'machinekit-dev' is not installed, so not removed
0 upgraded, 0 newly installed, 0 to remove and 7 not upgraded.
```

ArcEye, 2016-01-09:
Have now built a RIP on xenomai and rt-preempt under Wheezy, both as a user and root.

Cannot reproduce your error on a RIP build and suspect that whatever apt says you must have a remnant of the machinekit-xenomai package still installed to get the same error.

```
mick@Sempron:~/src/machinekit$ uname -a
Linux Sempron 3.2.0-4-rt-amd64 #1 SMP PREEMPT RT Debian 3.2.73-2+deb7u2 x86_64 GNU/Linux

mick@Sempron:~/src/machinekit$ halrun
msgd:0 stopped
rtapi:0 stopped
halcmd: newinst xor2 xr
halcmd: show pin
Component Pins:
  Comp   Inst Type  Dir         Value  Name                             Epsilon         Flags
    73     74 bit   IN          FALSE  xr.in0                        			0
    73     74 bit   IN          FALSE  xr.in1                        			0
    73     74 bit   OUT         FALSE  xr.out                        			0
    73     74 s32   OUT             0  xr.time                       			0
```

FYI, on this machine latency on servo thread under xenomai is 160K and under rt-preempt is 48K, which is why I stopped using xenomai a long time ago.

I have also installed the machinekit-rt-preempt package and run from the package user system install and that works too.

**The suspect in the original problem is the machinekit-xenomai package therefore I suggest.**

Can only suggest you wipe your partition, do a fresh install and either use rt-preempt, or log each stage of the xenomai install, so that it is reproducible if a RIP build still does not work.
All the problems I am aware of with the RIP build are to do with zmq / czmq versions etc and will prevent a build, not produce errors after a successful build

regards

sirop, 2016-01-09:
>Cannot reproduce your error on a RIP build and suspect that whatever apt says you must have a   **remnant of the machinekit-xenomai package** still installed to get the same error.  

It is also my suspicion. I'll do a full new install of debian and so on somewhen next week.  
>FYI, on this machine latency on servo thread under xenomai is 160K and under rt-preempt is 48K, which is why I stopped using xenomai a long time ago.  

Guess you mean Max Jitter (ns) ? And you started  `latency-test` without any additonal options?
On my machine with the machinekit-xenomai of Nov 1  it is about 20K ns with some load due to an active x11vnc server connection.
If I watch an mp4 video, the Max Jitter is about 27K ns  but then the vnc connection stops.
`sudo lshw -short` output for better hardware comparison: https://gist.github.com/sirop/47d19d9e2da3039e93cb  
Have to confess that I have not yet solved the SMI issue on this machine. So if the SMI issue could be solved to some degree, the max jitter would be lower then.  

What really keeps me attached to xenomai is that I donot know if and how RTDM was ported
to rt-preempt.  

Many thanks  for your help.

ArcEye, 2016-01-09:
>  Guess you mean Max Jitter (ns) ? And you started latency-test without any additonal options?

Yes max jitter, I ran the test with both no args to just get servo thread and with `latency-test 25us 1ms`
to get a fast base thread, which often produces much better results on the servo thread

Because the processor is just a dual core AMD sempron, I did not get the cache-effect improvement often seen with newer multi-processor machines when running the second option.

Hopefully a fresh install will sort it when you have time

regards

mhaberler, 2016-01-11:
is this still an issue? if so - which one? the headline does not reflect any

if not - please close

ArcEye, 2016-01-11:
Issue is that the Wheezy machinekit-xenomai-x86_amd64 package does not run.
Appears complete but rtapi.so does not load when realtime started.

A RIP build using the same Wheezy xenomai-x86_amd64 kernel package works fine, as does a complete package install for x86 using the corresponding Wheezy machinekit-xenomai-x86-pae package.

sirop, 2016-01-11:
@mhaberler 
At least last week it was possible to  use debian packages for wheezy 3.8-1-xenomai.x86-amd64,
but there was an error
```
$ halrun
msgd:0 stopped
rtapi:0 stopped
halcmd: hal_init() failed: -22
NOTE: 'rtapi' module must be loaded
```
when one started  `halrun` as both ArcEye and I proved. 

If built from sources  for wheezy 3.8-1-xenomai.x86-amd64, `halrun` did not yield any errors when started - so in ArcEye's case. **In my case** some remnants of machinekit-xenomai packages are probably to blame as the above mentioned error does come up again.

*I'll try to edit the headline to be more specific.*

sirop, 2016-01-17:
Sorry to say, but  the RIP build does not get better on a fresh installed wheezy.
`make` stops with the error message:
```
Linking ../rtlib/xenomai/message.so
Depending halcomp-srcs/hal/components/rtfault.c
Compiling realtime halcomp-srcs/hal/components/rtfault.c
hal/components/rtfault.comp: In function ‘_’:
hal/components/rtfault.comp:34:2: error: implicit declaration of function ‘printf’ [-Werror=implicit-function-declaration]
hal/components/rtfault.comp:34:2: warning: incompatible implicit declaration of built-in function ‘printf’ [enabled by default]
cc1: some warnings being treated as errors
make[1]: *** [objects/xenomai/halcomp-srcs/hal/components/rtfault.o] Error 1
make[1]: Leaving directory `/home/slave/machinekit/src'
make: *** [modules] Error 2
make: Leaving directory `/home/slave/machinekit/src'
```

Some basics about my system:
```
slave@debian-slave:~/machinekit/src$ uname -a
Linux debian-slave 3.8-1-xenomai.x86-amd64 #1 SMP Debian 3.8.13-12~1wheezy~1da x86_64 GNU/Linux

slave@debian-slave:~/machinekit/src$ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/4.7/lto-wrapper
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Debian 4.7.2-5' --with-bugurl=file:///usr/share/doc/gcc-4.7/README.Bugs --enable-languages=c,c++,go,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-4.7 --enable-shared --enable-linker-build-id --with-system-zlib --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --with-gxx-include-dir=/usr/include/c++/4.7 --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --enable-gnu-unique-object --enable-plugin --enable-objc-gc --with-arch-32=i586 --with-tune=generic --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 4.7.2 (Debian 4.7.2-5)

slave@debian-slave:~/machinekit/src$ git log | head -5
commit 5257e2826d1698bd40b85d770be4aed72f52cacc
Merge: 84c8add bb4d58c
Author: Bas de Bruijn <bdebruijn@luminize.nl>
Date:   Tue Jan 12 15:32:01 2016 +0100
```

`./configure` does not produce any errros, but *config.log* does contain some errors,
for example:
```
configure:7056: checking for ANSI C header files
configure:7076: gcc -c -g -O2 -I/usr/include/xenomai -D_GNU_SOURCE -D_REENTRANT -Wall -Werror-implicit-function-declaration -pipe -D__XENO__  -I/usr/include/xenomai -D_GNU_SOURCE -D_REENTRANT -Wall -Werror-implicit-function-declaration -pipe -D__XENO__ conftest.c >&5
configure:7076: $? = 0
configure:7149: gcc -o conftest -g -O2 -I/usr/include/xenomai -D_GNU_SOURCE -D_REENTRANT -Wall -Werror-implicit-function-declaration -pipe -D__XENO__  -I/usr/include/xenomai -D_GNU_SOURCE -D_REENTRANT -Wall -Werror-implicit-function-declaration -pipe -D__XENO__  conftest.c -lnative  -lnative -L/usr/lib -lxenomai -lpthread -lrt >&5
configure:7149: $? = 0
configure:7149: ./conftest
Xenomai: binding failed: Operation not permitted.
configure:7149: $? = 1
configure: program exited with status 1
configure: failed program was:
| /* confdefs.h */
```
Full *config.log* : https://gist.github.com/sirop/bff6a7b2fe0a5bf81324  

I'll try to build on rt-preemt later today.

sirop, 2016-01-17:
**Building on rt-preempt**  

Although i did  only `debian/configure -pr` instead of  `debian/configure -px`

`./configure` yields:
```
checking whether to build Xenomai userland threads... yes
```
and `make` stops with
```
Depending halcomp-srcs/hal/components/rtfault.c
Compiling realtime halcomp-srcs/hal/components/rtfault.c
hal/components/rtfault.comp: In function ‘_’:
hal/components/rtfault.comp:34:2: error: implicit declaration of function ‘printf’ [-Werror=implicit-function-declaration]
hal/components/rtfault.comp:34:2: warning: incompatible implicit declaration of built-in function ‘printf’ [enabled by default]
cc1: some warnings being treated as errors
make[1]: *** [objects/xenomai/halcomp-srcs/hal/components/rtfault.o] Error 1
make[1]: Leaving directory `/home/slave/machinekit-rt-preempt/src'
make: *** [modules] Error 2
```

My system again:
```
slave@debian-slave:~/machinekit-rt-preempt$ uname -a
Linux debian-slave 3.2.0-4-rt-amd64 #1 SMP PREEMPT RT Debian 3.2.73-2+deb7u2 x86_64 GNU/Linux

slave@debian-slave:~/machinekit-rt-preempt$ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/4.7/lto-wrapper
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Debian 4.7.2-5' --with-bugurl=file:///usr/share/doc/gcc-4.7/README.Bugs --enable-languages=c,c++,go,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-4.7 --enable-shared --enable-linker-build-id --with-system-zlib --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --with-gxx-include-dir=/usr/include/c++/4.7 --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --enable-gnu-unique-object --enable-plugin --enable-objc-gc --with-arch-32=i586 --with-tune=generic --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 4.7.2 (Debian 4.7.2-5)

slave@debian-slave:~/machinekit-rt-preempt$ git log | head -5
commit 5257e2826d1698bd40b85d770be4aed72f52cacc
Merge: 84c8add bb4d58c
Author: Bas de Bruijn <bdebruijn@luminize.nl>
Date:   Tue Jan 12 15:32:01 2016 +0100
```

ArcEye, 2016-01-17:
>  Although i did only debian/configure -pr instead of debian/configure -px

There is no need to do debian/ anything, I have never understood the instructions that advocate it.

From src/ do a **make clean**
From / **git clean -xdf**

**cd src
./autogen.sh
./configure
make -j(core num plus 1)
make setuid
cd ../
. ./scripts/rip-environment**

rtfault is pretty much unused AFAIK

I don't know why you are getting that error, but the insertion as below should prevent it

```
component rtfault "generate RT faults on purpose";

pin in  bit fault "cause a series of count RT faults" ;
pin in  u32 count "number of faults";
pin in  u32 delay "ms delay on fault ⁽less than 1000)";

option singleton yes;
function _;
license "GPL";
;;

#include "rtapi.h"
#if defined(BUILD_SYS_USER_DSO)
#include <time.h>
```
#include \<stdio.h\>
```
#endif
#include <stdbool.h>
static bool previous_fault;
static int n;

FUNCTION(_) {
    if (previous_fault ^ fault) {
	n = count;
    }
    previous_fault = fault;  
    if (n) {
#if defined(BUILD_SYS_USER_DSO)
	// printf() eventually calls write(), which will cause
	// a Xenomai domain switch

	struct timespec nap = {0, delay * 1000 * 1000};
	if (delay)
	    nanosleep(&nap, NULL);
	printf("fault # %d\n", n--);
#else
	do {
	    rtapi_delay(delay * 1000 * 1000);
	    n--;
	} while (n > 0);
	rtapi_print_msg(RTAPI_MSG_ERR, "RT delay ..\n");
#endif
    }
}
```

sirop, 2016-01-17:
Thanks.
It's got better.
So I did:
```
cd src
make clean
cd  ../src  
git clean -xdf

cd src
./autogen.sh
./configure  --with-rt-preempt  --with-posix  --without-xenomai
make -j3
sudo make setuid
cd ../
. ./scripts/rip-environment
```
Then  
```
~/machinekit-rt-preempt$ /home/slave/machinekit-rt-preempt/scripts/halrun
msgd:0 stopped
rtapi:0 stopped
halcmd: cant connect to rtapi_app: -1 (uri= uuid=a42c8c6b-4025-4f83-ba28-dad21114744a): rtapi_rpc(): reply timeout

halcmd: the rtapi:0 RT demon is not running - please investigate /var/log/linuxcnc.log
halcmd: the msgd:0 logger demon is not running - please investigate /var/log/linuxcnc.log
halcmd: cant connect to rtapi_app: -1 (uri= uuid=a42c8c6b-4025-4f83-ba28-dad21114744a): rtapi_rpc(): reply timeout
```

```
~/machinekit-rt-preempt$ more /var/log/linuxcnc.log
Jan 17 06:11:18 debian-slave slave: logtest:1f61250ee7e2d4ac957e495bd952a847
Jan 17 06:12:06 debian-slave msgd:0: mlock(global) failed: 12 'Cannot allocate memory'
Jan 17 06:12:06 debian-slave msgd:0: Locked memory limit is 64KiB, recommended at least 20480KiB.
Jan 17 06:12:06 debian-slave msgd:0: This can cause the error 'Cannot allocate memory'.
Jan 17 06:12:06 debian-slave msgd:0: For more information, see http://wiki.linuxcnc.org/cgi-bin/emcinfo.pl?LockedMemory
...

~/machinekit-rt-preempt$ sudo sh -c "ulimit -l 20480 && exec su $LOGNAME"

~/machinekit-rt-preempt$ . ./scripts/rip-environment
slave@debian-slave:~/machinekit-rt-preempt$ halrun
msgd:0 stopped
rtapi:0 stopped
halcmd: cant connect to rtapi_app: -1 (uri= uuid=a42c8c6b-4025-4f83-ba28-dad21114744a): rtapi_rpc(): reply timeout

halcmd: the rtapi:0 RT demon is not running - please investigate /var/log/linuxcnc.log
halcmd: the msgd:0 logger demon is not running - please investigate /var/log/linuxcnc.log

~/machinekit-rt-preempt$ more /var/log/linuxcnc.log
Jan 17 06:42:36 debian-slave msgd:0: startup pid=22289 flavor=rt-preempt rtlevel=1 usrlevel=1 halsize=524288 shm=Posix gcc=4.7.2 version=unknown
Jan 17 06:42:36 debian-slave msgd:0: ØMQ=4.0.5 czmq=2.2.0 protobuf=2.4.1 libwebsockets=<no version symbol>
Jan 17 06:42:36 debian-slave msgd:0: configured: sha=5257e28
Jan 17 06:42:36 debian-slave msgd:0: built:      Jan 17 2016 06:01:54 sha=5257e28
Jan 17 06:42:36 debian-slave msgd:0: zeroconf: Failed to create avahi client object: Daemon not running
Jan 17 06:42:36 debian-slave msgd:0: zeroconf: unregistering 'Log service on debian-slave.local pid 22289'
Jan 17 06:42:36 debian-slave msgd:0: failed to start zeroconf  publisher for 'Log service on debian-slave.local pid 22289'
Jan 17 06:42:41 debian-slave rtapi:0: rtapi_app:0: ERROR: global segment magic not changing to ready: magic=0xeadbeef
Jan 17 06:42:41 debian-slave rtapi:0: rtapi:0: FATAL - failed to attach to global segment
```

I do not know how to deal with that. Would `sudo apt-get install avahi-daemon` suffice?

ArcEye, 2016-01-17:
It looks like you have not set up things after **make setuid**

That should call *scripts/check-system-configuration* which will output the lines you can cut and paste to set up ulimits, logging etc

Try opening up that file, use its contents to set up rsyslog, ulimits etc.  reboot and try again

ArcEye, 2016-01-17:
>  I do not know how to deal with that. Would sudo apt-get install avahi-daemon suffice?

When cloning a new partition, I capture the output of 
**ls -A1 /var/cache/apt/archives/ > packages.txt**
run on a working one and duplicate that on the new partition

This is my output for my Wheezy x86_amd64 rt-preempt partition
```
autoconf_2.69-1_all.deb
automake_1%3a1.11.6-1_all.deb
autopoint_0.18.1.1-9_all.deb
autotools-dev_20120608.1_all.deb
avahi-discover_0.6.31-2_all.deb
binutils_2.22-8+deb7u2_amd64.deb
blt_2.4z-4.2_amd64.deb
build-essential_11.5_amd64.deb
bwidget_1.9.5-1_all.deb
cmake_2.8.9-1_amd64.deb
cmake-data_2.8.9-1_all.deb
cython_0.19.1+git34-gac3e3a2-1~bpo70+1_amd64.deb
debhelper_9.20120909_all.deb
dpkg-dev_1.16.17_all.deb
dvipng_1.14-1+b1_amd64.deb
ed_1.6-2_amd64.deb
elyxer_1.2.3-1_all.deb
emacsen-common_2.0.5_all.deb
evince-common_3.4.0-3.1_all.deb
evince-gtk_3.4.0-3.1_amd64.deb
fakeroot_1.18.4-2_amd64.deb
firmware-linux-free_3.2_all.deb
freeglut3_2.6.0-4_amd64.deb
g++_4%3a4.7.2-1_amd64.deb
g++-4.7_4.7.2-5_amd64.deb
gcc_4%3a4.7.2-1_amd64.deb
gcc-4.7_4.7.2-5_amd64.deb
gccxml_0.9.0+cvs20120420-4_amd64.deb
gettext_0.18.1.1-9_amd64.deb
git_1%3a1.7.10.4-1+wheezy2_amd64.deb
git-man_1%3a1.7.10.4-1+wheezy2_all.deb
groff_1.21-9_amd64.deb
hostmot2-firmware-3x20-1_0.8.10.g3e97b6f_all.deb
hostmot2-firmware-4i65_0.8.10.g3e97b6f_all.deb
hostmot2-firmware-4i68_0.8.10.g3e97b6f_all.deb
hostmot2-firmware-5i20_0.8.10.g3e97b6f_all.deb
hostmot2-firmware-5i22-1_0.8.10.g3e97b6f_all.deb
hostmot2-firmware-5i22-1.5_0.8.10.g3e97b6f_all.deb
hostmot2-firmware-5i23_0.8.10.g3e97b6f_all.deb
hostmot2-firmware-7i43-2_0.8.10.g3e97b6f_all.deb
hostmot2-firmware-7i43-4_0.8.10.g3e97b6f_all.deb
hostmot2-firmware-all_0.8.10.g3e97b6f_all.deb
html2text_1.3.2a-15_amd64.deb
imagemagick_8%3a6.7.7.10-5+deb7u3_amd64.deb
intltool-debian_0.35.0+20060710.1_all.deb
javascript-common_7_all.deb
kernel-package_12.036+nmu3_all.deb
latex-beamer_3.10-2_all.deb
latex-xcolor_2.11-1.1_all.deb
libalgorithm-diff-perl_1.19.02-2_all.deb
libalgorithm-diff-xs-perl_0.04-2+b1_amd64.deb
libalgorithm-merge-perl_0.08-2_all.deb
libart-2.0-dev_2.3.21-2_amd64.deb
libatk1.0-dev_2.4.0-2_amd64.deb
libaudio2_1.9.3-5wheezy1_amd64.deb
libavahi-client-dev_0.6.31-2_amd64.deb
libavahi-common-dev_0.6.31-2_amd64.deb
libavahi-compat-libdnssd1_0.6.31-2_amd64.deb
libavahi-compat-libdnssd-dev_0.6.31-2_amd64.deb
libboost1.49-dev_1.49.0-3.2_amd64.deb
libboost-date-time1.49.0_1.49.0-3.2_amd64.deb
libboost-date-time1.49-dev_1.49.0-3.2_amd64.deb
[... 257 lines cut ...]
x11proto-kb-dev_1.0.6-2_all.deb
x11proto-randr-dev_1.3.2-2_all.deb
x11proto-render-dev_2%3a0.11.1-2_all.deb
x11proto-scrnsaver-dev_1.2.2-1_all.deb
x11proto-xext-dev_7.2.1-1_all.deb
x11proto-xinerama-dev_1.2.1-2_all.deb
xenomai-runtime_2.6.3-0.1mk~wheezy1_amd64.deb
xorg-sgml-doctools_1%3a1.10-1_all.deb
xtrans-dev_1.2.7-1_all.deb
zlib1g-dev_1%3a1.2.7.dfsg-13_amd64.deb
```

All I normally do is save it as a file called packages.txt and run a script like so
```
# !/bin/bash

    exec < ./packages.txt
		
    while read LINE ; do
	apt-get install -y $LINE ;
    done
```
To duplicate the packages required for a working build environment.

You can do that or cherry-pick the avahi packages as you wish.

sirop, 2016-01-17:
Thanks so far.

Now I can start start `halrun` on rt-preempt as RIP without errors.
I did some corrections for `sudo make setuid`. Do not know what happened there.
Maybe,  one reboot would have sufficed.  

```
slave@debian-slave:~/machinekit-rt-preempt$ ls -A1 /var/cache/apt/archives/ | grep avahi
avahi-daemon_0.6.31-2_amd64.deb
avahi-discover_0.6.31-2_all.deb
libavahi-client3_0.6.31-2_amd64.deb
libavahi-client-dev_0.6.31-2_amd64.deb
libavahi-common3_0.6.31-2_amd64.deb
libavahi-common-data_0.6.31-2_amd64.deb
libavahi-common-dev_0.6.31-2_amd64.deb
libavahi-core7_0.6.31-2_amd64.deb
python-avahi_0.6.31-2_amd64.deb
```
It seems I have overdone it and we do not need `avahi-daemon_0.6.31-2_amd64.deb`,
as you do not have it in your *packages.txt* .  

Anyway I stop "experimenting" for today and continue next week.

ArcEye, 2016-01-17:
>  Now I can start start halrun on rt-preempt as RIP without errors.

Glad you got running.

> I did some corrections for sudo make setuid. Do not know what happened there.
Maybe, one reboot would have sufficed. 

The rsyslog stuff does require that the daemon be stopped and restarted, so that might be part of it and why it worked after a reboot.

regards

mhaberler, 2016-02-14:
can this be closed now?

sirop, 2016-02-14:
I tried to install debian packages on wheezy 3.8-1-xenomai.x86-amd64 about a week ago,
this yielded the same error as before.

However, building from source works for both wheezy and jessie if there are no remnants of machinekit debian packets.

Therefore and wrt https://github.com/machinekit/machinekit/issues/863  I close this issue.

closed 2016-02-15 by sirop