machinekit/machinekit #863: package split undefined

mhaberler, 2016-01-18:
We [resolved to split up the machinekit package at the last meeting](http://blog.machinekit.io/2015/11/summary-of-open-discussion-at-nov-2015.html) but have not yet delivered on that decision. This issue is about organizing the split.

The determinants are:
- the future set of packages should enable smaller-footprint installations - like for instance  HAL-for-ROS-integration - without pulling in wads of package dependencies. 
- the legacy build system should be replaced with cmake. This permits introducing the new build method while the legacy build support is still in place and used, as cmake would be confined to a new build subdirectory.
- we should organize the split such that we arrive at a minimally working package again quickly, and can add individual binaries and libraries as small projects so many people can contribute parts with moderate time budget.

I think we should first concentrate on building a minimal HAL environment first which "runs again"; then add to this as we go along, and postpone the question of "what goes where" for now. The reason is that a by concentrating on a cmake-built machinekit-HAL-only package (which could be used for ROS integration) we very likely have the parts which will remain in the final machinekit-realtime part.

A first milestone could be a package which contains:
- rtapi_msg, rtapi_app*, flavor
- halcmd halscope
- the rtapi and hal_lib RT and userland shared libraries

The deliverable would be "realtime start/stop" works and halcmd can talk to rtapi_app.

Based on that we can add individual pieces to this new package by adding the proper cmake incantations, for instance halscope, components and supporting libraries like posemath,

Since we removed kernel builds and we have currently no kernel modules to be built, thread flavor support would be drastically simplified. I think thread support can be simplified to:

- always build posix and rt-preempt.
- if the Xenomai userland support is available, built the xenomai flavor too.

the part I'm not sure about is: how do we handle the middleware (proto, zmq, mdns) and its libraries
- halcmd does need protobuf artefacts and zeromq
- haltalk webtalk would need it, as do other apps

I _think_ that should eventually go into a middleware package but if everything is in CMakeLists.txt splitting ex-post should be easy.

Related issues: #336 #598

bobvanderlinden, 2016-01-18:
I think another requirement is that changes made in linuxcnc should be easily mergable back in Machinekit, even after the CMake change. At the moment a lot of the CMake changes are already working, but not in the original directory structure (I split .h to include/ and .c to src/ for each target). The cmake work is available here: https://github.com/bobvanderlinden/machinekit/tree/cmake

The following is already compiles with cmake:

* rtapi_app_posix
* io and iov2
* milltask
* HAL components
* linuxcncsvr
* halcmd
* Some of the configuration
* Some install rules

With the install rules it was possible to place executables and libraries in /usr and configuration in /etc, but it did not yet work together. At the moment I can't really remember why, but it had probably something to do with setuid. Once it works, we can look into integrating this into the machinekit repo, so that things are still mergable with linuxcnc.

ArcEye, 2016-01-27:
This issue is probably the key one.

It has so many overlaps and dependencies, that it is has become not worth beginning any work in some other areas - even documentation, until the split is implemented and the exact fate of the legacy autoconf build repo is decided.

I personally would like to see a good separation of the 'middle layer' and perhaps a couple of different implementation libraries of zmq / protobuf functions, to enable a light local machine, with no need for much of the haltalk / machinetalk / webtalk stuff and a full blown networked, multi remote GUI capable implementation.

@bobvanderlinden 
> I think another requirement is that changes made in linuxcnc should be easily mergable back in Machinekit

I personally do not regard that as a priority and it would be a severe limiting factor to any split.

A lot of the HAL code will soon have diverged so far from Linuxcnc ( when MultiKore is implemented in the main code ) that any straightforward merge would be impossible.

If the planner and interpreter changes plus task ticket implementation are introduced, heralding the end of NML, the will be very few points of convergence left.

Any beneficial code changes could be implemented at a file level and we would still have the old, fully compatible repo anyway?

kinsamanka, 2016-02-06:
Here's the steps on how to test the build using docker:
```
docker run -it --privileged=true --name cmake-test kinsamanka/cmake_mk_builder bash
su mk
cd ~
git clone -b cmake git://github.com/kinsamanka/machinekit
mkdir build
cd build
cmake ../machinekit
make -j5
export PATH=$(pwd)/bin:$PATH
sudo make setuid
halrun
```

Note that docker will complain if there is an already existing container named `cmake-test`

To return to the build after exiting docker, use the following command:
```
docker start -ia cmake-test
```

kinsamanka, 2016-02-06:
I think we can split machinekit into the following parts:
- mk-hal
- mk-cnc (all the targets under `src/emc`)
- mk-legacy-ui

mhaberler, 2016-02-06:
@bobvanderlinden @strahlex I guess we need to spin out a machinetalk package as well - please help specify what goes into this

TBD: HAL only, or HAL plus cnc stack

mhaberler, 2016-02-06:
just to clarify - @kinsamanka 's remark does not really say so clearly:

Gemi has the HAL build with cmake almost running!

ArcEye, 2016-02-06:
This is presumably being built on Wheezy

It fails on GTK2 and tcl/tk-8.5 dependencies running locally on Jessie, but I can see the process.

ArcEye, 2016-02-07:
Now cloned into a Wheezy amd64 partition

Cloned into /usr/src/mk-cmake and checked out cmake branch
Run locally from /usr/src/build  `cmake ../mk-cmake`

Builds 100% without errors :checkered_flag: 

Well done @kinsamanka !!

Hopefully get to experiment with it more tomorrow

mhaberler, 2016-02-09:
just looking into the 
`````
machinekit@debian:~/build$ halrun
msgd:0 stopped
rtapi:0 stopped
halcmd: loadrt abs
stat: No such file or directory
halcmd: 
`````
this comes from src/rtapi/rtapi_compat.c:get_elf_section() which halcmd calls to inspect flags kept in an elf section (is this module instantiable? etc) - something Mick and me need to look into:

`````
Breakpoint 1, 0x00007f08e597ca2e in get_elf_section () from /home/machinekit/build/lib/libmkhal.so
(gdb) backtrace 
#0  0x00007f08e597ca2e in get_elf_section () from /home/machinekit/build/lib/libmkhal.so
#1  0x00007f08e597ceb3 in get_caps () from /home/machinekit/build/lib/libmkhal.so
#2  0x00007f08e597d211 in rtapi_get_tags () from /home/machinekit/build/lib/libmkhal.so
#3  0x000000000040b824 in loadrt_cmd ()
#4  0x000000000040beff in do_loadrt_cmd ()
#5  0x00000000004079a4 in parse_cmd1 ()
#6  0x0000000000407c18 in halcmd_parse_cmd ()
#7  0x0000000000414414 in main ()
`````
very likely instantiation currently does not work properly

kinsamanka, 2016-02-09:
The hal drivers were originally built on a per flavor basis, this cmake
version shares the drivers between the 3 flavors.

Would this be the cause of the issue?

mhaberler, 2016-02-10:
no, it is not - having a look right now, but sharing the hal comps+drivers across thread flavors is not a good idea

it might work mostly, but there will definitely flavor-specific drivers and then sharing will break things; we do need per-flavor subdirs for comps, drivers, rtapi.so, hal_lib.so

mhaberler, 2016-02-14:
@kinsamanka  has made huge progress - I just ran his repo cmake branch on jessie and HAL/RT seems to build and run fine!

It is time we start hammering out the future package and repo structure

I think the first questions are:
- will there only be one machinekit repo, or several ones
- if there are several ones, how will they interact - side by side, ships in the night? or git subtrees?
- we need to decide which packages we eventuall build - I see machinekit-rt, machinekit-cnc (machinekit-cnc-legacy maybe), machinekit-talk (?)
- where will bindings go: node.js, Python - separate packages or machinekit-talk? - the cnc-level bindings will incur a dependency on machinekit-cnc, do we want that?
- decide on a directory (maybe sometimes file) basis what goes where

this is far-reaching stuff, whatever we decide here will need lots of explaining lateron - @bobvanderlinden @strahlex @zultron  please help hammering this out!

luminize, 2016-02-15:
fwiw, my 0.2ct,

I think that for people like me it should be kept simple and stupid so I'd not walk the git subtrees path. In addition to @kinsamanka https://github.com/machinekit/machinekit/issues/863#issuecomment-180632471 and @mhaberler https://github.com/machinekit/machinekit/issues/863#issuecomment-183895432 i'd do:

- `machinekit-rt-hal` (which includes all the service hatches like machinetalk etc)
- `machinekit-talk` (the service hatches)
- `machinekit-cnc' (where machinekit-rt-hal is not mandatory but you need machinekit-talk to talk to it local/remote)
- keep `machinekit` for now as a legacy repository, only fixing bugs

this way documentation and examples can be done on a per repository basis. No need for knowing about remapping or G-codes, or GUIs if you are only interested in the `machinekit-rt-hal` part

I would like to think that if we have `machinekit-rt-hal` and `machinekit-talk` the building blocks are in place to have the legacy `machinekit` evolve to machinekit-cnc and surpass `machinekit`

mhaberler, 2016-02-17:
@luminize 
the dependencies are such that
- cnc depends on hal/rt not machinetalk and we are not free to decide if it depends or not - it does (i.e. ioctl)
- only new UI's like qtquickvcp depend on machinetalk (protobuf+zeromq+mDNS)
- halcmd uses zeromq+protobuf so the basic machinetalk support needs to go into hal/rt even if haltalk/webtalk did not (which would not quite make sense then)

the basic question is how we handle this in terms of git repositories:
1. branches of the legacy repo (which would eventually suggest branches in the main mk/mk repo which we currently do not want - C4 does suggest one branch only but we might diverge from this rule)
2. fork the legacy repo into one per-package repo each, pruning out unnecessary files
3. decide if/how to integrate @strahlex ' Python machinetalk client
4. how to eventually deal with @bobvanderlinden 's nodejs work

not sure what you mean by service hatch

luminize, 2016-02-17:
> cnc depends on hal/rt not machinetalk and we are not free to decide if it depends or not - it does (i.e. ioctl)

I was thinking for this to be the non-blob future cnc after a split. Non legacy

> not sure what you mean by service hatch

Interaction with the base building block. Python cython machinetalk node etc.

mhaberler, 2016-02-17:
to set the scope: we are divying up an existing codebase against minimum-dependency fault lines - it is not a green field exercise redesigning or fixing the past, and we are not doing any _development_ here.

The goal IMO must be: 

- one can install the cnc package and all deps are pulled in and resolved. 
- one can install the rt/hal package and have something working _including_ the HAL bindings (they do not make sense somewhere else)
- ontop of the latter, one can install node.js and/or Python remote HAL bindings

the latter actually might be better placed in the rt/hal package proper the more I think of it

@strahlex - any good arguments for a machinetalk package _outside_ of rt/hal? I do not see it because the protobuf/zeromq stuff MUST be in rt/hal (halcmd uses it as do the cython bindings)

luminize, 2016-02-18:
> On 17 Feb 2016, at 23:52, Michael Haberler <notifications@github.com> wrote:
> 
> it is not a green field exercise redesigning or fixing the past, and we are not doing any development here.

Guess there is a misconception about my remark. I'm not proposing to redevelop the cnc application as part of this endeavor. Since there _will_ be a split, and we use cmake for the building, doesn't building the cnc stack need to be addressed too?

I guess we're saying the same in different words.

luminize, 2016-02-18:
I miss the split of the documentation in this thread.

I propose we discuss how we split up the documentation. So different parts go in different packages. No getting some docs from one repo into another. That just won't work and is hard to maintain and understand for most of the people. I even would go as far as getting the machinekit-docs back into the cnc package.
I'm not saying this needs to be done right now, but we must have the blueprint in place to get proper documentation _in_ the package it is for. IMO a different repo was a great idea a year ago to lessen the building process but the lack of people chiming in shows it doesn't work.

kinsamanka, 2016-02-18:
Here's the new test repo for cmake builds: https://github.com/kinsamanka/machinekit-NG

ArcEye, 2016-02-18:
Looks like you have applied cmake to the whole repo!!!

That should make things easier :+1: 

I tried it quickly on Jessie and it fell over at the ..src/hal/icomp/icomp-examples area

Haven't got time to chase it right now, output below.

```
mick@INTEL-i7:/usr/src/build$ cmake ../machinekit-NG/
-- The C compiler identification is GNU 4.9.2
-- The CXX compiler identification is GNU 4.9.2
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.28") 
-- checking for module 'protobuf>=2.4.1'
--   found protobuf, version 2.6.1
-- checking for module 'libczmq>=2.2.0'
--   found libczmq, version 2.2.0
-- checking for module 'avahi-client'
--   found avahi-client, version 0.6.31
-- checking for module 'jansson>=2.5'
--   found jansson, version 2.7
-- checking for module 'libwebsockets'
--   found libwebsockets, version 1.3
-- checking for module 'libssl'
--   found libssl, version 1.0.1k
-- checking for module 'liburiparser'
--   found liburiparser, version 0.8.0.1
-- checking for module 'uuid'
--   found uuid, version 2.25.0
-- checking for module 'libusb-1.0'
--   found libusb-1.0, version 1.0.19
-- checking for module 'libmodbus>=3'
--   found libmodbus, version 3.0.6
-- Looking for include file pthread.h
-- Looking for include file pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found PythonInterp: /usr/bin/python2.7 (found suitable version "2.7.9", minimum required is "2.7") 
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython2.7.so (found suitable version "2.7.9", minimum required is "2.7") 
-- Boost version: 1.55.0
-- Found the following Boost libraries:
--   python
--   serialization
--   thread
-- Found GTK2_GTK: /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so  
CMake Warning at cmake/modules/Options.cmake:6 (message):
  xenomai flavor disabled
Call Stack (most recent call first):
  CMakeLists.txt:32 (include)

-- Looking for include file linux/hidraw.h
-- Looking for include file linux/hidraw.h - not found
-- checking for module 'libudev'
--   found libudev, version 215
-- Configuring done
-- Generating done
-- Build files have been written to: /usr/src/build
[... 190 lines cut ...]
[  6%] Building C object src/rtapi/rtapi_math/CMakeFiles/rtapi_math.dir/s_scalbn.c.o
[  6%] Building C object src/rtapi/rtapi_math/CMakeFiles/rtapi_math.dir/s_sin.c.o
[  6%] Building C object src/rtapi/rtapi_math/CMakeFiles/rtapi_math.dir/s_tan.c.o
[  6%] Building C object src/rtapi/rtapi_math/CMakeFiles/rtapi_math.dir/w_acos.c.o
[  6%] Building CXX object src/machinetalk/proto/CMakeFiles/mk-pb2++.dir/preview.pb.cc.o
[  6%] [  6%] Building C object src/rtapi/rtapi_math/CMakeFiles/rtapi_math.dir/w_asin.c.o
Building CXX object src/machinetalk/proto/CMakeFiles/mk-pb2++.dir/rtapi_message.pb.cc.o
[  6%] Building CXX object src/machinetalk/proto/CMakeFiles/mk-pb2++.dir/rtapicommand.pb.cc.o
[  7%] Building C object src/rtapi/rtapi_math/CMakeFiles/rtapi_math.dir/w_atan2.c.o
[  7%] Building C object src/rtapi/rtapi_math/CMakeFiles/rtapi_math.dir/w_fmod.c.o
[  7%] Building C object src/rtapi/rtapi_math/CMakeFiles/rtapi_math.dir/w_hypot.c.o
[  7%] [  7%] Building CXX object src/machinetalk/proto/CMakeFiles/mk-pb2++.dir/status.pb.cc.o
Building C object src/rtapi/rtapi_math/CMakeFiles/rtapi_math.dir/w_pow.c.o
[  7%] Building C object src/rtapi/rtapi_math/CMakeFiles/rtapi_math.dir/w_sqrt.c.o
[  7%] Building CXX object src/machinetalk/proto/CMakeFiles/mk-pb2++.dir/task.pb.cc.o
Linking C shared library ../../../lib/librtapi_math.so
[  7%] Built target rtapi_math
[  7%] Building CXX object src/machinetalk/proto/CMakeFiles/mk-pb2++.dir/test.pb.cc.o
[  8%] Building CXX object src/machinetalk/proto/CMakeFiles/mk-pb2++.dir/types.pb.cc.o
[  8%] Building CXX object src/machinetalk/proto/CMakeFiles/mk-pb2++.dir/value.pb.cc.o
Linking CXX shared library ../../../lib/libmk-pb2++.so
[  8%] Built target mk-pb2++
Makefile:117: recipe for target 'all' failed
make: *** [all] Error 2
```

ArcEye, 2016-02-18:
@mhaberler  wanted me to echo a suggestion for proceeding on this issue here

Because of the way @kinsamanka  did his cmake stuff for the HAL split,  ( and now the whole repo apparantly! ) which was to put the scripts etc into the existing repo source tree,
and build from outside it, a quick and easy way to try out things should be:

Clone a repo of MK to another location
Have 3 branches derived from it, say hal, middleware and cnc
Copy over Gemi's cmake additions into the hal branch and then use those as a basis for the other 2 branches
(We now have a lot more to play with)

This has the advantage of discarding nothing initially, what is included is decided by the scripts.
One can change ones mind about what goes where, until it is all settled and then the physical split can be initiated.

In the interim the existing repo carries on as before and if necessary updates can be merged across easily to the new one.

kinsamanka, 2016-02-18:
Actually the cmake stuff is built on top of the original repo.

The original repo is given with a tag `base`, so any changes to the original repo can be easily merged to the cmake repo.

I'll rename the master branch to `cmake-hal` and keep the master pointed to the original master.

luminize, 2016-02-18:
On 18 Feb 2016, at 10:18, ArcEye <notifications@github.com> wrote:

> Clone a repo of MK to another location 
> Have 3 branches derived from it, say hal, middleware and cnc

Shall we then clone this repo under the Machinekit organisation labelling it “machinekit-wip” as in “work in progress” (or we might call this working repository “machinekit-split” or whatever, trivial as long as it’s clear this is a temporary repository) where we can follow your proposal and make branches?

> This has the advantage of discarding nothing initially, what is included is decided by the scripts.
> One can change ones mind about what goes where, until it is all settled and then the physical split can be initiated.

I like this. I can understand this. Puts focus on the work, and not on git subtree/module-fu

kinsamanka, 2016-02-18:
The cmake hal stuff is now on branch `cmake-hal`

kinsamanka, 2016-02-18:
@ArcEye , here's the output of the `cmake` step: https://gist.github.com/kinsamanka/5235de4ff628612e0fff

This is the main difference:
```
-- Looking for include file linux/hidraw.h
-- Looking for include file linux/hidraw.h - not found
```

ArcEye, 2016-02-18:
Don't know why it does not find it.
It is present in /usr/include/linux and in the kernel sources at /usr/src/linux/include/linux

mhaberler, 2016-02-18:
@kinsamanka congratulations - ubuntu 14.04 lts build works for me too!

ArcEye, 2016-02-18:
In the CMakeCache.txt it has
```
//Flags used by the compiler during all build types.
CMAKE_C_FLAGS:STRING='-I/usr/src/linux/include '
```
so don't know why it cannot find it.

I will do a fresh clone and build in a bit

mhaberler, 2016-02-18:
@ArcEye - that path business for kernel module builds only AFAICT !

ArcEye, 2016-02-18:
OK, found it.

All the stuff I had been doing with the NANO-SoC, cross-builds, Quartus etc had left an env variable which was screwing things up.

Just cloned the new repo, checked out cmake-hal and all builds properly again.

I see you have trimmed this branch down considerably.
We just need to decide if we are going to work with branches of the whole repo and pick out what we want in the cmake scripts until the split is settled, or adopt this for the HAL repo, if we are sure nothing else might have to go back in.

mhaberler, 2016-02-18:
for others - debug builds (linenumber info etc for gdb): 
`cmake -DCMAKE_BUILD_TYPE=Debug ...``

mhaberler, 2016-02-18:
oops

mhaberler, 2016-02-18:
re Ubuntu 14/32bit: I would specifically be interested in results by other folks after:

````
halrun
halcmd: newthread fast 1000000 fp
halcmd: show thread
`````
newthread segfaults with

`````
(gdb) backtrace
#0  0xb740aa16 in allocate_stack (stack=<synthetic pointer>, pdp=<synthetic pointer>, attr=<optimized out>) at allocatestack.c:415
#1  __pthread_create_2_1 (newthread=newthread@entry=0xb7546c8c <extra_task_data+76>, attr=attr@entry=0xbf910380,
    start_routine=start_routine@entry=0xb753909d <realtime_thread>, arg=arg@entry=0xb75440e4 <local_rtapi_data+2468>) at pthread_create.c:500
#2  0xb740ae01 in __pthread_create_2_0 (newthread=0xb7546c8c <extra_task_data+76>, attr=0xbf910380,
    start_routine=0xb753909d <realtime_thread>, arg=0xb75440e4 <local_rtapi_data+2468>) at pthread_create.c:656
#3  0xb75394fe in _rtapi_task_start_hook (task=0xb75440e4 <local_rtapi_data+2468>, task_id=1)
    at /home/mah/machinekit-NG/src/rtapi/rt-preempt.c:372
#4  0xb7532f49 in _rtapi_task_start (task_id=1, period_nsec=1000000) at /home/mah/machinekit-NG/src/rtapi/rtapi_task.c:352
#5  0xb6e2e6ea in hal_create_xthread (args=0xbf9105a0) at /home/mah/machinekit-NG/src/hal/lib/hal_thread.c:213
#6  0x08054611 in rtapi_request (loop=0x942f9c8, poller=0x9437ba8, arg=0x0) at /home/mah/machinekit-NG/src/rtapi/rtapi_app.cc:899
#7  0xb73d2010 in zloop_start () from /usr/lib/i386-linux-gnu/libczmq.so.1
#8  0x08055847 in mainloop (argc=2, argv=0xbf910e94) at /home/mah/machinekit-NG/src/rtapi/rtapi_app.cc:1286
#9  0x0805631d in main (argc=2, argv=0xbf910e94) at /home/mah/machinekit-NG/src/rtapi/rtapi_app.cc:1643
````

I cannot reproduce that  on jessie64

ArcEye, 2016-02-18:
I tried it on Jessie too, plus a whole lot more, adding component, starting thread, stopping etc

All worked fine.

Don't have any Ubuntu partitions or VMs to try.

ArcEye, 2016-02-18:
Been having a browse through the cmake stuff, it is quite complete now.

It had not occurred to me that the instcomp / comp workflow would need changing, but of course they use Makefile.modinc to write a Makefile 'on-the-fly', to compile the module.
Thus using them to generate the C code , then using cmake to build that code makes sense.

Have to think about whether there is any point producing a cmake version of them.  As it is usable via cmake or the current build system, probably not.

The overall build work flow should be simple, if the repo is organised with branches as previously mooted.

The `../build` directory above the repo is a naming convention, not a requirement, cmake will use the PWD whatever it is called.

So having 3 build directories, `build-hal, build-middle and build-cnc`, one level above the repo will just require a cycle through, changing branches and build dirs, to build all the sections of MK.

Need some guidance as to what exactly needs to go where, before attempting to adapt the existing scripts to another set of targets.

mhaberler, 2016-02-18:
@ArcEye - jessie: I assume this was 64bit? it seems the issue with the pthread_create crash is 32bit only

mhaberler, 2016-02-18:
well to make progress, I propose:

@kinsamanka - transfer your machinekit-NG repo to the machinekit organisation and that becomes the development repo which will contain all the packages as separate branches

branchnames for now:

`cmake-hal` for the rt/hal package
`cmake-cnc` for the cnc stack

(maybe master for the legacy tree?)

@strahlex @bobvanderlinden PLEASE comment on the middleware package - what should be in it or not?

the way I understand it we would have:

mk-talk providing protobuf/zeromq/dnssd etc, and would contain pymachinetalk
mk-rt-hal using/depending on mk-talk
mk-cnc depends on mk-rt-hal and hence implicitly on mk-talk

remote using code will have to use only mk-talk to talk to rt/hal remotely
for local ops it would need mk-rt-hal as well

@strahlex - is https://github.com/strahlex/pymachinetalk/tree/master/pymachinetalk ready to merge?

@kinsamanka - sorry I goofed with the split when we talked - to make remote ops possible without importing all of hal/rt we need the above structure

ArcEye, 2016-02-18:
>  @ArcEye - jessie: I assume this was 64bit? it seems the issue with the pthread_create crash is 32bit only

Correct amd64 rt-preempt

kinsamanka, 2016-02-19:
Fixed the 32 bit issue. The problem was missing cflags and ldflags.

I've also pushed the preliminary code for deb packaging.

@ArcEye regarding comp/icomp, I think we keep using Makefile for building comps but instead of using Makefile.modinc we integrate the Makefile inside the script. 

The Makefile won't be too complex as it doesn't need to handle kernel modules anymore.

ArcEye, 2016-02-19:
> @ArcEye regarding comp/icomp, I think we keep using Makefile for building comps but instead of using Makefile.modinc we integrate the Makefile inside the script.

>The Makefile won't be too complex as it doesn't need to handle kernel modules anymore.

It works fine at present, 90% of the code is concerned with C code and man page generation so just using it to create code then building that with cmake, is a simple option.

I am thinking further forward to its use to build individual custom components, within a cmake generated system, without a complete rebuild.
Will not matter for RIP builds, just rebuild, only takes a minute, but if we get to a distributed .deb based system, it will be required.

One for the future anyway.

kinsamanka, 2016-02-19:
Actually it should not work as the current cmake doesn’t update the modinc file.

ArcEye, 2016-02-19:
> Actually it should not work as the current cmake doesn’t update the modinc file.

No it won't work.

The easiest option may well be to just have a build script for components, that does what your macro does, generates the C code with instcomp and compiles with cmake.

Backward compatibility might actually require renaming instcomp and having a wrapper script called instcomp, which achieves what is expected by the user, in an opaque fashion.

It is a detail really.

bobvanderlinden, 2016-02-19:
@mhaberler Sorry for the delay, I hadn't checked Github/gitter after my vacation until now.

About Machinetalk protobuf language bindings:
The language bindings are already handled in Machinetalk-protobuf. These bindings are automatically generated from the protobuf source files. I'm still not sure whether having all languages in the same repo is a good idea, but it works fine for now.

About Machinetalk client-libraries:
The Machinetalk-client libraries for Python ([pymachinetalk](https://github.com/strahlex/pymachinetalk)) and NodeJS ([node-machinetalk](https://github.com/bobvanderlinden/node-machinetalk)) are in separate repositories now. I think that's a good approach. These libraries are at the moment hand-written.

@strahlex was working on a way to generate the client-libraries for various languages ([machinetalk-gsl](https://github.com/strahlex/machinetalk-gsl)). I'm not sure yet how we're going to structure the output of that, but it might be integrated into pymachinetalk and node-machinetalk. We'll have to see once machinetalk-gsl is stable enough.

About integrating client-libraries into Machinekit:
The only shared dependency that the client-libraries have with Machinekit is the Machinetalk API. If that API doesn't change significantly regularly, then I think there is no need to integrate the client-libraries into the Machinekit repository.

About Machinetalk remote-HAL:
I would think the Python and NodeJS libraries will both include remote-HAL functionality. What I don't know is whether remote-HAL components are needed in Machinekit itself. If it is not, then the repository-split (how things are now) is fine.

About machinekit-docs:
I agree with @luminize integrating documentation into the Machinekit repository (or machinekit-cnc repository). However, only documentation related to machinekit-cnc. If rt or hal is going to be split off, documentation should move with it. The benefit is having documentation that is always in sync with the implementation. It ensures the documentation is always at the same version as the code.
This has the potential to read the documentation of the version you are running, instead of reading documentation of the latest version while running an older one.

That said, it would be nice to have an easily accessible way for people to see and find the documentation. We could generate and host the resulting HTML on machinekit.io or let the generating and hosting be handled by something like readthedocs.org. I think this is a separate issue and we can decide this one the split/merge has been made.

mhaberler, 2016-02-19:
@bobvanderlinden the RT/HAL core part uses protobuf/zeromq to talk between halcmd and rtapi_app but only over Unix domain sockets 
so technically not remote really

strahlex, 2016-02-19:
Webtalk -- I would completely strip out this package, I pretty much reimplemented this function in a few hundred lines of node.js code. Webtalk is a broker an therefore should not be part of the main package. The C++ websockets dependency is really big one.

Machinetalk-Protobuf is already in its own repository. There is no need to create another package.

The rest of Machinetalk should be part of the Machinekit core package. There is no difference between local and remote. Everything should use Machinetalk sooner or later. ZeroMQ and Protobuf are pretty slim dependencies, avahi on Linux too. All these dependencies are available in all major distributions out-of-the-box now.

mhaberler, 2016-02-19:
@strahlex - fine with dropping webtalk, assuming your node.js work will be merged downstream

in terms of what goes where:  yes, the _core_ package (I assume you mean hal/rt) uses protobuf/zeromq

HOWEVER any remote clients (say a ROS adapter running on a remote host) will need the zeromq/protobuf/machinetalk stuff to talk to the RT instance, and it does not make sense to burden a remote client with the prerequisites for RT if HAL is not running there

that was the thinking behind a separate mtalk package - proto files built, protobuf, zeromq, the library - the minimum stuff to get going on a remote host to talk to hal/rt

a remote client would just install mtalk
the RT instance would install hal/rt which pulls in mtalk for its own internal use (halcmd)

ArcEye, 2016-02-19:
I have been playing with CMake and instcomp.
Good way to find my way around CMake if nothing else.

The aim is to see how easy it is to create components and rebuild existing ones from the new
'package' that CMake creates in ../build-hal

That prompts the question - what do we want in that ../build-hal 'package'

Are we looking at a stripped 'package' with a separate `MK-HAL-dev` add on, to get functionality required for building components?

Currently the .icomp sources are not in it, just the C files - fixed that.
I have CMake creating a makefile.modinc equivalent, using the CMake build environment, which will then allow building or rebuilding components 

The headers are not copied over to ../build-hal/include, so no builds will work within its own environment. Fixed temporarily.

IMHO we should have everything in it for a fully functioning ../build-hal 'package' initially.
Any thought of splitting into essential and additional stuff is really for the later .deb packaging
stage - if we get to that.

mhaberler, 2016-02-19:
@ArcEye - good point

personally I think this -dev distinction does not make much sense - it is just a hurdle to be taken as soon as comp/instcomp is used

however it will require pulling in build-essential with the bare rt/hal (build-hal) package, probably headers and such, however only as far as required for building a halcomp, not the whole cnc stack (where most of the deps are)

we're shedding a lot of dependencies when doing just the rt/hal package so that one isnt too much to ask

I am open either way - it means a fourth package though: rt-hal-dev

**decision item:** what are the opinions on "separate rt-hal-dev" versus "all deps for comp/instcomp included in rt-hal" ?

strahlex, 2016-02-19:
@mhaberler Yes, clients need the Machinetalk bindings as well. However, the client bindings are currently different from the Haltalk and mkwrapper implementations. Ideally both implementations, server and client, will use the same bindings in the future. These bindings should be packaged into a separate package. But, keep in mind that the bindings are language dependent. E.g. there will be Python bindings, C++ bindings, Node.js bindings, ... So it might make more sense to keep them in the language dependent packages or the native packages managers (e.g. pip for python, npm for Node.js).

So I would conclude that Machinekit core should contain the Haltalk mkwrapper servers. Once abstraction of the bindings is completed Machinekit core may depend on the bindings package. Currently there is no such thing.

strahlex, 2016-02-19:
Regarding Webtalk: People who want to use WebSockets will most likely use Node.js anyway.

ArcEye, 2016-02-20:
> decision item: what are the opinions on "separate rt-hal-dev" versus "all deps for comp/instcomp included in rt-hal" ? 

In the absence of any strong opinions either way, I will continue incorporating into the cmake-hal branch.

I will however put the whole lot into one merged commit for PR, to make it easier to move around if required later.
It is all additions to the existing scripts, files etc rather than alterations, so should unplug as easily.

ArcEye, 2016-02-20:
I have noticed when building with `make -j9` that the build from the CMake generated Makefile, quite often falls over.
If you repeat the command, it completes successfully.
Looks like one core getting ahead of another and trying to build or link with dependencies that are not yet completed on another core.
Doesn't happen on the old make system.

kinsamanka, 2016-02-20:
There's an issue on the python build scripts. I still haven't tracked it
down but it is likely a missing dependency link on the add_custom_command.

I'm still stuck on the 32bit rtapi issue :-/

mhaberler, 2016-02-23:
happy to note @kinsamanka just nailed the rtapi issue - cheers!

ArcEye, 2016-02-24:
I have just done a fresh pull before adding some files.

Now fails to compile at the /src/hal/components/sampler.c stage

At a guess, -DULAPI is not being set any more, will try to find the change.

```
Scanning dependencies of target halsampler
[ 40%] Building C object src/hal/components/CMakeFiles/halsampler.dir/sampler_usr.c.o
In file included from /usr/src/machinekit-NG/src/hal/components/sampler_usr.c:73:0:
/usr/src/machinekit-NG/include/rtapi.h:70:2: error: #error "Please define either RTAPI or ULAPI!"
 #error "Please define either RTAPI or ULAPI!"
  ^
In file included from /usr/src/machinekit-NG/src/hal/components/sampler_usr.c:74:0:
/usr/src/machinekit-NG/include/hal.h:130:2: error: #error HAL needs RTAPI/ULAPI, check makefile and flags
 #error HAL needs RTAPI/ULAPI, check makefile and flags
  ^
src/hal/components/CMakeFiles/halsampler.dir/build.make:54: recipe for target 'src/hal/components/CMakeFiles/halsampler.dir/sampler_usr.c.o' failed
make[2]: *** [src/hal/components/CMakeFiles/halsampler.dir/sampler_usr.c.o] Error 1
CMakeFiles/Makefile2:5394: recipe for target 'src/hal/components/CMakeFiles/halsampler.dir/all' failed
make[1]: *** [src/hal/components/CMakeFiles/halsampler.dir/all] Error 2
Makefile:117: recipe for target 'all' failed
make: *** [all] Error 2
```

kinsamanka, 2016-02-24:
I think you're either on the wrong repo or branch.

What does `git log --pretty=format:'%h' -n 1` show? It should be `5d8087c`

ArcEye, 2016-02-24:
There is something wrong with this pull, it doesn't have the correct commit at HEAD~

I will delete the fork completely and do it over again.

ArcEye, 2016-02-24:
The last 2 commits were screwed in that pull.
There was a dramatic slowdown in the connection to github during it, must have been a glitch.

All good now :smile:

mhaberler, 2016-02-25:
@kinsamanka - some more testing on arm's after yesterdays commits - all looking good (newthread,  nosetests ok):

jessie cubox-i iMX6 quadcore posix 3.14.57 
Altera nano - Michael Brown debian image, rt-preempt 3.10.37-ltsi-rt37-05836-g0de1eb3-dirty
BBB bb-ubuntu 3.8.13-xenomai-r78 Ubuntu 14.04.3 LTS  rcn-ee.net console Ubuntu Image 2015-12-11

ArcEye, 2016-02-27:
I don't think @kinsamanka  was too happy at my pulling headers into the build dir, so that components could be rebuilt.  
It wasn't an ideal solution.

Had a long think and decided the root problem is, that we should not be building outside the repo to build a RIP build, even though we are using CMake.

Have a look here https://github.com/ArcEye/machinekit-HAL-inrepo

This is a copy of the existing -NG repo, in which I have changed all the paths so that the cmake build is carried out from a dir called `/build` within the repo and all the binaries, scripts libs etc are copied to where they used to be, when doing a RIP build.

Another advantage of this is backward compatibility.
I have adjusted /scripts/rip-environment.in and now that can be used to set the environment, 
so with the exception of calling 
`cmake ../ && make -jX && make setuid`
from the /build dir instead of the usual incantations from /src, it is pretty familiar.

[9 more replies]

closed 2018-08-01 by ArcEye