machinekit/machinekit #865: mksocfpga hm2 port needs machinekit driver implemention.

the-snowwhite, 2016-01-26:
The new Nano / Atlas Mesa based socfpga port presents itself to machinekit
via an custom uio_irq driver where the irq part for now is unimplemented / disabled to keep things as simple as possible.

loading the uio kernel module
https://github.com/the-snowwhite/mksoc/tree/master/Software/hm2reg_uio-module

will then present a /dev/uio device, that is asscessable from user space via this small procedure:
taken from:
http://blog.idv-tech.com/wp-content/uploads/2014/09/drivers-session3-uio-4public.pdf
page 13.

    #define UIO_SIZE "/sys/class/uio/uio0/maps/map0/size"

    int main( int argc, char **argv ) {
    int uio_fd;
    unsigned int uio_size;
    FILE *size_fp;
    void *base_address;
    uio_fd = open( “/dev/uio0”, O_RDWR);
    size_fp = fopen(UIO_SIZE, O_RDONLY);
    fscanf(size_fp, “0x%08X”, &uio_size);
    base_address = mmap(NULL, uio_size,
    PROT_READ | PROT_WRITE,
    MAP_SHARED, uio_fd, 0);

// Access to the hardware can now occur....

    munmap(base_address, uio_size);

    }

This above is all that is needed to talk to the hostmot2 vhdl ip-core on the fpga part of the Nano-board and allready in place.

--------

The Machinekit required parts of the procedure of creating a new hostmot2 driver I so far have been able to decode are this:

the driver needs to export at least these functions initially:

    hm2_cvsoc_read(hm2_lowlevel_io_t *this, u32 addr, void *buffer, int size) {

    // there are read / write templates with allignment chekking in the current commit and hm2_pci.c

    }

    static int hm2_cvsoc_write(hm2_lowlevel_io_t *this, u32 addr, void *buffer, int size) {
    }

    static void hm2_cvsocfpga_cleanup(void) {
        int i;
        munmap(base_address, uio_size);
        // NOTE: hal_malloc() doesnt have a matching free
    
        for (i = 0; i < num_ipcores; i ++) {
            hm2_lowlevel_io_t *this = &ipcore[i].llio;
            THIS_PRINT("releasing ipcore\n");
            hm2_unregister(this);
            hal_parport_release(&ipcore[i].port);
        }
    }

these can be added later:

    static int hm2_cvsoc_program_fpga(hm2_lowlevel_io_t *this, const bitfile_t *bitfile) {
    source code.
    }

    static int hm2_cvsoc_probe(struct cvsoc_dev *dev, const struct dts_device_id *id) {
    }

Plus have these 2 main function(minimal epp driver as example with cvsocfpga names):

        int rtapi_app_main(void) {
        int r = 0;
    
        comp_id = hal_init(HM2_LLIO_NAME);
        if (comp_id < 0) return comp_id;
    
        r = hm2_cvsocfpga_setup();
        if (r) {
            hm2_cvsocfpga_cleanup();
            hal_exit(comp_id);
        } else {
            hal_ready(comp_id);
        }
    
        return r;
    }
  

    void rtapi_app_exit(void) {
        hm2_cvsocfpga_cleanup();
        hal_exit(comp_id);
        LL_PRINT("driver unloaded\n");
    }

OR:

    void rtapi_app_exit(void) {
        uio_unregister_driver(&hm2_cvsoc_driver);
        LL_PRINT("driver unloaded\n");
        hal_exit(comp_id);
    }

Following new dts data structure can be added for the embedded platforms for probing of hardware capabilities / resources available:

this one is loaded with the cvsocfpga hm2 interface info.

    static struct dts_device_id hm2_cvsoc_tbl[] = {
        
        // 5i25
        {   
            .address_width = 14,        //0x0000000E
            .clocks = 2,                //0x00000002 number of clocks ?
            .compatible {
                .vendor = machkt,       //6D 61 63 68 6B 74 2C
                .name = hm2reg-io,      //68 6D 32 72 65 67 2D 69 6F 2D 31 2E 30 00
            } // (machkt,hm2reg-io-1.0.)
            .data_width = 32,           // 0x00000020
            .name = hm2-socfpga,        //68 6D 32 2D 73 6F 63 66 70 67 61 00 (hm2-socfpga.)
            .reg = 0x00000001, 0x00040000, 0x00010000,// ?, address offset from bridge, address span (= max address +1)
        },
        {0,},
    };

the-snowwhite, 2016-01-26:
The reward for assembling this correctly is enabling the mksoc port to function in the real world .... :-)

cdsteinkuehler, 2016-01-26:
Incredible work, Michael!

I ported the hostmot2 driver to user-mode PCI, so I can probably handle integration of the uio driver.  Hardware interrupts are (currently) not used anywhere else in HAL or RTAPI, so there's no need to worry about that complexity...there's a lot of other infrastructure that needs to be built-up to properly use hardware interrupts to drive timing.

I'll see if I can build your mksoc image and get my DE0-Nano up and running, then try to make progress on the driver.  It will probably take at least a week or two...I don't seem to make progress as fast as you do!

ArcEye, 2016-01-26:
Forgive my unfamiliarity with this area.

You refer to the DE0-NANO-SOC which I assume is this one
http://www.mouser.co.uk/ProductDetail/Terasic/P0286/?qs=%2ff49I7iJAtHwNTYwSQVJ%2fA%3d%3d
(or at least a constituent part of the kit)

If I understand correctly, that would mean running the whole system on an 800MHz A9

Is it also possible to program the DE0-NANO FPGA board
http://www.digikey.co.uk/en/product-highlight/t/terasic-tech/de0-nano-dev-board?WT.srch=1&mkwid=sLaslpWaX&pcrid=71112000506&pkw=_cat%3Adigikey.co.uk&pmt=b&pdv=c
the same way and use it as a 5i25 replacement module with an existing system?

That would be a major breakthrough indeed.

the-snowwhite, 2016-01-26:
@cdsteinkuehler 
Thanks Great to hear you are ready to move the project through it's final fase.
Hmmm Time is relative It did take me 4 months of preparation to be able to to tune in to that place from where I was able to "see" what needed to be done and actually do it at the same time.
Working at these seemingly impressive levels of velocity and abstraction Is something I have learned to do my best to avoid as the stress induced can have serious impact afterwards.......
So I'm happy I managed to see the stop signals and gear down before I tried to do something that normally takes a week or 2 in less than one day .. :-)

@ArcEye 
Yes 
It's actually a dual core 800-900Mhz arm A9 with integrated pga.
The actual 5i25 design Charles and I agreed to start with uses only around 25% of the fpga's resources.

But as is is now it should be fully able to function as a pc with a 5i25 card, once the machine kit driver is in place.

Hardware interfaces: 
I have an extra cramps board reserved for the purpose of being able to connect / wire up to some hardware in a fairly safe way.

But I'm quite positive that once the board it up and running It will put enough interrest to motivate some hardware interface designs. I personally have not been doing any circuits since I tried to build an integrated micro stepper controller that keeps dying as soon as the step rate got too high and I ended up having to import some built with the same chips (at a reasonable price) from the states instead... several years ago ......way before the reprap movement was born.

unseenlaser, 2016-01-26:
i'll be looking at hardware interfaces , once i get my nano up and running .
so no doubt there will be a few ideas floating ,

-- 

The information contained in this message is confidential and is intended
for the addressee only. If you have received this message in error or there
are any problems please notify the originator immediately. The unauthorised
use, disclosure, copying or alteration of this message is strictly
forbidden. This mail and any attachments have been scanned for viruses
prior to leaving the RcTechnix network. RcTechnix will not be liable for
direct, special, indirect or consequential damages arising from alteration
of the contents of this message by a third party or as a result of any
virus being passed on.

RcTechnix reserves the right to monitor and record e-mail messages being
sent to and from this address for the purposes of investigating or
detecting any unauthorised use of its system and ensuring effective
operation.

(c) RcTechnix

ArcEye, 2016-01-28:
@the-snowwhite 
OK, you didn't actually answer my question, but I suspect the FPGA only board will be able to be programmed as a 5i25 clone and interface to another controller

I have ordered a DE0-NANO-SoC for now and will be trying to catch you up soon!

Great work.

the-snowwhite, 2016-01-28:
@ArcEye Ok I overlooked you also mentioned the (old) de0 nano without SOC.
Absolutely not recommendable,  while yes you can put the fpga design in it you have no direct way to connect / interface it to the machinekit software.... As you have no arm CPU on-board. And no pci interface if you want to put it in a pc. Trying to interface via lån is also not directly possible.
So that would not be much fun without the arm / hps.

ArcEye, 2016-01-28:
@the-snowwhite 
Thanks for the reply - I have the same board as you on order now anyway - so see how it goes (when it arrives) :smile:

mhaberler, 2016-01-30:
@the-snowwhite - congratulations! I must have been asleep while on the road ;)

terasic boards are in the pipe - very curious to reproduce!

mhaberler, 2016-01-30:
not urgent - just something on my list I would love to see ticked eventually (#687):

if you guys can (eventually) cause the firmware to issue an interrupt [like here](https://github.com/mhaberler/asciidoc-sandbox/wiki/Generating-clock-interrupts-from-Mesanet-cards)  I will take on the IRQ support in the driver(s) and do the RTAPI+HAL plumbing so a funct chain can be executed triggered by the firmware IRQ

cdsteinkuehler, 2016-01-30:
On 1/30/2016 1:41 PM, Michael Haberler wrote:
> not urgent - just something on my list I would love to see ticked
> eventually (#687):
>
> if you guys can (eventually) cause the firmware to issue an
> interrupt [like here] [...]

It's super easy to generate interrupts in hardware, and IIRC the hm2
code already has some interrupt generation logic that should tie
pretty simply to the SoC interrupts.  The problem is handling them on
the Linux side, which is pretty complex.

-- 
Charles Steinkuehler
charles@steinkuehler.net

the-snowwhite, 2016-02-08:
OK
I have following commits ready
Hm2 driver:
https://github.com/the-snowwhite/machinekit/commit/bfae9ef5ba7657862bdd74f6a0abee9f46ad1c0b

config file changes:
https://github.com/the-snowwhite/machinekit/commit/7cf765e4e98a7736df22e8980cb2bf3c981c550b

And some build stuff:
https://github.com/the-snowwhite/mksocfpga/commit/c416b9ba267816466d28d02becfb9a5b37c6db42

   So how do we test this ?

the-snowwhite, 2016-02-10:
@RobertCNelson The 4.4,1 ltsi-rt driver came out a few days ago so I gave it a shot, however
I have not been able to get the eth0 ipv4 lan / internet to ping anything outside the adapters own ip address, what a pita....
So I have reverted the kernel once again to the altera-socfpga 3.10-rt-ltsi repo version.

@cdsteinkuehler I expect to be able to put a new Nano test image online later this evening Including a rip mk install with the uio driver module and the hm2cvsoc hal module compiled.

all the sources are online in:
https://github.com/the-snowwhite/mksocfpga  master
https://github.com/the-snowwhite/machinekit/tree/mksoc  mksoc branch

Testers and Test procedures are welcome

unseenlaser, 2016-02-10:
Thanks Michael

i'm hoping to get everything installed this evening or latest tomorrow , to
test out ,

-- 

The information contained in this message is confidential and is intended
for the addressee only. If you have received this message in error or there
are any problems please notify the originator immediately. The unauthorised
use, disclosure, copying or alteration of this message is strictly
forbidden. This mail and any attachments have been scanned for viruses
prior to leaving the RcTechnix network. RcTechnix will not be liable for
direct, special, indirect or consequential damages arising from alteration
of the contents of this message by a third party or as a result of any
virus being passed on.

RcTechnix reserves the right to monitor and record e-mail messages being
sent to and from this address for the purposes of investigating or
detecting any unauthorised use of its system and ensuring effective
operation.

(c) RcTechnix

the-snowwhite, 2016-02-11:
Ok :-)

Since the rootfs build is quite messy (qemu debootstrap script needs to be edited set -e --> set -v and then stage 2 started by hand + some unmounting and script changes afterwards), I have put an image containing only the full rootfs online here:
https://drive.google.com/file/d/0BwyLvgyVIdi8cjM3MlYyWG9xY1E/view?usp=sharing
https://drive.google.com/file/d/0BwyLvgyVIdi8TkI5NHNBQlhnSms/view?usp=sharing

then there is only to copy boot-part files, hm2 uio driver folder and socfpga_defconfig file --> kernel folder (uio enable). Into your new work folder together with the provided rootfs sd-image.
Run the image gen script from the external mksoc repofolder with cd in you work folder and add any character as parameter to the end to get it to run.
Then run the sd-image build script with u-boot, kernel-build, install files (remember to check image-gen is disabled at bottom).

Last remember to have a usb serial boot console up and running on first boot press space within 5 sek of boot / power on and add(paste) the env variables (e-net vars not required) from the md file in the doc folder in mksocfpga repo.

the-snowwhite, 2016-02-11:
this is current run status:

Last login: Thu Feb 11 00:54:30 2016 from debian9-ws.holotronic.lan
machinekit@mksoc:~$ source ~/machinekit/scripts/rip-environment
machinekit@mksoc:~$ linuxcnc
MACHINEKIT - 0.1
Machine configuration directory is '/home/machinekit/machinekit/configs/hm2-stepper'
Machine configuration file is '5i25.ini'
Starting Machinekit...
io started
halcmd loadusr io started
emc/iotask/ioControl.cc 752: can't load tool table.
hm2-stepper-5i25.hal:43: module 'hm2_cvsoc' not loaded
Shutting down and cleaning up Machinekit...
Cleanup done
Machinekit terminated with an error.  You can find more information in the log:
    /home/machinekit/linuxcnc_debug.txt
and
    /home/machinekit/linuxcnc_print.txt
as well as in the output of the shell command 'dmesg' and in the terminal
machinekit@mksoc:~$ dmesg
....
[    3.802479] systemd-udevd[160]: starting version 215
[    4.255647] hm2cvsoc ff240000.hm2-socfpga: setting: uioinfo->irq = UIO_IRQ_NONE 
[    5.179558] systemd-journald[154]: Received request to flush runtime journal from PID 1
[    5.810675] eth0: device MAC address e6:a0:86:d5:a1:00
[    7.813559] libphy: stmmac-1:01 - Link is Up - 1000/Full

---
machinekit@mksoc:~$ cat /home/machinekit/linuxcnc_debug.txt
1653
  PID TTY      STAT   TIME COMMAND
Stopping realtime threads
Unloading hal components
machinekit@mksoc:~$ cat /home/machinekit/linuxcnc_print.txt
RUN_IN_PLACE=yes
LINUXCNC_DIR=
LINUXCNC_BIN_DIR=/home/machinekit/machinekit/bin
LINUXCNC_TCL_DIR=/home/machinekit/machinekit/tcl
LINUXCNC_SCRIPT_DIR=
LINUXCNC_RTLIB_DIR=/home/machinekit/machinekit/rtlib
LINUXCNC_CONFIG_DIR=
LINUXCNC_LANG_DIR=/home/machinekit/machinekit/src/objects
INIVAR=/home/machinekit/machinekit/libexec/inivar
HALCMD=halcmd
LINUXCNC_EMCSH=/usr/bin/wish8.6
INIFILE=/home/machinekit/machinekit/configs/hm2-stepper/5i25.ini
PARAMETER_FILE=hm2-stepper.var
TASK=milltask
HALUI=
DISPLAY=axis
Starting Machinekit server program: linuxcncsvr
Loading Real Time OS, RTAPI, and HAL_LIB modules
Starting Machinekit IO program: io
Killing task linuxcncsvr, PID=1653
Removing HAL_LIB, RTAPI, and Real Time OS modules
Removing NML shared memory segments
machinekit@mksoc:~$ nano hm2-stepper-5i25.hal
machinekit@mksoc:~$ nano /home/machinekit/machinekit/configs/hm2-stepper/hm2-stepper-5i25.hal
machinekit@mksoc:~$ nano /home/machinekit/machinekit/configs/hm2-stepper/5i25.ini            
machinekit@mksoc:~$ linuxcnc
MACHINEKIT - 0.1
Machine configuration directory is '/home/machinekit/machinekit/configs/hm2-stepper'
Machine configuration file is '5i25.ini'
Starting Machinekit...
io started
emc/iotask/ioControl.cc 752: can't load tool table.
halcmd loadusr io started
hm2-stepper-5i25.hal:43: module 'hm2_cvsoc' not loaded
Shutting down and cleaning up Machinekit...
Cleanup done
Machinekit terminated with an error.  You can find more information in the log:
    /home/machinekit/linuxcnc_debug.txt
and
    /home/machinekit/linuxcnc_print.txt
as well as in the output of the shell command 'dmesg' and in the terminal
machinekit@mksoc:~$ cat /home/machinekit/linuxcnc_print.txt
RUN_IN_PLACE=yes
LINUXCNC_DIR=
LINUXCNC_BIN_DIR=/home/machinekit/machinekit/bin
LINUXCNC_TCL_DIR=/home/machinekit/machinekit/tcl
LINUXCNC_SCRIPT_DIR=
LINUXCNC_RTLIB_DIR=/home/machinekit/machinekit/rtlib
LINUXCNC_CONFIG_DIR=
LINUXCNC_LANG_DIR=/home/machinekit/machinekit/src/objects
INIVAR=/home/machinekit/machinekit/libexec/inivar
HALCMD=halcmd
LINUXCNC_EMCSH=/usr/bin/wish8.6
INIFILE=/home/machinekit/machinekit/configs/hm2-stepper/5i25.ini
PARAMETER_FILE=hm2-stepper.var
TASK=milltask
HALUI=
DISPLAY=axis
Starting Machinekit server program: linuxcncsvr
Loading Real Time OS, RTAPI, and HAL_LIB modules
Starting Machinekit IO program: io
Killing task linuxcncsvr, PID=2276
Removing HAL_LIB, RTAPI, and Real Time OS modules
Removing NML shared memory segments

the-snowwhite, 2016-02-11:
I have a board meeting tomorrow(today) and will first be back online  in about 16 hours or so.

unseenlaser, 2016-02-11:
the first error about the tool table , usually is that tool.tbl as referenced in the ini , is not there , so file not found ,  are the config hal and ini files available ? is tool.tbl file in the config

the-snowwhite, 2016-02-11:
Well as I havn't been doing any kind of hal stuff since september the only kind of quick test I was able to invent last night was to do:
ssh -X machinekit@mksoc
...then execute the rip script ...
linuxcnc
in the gui select 5i25 config in hardware group
then: nano ...5i25.ini  --> change hm2_pci to hm2_cvsoc
then nano: ...5i25...hal --> remove pin config file from hm2 module loader

...--> paste error messages ... and goto sleep ...zzzz
:-)

the-snowwhite, 2016-02-11:
BTW
I have added a quickstart guide in the mksocfpga repo and.
a redirection to mksocfpga repo in the mksoc repo.

the-snowwhite, 2016-02-12:
I changed the module name to hm2_soc as suggested and put that commit into the pr pipe.

I have tried with following changes in the config files:

machinekit/configs/hm2-stepper/5i25.ini

    [HOSTMOT2]
    DRIVER=hm2_soc
    BOARD=5i25
    #CONFIG="firmware=hm2/5i20/SVST8_4.BIT num_encoders=0 num_pwmgens=0 num_stepgens=3"

    [EMC]

    # Name of machine, for use with display, etc.
    MACHINE =               HM2-Stepper

    # Debug level, 0 means no messages. See src/emc/nml_int/emcglb.h for others
    #DEBUG =                0x00000003
    DEBUG =                0x00000007
    #DEBUG = 0

    # motion controller, get name and thread periods from ini file
    # trajectory planner
    loadrt tp
    loadrt [EMCMOT]EMCMOT servo_period_nsec=[EMCMOT]SERVO_PERIOD num_joints=[TRAJ]AXES# tp=tp kins=trivkins

However the module still does not load and Im unable to see any messages from the module load in any of the log files.

    machinekit@mksoc:~$ cat /home/machinekit/linuxcnc_print.txt
    RUN_IN_PLACE=yes
    LINUXCNC_DIR=
    LINUXCNC_BIN_DIR=/home/machinekit/machinekit/bin
    LINUXCNC_TCL_DIR=/home/machinekit/machinekit/tcl
    LINUXCNC_SCRIPT_DIR=
    LINUXCNC_RTLIB_DIR=/home/machinekit/machinekit/rtlib
    LINUXCNC_CONFIG_DIR=
    LINUXCNC_LANG_DIR=/home/machinekit/machinekit/src/objects
    INIVAR=/home/machinekit/machinekit/libexec/inivar
    HALCMD=halcmd
    LINUXCNC_EMCSH=/usr/bin/wish8.6
    INIFILE=/home/machinekit/machinekit/configs/hm2-stepper/5i25.ini
    PARAMETER_FILE=hm2-stepper.var
    TASK=milltask
    HALUI=
    DISPLAY=axis
    Starting Machinekit server program: linuxcncsvr
    Loading Real Time OS, RTAPI, and HAL_LIB modules
    Starting Machinekit IO program: io
    Killing task linuxcncsvr, PID=2456
    Removing HAL_LIB, RTAPI, and Real Time OS modules
    Removing NML shared memory segments

It is these messages I would expect to be able to see somewhere ?
https://github.com/the-snowwhite/machinekit/commit/6a02948b0d3c5f9e8be244c782e650824cb27ef8#diff-abfe171692b79936d1cee36d274edb06R441

unseenlaser, 2016-02-12:
> the-snowwhite@6a02948#diff-abfe171692b79936d1cee36d274edb06R441

take a look at the /var/log/linuxcnc logfile , if nothing shows up with
dmesg , this is the default location
you may need to change the debug level at the top of the ini file .,

i'm just not fully installed here at the moment to do it in realtime

-- 

The information contained in this message is confidential and is intended
for the addressee only. If you have received this message in error or there
are any problems please notify the originator immediately. The unauthorised
use, disclosure, copying or alteration of this message is strictly
forbidden. This mail and any attachments have been scanned for viruses
prior to leaving the RcTechnix network. RcTechnix will not be liable for
direct, special, indirect or consequential damages arising from alteration
of the contents of this message by a third party or as a result of any
virus being passed on.

RcTechnix reserves the right to monitor and record e-mail messages being
sent to and from this address for the purposes of investigating or
detecting any unauthorised use of its system and ensuring effective
operation.

(c) RcTechnix

the-snowwhite, 2016-02-12:
OK now I do find the messages in:
/var/log/linuxcnc.log

I'm just running halcmd and running the hal file commands from the top.
problem here:

    halcmd: loadrt hm2_soc
    <stdin>:9: insmod failed, returned -1:

cat /var/log/linuxcnc.log

    Feb 12 13:48:02 mksoc msgd:0: hal_lib:1333:rt hm2_soc: PRINT: loading Mesa AnyIO HostMot2 socfpgs driver version 0.8
    Feb 12 13:48:02 mksoc msgd:0: hal_lib:1333:rt hm2_soc: error registering UIO driver

this came after i tried:
    halcmd: loadrt hm2_soc

the second time
    Feb 12 13:55:38 mksoc rtapi:0: hm2_soc: already loaded

the-snowwhite, 2016-02-12:
Seems like I made a mistake in the return value.. recompiling

unseenlaser, 2016-02-12:
great , at least progress

-- 

The information contained in this message is confidential and is intended
for the addressee only. If you have received this message in error or there
are any problems please notify the originator immediately. The unauthorised
use, disclosure, copying or alteration of this message is strictly
forbidden. This mail and any attachments have been scanned for viruses
prior to leaving the RcTechnix network. RcTechnix will not be liable for
direct, special, indirect or consequential damages arising from alteration
of the contents of this message by a third party or as a result of any
virus being passed on.

RcTechnix reserves the right to monitor and record e-mail messages being
sent to and from this address for the purposes of investigating or
detecting any unauthorised use of its system and ensuring effective
operation.

(c) RcTechnix

the-snowwhite, 2016-02-12:
I dont know if its possible to just compile the module with instcomp or comp in hal or
if its possible just to compile the 2 sourcefile separate (hm2_soc.c, hm2_soc.h) and just link them in ?

as it is now I cross-compile the full machinekit folder, compress it and the replace the full machinekit folder in the sd-image rootfs part /home/machinekit/machinekit, which takes about 20 min or so...

Hmmm

the-snowwhite, 2016-02-12:
OK Now I have this to correct:

    machinekit@mksoc:~$ source ~/machinekit/scripts/rip-environment
    machinekit@mksoc:~$ halrun
    msgd:0 stopped
    rtapi:0 stopped
    halcmd: loadrt trivkins
    halcmd: loadrt tp
    halcmd: 
    halcmd: loadrt motmod servo_period_nsec=1000000 num_joints=3
    halcmd: 
    halcmd: loadrt hostmot2
    halcmd: loadrt hm2_soc
    <stdin>:7: insmod failed, returned -1:
    rtapi_rpc(): reply timeout
    See /var/log/linuxcnc.log for more information.
    halcmd: 
    

    "instance=3f0e763c-d197-11e5-b09e-5241648c8fa3" "service=log" "dsn=ipc:///tmp/0.log.a42c8c6b-4025-4f83-ba28-dad21114744a"
    Feb 12 14:45:37 mksoc msgd:0: hal_lib:1366:rt hm2: loading Mesa HostMot2 driver version 0.15
    Feb 12 14:45:42 mksoc msgd:0: hal_lib:1366:rt hm2_soc: PRINT: loading Mesa AnyIO HostMot2 socfpgs driver version 0.8

    Feb 12 14:45:42 mksoc msgd:0: rtapi_app:1366:user signal 11 - 'Segmentation fault' received, dumping core (current dir=/home/machinekit)
    Feb 12 14:45:42 mksoc msgd:0: rtapi_app:1366:user  --- rtapi_app backtrace: ---
    Feb 12 14:45:42 mksoc msgd:0: rtapi_app:1366:user 75601b21 hm2_soc_read     (hal/drivers/mesa-hostmot2/hm2_soc.c:172)
    Feb 12 14:45:42 mksoc msgd:0: rtapi_app:1366:user 751eff69 hm2_register     (hal/drivers/mesa-hostmot2/hostmot2.c:1297)
    Feb 12 14:45:42 mksoc msgd:0: rtapi_app:1366:user 75601cf9 hm2_soc_mmap     (hal/drivers/mesa-hostmot2/hm2_soc.c:411)
    Feb 12 14:45:42 mksoc msgd:0: rtapi_app:1366:user 75601e2b rtapi_app_main   (hal/drivers/mesa-hostmot2/hm2_soc.c:452)
    Feb 12 14:45:42 mksoc msgd:0: rtapi_app:1366:user 00015591 do_load_cmd      (rtapi/rt-preempt/rtapi_app.cc:571)
    Feb 12 14:45:42 mksoc msgd:0: rtapi_app:1366:user 000174b1 rtapi_request    (rtapi/rt-preempt/rtapi_app.cc:837)
    Feb 12 14:45:42 mksoc msgd:0: rtapi_app:1366:user  --------------------
    Feb 12 14:45:42 mksoc msgd:0: rtapi_app exit detected - scheduled shutdown
    Feb 12 14:45:44 mksoc msgd:0: msgd shutting down
    Feb 12 14:45:44 mksoc msgd:0: zeroconf: unregistering 'Log service on mksoc.local pid 1360'
    Feb 12 14:45:44 mksoc msgd:0: log buffer hwm: 0% (7 msgs, 700 bytes out of 524288)
    Feb 12 14:45:44 mksoc msgd:0: normal shutdown - global segment detached

the-snowwhite, 2016-02-12:
Obviously I have done something wrong in:
static int hm2_soc_mmap(void) {

but what ?

the-snowwhite, 2016-02-12:
Ok So it's the hm2 register:

    Feb 12 15:24:08 mksoc msgd:0: hal_lib:1380:rt hm2_soc: PRINT: will now run hm2_register
    Feb 12 15:24:08 mksoc msgd:0: rtapi_app:1380:user signal 11 - 'Segmentation fault' received, dumping core (current dir=/home/machinekit)
    Feb 12 15:24:08 mksoc msgd:0: rtapi_app:1380:user  --- rtapi_app backtrace: ---
    Feb 12 15:24:08 mksoc msgd:0: rtapi_app:1380:user 75245b21 hm2_soc_read     (hal/drivers/mesa-hostmot2/hm2_soc.c:172)
    Feb 12 15:24:08 mksoc msgd:0: rtapi_app:1380:user 75609f69 hm2_register     (hal/drivers/mesa-hostmot2/hostmot2.c:1297)
    Feb 12 15:24:08 mksoc msgd:0: rtapi_app:1380:user 75245d0d hm2_soc_mmap     (hal/drivers/mesa-hostmot2/hm2_soc.c:414)
    Feb 12 15:24:08 mksoc msgd:0: rtapi_app:1380:user 75245e47 rtapi_app_main   (hal/drivers/mesa-hostmot2/hm2_soc.c:455)
    Feb 12 15:24:08 mksoc msgd:0: rtapi_app:1380:user 00015591 do_load_cmd      (rtapi/rt-preempt/rtapi_app.cc:571)
    Feb 12 15:24:08 mksoc msgd:0: rtapi_app:1380:user 000174b1 rtapi_request    (rtapi/rt-preempt/rtapi_app.cc:837)
    Feb 12 15:24:08 mksoc msgd:0: rtapi_app:1380:user  --------------------

here:
https://github.com/the-snowwhite/machinekit/commit/6a02948b0d3c5f9e8be244c782e650824cb27ef8#diff-abfe171692b79936d1cee36d274edb06R411

mhaberler, 2016-02-12:
to debug rtapi_app with gdb, proceed like so:

DEBUG=5 NOSIGHDLR=1 realtime start
in separate window, find the pid of process 'rtapi:0'
sudo gdb -p <pid>
continue
then in halcmd source the hal file which produces the crash

the-snowwhite, 2016-02-12:
Thanks
Problem is that I'm in way over my head and I have never been able to fully understand the c++ objects and pointer stuff:
the seg fault is triggered by this function:

        r = hm2_register(&board->llio, config[0]);

Which I have tried to clone from the hm2_test.c   instead of hm2_pci.c which has way too much pci driver overhead to be in any way compressive for me.

I have however tried to retain the option for more than 1 board (ip interface core) in the config structure + some uio dts probe config structs. Maybe I should just cut that out.

IMHO gdb debugging seems a bit overkill for something as easy and simple as @cdsteinkuehler describes it ?

the-snowwhite, 2016-02-12:
I most likely have made an pointer error or forgotten or added an & wrongly.
However thats outside my expertice and my only way to solve these things are by trial and error until it works ...

the-snowwhite, 2016-02-12:
Hmmmm

    machinekit@mksoc:~$ source /home/machinekit/machinekit/configs/hm2-stepper/hm2-stepper-5i25.hal

    -bash: loadrt: command not found

    -bash: loadrt: command not found

    -bash: loadrt: command not found

    -bash: loadrt: command not found

    -bash: loadrt: command not found

    -bash: /home/machinekit/machinekit/configs/hm2-stepper/hm2-stepper-5i25.hal: line 43: syntax error near unexpected token `('

    -bash: /home/machinekit/machinekit/configs/hm2-stepper/hm2-stepper-5i25.hal: line 43: `loadrt 
    [HOSTMOT2](DRIVER)'

I need to try again after a short break ,,, this is turning out as enexpected painfull .. :-(

the-snowwhite, 2016-02-12:
OK I could not find or remember a way to source a hal file with parameters in an ini file
so I just ran the first 5 lines with halcmd prepended:

  loadrt trivkins
  loadrt tp
  loadrt motmod servo_period_nsec=1000000 num_joints=3
  loadrt hostmot2
  
halcmd loadrt hm2_soc
-->
   
    (gdb) continue
    Continuing.
    [New Thread 0x468e0 (LWP 1469)]

    Program received signal SIGSEGV, Segmentation fault.
    0x75245b22 in hm2_soc_read (this=0x75256654 <hm2_soc_board+16>, addr=256, buffer=0x7ef28898, size=4) at hal/drivers/mesa-hostmot2/hm2_soc.c:172
    172             dst[i] = src[i];

the-snowwhite, 2016-02-12:
Well as this function is copied unchanged from the hm2_pci.c
The seg fault must be due to faulty src or dst memory locations / pointers
here:
    static int hm2_soc_read(hm2_lowlevel_io_t *this, u32 addr, void *buffer, int size) {
        hm2_soc_t *board = this->private;
        int i;
        u32* src = (u32*) (board->base + addr);
        u32* dst = (u32*) buffer;

ArcEye, 2016-02-12:
>  machinekit@mksoc:~$ source /home/machinekit/machinekit/configs/hm2-stepper/hm2-stepper-5i25.hal

Will not work unfortunately because to execute the file requires a second part to the call ie
`halcmd -f /home/machinekit/machinekit/configs/hm2-stepper/hm2-stepper-5i25.hal`
otherwise loadrt etc are not resolved as they are internal halcmd commands.

> the seg fault is triggered by this function:
    r = hm2_register(&board->llio, config[0]);

The function hm2_register() is declared as
hm2_register(hm2_lowlevel_io_t *llio, char *config_string);

Given your 'board' structure or whatever appears to be a pointer given the -> C++ addressing syntax, you may well not need the address of (&) operator.
The fact that it will pass an address of something, albeit wrong, could be why the function is not 
rejecting it on the first test, as it only tests for NULL pointers.

Not had an in-depth look yet, only just managed to get the Altera software working
_(Quartus 15.1.1 fails completely in some generations on Jessie and locks up a VM in compilation no matter how much processor and memory you give it - just managed to get it running properly on windoze 10 :snail: )_

Looks like I need to learn to type faster - you have done something else;)

the-snowwhite, 2016-02-12:
:-) yeah I'm a slow typer also

    machinekit@mksoc:~$ halcmd -f /home/machinekit/machinekit/configs/hm2-stepper/hm2-stepper-5i25.hal
    /home/machinekit/machinekit/configs/hm2-stepper/hm2-stepper-5i25.hal:29: Realtime module 'trivkins' loaded
    /home/machinekit/machinekit/configs/hm2-stepper/hm2-stepper-5i25.hal:33: Realtime module 'tp' loaded
    /home/machinekit/machinekit/configs/hm2-stepper/hm2-stepper-5i25.hal:34: Ini variable '[EMCMOT]EMCMOT' not found.

Just need how to include the .ini file

Quartus 15.1.1 compiles fine in (native) Jessie and Stretch however I have never had the urge to test quartus out in a VM ...??

(Quartus will also not utilize any kind of hyperthreaded cpu cores, so with 16 (intel x2)cores it uses only 8)

Note: I have included the quartus outputs in the mksocfpga repo so you do not need to quartus compile to get the board up.
only u-boot and kernel need compilation.

I would be great if you can help correct the:

   hm2_register(hm2_lowlevel_io_t *llio, char *config_string);

as I obviously have no clue to what I'm talking about trying to describe how it works :-)

ArcEye, 2016-02-12:
> Just need how to include the .ini file

halcmd -i /path/to/inifile -f /path/to/halfile

unseenlaser, 2016-02-12:
hoping to catch up with you shortly Michael , both Arceye and myself are
fighting hardware Lol
i'm building on Jessie x64 , so lets see .

catching up slowly

-- 

The information contained in this message is confidential and is intended
for the addressee only. If you have received this message in error or there
are any problems please notify the originator immediately. The unauthorised
use, disclosure, copying or alteration of this message is strictly
forbidden. This mail and any attachments have been scanned for viruses
prior to leaving the RcTechnix network. RcTechnix will not be liable for
direct, special, indirect or consequential damages arising from alteration
of the contents of this message by a third party or as a result of any
virus being passed on.

RcTechnix reserves the right to monitor and record e-mail messages being
sent to and from this address for the purposes of investigating or
detecting any unauthorised use of its system and ensuring effective
operation.

(c) RcTechnix

ArcEye, 2016-02-12:
> r = hm2_register(&board->llio, config[0]);

The addressing appears correct, because llio is not in itself a pointer
https://github.com/the-snowwhite/machinekit/blob/mksoc/src/hal/drivers/mesa-hostmot2/hm2_soc.h#L56

ArcEye, 2016-02-12:
I will need to look properly tomorrow and compare to the files they are based upon.

There is stuff in there that is making hairs on my neck stand up already, because it uses a pointer called `this`, which is a protected C++ class instance pointer and never used for anything else.

Makes no odds in C, has probably worked like that for years, but immediately caught my eye.

the-snowwhite, 2016-02-12:
Ok
I thought I would give that (&) change a spin meanwhile, and it gave an interesting result:

    machinekit@mksoc:~$ source ~/machinekit/scripts/rip-environment
    machinekit@mksoc:~$ halrun
    Warning - /home/machinekit/machinekit/libexec/rtapi_app_posix not setuid
    'sudo make setuid' missing?
    msgd:0 stopped
    rtapi:0 stopped
    Warning - /home/machinekit/machinekit/libexec/rtapi_app_posix not setuid
    'sudo make setuid' missing?
    halcmd: cant connect to rtapi_app: -1 (uri= uuid=a42c8c6b-4025-4f83-ba28-dad21114744a): rtapi_rpc(): reply timeout

??? seems totally unrelated, but I will push the undo button on that change

BTW:
there seems to be an old config change that has been missed in the 5i25 config:

/home/machinekit/machinekit/configs/hm2-stepper/hm2-stepper-5i25.hal
https://github.com/machinekit/machinekit/blob/master/configs/hm2-stepper/hm2-stepper-5i25.hal#L34

    loadrt [EMCMOT]EMCMOT servo_period_nsec=[EMCMOT]SERVO_PERIOD num_joints=[TRAJ]AXES tp=tp kins=trivkins

-->

    loadrt [EMCMOT]EMCMOT servo_period_nsec=[EMCMOT]SERVO_PERIOD num_joints=[TRAJ]AXES

???

the-snowwhite, 2016-02-12:
-->      hm2_lowlevel_io_t

This struct is actually not used in this function and something I have carried over from the pci driver as I have been unable to figure out if it were needed or not....

the-snowwhite, 2016-02-12:
If you look at hm2_soc.h I have added this struct to reflect the info provided in the device tree,
coming from the hm2soc ip core that im able to customize and add more info to if needed.
The idea is to use it for later dts probing:

    
    struct dts_device_id {
        unsigned short address_width;           //address width /
        unsigned short clocks;                  // clocks /
        struct compatible compatible;
        unsigned short data_width;       /* (class,subclass,prog-if) triplet */
        char name[32];
        unsigned long reg[3];
    };
 
The other structures in that include file are stuff I really would like to exclude and get rid of if possible ?

nilrods, 2016-02-13:
Michael,
I got to the same point you were at today. I had some an issue with the defconfig file not being updated and kernel module not working, but got that fixed. 

Tried an few things and found that it looks to be failing in the hm2_cvsoc_read. I guess that the call to hm2_register must make a call to the hm2_cvsoc_read.

I will see else I can figure out.

Thanks,
Chris

ArcEye, 2016-02-13:
I need to get my board running with the .sof and the SD image before I can do much.
I'll concentrate on that.

ArcEye, 2016-02-13:
Do you have a pre-built .sof file for flashing the Nano?

The design loads in Quartus, but lots of errors reported when I try to generate it.
I did update the IP modules from 14.1 to 15.1, don't know if connected?
Don't know enough about it to figure what is wrong

```
.............
Warning: hps_0.f2h_irq0: Cannot connect clock for irq_mapper.sender
Warning: hps_0.f2h_irq0: Cannot connect reset for irq_mapper.sender
Warning: hps_0.f2h_irq1: Cannot connect clock for irq_mapper_001.sender
Warning: hps_0.f2h_irq1: Cannot connect reset for irq_mapper_001.sender
Info: button_pio: Starting RTL generation for module 'soc_system_button_pio'
Info: button_pio:   Generation command is [exec /media/sde/sde2/root/altera/15.1/quartus/linux64/perl/bin/perl -I /media/sde/sde2/root/altera/15.1/quartus/linux64/perl/lib -I /media/sde/sde2/root/altera/15.1/quartus/sopc_builder/bin/europa -I /media/sde/sde2/root/altera/15.1/quartus/sopc_builder/bin/perl_lib -I /media/sde/sde2/root/altera/15.1/quartus/sopc_builder/bin -I /media/sde/sde2/root/altera/15.1/quartus/../ip/altera/sopc_builder_ip/common -I /media/sde/sde2/root/altera/15.1/quartus/../ip/altera/sopc_builder_ip/altera_avalon_pio -- /media/sde/sde2/root/altera/15.1/quartus/../ip/altera/sopc_builder_ip/altera_avalon_pio/generate_rtl.pl --name=soc_system_button_pio --dir=/tmp/alt6844_9157043160967381239.dir/0001_button_pio_gen/ --quartus_dir=/media/sde/sde2/root/altera/15.1/quartus --verilog --config=/tmp/alt6844_9157043160967381239.dir/0001_button_pio_gen//soc_system_button_pio_component_configuration.pl  --do_build_sim=0  ]
Info: button_pio: perl: warning: Setting locale failed.
Info: button_pio: perl: warning: Please check that your locale settings:
Info: button_pio: 	LANGUAGE = "en_GB:en",
Info: button_pio: 	LC_ALL = (unset),
Info: button_pio: 	LC_CTYPE = "en_US.UTF-8",
Info: button_pio: 	LANG = "C"
Info: button_pio:     are supported and installed on your system.
Info: button_pio: perl: warning: Falling back to the standard locale ("C").
Error: button_pio: Failed to generate module soc_system_button_pio
Info: button_pio: Done RTL generation for module 'soc_system_button_pio'
Info: button_pio: "soc_system" instantiated altera_avalon_pio "button_pio"
Error: Generation stopped, 20 or more modules remaining
Info: soc_system: Done "soc_system" with 18 modules, 2 files
Error: qsys-generate failed with exit code 1: 2 Errors, 4 Warnings
Info: Finished: Create HDL design files for synthesis
```

I will try it again on windoze, in case this is the same core problem with Quartus on Jessie I had before.

nilrods, 2016-02-13:
Arceye, I just followed the instructions Michael sent me from below. I am just newly working on the nano board so not sure what the sof file is. I just built the sd card image and run from there. I built on Centos so I did have an issue with the defconfig file but other than that went pretty smooth. 

https://github.com/the-snowwhite/mksocfpga

If you can't get that to work I can take a copy of my sd card and try to upload somewhere for you.

ArcEye, 2016-02-13:
OK, It is a different work flow to that I have been using.
I have been flashing the board with jtag and then accessing it from software in the booted image.

This looks like setting the board up from uboot and then booting to use it.

If all else fails, RTFM :laughing:

nilrods, 2016-02-13:
Ok, I think I got the driver to load(hopefully). But I could be way off here. I haven't done any work with the hostmot drivers before so I may have messed something up.
Michael, it looks like the base address for the driver wasn't being set in mmap function.

Here is what I get when I load a copy of the hm2-stepper driver.
I think I have something I need to change in the hal file now.

machinekit@mksoc:~/machinekit/src$ sudo tail -n 50 /var/log/linuxcnc.log
Feb 13 16:03:38 mksoc msgd:0: hal_lib:4698:rt hm2/hm2_5i25.1: 34 I/O Pins used:
Feb 13 16:03:38 mksoc msgd:0: hal_lib:4698:rt hm2/hm2_5i25.1:     IO Pin 000 (P3-01): IOPort
Feb 13 16:03:38 mksoc msgd:0: hal_lib:4698:rt hm2/hm2_5i25.1:     IO Pin 001 (P3-14): PWMGen #0, pin Out0 (PWM or Up) (Output)
Feb 13 16:03:38 mksoc msgd:0: hal_lib:4698:rt hm2/hm2_5i25.1:     IO Pin 002 (P3-02): StepGen #0, pin Step (Output)
Feb 13 16:03:38 mksoc msgd:0: hal_lib:4698:rt hm2/hm2_5i25.1:     IO Pin 003 (P3-15): IOPort
Feb 13 16:03:38 mksoc msgd:0: hal_lib:4698:rt hm2/hm2_5i25.1:     IO Pin 004 (P3-03): StepGen #0, pin Direction (Output)
Feb 13 16:03:38 mksoc msgd:0: hal_lib:4698:rt hm2/hm2_5i25.1:     IO Pin 005 (P3-16): IOPort
Feb 13 16:03:38 mksoc msgd:0: hal_lib:4698:rt hm2/hm2_5i25.1:     IO Pin 006 (P3-04): StepGen #1, pin Step (Output)
Feb 13 16:03:38 mksoc msgd:0: hal_lib:4698:rt hm2/hm2_5i25.1:     IO Pin 007 (P3-17): IOPort
Feb 13 16:03:38 mksoc msgd:0: hal_lib:4698:rt hm2/hm2_5i25.1:     IO Pin 008 (P3-05): StepGen #1, pin Direction (Output)
Feb 13 16:03:38 mksoc msgd:0: hal_lib:4698:rt hm2/hm2_5i25.1:     IO Pin 009 (P3-06): StepGen #2, pin Step (Output)
Feb 13 16:03:38 mksoc msgd:0: hal_lib:4698:rt hm2/hm2_5i25.1:     IO Pin 010 (P3-07): StepGen #2, pin Direction (Output)
Feb 13 16:03:38 mksoc msgd:0: hal_lib:4698:rt hm2/hm2_5i25.1:     IO Pin 011 (P3-08): StepGen #3, pin Step (Output)
Feb 13 16:03:38 mksoc msgd:0: hal_lib:4698:rt hm2/hm2_5i25.1:     IO Pin 012 (P3-09): StepGen #3, pin Direction (Output)
Feb 13 16:03:38 mksoc msgd:0: hal_lib:4698:rt hm2/hm2_5i25.1:     IO Pin 013 (P3-10): IOPort
Feb 13 16:03:38 mksoc msgd:0: hal_lib:4698:rt hm2/hm2_5i25.1:     IO Pin 014 (P3-11): IOPort
Feb 13 16:03:38 mksoc msgd:0: hal_lib:4698:rt hm2/hm2_5i25.1:     IO Pin 015 (P3-12): IOPort
Feb 13 16:03:38 mksoc msgd:0: hal_lib:4698:rt hm2/hm2_5i25.1:     IO Pin 016 (P3-13): IOPort
Feb 13 16:03:38 mksoc msgd:0: hal_lib:4698:rt hm2/hm2_5i25.1:     IO Pin 017 (P2-01): IOPort
Feb 13 16:03:38 mksoc msgd:0: hal_lib:4698:rt hm2/hm2_5i25.1:     IO Pin 018 (P2-14): PWMGen #1, pin Out0 (PWM or Up) (Output)
Feb 13 16:03:38 mksoc msgd:0: hal_lib:4698:rt hm2/hm2_5i25.1:     IO Pin 019 (P2-02): StepGen #4, pin Step (Output)
Feb 13 16:03:38 mksoc msgd:0: hal_lib:4698:rt hm2/hm2_5i25.1:     IO Pin 020 (P2-15): IOPort
Feb 13 16:03:38 mksoc msgd:0: hal_lib:4698:rt hm2/hm2_5i25.1:     IO Pin 021 (P2-03): StepGen #4, pin Direction (Output)
Feb 13 16:03:38 mksoc msgd:0: hal_lib:4698:rt hm2/hm2_5i25.1:     IO Pin 022 (P2-16): IOPort
Feb 13 16:03:38 mksoc msgd:0: hal_lib:4698:rt hm2/hm2_5i25.1:     IO Pin 023 (P2-04): StepGen #5, pin Step (Output)
Feb 13 16:03:38 mksoc msgd:0: hal_lib:4698:rt hm2/hm2_5i25.1:     IO Pin 024 (P2-17): IOPort
Feb 13 16:03:38 mksoc msgd:0: hal_lib:4698:rt hm2/hm2_5i25.1:     IO Pin 025 (P2-05): StepGen #5, pin Direction (Output)
Feb 13 16:03:38 mksoc msgd:0: hal_lib:4698:rt hm2/hm2_5i25.1:     IO Pin 026 (P2-06): StepGen #6, pin Step (Output)
Feb 13 16:03:38 mksoc msgd:0: hal_lib:4698:rt hm2/hm2_5i25.1:     IO Pin 027 (P2-07): StepGen #6, pin Direction (Output)
Feb 13 16:03:38 mksoc msgd:0: hal_lib:4698:rt hm2/hm2_5i25.1:     IO Pin 028 (P2-08): StepGen #7, pin Step (Output)
Feb 13 16:03:38 mksoc msgd:0: hal_lib:4698:rt hm2/hm2_5i25.1:     IO Pin 029 (P2-09): StepGen #7, pin Direction (Output)
Feb 13 16:03:38 mksoc msgd:0: hal_lib:4698:rt hm2/hm2_5i25.1:     IO Pin 030 (P2-10): IOPort
Feb 13 16:03:38 mksoc msgd:0: hal_lib:4698:rt hm2/hm2_5i25.1:     IO Pin 031 (P2-11): IOPort
Feb 13 16:03:38 mksoc msgd:0: hal_lib:4698:rt hm2/hm2_5i25.1:     IO Pin 032 (P2-12): IOPort
Feb 13 16:03:38 mksoc msgd:0: hal_lib:4698:rt hm2/hm2_5i25.1:     IO Pin 033 (P2-13): IOPort
Feb 13 16:03:38 mksoc msgd:0: hal_lib:4698:rt hm2/hm2_5i25.1: registered
Feb 13 16:03:38 mksoc msgd:0: hal_lib:4698:rt hm2_cvsoc: initialized AnyIO hm2_cvsoc_board
Feb 13 16:03:38 mksoc msgd:0: hal_lib:4712:user hal_add_funct_to_thread:269 HAL error: function 'hm2_5i25.0.read' not found
Feb 13 16:03:39 mksoc msgd:0: hal_lib:4698:rt hm2_cvsoc: UIO driver unloaded
Feb 13 16:03:39 mksoc msgd:0: hal_lib:4698:rt hm2: unloading
Feb 13 16:03:42 mksoc rtapi:0: unload: 'trivkins' not loaded

nilrods, 2016-02-13:
ok I must have done something a little different than the other hostmot drivers do. Looks like to hal my changes made the board instance is 1 based instead or 0 based for the 5i25 board.
I got the driver to fully load with modified hm2-stepper.hal and 5i25.ini and updated code.
Got axis to load but then got some gui error and so it shut down.  I guess I need to figure out how to run the mklauncher to run headless to see if things are actually working.

unseenlaser, 2016-02-13:
have you details of the gui error ?

-- 

The information contained in this message is confidential and is intended
for the addressee only. If you have received this message in error or there
are any problems please notify the originator immediately. The unauthorised
use, disclosure, copying or alteration of this message is strictly
forbidden. This mail and any attachments have been scanned for viruses
prior to leaving the RcTechnix network. RcTechnix will not be liable for
direct, special, indirect or consequential damages arising from alteration
of the contents of this message by a third party or as a result of any
virus being passed on.

RcTechnix reserves the right to monitor and record e-mail messages being
sent to and from this address for the purposes of investigating or
detecting any unauthorised use of its system and ensuring effective
operation.

(c) RcTechnix

nilrods, 2016-02-13:
Here is the error I get. I assume some gui dependency not on the image Michael created.
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
X Error of failed request:  GLXBadRenderRequest
  Major opcode of failed request:  148 (GLX)
  Minor opcode of failed request:  1 (X_GLXRender)
  Serial number of failed request:  4215
  Current serial number in output stream:  4216
Shutting down and cleaning up Machinekit...
Cleanup done
Machinekit terminated with an error.  You can find more information in the log:
    /home/machinekit/linuxcnc_debug.txt
and
    /home/machinekit/linuxcnc_print.txt
as well as in the output of the shell command 'dmesg' and in the terminal
machinekit@mksoc:~/machinekit$

unseenlaser, 2016-02-13:
Note:

ok my guess but based on a debian pc disto , not on an arm processor etc ,
but  might give you a clue

The swrast thing is the software renderer. That means it's not finding the
hardware driver for your graphics card. There's a bunch of libGL libraries
installed and a bunch of of symbolic links to those libraries. To see these
run this from the cmd line.

sudo find /usr -iname "*libGL.so*" -exec ls -l {} \;

Now the probable cause of your problem is that installing graphics drivers
sometimes break these symbolic links. (Specifically
/usr/local/lib/libGL.so.1.2.0 is likely to be either the wrong lib or a sym
link to the wrong link).

To work out what library the opengl programs are trying to run you can turn
on a bit of verbosity and run a simple opengl program. You can verify this
using the standard opengl test program:

export LIBGL_DEBUG=verbose
glxgears

Hopefully that will fail in the same way as SFML. With the LIBGL_DEBUG it
should tell you what opengl library it's trying to load. Furthermore.. the
lib it will be trying to load will almost certainly be
/usr/local/lib/libGL.so.1.2.0

So the solution (in this case) is to make sure that
/usr/local/lib/libGL.so.1.2.0 is a symbolic link pointing at the right
opengl library. In my case I have the nvidia 3.40 driver so I ran:

ln -s /usr/lib/nvidia-340/libGL.so.1 /usr/local/lib/libGL.so.1.2.0

But you'll want to point it at the opengl lib that is appropriate for you
(listed in the first find cmd).
In summary: installing (proprietary) graphics drivers can break the symbol
links used for opengl libs. To solve this problem manually fix the symbolic
links (fix /usr/local/lib/libGL.so.1.2.0 first).

-- 

The information contained in this message is confidential and is intended
for the addressee only. If you have received this message in error or there
are any problems please notify the originator immediately. The unauthorised
use, disclosure, copying or alteration of this message is strictly
forbidden. This mail and any attachments have been scanned for viruses
prior to leaving the RcTechnix network. RcTechnix will not be liable for
direct, special, indirect or consequential damages arising from alteration
of the contents of this message by a third party or as a result of any
virus being passed on.

RcTechnix reserves the right to monitor and record e-mail messages being
sent to and from this address for the purposes of investigating or
detecting any unauthorised use of its system and ensuring effective
operation.

(c) RcTechnix

the-snowwhite, 2016-02-13:
I have been to a seminar all day, so I'm trying to do some catch up now.
Great to see there has been some progress.

I put in the xorg package and a dummy monitor config to  be able to ssh -X into the board from my debian pc.

This does work with the machinekit / linuxcnc gui, weather the xorg thing and config really is needed on the nano or not.

However the nano setup is headlles as there is no free graphics core available, a framebuffer could be used (no opengl), however you would also be using most of the pins to interface to a monitor.

So eth0 --> ssh -X   provides the opengl graphich support instead (for machinekit client etc)

nilrods, 2016-02-13:
good info. I tried to run the glxgears with debug but that is not installed. the library looks ok.
sudo find /usr -iname "LibGL.so" -exec ls -l {} \;
lrwxrwxrwx 1 root root 14 Aug 19 22:28 /usr/lib/arm-linux-gnueabihf/libGL.so -> libGL.so.1.2.0

 I will probably just try getting  the mklauncher configured since even if I could get the opengl running the graphics would probably be slow anyway running on the board. My intention was always to run it with remote graphics anyway.

Michael, Ahh that makes sense. I was running on a vncserver session. Let me try a remote ssh -X session.

the-snowwhite, 2016-02-13:
ssh -X machinekit@mksoc
should connect you

the-snowwhite, 2016-02-13:
@ArcEye 
There are some msel dip settings near the power connector that need to be correctly setup for boot selection.
This dist is set to the  standard linux soc config style where it boots from sd-card and then loads the config (.sof -- rbf.) and the devicetree .dtb from the boot partition1 )(+ the zImage kernel).
therefore the need to add the u-boot env vars on first boot.

If you (jtag)flash the config AND want to use this for booting you are into a very different boot scenario,
which I would reserve to If you feel the need to boot from usb  pxe-net boot without-sd or something else more advanced...

nilrods, 2016-02-13:
Michael,
Thanks that did it, my bad.
Got axis running, but my X axis is just running away. When I try to toggle machine power it turns on logs a joint 0 following error and turns back off. I need to check my updated hal config and see if I messed something up when changing all the indexes for the board or if the data is coming from the fpga.

Attached is a screenshot of axis running.
[image]

the-snowwhite, 2016-02-13:
@nilrods 
I sounds like you are pointing to that the:

.h  
      
    volatile void *virtual_base;
   
.c
        
    virtual_base = mmap( NULL, HM2REG_IO_0_SPAN, ( PROT_READ | PROT_WRITE ), MAP_SHARED, uio_fd, 0);

needs some change's ?

I'm really in the backseat here until you are ready to share what changes enabled you to get the driver to load..

[187 more replies]

closed 2016-04-16 by the-snowwhite