machinekit/machinekit #869: armhf only hal module fails travis check

the-snowwhite, 2016-02-14:
this PR is a hal module designed for the armhf mksocfpga port / platform only:

https://github.com/machinekit/machinekit/pull/868

I have added an config option:

    CONFIG_ARGS=" --with-rt-preempt --with-posix --with-platform=socfpga"

cloned from the beaglebone setup.

     --with-platform=socfpga

Seems to work correctly with the rip build

https://github.com/the-snowwhite/mksocfpga/blob/master/Scripts/MK-Cross-build/mk-rip-build.sh#L88

However there seems to be something missing to get the travis package builder to understand ?

kinsamanka, 2016-02-14:
This build fails because the driver gets compiled when running on x86.

You need to add something like these: https://github.com/machinekit/machinekit/blob/master/src/Makefile#L1254-L1274

the-snowwhite, 2016-02-14:
Yes
This was what I was able to put in:

https://github.com/the-snowwhite/machinekit/blob/mksoc/src/Makefile#L1269

#ifdef TARGET_PLATFORM_SOCFPGA
#ifeq ($(BUILD_SYS),user-dso)
#obj-$(CONFIG_HAL_UIO) += ???.o
#hm2_soc-uio-objs := hal/drivers/???.o
#endif

the-snowwhite, 2016-02-14:
ifdef TARGET_PLATFORM_SOCFPGA
ifeq ($(BUILD_SYS),user-dso)
obj-$(CONFIG_HAL_UIO) += hm2_soc.o
hm2_soc-uio-objs := hal/drivers/hm2_soc.o
endif

the-snowwhite, 2016-02-14:
OK this would seem to be the idea ... commiting

kinsamanka, 2016-02-14:
hashtag means comment for makefiles ;-)

the-snowwhite, 2016-02-14:
:-)
Hmmm
I felt it would be best to comment these lines out until I was sure about what to put in instead ?:
obj-$(CONFIG_HAL_UIO) += ???.o
hm2_soc-uio-objs := hal/drivers/???.o

But may be I should just go ahead and be bold next time ?

the-snowwhite, 2016-02-14:
OK it looks like only the x64 checks are complaining now...:
The beaglebone and rpi use following configure options:

    # for the Beaglebone, use:
    #CONFIG_ARGS=" --with-xenomai --with-posix --with-platform=beaglebone"

    # for the raspberry, use this:
    #CONFIG_ARGS=" --with-xenomai --with-posix --with-platform=raspberry"

So I have tried to inject this parallel option for the mksocfpga platform:

    # for the cyclone v soc install use:
    #CONFIG_ARGS=" --with-rt-preempt --with-posix --with-extra-kernel-sources=/home/mib/Documents/Altera/WS2/test/rocketboards"
    CONFIG_ARGS=" --with-rt-preempt --with-posix --with-platform=socfpga"

where I'm hoping the 

    --with-platform=

is what's supposed to do all the magic ?

the-snowwhite, 2016-02-14:
Is travis supposed to be able to recognize the  socfpga platform ?

mhaberler, 2016-02-26:
you need to read through src/configure.ac to see what the --with-platform flag does

the-snowwhite, 2016-03-01:
Thanks (passing checks now)

Due to being stuck in a 4.4 soc kernel build without network connection, and a crosscompile setup that keeps changing my hosts hostname (which puts me offline).

I would recommend someone running a test build before accepting the PR.
There may be some more work to do to get the needed hostmot2 objects to get pulled in....

the-snowwhite, 2016-03-07:
#865

closed 2016-03-07 by the-snowwhite