machinekit/machinekit #870: instantiatiable component `offset` "pincount" parameter is unknown

luminize, 2016-02-14:
@ArcEye @mhaberler 
I either need to load 5 offset instances, or (seems smarter) load 1 instance with 5 pins.
I'm following `man offset`:
```
USAGE SYNOPSIS

       loadrt offset

       newinst  offset  <newinstname> [ pincount=N | iprefix=prefix ] [instan‐
       ceparamX=X | argX=X ]
```
I first `loadrt offset`
then I try `newinst offset joint_offset pincount=5` but the parameter is unknown
```
newinst offset joint_offset pincount=5
<stdin>:11: rc=-1
Unknown parameter `pincount=5'
passing args for 'joint_offset' failed: 'pincount=5'halcmd: 
```
Looking at `src/hal.i_components/offset.icomp` i do not find a reference that this comp can generate more that 1 pincount. Is that correct?

ArcEye, 2016-02-14:
The man file header entry is generic in terms of possible arguments to newinst, does not apply to all components

I will change it to make that clearer later.

It would be easy to rewrite the component to provide a variable number of inputs and outputs, thus utilising the `pincount` parameter.

Arguably the current component in having separate update functions for output and feedback, is unnecessarily complex and risks those functions returning values at different times, when the values used to calculate them have changed.

Don't know if backward compatibility is required with this one.
There are no instances of its use in the sample configs, but no telling what anyone has in their own files.

ArcEye, 2016-02-14:
If you give me a while, I will write a multi-pin version for you to test, probably with the same functions to make it backwardly compatible for now.

Have to go out for a while

luminize, 2016-02-14:
no need to hurry, I manage with 5 different components for now. I was wondering if i was misunderstood.

ArcEye, 2016-02-14:
icomp man pages now read thus;

```
USAGE SYNOPSIS

       loadrt offsetn

       newinst offsetn <newinstname> [ pincount=N | iprefix=prefix ] [instanceparamX=X | argX=X ]

       ( args in [ ] denote possible args and parameters, may not be used in all components )
```

New component `offsetn` written, which has 1 - 8 pin sets

```
mick@INTEL-i7:/usr/src/machinekit/src$ DEBUG=5 realtime restart
mick@INTEL-i7:/usr/src/machinekit/src$ halcmd loadrt offsetn
<commandline>:0: Realtime module 'offsetn' loaded
mick@INTEL-i7:/usr/src/machinekit/src$ halcmd show pin
Component Pins:
  Comp   Inst Type  Dir         Value  Name                             Epsilon Flags  linked to:
    78     80 float IN              0  offsetn.0.0.fb-in             	0.000010	--l-
    78     80 float OUT             0  offsetn.0.0.fb-out            	0.000010	--l-
    78     80 float IN              0  offsetn.0.0.in                	0.000010	--l-
    78     80 float IN              0  offsetn.0.0.offset            	0.000010	--l-
    78     80 float OUT             0  offsetn.0.0.out               	0.000010	--l-
    78     80 s32   OUT             0  offsetn.0.update-feedback.time		----
    78     80 s32   I/O             0  offsetn.0.update-feedback.tmax		----
    78     80 bit   OUT         FALSE  offsetn.0.update-feedback.tmax		----
    78     80 s32   OUT             0  offsetn.0.update-output.time  		----
    78     80 s32   I/O             0  offsetn.0.update-output.tmax  		----
    78     80 bit   OUT         FALSE  offsetn.0.update-output.tmax-i		----

mick@INTEL-i7:/usr/src/machinekit/src$ halcmd newinst offsetn offset5 pincount=5
mick@INTEL-i7:/usr/src/machinekit/src$ halcmd show pin
Component Pins:
  Comp   Inst Type  Dir         Value  Name                             Epsilon Flags  linked to:
    78    102 float IN              0  offset5.0.fb-in               	0.000010	--l-
    78    102 float OUT             0  offset5.0.fb-out              	0.000010	--l-
    78    102 float IN              0  offset5.0.in                  	0.000010	--l-
    78    102 float IN              0  offset5.0.offset              	0.000010	--l-
    78    102 float OUT             0  offset5.0.out                 	0.000010	--l-
    78    102 float IN              0  offset5.1.fb-in               	0.000010	--l-
    78    102 float OUT             0  offset5.1.fb-out              	0.000010	--l-
    78    102 float IN              0  offset5.1.in                  	0.000010	--l-
    78    102 float IN              0  offset5.1.offset              	0.000010	--l-
    78    102 float OUT             0  offset5.1.out                 	0.000010	--l-
    78    102 float IN              0  offset5.2.fb-in               	0.000010	--l-
    78    102 float OUT             0  offset5.2.fb-out              	0.000010	--l-
    78    102 float IN              0  offset5.2.in                  	0.000010	--l-
    78    102 float IN              0  offset5.2.offset              	0.000010	--l-
    78    102 float OUT             0  offset5.2.out                 	0.000010	--l-
    78    102 float IN              0  offset5.3.fb-in               	0.000010	--l-
    78    102 float OUT             0  offset5.3.fb-out              	0.000010	--l-
    78    102 float IN              0  offset5.3.in                  	0.000010	--l-
    78    102 float IN              0  offset5.3.offset              	0.000010	--l-
    78    102 float OUT             0  offset5.3.out                 	0.000010	--l-
    78    102 float IN              0  offset5.4.fb-in               	0.000010	--l-
    78    102 float OUT             0  offset5.4.fb-out              	0.000010	--l-
    78    102 float IN              0  offset5.4.in                  	0.000010	--l-
    78    102 float IN              0  offset5.4.offset              	0.000010	--l-
    78    102 float OUT             0  offset5.4.out                 	0.000010	--l-
    78    102 s32   OUT             0  offset5.update-feedback.time  		----
    78    102 s32   I/O             0  offset5.update-feedback.tmax  		----
    78    102 bit   OUT         FALSE  offset5.update-feedback.tmax-i		----
    78    102 s32   OUT             0  offset5.update-output.time    		----
    78    102 s32   I/O             0  offset5.update-output.tmax    		----
    78    102 bit   OUT         FALSE  offset5.update-output.tmax-inc		----
    78     80 float IN              0  offsetn.0.0.fb-in             	0.000010	--l-
    78     80 float OUT             0  offsetn.0.0.fb-out            	0.000010	--l-
    78     80 float IN              0  offsetn.0.0.in                	0.000010	--l-
    78     80 float IN              0  offsetn.0.0.offset            	0.000010	--l-
    78     80 float OUT             0  offsetn.0.0.out               	0.000010	--l-
    78     80 s32   OUT             0  offsetn.0.update-feedback.time		----
    78     80 s32   I/O             0  offsetn.0.update-feedback.tmax		----
    78     80 bit   OUT         FALSE  offsetn.0.update-feedback.tmax		----
    78     80 s32   OUT             0  offsetn.0.update-output.time  		----
    78     80 s32   I/O             0  offsetn.0.update-output.tmax  		----
    78     80 bit   OUT         FALSE  offsetn.0.update-output.tmax-i		----
```

https://github.com/ArcEye/machinekit/commit/2c5b2534e36336abc8f1ad58ad8a493191963705

When you are happy with it I can do a PR

ArcEye, 2016-02-14:
Proof of concept and shows addressing syntax

```
mick@INTEL-i7:/usr/git/Machinekit/machinekit$ halcmd loadrt offsetn
<commandline>:0: Realtime module 'offsetn' loaded
mick@INTEL-i7:/usr/git/Machinekit/machinekit$ halcmd show pin
Component Pins:
  Comp   Inst Type  Dir         Value  Name                             Epsilon Flags  linked to:
   156    158 float IN              0  offsetn.0.0.fb-in             	0.000010	--l-
   156    158 float OUT             0  offsetn.0.0.fb-out            	0.000010	--l-
   156    158 float IN              0  offsetn.0.0.in                	0.000010	--l-
   156    158 float IN              0  offsetn.0.0.offset            	0.000010	--l-
   156    158 float OUT             0  offsetn.0.0.out               	0.000010	--l-
   156    158 s32   OUT             0  offsetn.0.update-feedback.time		----
   156    158 s32   I/O             0  offsetn.0.update-feedback.tmax		----
   156    158 bit   OUT         FALSE  offsetn.0.update-feedback.tmax		----
   156    158 s32   OUT             0  offsetn.0.update-output.time  		----
   156    158 s32   I/O             0  offsetn.0.update-output.tmax  		----
   156    158 bit   OUT         FALSE  offsetn.0.update-output.tmax-i		----

mick@INTEL-i7:/usr/git/Machinekit/machinekit$ halcmd newthread servo 1000000 fp
mick@INTEL-i7:/usr/git/Machinekit/machinekit$ halcmd addf offsetn.0.update-output servo
<commandline>:0: Function 'offsetn.0.update-output' added to thread 'servo', rmb=0 wmb=0
mick@INTEL-i7:/usr/git/Machinekit/machinekit$ halcmd addf offsetn.0.update-feedback servo
<commandline>:0: Function 'offsetn.0.update-feedback' added to thread 'servo', rmb=0 wmb=0

mick@INTEL-i7:/usr/git/Machinekit/machinekit$ halcmd setp offsetn.0.0.in 21.5
<commandline>:0: setting parameter 'offsetn.0.0.in' to '21.5'
<commandline>:0: Pin 'offsetn.0.0.in' set to 21.5
mick@INTEL-i7:/usr/git/Machinekit/machinekit$ halcmd setp offsetn.0.0.offset 3.0
<commandline>:0: setting parameter 'offsetn.0.0.offset' to '3.0'
<commandline>:0: Pin 'offsetn.0.0.offset' set to 3.0
mick@INTEL-i7:/usr/git/Machinekit/machinekit$ halcmd setp offsetn.0.0.fb-in 122.6
<commandline>:0: setting parameter 'offsetn.0.0.fb-in' to '122.6'
<commandline>:0: Pin 'offsetn.0.0.fb-in' set to 122.6
mick@INTEL-i7:/usr/git/Machinekit/machinekit$ halcmd start
<commandline>:0: Realtime threads started

mick@INTEL-i7:/usr/git/Machinekit/machinekit$ halcmd show pin
Component Pins:
  Comp   Inst Type  Dir         Value  Name                             Epsilon Flags  linked to:
   156    158 float IN          122.6  offsetn.0.0.fb-in             	0.000010	--l-
   156    158 float OUT         119.6  offsetn.0.0.fb-out            	0.000010	--l-
   156    158 float IN           21.5  offsetn.0.0.in                	0.000010	--l-
   156    158 float IN              3  offsetn.0.0.offset            	0.000010	--l-
   156    158 float OUT          24.5  offsetn.0.0.out               	0.000010	--l-
   156    158 s32   OUT           289  offsetn.0.update-feedback.time		----
   156    158 s32   I/O          2305  offsetn.0.update-feedback.tmax		----
   156    158 bit   OUT         FALSE  offsetn.0.update-feedback.tmax		----
   156    158 s32   OUT           390  offsetn.0.update-output.time  		----
   156    158 s32   I/O          1651  offsetn.0.update-output.tmax  		----
   156    158 bit   OUT         FALSE  offsetn.0.update-output.tmax-i		----
    66        s32   OUT       1001074  servo.curr-period             		----
    66        s32   OUT           679  servo.time                    		----
    66        s32   I/O          3072  servo.tmax                    		----

mick@INTEL-i7:/usr/git/Machinekit/machinekit$ halcmd stop
<commandline>:0: Realtime threads stopped
mick@INTEL-i7:/usr/git/Machinekit/machinekit$ halcmd unloadrt offsetn
<commandline>:0: Realtime module 'offsetn' unloaded
mick@INTEL-i7:/usr/git/Machinekit/machinekit$ 
```

luminize, 2016-02-14:
wonderfull!. Can I pull your branch somewhere?

ArcEye, 2016-02-15:
>  ArcEye@2c5b253

Got truncated by github and buried somewhat in the other text of a couple of posts back

luminize, 2016-02-20:
@ArcEye I've cherry-picked your commit. Had an error with compiling where there was a `local_pincount` variable uninitialised.
I made this commit https://github.com/luminize/machinekit/commit/770247dc2d2366361851605c3a7deb71d79b9a6e to fix that but I'm unsure if this was correct since I don't get correct `joint_offset.n.out` pin values (can't replicate your example above).

my `joint_interpolator.n.curr-pos` feeds the `joint_offset.in` pins
```
471    473 float OUT             0  joint_interpolator.0.curr-acc 	0.000010	0
471    473 float OUT     0.7060226  joint_interpolator.0.curr-pos 	0.000010	0 ==> s_joint_val0
471    473 float OUT             0  joint_interpolator.0.curr-vel 	0.000010	0
471    473 float OUT    -0.5018006  joint_interpolator.0.end-acc  	0.000010	0
471    473 float OUT     0.7060226  joint_interpolator.0.end-pos  	0.000010	0
471    473 float OUT             0  joint_interpolator.0.end-vel  	0.000010	0
471    473 float OUT             0  joint_interpolator.1.curr-acc 	0.000010	0
471    473 float OUT    -0.2421496  joint_interpolator.1.curr-pos 	0.000010	0 ==> s_joint_val1
471    473 float OUT             0  joint_interpolator.1.curr-vel 	0.000010	0
471    473 float OUT     0.3289144  joint_interpolator.1.end-acc  	0.000010	0
471    473 float OUT    -0.2421496  joint_interpolator.1.end-pos  	0.000010	0
471    473 float OUT             0  joint_interpolator.1.end-vel  	0.000010	0
471    473 u32   IN     0x00000005  joint_interpolator.degree     			0
471    473 float OUT     0.5922621  joint_interpolator.duration   	0.000010	0
471    473 float IN         0.0001  joint_interpolator.epsilon    	0.000010	0
471    473 bit   IN           TRUE  joint_interpolator.jitter-corr			0
471    473 float OUT        15.892  joint_interpolator.progress   	0.000010	0
471    473 s32   OUT             0  joint_interpolator.serial     			0
471    473 s32   OUT          4320  joint_interpolator.update.time			0
520    521 float IN              0  joint_offset.0.fb-in          	0.000010	0 <== s_joint_off0-fb
520    521 float OUT             0  joint_offset.0.fb-out         	0.000010	0
520    521 float IN      0.7060226  joint_offset.0.in             	0.000010	0 <== s_joint_val0
520    521 float IN              0  joint_offset.0.offset         	0.000010	0
520    521 float OUT             0  joint_offset.0.out            	0.000010	0 ==> s_joint_off0-pos
520    521 float IN             -0  joint_offset.1.fb-in          	0.000010	0 <== s_joint_off1-fb
520    521 float OUT             0  joint_offset.1.fb-out         	0.000010	0
520    521 float IN     -0.2421496  joint_offset.1.in             	0.000010	0 <== s_joint_val1
520    521 float IN              0  joint_offset.1.offset         	0.000010	0
520    521 float OUT             0  joint_offset.1.out            	0.000010	0 ==> s_joint_off1-pos
520    521 float IN             -0  joint_offset.2.fb-in          	0.000010	0 <== s_joint_off2-fb
520    521 float OUT             0  joint_offset.2.fb-out         	0.000010	0
520    521 float IN      0.6933603  joint_offset.2.in             	0.000010	0 <== s_joint_val2
520    521 float IN              0  joint_offset.2.offset         	0.000010	0
520    521 float OUT             0  joint_offset.2.out            	0.000010	0 ==> s_joint_off2-pos
```

I've added the component to the thread as seen here:
```
halcmd: show thread
Realtime Threads (flavor: rt-preempt) :
     Period  FP     Name               (     Time, Max-Time ) flags
    1000000  YES          servo_thread (    91992,   283080 ) 
                  1 hm2_5i25.0.read
                  2 siggen.0.update
                  3 joint_interpolator.update
                  4 joint_stream
                  5 joint_offset.update-output
                  6 joint_offset.update-feedback
                  7 hm2_5i25.0.write
                  8 hm2_5i25.0.pet_watchdog
                  9 scope.sample
```

ArcEye, 2016-02-20:
The `local_pincount` means I was probably using the `ck-integration` branch that @mhaberler and I did, which still has not yet been merged.
`pincount should work.

I will have to look at the `MAXCOUNT` macro too, think that was probably in `ck-integration` too, so won't work on a normal branch.

I'll have to come back to you, when I have built it again on the current _master_

luminize, 2016-02-20:
ok, thanks

ArcEye, 2016-02-20:
Found it.

`local_pincount` did not commence until `ck-integration` branch, but the master already has the mechanism as `localpincount`
It also has `MAXCOUNT`

The underscore between the words began in `ck-integration`, when other local variables were added
to the ip-> struct using the same format.

By using `pincount` you were accessing the kernel parameter value, but that is zeroed after having been transferred to localpincount, to prevent kernel param persistence problems we had, when the user creates another instance of the same component.
That is why the values did not change.

So change 'pincount' to 'localpincount' and all should work as intended.
I will amend my commit, but I have tested similar to my previous and the pins update.

luminize, 2016-02-20:
> On 20 Feb 2016, at 14:36, ArcEye <notifications@github.com> wrote:
> 
> So change 'pincount' to 'localpincount' and all should work as intended.

Will do.

ArcEye, 2016-02-20:
I will be glad when all the new hal stuff is in the main repo, gets very confusing remembering which branch you are in and what the differences are between them.

I rewrote all my access libraries to use the new HAL functions etc. so tend to use that branch even though it is not mainstream.

Must label my various clones better, so I can see exactly what I am using :frowning:

PR #879 (2016-02-23): Edit to instcomp man page generation and add new offsetn component
Man page edit, clarifies that not all possible parameters and args
may be in use by all components

offsetn - multi-pin version of offset allowing 1-9 sets of pins
All pin sets are individual, allowing unique offsets per n axes
to be applied within same component

Fixes #870
closed 2016-02-23 by luminize