machinekit/machinekit #871: Xenomai problem on x86 Wheezy

kinsamanka, 2016-02-15:
Installing machinekit with xenomai kernel on virtualbox shows this error:
```
mk@debian:~$ ULAPI_DEBUG=5 realtime start
halcmd: hal_init() failed: -22
NOTE: 'rtapi' module must be loaded
```
Logs:
```
Feb 15 08:20:32 debian msgd:0: startup pid=3028 flavor=xenomai rtlevel=1 usrlevel=1 halsize=524288 shm=Posix gcc=4.7.2 version=unknown
Feb 15 08:20:32 debian msgd:0: ØMQ=4.0.5 czmq=2.2.0 protobuf=2.4.1 libwebsockets=<no version symbol>
Feb 15 08:20:32 debian msgd:0: configured: sha=5257e28
Feb 15 08:20:32 debian msgd:0: built:      Jan 12 2016 14:51:58 sha=5257e28
Feb 15 08:20:32 debian msgd:0: register_stuff: actual hostname as announced by avahi='debian-2.local'
Feb 15 08:20:32 debian msgd:0: zeroconf: registering: 'Log service on debian.local pid 3028'
Feb 15 08:20:32 debian msgd:0: rtapi_app:3038:user accepting commands at ipc:///tmp/0.rtapi.a42c8c6b-4025-4f83-ba28-dad21114744a
Feb 15 08:20:32 debian msgd:0: ulapi:3042:user HAL_LIB:0 FATAL - dlopen(/usr/lib/linuxcnc/ulapi-xenomai.so) failed: libnative.so.3: shared object cannot be dlopen()ed
Feb 15 08:20:32 debian msgd:0: ulapi:3042:user hal_xinit:95 HAL error: rtapi init(HAL_hal_lib3042) failed
Feb 15 08:20:33 debian msgd:0: zeroconf: registered 'Log service on debian.local pid 3028' _machinekit._tcp 0 TXT "uuid=a42c8c6b-4025-4f83-ba28-dad21114744a" "instance=ecf1b5f0-d379-11e5-bdcb-080027009552" "service=log" "dsn=ipc:///tmp/0.log.a42c8c6b-4025-4f83-ba28-dad21114744a"
```
kernel:
```
mk@debian:~$ uname -a
Linux debian 3.8-1-xenomai.x86-amd64 #1 SMP Debian 3.8.13-12~1wheezy~1da x86_64 GNU/Linux
```
```
mk@debian:~$ apt-cache show linux-image-3.8-1-xenomai.x86-amd64
Package: linux-image-3.8-1-xenomai.x86-amd64
Source: linux
Version: 3.8.13-12~1wheezy~1da
Architecture: amd64
Maintainer: John Morris <repos@zultron.com>
Installed-Size: 143345
Pre-Depends: debconf | debconf-2.0
Depends: kmod | module-init-tools, linux-base (>= 3~), initramfs-tools (>= 0.99~) | linux-initramfs-tool
Recommends: firmware-linux-free (>= 3~)
Suggests: linux-doc-3.8, debian-kernel-handbook, grub-pc | extlinux | lilo
Breaks: at (<< 3.1.12-1+squeeze1), initramfs-tools (<< 0.99~)
Provides: linux-image, linux-image-xenomai, linux-modules-3.8-1-xenomai.x86-amd64
Priority: optional
Section: kernel
Filename: pool/main/l/linux/linux-image-3.8-1-xenomai.x86-amd64_3.8.13-12~1wheezy~1da_amd64.deb
Size: 33740548
SHA256: b212382c9af9be5f8c80d6b6ed7ed265a4a198690315ca8d19dd4ef143a84ff8
SHA1: 16f4a6b44acee41b991d2ea25f1c2d66d867de59
MD5sum: b55b6ef7c269a98fb718e6d926644196
Description: Linux 3.8 for 64-bit PCs, Xenomai x86
 The Linux kernel 3.8 and modules for use on PCs with AMD64, Intel 64 or
 VIA Nano processors.
 .
 This kernel includes the Xenomai realtime patch set for x86 architectures.
```

This issue is not present on Jessie

sirop, 2016-02-15:
Ok. I was too fast to close https://github.com/machinekit/machinekit/issues/856 .

Look  through https://github.com/machinekit/machinekit/issues/856 .
Try to build from sources, this should work also on wheezy as I tried it out two days ago.

There is some strange problem with remnants of machinekit debian packages
which prevent your from building from source successfully.
So if you build from source, do it with a fresh clean wheezy install.

kinsamanka, 2016-02-15:
This is a problem with the xenomai kernel:
```
HAL_LIB:0 FATAL - dlopen(/usr/lib/linuxcnc/ulapi-xenomai.so) failed: libnative.so.3: shared object cannot be dlopen()ed
```
This issue is similar to this one: https://sourceforge.net/p/emc/mailman/message/30408147/

sirop, 2016-02-15:
If so, then it is rather a problem with xenomai kernel **packages**.
I just rebuilt a RIP machinekit:
```
cml@debian-cml2:~/machinekit-sirop$ uname -a
Linux debian-cml2 3.8-1-xenomai.x86-amd64 #1 SMP Debian 3.8.13-9 x86_64 GNU/Linux
cml@debian-cml2:~/machinekit-sirop$ halrun -I -k -f pktuart.hal
msgd:0 stopped
rtapi:0 stopped
halcmd: show comp
Loaded HAL Components:
    ID  Type Flags Inst Name                                      PID   State
   358  User          0 halcmd17756                               17756 ready, u1:0 u2:0
   357  uHAL   i      0 hal_lib17756                              17756 ready, u1:0 u2:0
   355  RT            0 threads                                   RT    ready, u1:0 u2:0
   342  RT            0 mesa_pktgyro                              RT    ready, u1:0 u2:0
    74  RT            0 hm2_pci                                   RT    ready, u1:0 u2:0
    73  RT            0 hostmot2                                  RT    ready, u1:0 u2:0
    66  rHAL   i      0 hal_lib                                   0     ready, u1:0 u2:0

halcmd:
```

Maybe the list of  the xenomai packages installed  would be helpful:
```
cml@debian-cml2:~$ dpkg-query -l | grep xenomai
ii  libxenomai-dev                               2.6.3-0.1mk~wheezy1                        amd64        Headers and static libs for Xenomai
ii  libxenomai1                                  2.6.3-0.1mk~wheezy1                        amd64        Shared libraries for Xenomai
ii  linux-headers-3.8-1-common-xenomai.x86       3.8.13-9da~wheezy1                         amd64        Common header files for Linux 3.8-1-xenomai.x86
ii  linux-headers-3.8-1-xenomai.x86-amd64        3.8.13-9da~wheezy1                         amd64        Header files for Linux 3.8-1-xenomai.x86-amd64
ii  linux-headers-xenomai.x86-amd64              3.8+59-1da~wheezy1                         amd64        Header files for Linux xenomai.x86-amd64 configuration (meta-package)
ii  linux-image-3.8-1-xenomai.x86-amd64          3.8.13-9da~wheezy1                         amd64        Linux 3.8 for 64-bit PCs, Xenomai x86
ii  linux-image-xenomai.x86-amd64                3.8+59-1da~wheezy1                         amd64        Linux for 64-bit PCs (meta-package), Xenomai x86
ii  machinekit-xenomai                           0.1.1502-1da.bbot.pr779.git0edd3c0~1wheezy amd64        PC based motion controller for real-time Linux
ii  xenomai-runtime                              2.6.3-0.1mk~wheezy1                        amd64        Xenomai runtime utilities
```

```
cml@debian-cml2:~$ grep install /var/log/dpkg.log.4 | grep xenomai
...
2015-10-08 04:56:25 status installed libxenomai1:amd64 2.6.3-0.1mk~wheezy1
2015-10-08 04:56:25 status installed libxenomai-dev:amd64 2.6.3-0.1mk~wheezy1
2015-10-08 06:08:33 status installed linux-image-3.8-1-xenomai.x86-amd64:amd64 3.8.13-9da~wheezy1
2015-10-08 06:08:34 status installed linux-image-xenomai.x86-amd64:amd64 3.8+59-1da~wheezy1
2015-10-08 06:12:49 status installed xenomai-runtime:amd64 2.6.3-0.1mk~wheezy1

```

```
cml@debian-cml2:~$ grep install /var/log/dpkg.log.3 |  grep xenomai
...
2015-11-19 09:06:36 status installed linux-headers-3.8-1-xenomai.x86-amd64:amd64 3.8.13-9da~wheezy1
2015-11-24 06:42:57 status installed machinekit-xenomai:amd64 0.1.1502-1da.bbot.pr779.git0edd3c0~1wheezy
```

mhaberler, 2016-02-25:
@kinsamanka - looking into https://github.com/machinekit/machinekit/issues/871#issuecomment-184079845

I do not think this is related to  https://sourceforge.net/p/emc/mailman/message/30408147/, that is fixed long since - also not kernel or "kernel package" issue, this kernel and pkgs are stable since a long time

platform j1900 3.8-1-xenomai.x86-686-pae Ubuntu 14.04

`````
ULAPI_DEBUG=5 realtime start
halcmd: hal_init() failed: -22
NOTE: 'rtapi' module must be loaded

/var/log/syslog says:
Feb 25 13:52:48 j1900 msgd:0: ulapi:2294:user HAL_LIB:0 FATAL - dlopen(ulapi-xenomai.so) failed: /home/mah/build/lib/machinekit/ulapi-xenomai.so: undefined symbol: rt_timer_read

looking at ulapi-xenomai.so - no wonder, libnative.so isnt there - so cant resolve the rt_* functions

mah@j1900:~/build$ ldd lib/machinekit/ulapi-xenomai.so
	linux-gate.so.1 =>  (0xb7742000)
	libmkshm.so.1.0 => /home/mah/build/lib/libmkshm.so.1.0 (0xb7728000)
	libmkhal.so.1.0 => /home/mah/build/lib/libmkhal.so.1.0 (0xb76fd000)
	libc.so.6 => /lib/i386-linux-gnu/libc.so.6 (0xb7528000)
	librt.so.1 => /lib/i386-linux-gnu/librt.so.1 (0xb751f000)
	libmk-pb2++.so.1.0 => /home/mah/build/lib/libmk-pb2++.so.1.0 (0xb7329000)
	libdl.so.2 => /lib/i386-linux-gnu/libdl.so.2 (0xb7324000)
	libprotobuf.so.8 => /usr/lib/i386-linux-gnu/libprotobuf.so.8 (0xb7224000)
	libpthread.so.0 => /lib/i386-linux-gnu/libpthread.so.0 (0xb7207000)
	libstdc++.so.6 => /usr/lib/i386-linux-gnu/libstdc++.so.6 (0xb711f000)
	libgcc_s.so.1 => /lib/i386-linux-gnu/libgcc_s.so.1 (0xb7102000)
	/lib/ld-linux.so.2 (0xb7743000)
	libz.so.1 => /lib/i386-linux-gnu/libz.so.1 (0xb70e8000)
	libm.so.6 => /lib/i386-linux-gnu/libm.so.6 (0xb70a2000)
````

I am pretty sure that is again a flags ordering issue during linking of ulapi-xenomai.so, and/or an no-as-needed thing:

`````
deleting ulapi-xenomai.so, VERBOSE=1 make:

make -f src/rtapi/CMakeFiles/ulapi-xenomai.dir/build.make src/rtapi/CMakeFiles/ulapi-xenomai.dir/build
make[2]: Entering directory `/home/mah/build'
Linking C shared library ../../lib/machinekit/ulapi-xenomai.so
cd /home/mah/build/src/rtapi && /usr/bin/cmake -E cmake_link_script CMakeFiles/ulapi-xenomai.dir/link.txt --verbose=1
/usr/bin/cc  -fPIC  -fPIC -fwrapv -fno-strict-overflow -std=gnu99 -fgnu89-inline -g -lnative -L/usr/lib -lxenomai -lpthread -lrt   -shared -Wl,-soname,ulapi-xenomai.so -o ../../lib/machinekit/ulapi-xenomai.so CMakeFiles/ulapi-xenomai.dir/rtapi_common.c.o CMakeFiles/ulapi-xenomai.dir/rtapi_task.c.o CMakeFiles/ulapi-xenomai.dir/rtapi_shmem.c.o CMakeFiles/ulapi-xenomai.dir/rtapi_time.c.o CMakeFiles/ulapi-xenomai.dir/rtapi_exception.c.o CMakeFiles/ulapi-xenomai.dir/ulapi_main.c.o CMakeFiles/ulapi-xenomai.dir/xenomai.c.o CMakeFiles/rtapi_heap.dir/rtapi_heap.c.o CMakeFiles/rtapi_support.dir/rtapi_support.c.o CMakeFiles/rtapi_support.dir/rtapi_hexdump.c.o ../../lib/libmkshm.so.1.0 ../../lib/libmkhal.so.1.0 ../../lib/libmk-pb2++.so.1.0 -lprotobuf -lpthread -ldl -lrt -Wl,-rpath,/home/mah/build/lib:

`````
I reshuffled the link to
````
mah@j1900:~/build$ sh xxx
+ cd /home/mah/build/src/rtapi
+ /usr/bin/cc -fPIC -fPIC -fwrapv -fno-strict-overflow -std=gnu99 -fgnu89-inline -g -shared -Wl,-soname,ulapi-xenomai.so -o ../../lib/machinekit/ulapi-xenomai.so CMakeFiles/ulapi-xenomai.dir/rtapi_common.c.o CMakeFiles/ulapi-xenomai.dir/rtapi_task.c.o CMakeFiles/ulapi-xenomai.dir/rtapi_shmem.c.o CMakeFiles/ulapi-xenomai.dir/rtapi_time.c.o CMakeFiles/ulapi-xenomai.dir/rtapi_exception.c.o CMakeFiles/ulapi-xenomai.dir/ulapi_main.c.o CMakeFiles/ulapi-xenomai.dir/xenomai.c.o CMakeFiles/rtapi_heap.dir/rtapi_heap.c.o CMakeFiles/rtapi_support.dir/rtapi_support.c.o CMakeFiles/rtapi_support.dir/rtapi_hexdump.c.o ../../lib/libmkshm.so.1.0 ../../lib/libmkhal.so.1.0 ../../lib/libmk-pb2++.so.1.0 -lprotobuf -lpthread -ldl -lnative -L/usr/lib -lxenomai -lpthread -lrt -Wl,-rpath,/home/mah/build/lib:
`````

and it works!

kinsamanka, 2016-02-25:
Not reproducible on new VM, closing.

Thanks for fixing the cmake issue! ;-)

closed 2016-02-25 by kinsamanka

sirop, 2016-03-15:
Just for the record:
If configured with 
```
$ ./configure  --without-posix  --without-rt-preempt  --with-xenomai
```
on
```
$ uname -a
Linux debian-slave 3.8-1-xenomai.x86-amd64 #1 SMP Debian 3.8.13-12~1wheezy~1da x86_64 GNU/Linux
```
then no need to reshuffle the linking of `ulapi-xenomai.so` .

Built with
```
$ git log | head -5
commit 5257e2826d1698bd40b85d770be4aed72f52cacc
Merge: 84c8add bb4d58c
Author: Bas de Bruijn <bdebruijn@luminize.nl>
Date:   Tue Jan 12 15:32:01 2016 +0100
```