machinekit/machinekit #872: DSO missing from command line

FlashPackets, 2016-02-15:
Running the src install scripts resulted in the following frustration:

Creating shared library liblinuxcnchal.so.0
Linking webtalk
/usr/bin/ld: /usr/lib/gcc/i686-linux-gnu/5/../../../../lib/libwebsockets.a(client.c.o): undefined reference to symbol 'BIO_ctrl@@OPENSSL_1.0.0'
/lib/i386-linux-gnu/libcrypto.so.1.0.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
machinetalk/webtalk/Submakefile:53: recipe for target '../bin/webtalk' failed
make: *** [../bin/webtalk] Error 1

Sage advice on dealing with this would be appreciated.

luminize, 2016-02-15:
You dont provide a lot of info.
What installation path did you follow? Platform? Log files?

Have a read thru here:
https://github.com/machinekit/machinekit-docs/blob/master/machinekit-documentation/getting-help.asciidoc

Cheers

mhaberler, 2016-02-16:
is the environment confidential information or will be there more forthcoming?

closed 2016-02-17 by mhaberler

FlashPackets, 2016-02-18:
My error was in not fully appreciating the importance of using the Wheezy distribution. So... after reading the logs and finding nothing that made any sense, I bit the bullet and blew away the partition and loaded up Wheezy. From the correct distro everything compiled fine. So my sincere apologies for wasting your time on that one.

Unfortunately, after all that, when I got to running the motors, everything was inert. In each case, loading up and compiling for the three use cases, RT-Preempt, Xenomai, and RTAI had the same result ... no comforting hum from a motor spinning away.

Learning my lesson, I loaded up the latest LinuxCNC from the official site, sharing the partition. with Wheezy. Now, everything is working at least with a pure LinuxCNC. 

From reading  mhaberler's blog notes it's pretty clear that I should order up a Mesanet PCIe card and try again without the now obsolete printer port, that nevertheless, is presently working, well except for a much more sensitive home/endstop signal jitter. What I don't understand yet is exactly how Machinekit and LinuxCNC work together or individually.

I've really become quite enamored with 0MQ and protobuf. I'm sure the effort, whatever it may be will have a high reward. 

Regards,

I hope my next question will be more insightful !

Flash

mhaberler, 2016-02-19:
@FlashPackets - this is the machinekit project, not the LinuxCNC project. It was forked some three years ago. So your choice is either - or.

I think such basics should be resolved on the mailing list.